merbanan/rtl_433 #2967: Wrong temperature on negative values for Klimalogg Pro sensor

Rob-2, 2024-06-12:
Hello,
thank you for all the great work!

I am using a TFA Klimalogg Pro and have roughly 15 sensors in use. Some 30.3180.IT (temp+hum) and some  30.3181.IT (temp with wire sensor, no humidity). I'm using a RTLSDR since ~3 years with the TFREC project.

However, I would like to switch to rtl_433, because I would like to add a ecowitt weather station and I am hoping, that rel_433 can receive both, the TFA sensors and the Ecowitt in future, so that I don't need to maintain two SDRs.

I've set up rtl_433 and it works very good so far. One strange thing I've noticed.

I have one 30.3181.IT which I am using as binary sensor, so this sensor sends -1.2°C in 99% of the time. This works fine. The sensor shows -1.2°C, tfrec receives -1.2°C:
**#030 1718215951  2d d4 21 5c 83 88 6a 60 00 56 58           TFA1 ID 215c -1.2 0% seq 0 lowbat 0 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 215c -1.2 0 0 0 80 0 1718215951**

However, rtl_433 decodes this into -2.8°C
**time      : 2024-06-12 20:11:59
model     : Klimalogg Pro Id        : 215c
Battery   : 1            Temperature: -2.8 C       Humidity  : 106           Sequence Number: 0        Integrity : CRC
Modulation: ASK          Freq      : 868.4 MHz
RSSI      : -0.4 dB      SNR       : 31.3 dB       Noise     : -31.7 dB**

Now I wonder why the 2 libs decode this differently while still reporting CRC OK. Is it because it's an 30.3181.IT without humidity?
Is there an option to see the decoded output as hex as on the tfrec example?

According to the tfrec source the telegram format is:
**// 0x2d 0xd4 ID ID sT TT HH BB SS 0x56 CC
// 2d d4: Sync bytes
// ID(14:0): 15 bit ID of sensor (printed on the back and displayed after powerup)
// ID(15) is either 1 or 0 (fixed, depends on the sensor)
// s(3:0): Learning sequence 0...f, after learning fixed 8
// TTT: Temperatur in BCD in .1degC steps, offset +40degC (-> -40...+60)
// HH(6:0): rel. Humidity in % (binary coded, no BCD!)
// BB(7): Low battery if =1
// BB(6:4): 110 or 111 (for 3199)
// SS(7:4): sequence number (0...f)
// SS(3:0): 0000 (fixed)
// 56: Type?
// CC: CRC8 from ID to 0x56 (polynome x^8 + x^5 + x^4  + 1)**

Splitting the tfrec received telegram (2d d4 21 5c 83 88 6a 60 00 56 58):
ID: 2d d4 21 5c 
s: 8
TTT: 3 88  (in BCD!)
HH: 6a      (binary coded, not BCD)
BB: 60
SS: 00
56: 56
CC: 58

So temp is 38.8°C - 40°C Offset, so -1.2°C.

Now rtl_433 decodes this obviously to -2.8°C, plus 40°C offset, so I assume the rtl_433 decoded raw value for temp was 3 72.

The next 2 bytes are humidity, the 30.3181.IT doesn't have humidity, it usually sends 0x6a as seen above. rtl433 reads this 0x6a as 106, which is fine/correct. So the bit's are not shifted.

Do you have any idea why this is decoded differenly (wrong) in rtl_433?

BR
Rob

PS: Just a note: One challenge I still have is the same as with tfrec... I need set gain to auto to receive the far away sensors, but with this setting very close sensor come in to strong and can't be decoded. When I cover the close sensors to lower the signal strength, they are also received fine. I read someone here was confused, that he only could receive 3 out of 5 sensors and I wanted so say, with these sensors and RTL, it might be that they are to close and/or the gain is set to high. But I need the high gain to receive the very far out sensors.

zuckschwerdt, 2024-06-12:
Do your 30.3181.IT's always report `6a` (106) for the humidity? We should add that error code to the decoder and then skip the humidity output.

ProfBoc75, 2024-06-12:
@Rob-2 : 
Since this decoder got several updates and the model is now "Klimalogg-Pro" not "Klimalogg Pro", looks like you have an older version of rtl_433.

Can you check if you have the last version of rtl_433 as it seems to work at @zuckschwerdt  level ?

Rob-2, 2024-06-13:
Hello Christian, hello Bruno,

thank you very much for your respone. Bruno, you were spot on, I also noticed that the commands from Christian (thanks for the example. :) ) were somewhat unknown.
I had it installed with apt install on a raspbian/debian 11 and the version was 20.11. I didn't noticed that it's so old, before. Tried to install a newer version by hand, but then the dependencies were to old. So I upgraded to raspian 12, which had 22.11 and then installed 23.11 by hand.

**And then the signal decoded fine**. Sorry, that I didn't thought about the version. So the issue can be closed.

I'll try the suggested gain settings, as soon as I find some time.

@zuckschwerdt  Yes, the 30.3181.IT always report 6a for the humidity. However, there's a catch. When a 30.3180.IT's humidity goes into saturation (>99%), it also reports 6a (106), not 100 as one might have assumed. So there's the question how to handle hum=6a. It could be a saturated, humidity=100% 30.3180.IT sensor OR it could be a 30.3180.IT sensor.

tfrec's code sets humidity output value to 0 if it see's hum=6a. However, that was annoying, because in grafana my humidity value then jumped between 99% and 0%, and not between 99% and 100%. I solved this in my personal tfrec compile by an addition "if" check agains my personal used sensor id's. And if it's an id of a 30.3180.IT, it outputs 100% if it decodes 6a. But that's not a general solution because one would need to have the IDs o the 30.3180.IT's.

I checked the left over / unknown bits output of the raw output of my sensors, but I didn't find an indicator to 100% reliably distinguish between a 30.3180.IT and a 30.3181.IT (because hum=6a is not enough).

I was curious what the Klimalogg display unit shows on a 100% saturated 30.3180.IT and checked the history. It then displays "-" for humidity (not 100%). It also shows "-" for 30.3181.ITs.

So yeah, I don't know whats the best way to handle 6a. Skipping humidity output on 6a would be fine for 30.3181.ITs, but if you have a 30.3180.IT in the winter, it could be that it would jump between 0x64 (99%) and 0x6a (>99%). A little bit annoying.

So thank you again for your help and sorry that I claimed your time.

If you need samples recordings of the Klimalogg sensors for whatever resons, feel free to ping me. Have a good amount in use. :)

BR
Rob

zuckschwerdt, 2024-06-13:
Thanks for the very detailed analysis and writeup on humidity oddities. I guess we should just document the behaviour and not suppress the 6a value then.

I'll close this upon adding those notes.

Rob-2, 2024-06-13:
If you want to have a look at the different sensor raw output, I've just recorded a ~20s sample with tfrec and directly after a ~20s sample with rtl_433.

./tfrec -g -1 -t 450 -f 868325 -T 1 -e "publish-tfrec-to-mqtt.sh" -D
Registering demod for TFA_1 KlimaLoggPro
Detached kernel driver
Found Rafael Micro R820T tuner
AUTO GAIN
Frequency 868.3250MHz
Samplerate 1536000
START READ THREAD
Allocating 15 zero-copy buffers
#000 1718306981  2d d4 21 73 86 80 6a 60 80 56 72           TFA1 ID 2173 +28.0 0% seq 8 lowbat 0 RSSI 66
EXEC publish-tfrec-to-mqtt.sh 2173 +28.0 0 8 0 66 0 1718306981
#001 1718306982  2d d4 6f 6f 86 05 6a 60 00 56 45           TFA1 ID 6f6f +20.5 0% seq 0 lowbat 0 RSSI 77
EXEC publish-tfrec-to-mqtt.sh 6f6f +20.5 0 0 0 77 0 1718306982
#002 1718306983  2d d4 0a 6a 85 49 38 60 b0 56 99           TFA1 ID 0a6a +14.9 56% seq b lowbat 0 RSSI 76
EXEC publish-tfrec-to-mqtt.sh 0a6a +14.9 56 11 0 76 0 1718306983
#003 1718306984  2d d4 42 63 86 30 28 60 00 56 1c           TFA1 ID 4263 +23.0 40% seq 0 lowbat 0 RSSI 81
EXEC publish-tfrec-to-mqtt.sh 4263 +23.0 40 0 0 81 0 1718306984
#004 1718306985  2d d4 32 a7 86 35 28 60 00 56 58           TFA1 ID 32a7 +23.5 40% seq 0 lowbat 0 RSSI 81
EXEC publish-tfrec-to-mqtt.sh 32a7 +23.5 40 0 0 81 0 1718306985
#005 1718306985  2d d4 21 d6 86 00 6a 60 00 56 6f           TFA1 ID 21d6 +20.0 0% seq 0 lowbat 0 RSSI 81
EXEC publish-tfrec-to-mqtt.sh 21d6 +20.0 0 0 0 81 0 1718306985
#006 1718306985  2d d4 21 5c 83 88 6a 60 00 56 58           TFA1 ID 215c -1.2 0% seq 0 lowbat 0 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 215c -1.2 0 0 0 80 0 1718306985
#007 1718306986  2d d4 22 96 86 60 6a e0 00 56 06           TFA1 ID 2296 +26.0 0% seq 0 lowbat 1 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 2296 +26.0 0 0 1 80 0 1718306986
#008 1718306987  2d d4 51 92 85 81 6a 60 70 56 f9           TFA1 ID 5192 +18.1 0% seq 7 lowbat 0 RSSI 58
EXEC publish-tfrec-to-mqtt.sh 5192 +18.1 0 7 0 58 0 1718306987
#009 1718306988  2d d4 71 b1 85 55 6a 60 f0 56 46           TFA1 ID 71b1 +15.5 0% seq f lowbat 0 RSSI 74
EXEC publish-tfrec-to-mqtt.sh 71b1 +15.5 0 15 0 74 0 1718306988
#010 1718306988  2d d4 06 f2 87 42 6a 60 00 56 d0           TFA1 ID 06f2 +34.2 0% seq 0 lowbat 0 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 06f2 +34.2 0 0 0 80 0 1718306988
#011 1718306989  2d d4 20 3b 85 46 37 60 00 56 cb           TFA1 ID 203b +14.6 55% seq 0 lowbat 0 RSSI 77
EXEC publish-tfrec-to-mqtt.sh 203b +14.6 55 0 0 77 0 1718306989
#012 1718306990  2d d4 78 3d 85 40 6a 60 00 56 19           TFA1 ID 783d +14.0 0% seq 0 lowbat 0 RSSI 77
EXEC publish-tfrec-to-mqtt.sh 783d +14.0 0 0 0 77 0 1718306990
#013 1718306990  2d d4 6c 43 86 22 6a 60 00 56 3a           TFA1 ID 6c43 +22.2 0% seq 0 lowbat 0 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 6c43 +22.2 0 0 0 80 0 1718306990
#014 1718306990  2d d4 1e ef 85 61 32 60 00 56 36           TFA1 ID 1eef +16.1 50% seq 0 lowbat 0 RSSI 79
EXEC publish-tfrec-to-mqtt.sh 1eef +16.1 50 0 0 79 0 1718306990
#015 1718306991  2d d4 40 93 85 39 6a 60 00 56 3f           TFA1 ID 4093 +13.9 0% seq 0 lowbat 0 RSSI 77
EXEC publish-tfrec-to-mqtt.sh 4093 +13.9 0 0 0 77 0 1718306991
#016 1718306991  2d d4 26 3b 85 31 6a 60 00 56 03           TFA1 ID 263b +13.1 0% seq 0 lowbat 0 RSSI 63
EXEC publish-tfrec-to-mqtt.sh 263b +13.1 0 0 0 63 0 1718306991
#017 1718306991  2d d4 21 73 86 80 6a 60 90 56 1c           TFA1 ID 2173 +28.0 0% seq 9 lowbat 0 RSSI 65
EXEC publish-tfrec-to-mqtt.sh 2173 +28.0 0 9 0 65 0 1718306991
#018 1718306992  2d d4 6f 6f 86 05 6a 60 00 56 45           TFA1 ID 6f6f +20.5 0% seq 0 lowbat 0 RSSI 77
EXEC publish-tfrec-to-mqtt.sh 6f6f +20.5 0 0 0 77 0 1718306992
#019 1718306994  2d d4 0a 6a 85 49 38 60 c0 56 a2           TFA1 ID 0a6a +14.9 56% seq c lowbat 0 RSSI 76
EXEC publish-tfrec-to-mqtt.sh 0a6a +14.9 56 12 0 76 0 1718306994
#020 1718306994  2d d4 42 63 86 31 28 60 00 56 cf           TFA1 ID 4263 +23.1 40% seq 0 lowbat 0 RSSI 81
EXEC publish-tfrec-to-mqtt.sh 4263 +23.1 40 0 0 81 0 1718306994
#021 1718306995  2d d4 21 d6 86 00 6a 60 00 56 6f           TFA1 ID 21d6 +20.0 0% seq 0 lowbat 0 RSSI 81
EXEC publish-tfrec-to-mqtt.sh 21d6 +20.0 0 0 0 81 0 1718306995
#022 1718306995  2d d4 32 a7 86 35 28 60 00 56 58           TFA1 ID 32a7 +23.5 40% seq 0 lowbat 0 RSSI 81
EXEC publish-tfrec-to-mqtt.sh 32a7 +23.5 40 0 0 81 0 1718306995
#023 1718306996  2d d4 21 5c 83 88 6a 60 00 56 58           TFA1 ID 215c -1.2 0% seq 0 lowbat 0 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 215c -1.2 0 0 0 80 0 1718306996
#024 1718306996  2d d4 22 96 86 60 6a e0 00 56 06           TFA1 ID 2296 +26.0 0% seq 0 lowbat 1 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 2296 +26.0 0 0 1 80 0 1718306996
#025 1718306997  2d d4 51 92 85 81 6a 60 80 56 e1           TFA1 ID 5192 +18.1 0% seq 8 lowbat 0 RSSI 58
EXEC publish-tfrec-to-mqtt.sh 5192 +18.1 0 8 0 58 0 1718306997
#026 1718306999  2d d4 71 b1 85 55 6a 60 00 56 5e           TFA1 ID 71b1 +15.5 0% seq 0 lowbat 0 RSSI 75
EXEC publish-tfrec-to-mqtt.sh 71b1 +15.5 0 0 0 75 0 1718306999
#027 1718306999  2d d4 06 f2 87 42 6a 60 00 56 d0           TFA1 ID 06f2 +34.2 0% seq 0 lowbat 0 RSSI 80
EXEC publish-tfrec-to-mqtt.sh 06f2 +34.2 0 0 0 80 0 1718306999
#028 1718307000  2d d4 20 3b 85 46 37 60 00 56 cb           TFA1 ID 203b +14.6 55% seq 0 lowbat 0 RSSI 77
EXEC publish-tfrec-to-mqtt.sh 203b +14.6 55 0 0 77 0 1718307000
#029 1718307000  2d d4 6c 43 86 22 6a 60 00 56 3a           TFA1 ID 6c43 +22.2 0% seq 0 lowbat 0 RSSI 80
[... 192 lines cut ...]
Modulation: ASK          Freq      : 868.4 MHz
[... cut ...]
codes     : {273}80000000000000000000000000000000000000000b42b4469619a5607006af6000000
time      : 2024-06-13 21:30:31
model     : Klimalogg-Pro Id        : 6c43
Battery   : 1            Temperature: 22.2 C       Humidity  : 106           Sequence Number: 0        Integrity : CRC
Modulation: ASK          Freq      : 868.5 MHz
RSSI      : -0.2 dB      SNR       : 32.0 dB       Noise     : -32.1 dB
[pulse_slicer_nrzs] Klimalogg
codes     : {273}80000000000000000000000000000000000000000b42b36c261445606006a5c000000

commit 321f231c (2024-06-14): Change Klimalogg-Pro max humidity (closes #2967)
closed 2024-06-14 by zuckschwerdt