merbanan/rtl_433 #2973: WH40 id is incorrectly calculated

Steven-Jones911, 2024-06-18:
I have a new WH40 with a labelled ID of 0x1887F.

ambientweather_wh31e.c
Line 293 reads now:
int id         = (b[2] << 8) | b[3];
Line 293 should read:
int id         = (b[1] & 0xf) << 16 | (b[2] << 8) | b[3];

The commentary should also be changed at line 118 to change byte[1] from 00 to II indicating the fact byte[1] makes up parts of the ID.

ProfBoc75, 2024-06-19:
Hi @Steven-Jones911: There is a little risk to break the decoding for others, their ID could change, very small since it was 0x00 .

Can you clarify one point. You proposed to update the layout with II instead of 00 and you extract only lower nibble with the 0xf.

So layout 0I II II with you current code .... or II II II without & 0xf... ?

Steven-Jones911, 2024-06-19:
Yes, of course, I miswrote 00. 0I would be correct in the comment.
I do not know whether the upper nibble is used for an ID in other units. My previous unit had 4 hex digits, this new one has 5.

gdt, 2024-06-20:
I suppose there  is some risk, but this is pretty compelling evidence that the previous decoding is not right.   It is not 100% convincing that the missing 0x1 nibble is that one, but maybe that's the only choice.   FWIW, I think all decoders should output variables for undecoded bits so that they get logged and people are aware.

All in all, I'd be ok with a PR to change this.  I weight correct behavior higher than compat.