merbanan/rtl_433 #2978: Replicating Renault TPMS signal

avicarmeli, 2024-06-21:
For development purpose I created esp32 simulator using CC1101 to simulate Renault TPMS sensor. 
After reading the Renault [TPMS decoder code](https://github.com/merbanan/rtl_433/blob/master/src/devices/tpms_renault.c) I came up withe the following code for esp32:

```
#include <ELECHOUSE_CC1101_SRC_DRV.h>

byte transmitt_byte[13] = {0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA , 0xAA};

void setup() {

    Serial.begin(115200);

    if (ELECHOUSE_cc1101.getCC1101()){      // Check the CC1101 Spi connection.
    Serial.println("Connection OK");
    }else{
    Serial.println("Connection Error");
    }
 
    ELECHOUSE_cc1101.Init();                // must be set to initialize the cc1101!
    ELECHOUSE_cc1101.setCCMode(1);          // set config for internal transmission mode.
    ELECHOUSE_cc1101.setModulation(0);      // set modulation mode. 0 = 2-FSK, 1 = GFSK, 2 = ASK/OOK, 3 = 4-FSK, 4 = MSK.
    ELECHOUSE_cc1101.setMHZ(433.92);        // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
    ELECHOUSE_cc1101.setDeviation(47.60);   // Set the Frequency deviation in kHz. Value from 1.58 to 380.85. Default is 47.60 kHz.
    ELECHOUSE_cc1101.setChannel(0);         // Set the Channelnumber from 0 to 255. Default is cahnnel 0.
    ELECHOUSE_cc1101.setChsp(199.95);       // The channel spacing is multiplied by the channel number CHAN and added to the base frequency in kHz. Value from 25.39 to 405.45. Default is 199.95 kHz.
    ELECHOUSE_cc1101.setRxBW(812.50);       // Set the Receive Bandwidth in kHz. Value from 58.03 to 812.50. Default is 812.50 kHz.
    ELECHOUSE_cc1101.setDRate(19.97);       // Set the Data Rate in kBaud. Value from 0.02 to 1621.83. Default is 99.97 kBaud!
    ELECHOUSE_cc1101.setPA(10);             // Set TxPower. The following settings are possible depending on the frequency band.  (-30  -20  -15  -10  -6    0    5    7    10   11   12) Default is max!
    ELECHOUSE_cc1101.setSyncMode(0);        // Combined sync-word qualifier mode. 0 = No preamble/sync. 1 = 16 sync word bits detected. 2 = 16/16 sync word bits detected. 3 = 30/32 sync word bits detected. 4 = No preamble/sync, carrier-sense above threshold. 5 = 15/16 + carrier-sense above threshold. 6 = 16/16 + carrier-sense above threshold. 7 = 30/32 + carrier-sense above threshold.
    ELECHOUSE_cc1101.setSyncWord(211, 145); // Set sync word. Must be the same for the transmitter and receiver. (Syncword high, Syncword low)
    ELECHOUSE_cc1101.setAdrChk(0);          // Controls address check configuration of received packages. 0 = No address check. 1 = Address check, no broadcast. 2 = Address check and 0 (0x00) broadcast. 3 = Address check and 0 (0x00) and 255 (0xFF) broadcast.
    ELECHOUSE_cc1101.setAddr(0);            // Address used for packet filtration. Optional broadcast addresses are 0 (0x00) and 255 (0xFF).
    ELECHOUSE_cc1101.setWhiteData(0);       // Turn data whitening on / off. 0 = Whitening off. 1 = Whitening on.
    ELECHOUSE_cc1101.setPktFormat(0);       // Format of RX and TX data. 0 = Normal mode, use FIFOs for RX and TX. 1 = Synchronous serial mode, Data in on GDO0 and data out on either of the GDOx pins. 2 = Random TX mode; sends random data using PN9 generator. Used for test. Works as normal mode, setting 0 (00), in RX. 3 = Asynchronous serial mode, Data in on GDO0 and data out on either of the GDOx pins.
    ELECHOUSE_cc1101.setLengthConfig(1);    // 0 = Fixed packet length mode. 1 = Variable packet length mode. 2 = Infinite packet length mode. 3 = Reserved
    ELECHOUSE_cc1101.setPacketLength(0);    // Indicates the packet length when fixed packet length mode is enabled. If variable packet length mode is used, this value indicates the maximum packet length allowed.
    ELECHOUSE_cc1101.setCrc(0);             // 1 = CRC calculation in TX and CRC check in RX enabled. 0 = CRC disabled for TX and RX.
    ELECHOUSE_cc1101.setCRC_AF(0);          // Enable automatic flush of RX FIFO when CRC is not OK. This requires that only one packet is in the RXIFIFO and that packet length is limited to the RX FIFO size.
    ELECHOUSE_cc1101.setDcFilterOff(0);     // Disable digital DC blocking filter before demodulator. Only for data rates ≤ 250 kBaud The recommended IF frequency changes when the DC blocking is disabled. 1 = Disable (current optimized). 0 = Enable (better sensitivity).
    ELECHOUSE_cc1101.setManchester(0);      // Enables Manchester encoding/decoding. 0 = Disable. 1 = Enable.
    ELECHOUSE_cc1101.setFEC(0);             // Enable Forward Error Correction (FEC) with interleaving for packet payload (Only supported for fixed packet length mode. 0 = Disable. 1 = Enable.
    ELECHOUSE_cc1101.setPRE(0);             // Sets the minimum number of preamble bytes to be transmitted. Values: 0 : 2, 1 : 3, 2 : 4, 3 : 6, 4 : 8, 5 : 12, 6 : 16, 7 : 24
    ELECHOUSE_cc1101.setPQT(0);             // Preamble quality estimator threshold. The preamble quality estimator increases an internal counter by one each time a bit is received that is different from the previous bit, and decreases the counter by 8 each time a bit is received that is the same as the last bit. A threshold of 4∙PQT for this counter is used to gate sync word detection. When PQT=0 a sync word is always accepted.
    ELECHOUSE_cc1101.setAppendStatus(0);    // When enabled, two status bytes will be appended to the payload of the packet. The status bytes contain RSSI and LQI values, as well as CRC OK.

    Serial.println("Tx Mode");
}
void loop() {
ELECHOUSE_cc1101.SendData(transmitt_byte, 13, 120);
 Serial.println("Byte signal sent.");
delay(2000);
}

```

When transmitting that rtl_433 would not recognize it as Renault TPMS.
Using flex decoder : 
`>rtl_433  -X "n=name,m=FSK_PCM,s=48,l=48,r=49152"`
I get: 

```
time      : 2024-06-21 08:25:50
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 114          data      : 0daaaaaaa9123456784b00c81e45c
codes     : {114}0daaaaaaa9123456784b00c81e45c
```
Which is what I sent.
Bit bucket:

[image]

when I look at the[ test data for Renault TPMS ](https://triq.org/explorer/)it looks much different and I do not understand why? I can't spot the preamble. 
it looks like that:
[image]

Can anyone elaborate?

zuckschwerdt, 2024-06-21:
You are missing the Manchester coding. The example should [look like this](https://triq.org/pdv/#AAB0110701000000340068000C016C00F800008455+AAB05D0701000000340068000C016C00F800009191919191919191919191919191A19192A29191A2919191A191929191A192A191919291A291A1919191919291A2A2919191A1919191919191919191919191919191919192919191A1919295B055). I guess the explorer data is a bad decode (only the third packet in the example file is the TPMS data).

avicarmeli, 2024-06-22:
When I open the explorer data that is what I see:
gfile070.cu8:
[image]

gfile136.cu8:
[image]

gfile091.cu8:
[image]

gfile079.cu8:
[image]
 
So non of the data sets has actual data, as far as I understand. Or maybe it is some kind of bug or database mismatch in the explorer?

zuckschwerdt, 2024-06-22:
Explorer bug. I shows only the first packet, which is not data but noise. The files have multiple packets.

I'll see if I can fix or workaround that. Thanks for pointing at and example!

avicarmeli, 2024-06-22:
Here is some [recorded data ](https://drive.google.com/drive/folders/1KKbsOjeAd20V5xeI-bXDQ2Tar4bNxfQe?usp=drive_link) for Renault TPMS.
It looks like[ that](https://triq.org/pdv/#AAB1040000003000640108819191919191919191919191919191A19192A19291A291A291A19192A2919191A291A19291A291919191A2A291919191919191A192A2A2919191A2A1919192A19192A19191929355):
[image]

The decoder is able to decode it with the next result:
```
time      : @0.711584s
model     : Renault      type      : TPMS          id        : 031426
flags     : 36           pressure_kPa: 220.5 kPa   temperature_C: 33 C       Integrity : CRC
```
When I try to manually decode it I having hard time:
first I cant find the preamble.
second other than the flags nothing make sense.
What am I doing wrong?

zuckschwerdt, 2024-06-22:
Use PCM 50/50 in that pdv link. Copy the code of `{181}55555556A696596A655969655665555A66559AA6A6A900` to BitBench, align to preamble of `5556` then use MC decode. See [this BitBench](https://triq.net/bitbench#c=%7B181%7D%2055%2055%2055%2056%20A6%2096%2059%206A%2065%2059%2069%2065%2056%2065%2055%205A%2066%2055%209A%20A6%20A6%20A9%2000&f=F6b%20P10d%20T8d%20I%3C24h%20%3F8h%20%3F8h%20C8h&z=1&a=Preamble&m=5556&d=MC&cw=4).
Note that we didn't invert and thus didn't use `aaa9` as preamble. But we picked G.E.T. MC which does the same (vs IEEE MC).

avicarmeli, 2024-06-22:
Thank you very much for your patient, I will try to push it farther... 
Now I send what I copied wrongly from PVD: { 0x00 ,0x00 , 0x00 , 0xEC , 0x93 , 0xA1 , 0x32 , 0x0A ,0x01 , 0xA8 , 0x5E , 0xEF , 0x00};
Never the less RTL_433 decodes it as if it is legal Renault TPMS [data](https://triq.org/pdv/#AAB010040100000034006C002881939191A19255+AAB00B040100000034006C0028A255+AAB024040100000034006C00289191919191919191919191919191919191919191919191A1919255+AAB00C040100000034006C0028A19255+AAB00C040200000034006C002891A255+AAB00E040100000034006C002891A1919255+AAB00B040100000034006C0028A255+AAB00E040100000034006C0028919191A255+AAB00D040100000034006C002891A19255+AAB00C040100000034006C002891A255+AAB00F040100000034006C002891919191A255+AAB00B040100000034006C0028A255+AAB013040100000034006C002891919191919191A19255+AAB00B040200000034006C0028A255+AAB00E040100000034006C0028919191A255+AAB00E040100000034006C0028A191919255+AAB00D040100000034006C0028A1919255+AAB00E040100000034006C0028A191919255+AAB012040100000034006C002891919191919191A155).

As seen in [Bitbench](https://triq.net/bitbench#c=%7B181%7D%2055%2055%2055%2056%20A6%2096%2059%206A%2065%2059%2069%2065%2056%2065%2055%205A%2066%2055%209A%20A6%20A6%20A9%2000&f=F6b%20P10d%20T8d%20I%3C24h%20%3F8h%20%3F8h%20C8h&z=1&a=Preamble&m=5556&d=MC&cw=4).

How come there is a preamble where non sent?
I am puzzled.

zuckschwerdt, 2024-06-22:
The `0x00 ,0x00 , 0x00` in Manchester coding is `5555 5555 5555` ;)

zuckschwerdt, 2024-06-22:
Explorer bug is fixed now, all data for the samples will be shown.

avicarmeli, 2024-06-23:
OK now it works fine. I can simulate the Renault TPMS sensor with whatever data I choose. Thanks.

Code:

```
#include <ELECHOUSE_CC1101_SRC_DRV.h>

#define CC1101_FREQUENCY 433.92 // Frequency for TPMS

void setup() {
  Serial.begin(115200);
  Serial.println("Initializing CC1101...");

if (ELECHOUSE_cc1101.getCC1101()){      // Check the CC1101 Spi connection.
    Serial.println("Connection OK");
    }else{
    Serial.println("Connection Error");
    }
 
    ELECHOUSE_cc1101.Init();                // must be set to initialize the cc1101!
    ELECHOUSE_cc1101.setCCMode(1);          // set config for internal transmission mode.
    ELECHOUSE_cc1101.setModulation(0);      // set modulation mode. 0 = 2-FSK, 1 = GFSK, 2 = ASK/OOK, 3 = 4-FSK, 4 = MSK.
    ELECHOUSE_cc1101.setMHZ(CC1101_FREQUENCY);        // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
    ELECHOUSE_cc1101.setDeviation(47.60);   // Set the Frequency deviation in kHz. Value from 1.58 to 380.85. Default is 47.60 kHz.
    ELECHOUSE_cc1101.setChannel(0);         // Set the Channelnumber from 0 to 255. Default is cahnnel 0.
    ELECHOUSE_cc1101.setChsp(199.95);       // The channel spacing is multiplied by the channel number CHAN and added to the base frequency in kHz. Value from 25.39 to 405.45. Default is 199.95 kHz.
    ELECHOUSE_cc1101.setRxBW(812.50);       // Set the Receive Bandwidth in kHz. Value from 58.03 to 812.50. Default is 812.50 kHz.
    ELECHOUSE_cc1101.setDRate(18.5);       // Set the Data Rate in kBaud. Value from 0.02 to 1621.83. Default is 99.97 kBaud!
    ELECHOUSE_cc1101.setPA(10);             // Set TxPower. The following settings are possible depending on the frequency band.  (-30  -20  -15  -10  -6    0    5    7    10   11   12) Default is max!
    ELECHOUSE_cc1101.setSyncMode(0);        // Combined sync-word qualifier mode. 0 = No preamble/sync. 1 = 16 sync word bits detected. 2 = 16/16 sync word bits detected. 3 = 30/32 sync word bits detected. 4 = No preamble/sync, carrier-sense above threshold. 5 = 15/16 + carrier-sense above threshold. 6 = 16/16 + carrier-sense above threshold. 7 = 30/32 + carrier-sense above threshold.
    ELECHOUSE_cc1101.setSyncWord(211, 145); // Set sync word. Must be the same for the transmitter and receiver. (Syncword high, Syncword low)
    ELECHOUSE_cc1101.setAdrChk(0);          // Controls address check configuration of received packages. 0 = No address check. 1 = Address check, no broadcast. 2 = Address check and 0 (0x00) broadcast. 3 = Address check and 0 (0x00) and 255 (0xFF) broadcast.
    ELECHOUSE_cc1101.setAddr(0);            // Address used for packet filtration. Optional broadcast addresses are 0 (0x00) and 255 (0xFF).
    ELECHOUSE_cc1101.setWhiteData(0);       // Turn data whitening on / off. 0 = Whitening off. 1 = Whitening on.
    ELECHOUSE_cc1101.setPktFormat(0);       // Format of RX and TX data. 0 = Normal mode, use FIFOs for RX and TX. 1 = Synchronous serial mode, Data in on GDO0 and data out on either of the GDOx pins. 2 = Random TX mode; sends random data using PN9 generator. Used for test. Works as normal mode, setting 0 (00), in RX. 3 = Asynchronous serial mode, Data in on GDO0 and data out on either of the GDOx pins.
    ELECHOUSE_cc1101.setLengthConfig(1);    // 0 = Fixed packet length mode. 1 = Variable packet length mode. 2 = Infinite packet length mode. 3 = Reserved
    ELECHOUSE_cc1101.setPacketLength(0);    // Indicates the packet length when fixed packet length mode is enabled. If variable packet length mode is used, this value indicates the maximum packet length allowed.
    ELECHOUSE_cc1101.setCrc(0);             // 1 = CRC calculation in TX and CRC check in RX enabled. 0 = CRC disabled for TX and RX.
    ELECHOUSE_cc1101.setCRC_AF(0);          // Enable automatic flush of RX FIFO when CRC is not OK. This requires that only one packet is in the RXIFIFO and that packet length is limited to the RX FIFO size.
    ELECHOUSE_cc1101.setDcFilterOff(0);     // Disable digital DC blocking filter before demodulator. Only for data rates ≤ 250 kBaud The recommended IF frequency changes when the DC blocking is disabled. 1 = Disable (current optimized). 0 = Enable (better sensitivity).
    ELECHOUSE_cc1101.setManchester(0);      // Enables Manchester encoding/decoding. 0 = Disable. 1 = Enable.
    ELECHOUSE_cc1101.setFEC(0);             // Enable Forward Error Correction (FEC) with interleaving for packet payload (Only supported for fixed packet length mode. 0 = Disable. 1 = Enable.
    ELECHOUSE_cc1101.setPRE(0);             // Sets the minimum number of preamble bytes to be transmitted. Values: 0 : 2, 1 : 3, 2 : 4, 3 : 6, 4 : 8, 5 : 12, 6 : 16, 7 : 24
    ELECHOUSE_cc1101.setPQT(0);             // Preamble quality estimator threshold. The preamble quality estimator increases an internal counter by one each time a bit is received that is different from the previous bit, and decreases the counter by 8 each time a bit is received that is the same as the last bit. A threshold of 4∙PQT for this counter is used to gate sync word detection. When PQT=0 a sync word is always accepted.
    ELECHOUSE_cc1101.setAppendStatus(0);    // When enabled, two status bytes will be appended to the payload of the packet. The status bytes contain RSSI and LQI values, as well as CRC OK.

    Serial.println("Tx Mode");

[... cut ...]
  //ELECHOUSE_cc1101.Sidle();
  ELECHOUSE_cc1101.SendData(encodedData, sizeof(encodedData),100);
  Serial.println("TPMS signal sent.");

 

  delay(2000); // Wait 2 seconds before sending the next signal
}

```

The [hardware ](https://github.com/avicarmeli/TPMS-SDR/blob/main/Docs/ESP-SDR%20connection%20diagram.md)is pretty simple just esp32 devkit and cc1101 board connected only by SPI.

If you think it might help anyone I can generalize it to be used with rtl_433 decoders data to simulate any sensor.

zuckschwerdt, 2024-06-23:
Great to hear. Very nice! A TPMS simulator is sometimes requested. You should turn this into a full project and upload to a repo.
We should then link that in the wiki, e.g. at https://github.com/merbanan/rtl_433/wiki/Built-with-rtl_433!

avicarmeli, 2024-06-23:
Sure will do slowly. probably I will have to ask some while coding.

closed 2024-06-23 by avicarmeli