merbanan/rtl_433 #2980: Ecowitt WN32CP 433 Mhz

Jocala, 2024-06-22:
I'm trying to receive data from an Ecowitt WN32CP 433 Mhz Temp/Humidity/Pressure sensor. I don't find this sensor in any of the conf files.  It's definitely 433 Mhz, not 915 Mhz.  I'm running rtl_433 version 23.11 under debian.  Nothing from Ecowitt is decoded.  What now?

zuckschwerdt, 2024-06-22:
Exowitt might be a Fine Offset rebrand, but we only have the WN32B so far. Not sure how the CP might be different. Catch some samples and upload here as zip. https://triq.org/rtl_433/ANALYZE.html

Jocala, 2024-06-22:
after a compile from current git source, my device is now reported:

{"time" : "2024-06-22 13:17:13", "model" : "Fineoffset-WH32B", "id" : 173, "battery_ok" : 1, "temperature_F" : 81.140, "humidity" : 53, "pressure_inHg" : 29.958, "mic" : "CRC"}

Thank you!

closed 2024-06-22 by Jocala