merbanan/rtl_433 #2983: Dickert MAHS433-01 garage/gate remote decoding

klohner, 2024-06-26:
The Dickert MAHS433-01 remote contains a user-accessible bank of 10 dip switches labeled "1" to "10" and each tristate dip switch can be set to one of three positions.  These positions are labeled as "-" (down), "0" (half-way up), and "+" (up).  Based on the position of these switches, 59,049 (3^10) unique codes are possible.  There seems to be a model of this device "MAHS433-01" that has one button to trigger a repeating signal for the duration it is held, and there may be a "MAHS433-04" device with 4 buttons.

There's some photos and documentation on the [Dickert Electronic](https://dickert.com/de/mahs433-01-02004600.html) site.

This device was brought up in [this thread](https://groups.google.com/g/rtl_433/c/vouhWWwj-kE) and [this thread](https://groups.google.com/g/rtl_433/c/WwL8S3tE5gM/m/bJrHOyJxAQAJ) on the rtl_433 mailing list by a subscriber (`md`) and there was some interesting discussion to decode this signal.  Eventually, `md` created a draft of a decoder for this device for rtl_433, as well as an Arduino sketch using a CC1101 and the SmartRF library to send this signal. 

I'm opening this issue to capture some of the info discussed in that thread, as well as some of my own findings for this device to help refine an eventual rtl_433 decoder for this device.

While looking for more info on this device, I found [this StackExchange thread](https://electronics.stackexchange.com/questions/311355/clone-garage-door-opener)  that seems to be the same or a similar device with some nice photos.

I purchased a generic clone of this device from an AliExpress listing titled "433MHz For Dickert MAHS433 MAHS433-01 MAHS433-04 433 MHz Handheld Transmitter Pink Button LED Orange" which seems to be the correct device.

[image]

Note that there is also a version sold on AliExpress titled "DICKERT MAHS433-01 MAHS433-04 433mhz Remote Control Duplicator Fixed Code Copy DICKERT Gate Remote Control"  (note the "Fixed Code" in the description) which does **_not_** seem to send this signal -- instead it seems to be a generic EV1527 transmitter.

The signal itself is a bit unusual.  Logical bits each seem to be encoded over three symbols.  A logical "1" is encoded as "001" and a logical "0" is encoded as "011" which, although it looks like typical PWM, has each bit encoding starting with a ASK/OOK gap, then ending with the PWM pulse.  The start of the signal is a single "1" pulse symbol.

For example, here is `{36}f4f075515` (representing `++0-++--0+`) when encoded into this signal:
[image]

After decoding, there are 36 logical bits.  The first 20 are 10 sets of 2 bits encoding the state of the 10 tristate dip switches.  A "-" state is "00", a "0" state is "01" and a "+" state is "11".  "10" is never observed and seems to be invalid.  I would expect the final 16 bits to be some kind of per-device ID.  

Surprisingly, the final 16 bits on my device is 0x5515, which is identical to the final 16 bits reported on `md`'s device.  So maybe it's some static code to help identify the signal itself?  Is it the same on all copies of this device?

The manual states: "The sender works with 387 million internal coding permutations, of which 59049 can be freely set using the 10-pin tristate coding switch".  I assume "387 million" is somehow related to `(3^10 * 2^16) / 10 = 386,983,526` but I'm not exactly sure if or how that clears everything up.

- Here's a few .cu8 samples from this device: [dickert.zip](https://github.com/user-attachments/files/15982109/dickert.zip)

- Here's a [BitBench](https://triq.net/bitbench#c=%7B36%7Df4f075515%20%5B%2B%2B0-%2B%2B--0%2B%20on%20md's%20device%5D&c=&c=%7B36%7Df4f075515%20%5B%2B%2B0-%2B%2B--0%2B%20on%20my%20device%5D&c=%7B36%7D000005515%20%5B----------%5D&c=%7B36%7D400305515%20%5B0---------%5D&c=%7B36%7Dc00005515%20%5B%2B---------%5D&c=%7B36%7D000015515%20%5B---------0%5D&c=%7B36%7D000035515%20%5B---------%2B%5D&c=%7B36%7D000045515%20%5B--------0-%5D&c=%7B36%7D0000c5515%20%5B--------%2B-%5D&c=%7B36%7D000105515%20%5B-------0--%5D%20&c=%7B36%7D000305515%20%5B-------%2B--%5D%20&c=%7B36%7D555555515%20%5B0000000000%5D&c=%7B36%7Dfffff5515%20%5B%2B%2B%2B%2B%2B%2B%2B%2B%2B%2B%5D&f=2d2d2d2d2d2d2d2d2d2d%2016h&z=1&cw=4) of some values I generated.

- Here's a flex decoder that seems to return good results for me.

```
# Dickert_MAHS433
# Signal starts with a "1" symbol, then each trio of symbols is either "001" for a logical "1" bit, or "011" for a logical "0" bit.
# Signal encodes 36 logical bits.
# First 20 logical bits contain the user-selectable 10 digit "code", where each "digit" can be "-", "0", or "+".
# These "digits" are each encoded as two bits, "-" is 00, "0" is 01, and "+" is 11. A 10 bit pair is invalid in this section.
# The last 16 logical bits are always(?) 0x5515 (for all copies of this device?)
# Perhaps 0x5515 (decimal 21781) decoded at end of signal is a magic static code identifying this device?
# This was the case for my "433MHz For Dickert MAHS433 MAHS433-01 MAHS433-04 433 MHz Handheld Transmitter Pink Button LED Orange"
# purchased from AliExpress, and a device reported by "md" at https://groups.google.com/g/rtl_433/c/WwL8S3tE5gM/m/bJrHOyJxAQAJ
# Perhaps we can use this to reject any signal that does not contain 0x5515 encoded with this device's weird encoding. ({48}6596596d9659)
# # get = model:@20:{16}:[21781:MAHS433-01], # not necessary if using "match"
decoder {
    name        = Dickert_MAHS433,
    modulation  = OOK_PCM,
    short       = 414,
    long        = 414,
    reset       = 1242,
    bits       >= 109,
    bits       <= 116,
    match       = {48}6596596d9659,
    preamble    = {1}8,
    symbol_zero = {3}6,
    symbol_one  = {3}2,
    get         = _1_2:@0:{4}:[00:-- 01:-0 02:-X 03:-+ 04:0- 05:00 06:0X 07:0+ 08:X- 09:X0 10:XX 11:X+ 12:+- 13:+0 14:+X 15:++],
    get         = _3_4:@4:{4}:[00:-- 01:-0 02:-X 03:-+ 04:0- 05:00 06:0X 07:0+ 08:X- 09:X0 10:XX 11:X+ 12:+- 13:+0 14:+X 15:++],
    get         = _5_6:@8:{4}:[00:-- 01:-0 02:-X 03:-+ 04:0- 05:00 06:0X 07:0+ 08:X- 09:X0 10:XX 11:X+ 12:+- 13:+0 14:+X 15:++],
    get         = _7_8:@12:{4}:[00:-- 01:-0 02:-X 03:-+ 04:0- 05:00 06:0X 07:0+ 08:X- 09:X0 10:XX 11:X+ 12:+- 13:+0 14:+X 15:++],
    get         = _9_10:@16:{4}:[00:-- 01:-0 02:-X 03:-+ 04:0- 05:00 06:0X 07:0+ 08:X- 09:X0 10:XX 11:X+ 12:+- 13:+0 14:+X 15:++],
}
```

Does anyone else have this device and can test?

OevreFlataeker, 2024-06-26:
Thanks for documenting the journey so far. 
`Surprisingly, the final 16 bits on my device is 0x5515, which is identical to the final 16 bits reported on md's device. So maybe it's some static code to help identify the signal itself? Is it the same on all copies of this device?`

The 16 bit suffix is a factory code, which is the same for all devices of that particular batch. If you compare the MAHS433 to the other products like the S25 or S10 you see the article code is e.g. S10-433A1L00 whereas to the best of my knowledge the A1 stands for "one button model", whereas A4 denotes the "four button model", L is for "linear code" in contrast to "K" which denotes a Keeloq variant. Finally the 00 is the factory batch. I believe we (I also bought mine on Aliexpress) have the default batch 00 which also explains those last 2 bytes are identical for us. For larger orders Dickert can customize this code, making them incompatible for other senders out there.

It would be good to see if someone with a non Aliexpress sender sees the same suffix.
So it seems, at least the last 2 bytes can be checked against the prerequisite that they contain the same symbols like the first 20 bits, i.e. a "10" would be impossible it seems.

Please find attached the latest version of the working code. The Arduino is still heavy work in progress.

[dickert_mahs.zip](https://github.com/user-attachments/files/15983452/dickert_mahs.zip)
[dickert_mahs_ino.zip](https://github.com/user-attachments/files/15983538/dickert_mahs_ino.zip)

I am currently working on writing a decoder for the Flipper Zero SubGhz module.

If anyone has one of those S10 or S25 models I'd also be interested in some IQ raw recordings.

OevreFlataeker, 2025-07-04:
I've implemented by Decoder for rtl_433. Let me know if it is of general interest or too specific to be included. Happy to raise a PR.

https://github.com/OevreFlataeker/rtl_433/blob/master/src/devices/dickert_mahs.c

gdt, 2025-07-04:
My view is that if it was of enough interest to raise an issue, then it belongs in the sources - we're not running out of bytes in the repo, and the sources is where we store accumulated wisdom.

It does look like it lacks a CRC so should not be enabled by default.  It would be great for you to add the usual comment about what the bits mean.

I'm in favor of a PR.

OevreFlataeker, 2025-07-04:
Ok, then I´d got for Dickert-MAHS433 as you have suggested.
ID: Hm, I see... But basically you would need the full dip switch config in order to say "this one is identical to that one"... What about a CRC16 of the config? Would that be ok if I add this to the code? 

I've already opened the pull request. I am not sure: can I still add new commits or does the PR need to be canceled because the commits won´t be included then?

OevreFlataeker, 2025-07-07:
I think I have now included all your requests. I went for the initial proposal of the ID being an OR of some bit field values. Please find the PR here: https://github.com/merbanan/rtl_433/pull/3321

Please let me know if it's fine for you or you want something else changed.

zuckschwerdt, 2025-07-07:
Thanks, looks good. I think "id" can be `DATA_INT`, without the sprintf.

We might need to research and discuss if "dipswitches" and "facswitches" are good keys. We have other decoders with dip switches and trinary, we might want to have similar keys.