merbanan/rtl_433 #2985: Unknown Chinese TPMS external sensor support

cyberFIRM, 2024-06-26:
Hi,

First of all great work really appreciated. I had some unknown TPMS sensor in my garage for a lot of time. I tried running rtl-433 but it did not match with any of the present decoders. 

I tried debugging using URH to see the waveforms signals and tried to decode the data into Temp and Pressure but all in vain. Wondering if anyone can help.

**first pulse:** 
010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101011001100110100110101001011001011010101010100110100101011010100110100101100110010101010101100101011010100101010110011001010101011010100101010110101010011001

**nrz**

555555555555555555555555555555555545555555555555555555555555555555555599a6a596aa9a56a696655595a9566556a55a84c8

**manchester I**

ffffffffffffffffffffffffffffffffffca91b027135fb8f5f1e14

**second pulse**

010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101010101011001100110100110101001011001011010101010100110100101011010100110100101100110010101010101100101011010100101010110011001010101011010100101010110101010011001

**nrz**

555555555555555555555555555555555545555555555555555555555555555555555599a6a596aa9a56a696655595a9566556a55a84c8

**manchester I**

ffffffffffffffffffffffffffffffffffca91b027135fb8f5f1e14

Values monitored on LCD Display:
Temp: 42 C
Pressure: 25 PSI

zuckschwerdt, 2024-06-26:
That's an unexpectedly long preamble buthe MC with ~10 bytes looks right.
Try to capture a long sequence of codes while changing a value slowly. E.g. let the air out slowly (the sensor should notice and start to send codes rapidly).
With usual timings you should be able to grab codes with `rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=50,l=50,g=150,r=900,preamble=5556'

cyberFIRM, 2024-06-26:
Thank you for quick reply. I will work on the suggestion provided above.

Meanwhile please kindly take a look at the logs attached below:
[tpms_json.txt](https://github.com/user-attachments/files/15989346/tpms_json.txt)

I am using ESP32 Dev Module with CC1101 using library [RTL_433_ESP ](https://github.com/NorthernMan54/rtl_433_ESP).

klohner, 2024-06-26:
I pulled out what looked like the unique MC values in that `tpms_json.txt` file and put into [BitBench](https://triq.net/bitbench#c=%7B437%7Daaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab334d4b2d5534ad4d2ccaab2cb4ab4d2d2cb4cd11708%20%5B2024-06-25%2017%3A53%3A02.289%5D%20&c=%7B432%7Daaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab334d4b2d5534ad4d2ccaab2cb4ab4d2d2cb4cd6f55%20%5B2024-06-25%2017%3A53%3A02.931%5D%20&c=%7B417%7D55555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a5995565969569a5a59699a50%20%5B2024-06-25%2017%3A53%3A03.599%5D%20&c=%7B413%7D5555555555555555555555555555555555555555aaaaaaaaaaaaaaaaaaaaaaaaccd352cb554d56a6966555965995a956956a5a50%20%5B2024-06-25%2017%3A53%3A04.524%5D%20&c=%7B425%7Daaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab5555555555555555555555555556669a965aaa695a9a599556596656a55a55a9698%20%5B2024-06-25%2017%3A53%3A05.224%5D%20&c=%7B425%7Dd5556aaaaaaaaaaaaaaaaaaaaaaaaaab55555555aaaaaaaaaaaaaaaaaaaaaaaaaab334d4b2d5534ad4d2ceaab2cb32b52ad2ad4b498%20%5B2024-06-25%2017%3A53%3A05.952%5D%20&c=%7B429%7D55555555555555555555555555555556aaaaaaaaaab555555555555555555555555ccd352cb554d2b534acaab2cb4ab52ad2ad532e78%20%5B2024-06-25%2017%3A53%3A06.755%5D%20&c=%7B428%7Daaaaaaaaaaaaaaaaaaaaaa95555aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab5555aaab334d4b2d54d2b66966555965a55a956956a99541%20%5B2024-06-25%2017%3A53%3A07.550%5D%20&c=%7B421%7Daaaaaaaaaaaaab5555555555555555555555155555555555555556aaaaaaaaaaaccd352ad5534ad4d2ccaab2cb4ab52ad2ad532ca8%20%5B2024-06-25%2017%3A53%3A08.325%5D%20&c=%7B432%7D555555556aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab334d4b2d5534ad4d2ccaab2cb4ab52ad2ad532e53%20%5B2024-06-25%2017%3A53%3A09.064%5D%20&c=%7B431%7D5555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a599556596956a55a55aa65b26%20%5B2024-06-25%2017%3A53%3A09.818%5D%20&c=%7B417%7D2aaaaaaaaaaaaaaaaaaaaaaaaaaaaaab5555555555555155555555555555555556669a965aaa6cad4d599556596956a55a55aa650%20%5B2024-06-25%2017%3A53%3A10.493%5D%20&c=%7B427%7D55556aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaad55555555555555555555555556669a9cb554d2b534b32aacab4ab52ad2ad532966%20%5B2024-06-25%2017%3A53%3A11.217%5D%20&c=%7B409%7Dd5555555555555555555555555555555555555555555555555555555555aaaaaaccd352cb554d2b52599556596956a55a56a990%20%5B2024-06-25%2017%3A53%3A11.789%5D%20&c=%7B410%7Deaaaaaaaaaaaab555555556aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaccd352cb5534ad4d332ab965a55a956956d32c8%20%5B2024-06-25%2017%3A53%3A12.460%5D%20&c=%7B427%7D6aaaaaaab5555555555555555555555555555555ad555555555555555555555559ba6a596aa9a56a6966555965a55aa56a566a9a7e4%20%5B2024-06-25%2017%3A53%3A36.997%5D%20&c=%7B405%7Daaaaaaaaaaaaa2aaaaaaab55155555555555555555555555555555555555555666b52cb554d2b534b32ab2cb4ab54ad4ab3548%20%5B2024-06-25%2017%3A53%3A37.583%5D&f=v&z=1&a=Preamble&m=%7B14%7D5566&d=MC&cw=4) for easier initial review.  It looks like there's some bit errors, but there might be a few rows that decoded enough to get an idea of what's in there.

zuckschwerdt, 2024-06-26:
Shifting off 3 bits the values look a bit like a progression and the last byte could be a CRC. We will need more good codes.

cyberFIRM, 2024-06-27:
Attached are some more codes as requested. 
With usual timings, codes grabbed with `rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=50,l=50,g=150,r=900,preamble=5556'
[rtl-433-R0-X.txt](https://github.com/user-attachments/files/16011687/rtl-433-R0-X.txt)

Using URH, recorded the waveforms:
[RTL-SDR-20240627_143405-433_92MHz-1MSps-1MHz](https://drive.google.com/file/d/1QpQqq-id1dfM9P9A-417C2Cddi2HSX4Y/view?usp=sharing)

_Comments on waveforms:_
First 3 waveforms:   temp 35 C and pressure 21 PSI
Waveforms 4 to 8:   temp 36 C and pressure 21 PSI
Waveforms in rapid succession (slowly air leakage) : temp 36 C and pressure dropped from 20 to 18 PSI with passage of time  
Last couple of waveforms: temp 36 C and pressure 18 PSI

zuckschwerdt, 2024-06-27:
Note on the recorded IQ file: the level of around -24 dB is dangerously close to silent.
You need to rename the file to include the proper meta data: `…_433.92M_1000k.cs8` then looking at the noise levels you can preset a very low minimum level and use autolevel: `-Y minlevel=-45 -Y autolevel -Y magest -M level -M noise`

Here is [a BitBench of the data](https://triq.net/bitbench#c=%5B29%20C%2024%20PSI%5D&c=%7B153%7D669a965aaa695a9a599556569556a65a5956958%5B14%3A01%3A52%5D&c=%7B153%7D669a965aaa695a9a599556569556a65a5956950%5B14%3A01%3A52%5D&c=%7B153%7D669a965aaa695a9a599556569556a65a5956950%5B14%3A01%3A52%5D&c=%7B153%7D669a965aaa695a9a599556569556a55a5599aa8%5B14%3A01%3A54%5D&c=%7B153%7D669a965aaa695a9a599556569556a55a5599aa0%5B14%3A01%3A54%5D&c=%7B153%7D669a965aaa695a9a599556569556a55a5599aa0%5B14%3A01%3A54%5D&c=%7B153%7D669a965aaa695a9a599556569656a65a5565958%5B14%3A01%3A57%5D&c=%7B153%7D669a965aaa695a9a599556569656a65a5565958%5B14%3A01%3A57%5D&c=%7B153%7D669a965aaa695a9a599556569656a65a5565958%5B14%3A01%3A57%5D&c=%7B153%7D669a965aaa695a9a599556569656a65a5565958%5B14%3A01%3A59%5D&c=%7B154%7D669a965aaa695a9a599556569656a65a5565958%5B14%3A01%3A59%5D&c=%7B153%7D669a965aaa695a9a599556569656a65a5565958%5B14%3A01%3A59%5D&c=%5B35%20C%2024%20PSI%5D&c=%7B153%7D669a965aaa695a9a599556569556a65a5565698%5B14%3A02%3A01%5D&c=%7B153%7D669a965aaa695a9a599556569556a65a5565698%5B14%3A02%3A01%5D&c=%7B153%7D669a965aaa695a9a599556569556a65a5565690%5B14%3A02%3A01%5D&c=%7B153%7D669a965aaa695a9a599556569556a95a9555660%5B14%3A02%3A27%5D&c=%7B153%7D669a965aaa695a9a599556569556a95a9555668%5B14%3A02%3A27%5D&c=%7B153%7D669a965aaa695a9a599556569559555a95a56a0%5B14%3A03%3A31%5D&c=%7B153%7D669a965aaa695a9a599556569559555a95a56a8%5B14%3A03%3A31%5D&c=%7B153%7D669a965aaa695a9a599556569659565a9559558%5B14%3A04%3A35%5D&c=%7B153%7D669a965aaa695a9a599556569659565a9559558%5B14%3A04%3A35%5D&c=%7B153%7D669a965aaa695a9a599556569559595a955a998%5B14%3A05%3A40%5D&c=%7B153%7D669a965aaa695a9a599556569559595a955a998%5B14%3A05%3A40%5D&c=%7B153%7D669a965aaa695a9a599556569659595a955a658%5B14%3A06%3A44%5D&c=%7B154%7D669a965aaa695a9a599556569659595a955a658%5B14%3A06%3A44%5D&c=%7B153%7D669a965aaa695a9a5995565695595a5a95a65a0%5B14%3A07%3A48%5D&c=%7B153%7D669a965aaa695a9a5995565695595a5a95a65a8%5B14%3A07%3A48%5D&c=%7B153%7D669a965aaa695a9a5995565695595a5a95a65a0%5B14%3A08%3A53%5D&c=%7B154%7D669a965aaa695a9a5995565695595a5a95a65a4%5B14%3A08%3A53%5D&c=%7B153%7D669a965aaa695a9a5995565695595a5a95a65a0%5B14%3A09%3A57%5D&c=%7B153%7D669a965aaa695a9a5995565695595a5a95a65a8%5B14%3A09%3A57%5D&c=%7B153%7D669a965aaa695a9a5995565695595a5a95a65a0%5B14%3A11%3A01%5D&c=%7B153%7D669a965aaa695a9a5995565695595a5a95a65a0%5B14%3A11%3A01%5D&c=%5BLetting%20Air%20out%20slowly%5D&c=%5B36%20C%2022%2C%2021%2C%2020%20PSI%5D&c=%7B153%7D669a965aaa695a9a599556565a59655a956a558%5B14%3A12%3A06%5D&c=%7B153%7D669a965aaa695a9a599556565a59655a956a558%5B14%3A12%3A06%5D&c=%7B153%7D669a965aaa695a9a599556555959655a656a660%5B14%3A12%3A11%5D&c=%7B153%7D669a965aaa695a9a599556555959655a656a660%5B14%3A12%3A11%5D&c=%7B153%7D669a965aaa695a9a599556555959655a55a5960%5B14%3A12%3A17%5D&c=%7B153%7D669a965aaa695a9a599556555959655a55a5960%5B14%3A12%3A17%5D&c=%7B132%7Da9655a55a956aa996aa95a5aaa6a99a54%5B14%3A12%3A18%5D&c=%7B132%7Da9655a55a956aa996aa95a5aaa6a6695b%5B14%3A12%3A19%5D&c=%7B132%7Da9655a55a956aa996aa95a5aaa696566b%5B14%3A12%3A19%5D&c=%7B153%7D669a965aaa695a9a5995565555595a5a5556650%5B14%3A12%3A22%5D&c=%7B153%7D669a965aaa695a9a5995565555595a5a5556650%5B14%3A12%3A22%5D&c=%7B153%7D669a965aaa695a9a595556aaaa595a5a559aa58%5B14%3A12%3A27%5D&c=%7B153%7D669a965aaa695a9a595556aaaa595a5a559aa58%5B14%3A12%3A27%5D&c=%7B153%7D669a965aaa695a9a595556aaa959655a556a5a0%5B14%3A12%3A33%5D&c=%7B153%7D669a965aaa695a9a595556aaa959655a556a5a0%5B14%3A12%3A33%5D&c=%7B153%7D669a965aaa695a9a595556aaa659655a5569560%5B14%3A12%3A38%5D&c=%7B153%7D669a965aaa695a9a595556aaa659655a5569560%5B14%3A12%3A38%5D&c=%7B153%7D669a965aaa695a9a595556aaa5595a5a5599a90%5B14%3A12%3A44%5D&c=%7B153%7D669a965aaa695a9a595556aaa5595a5a5599a98%5B14%3A12%3A44%5D&c=%7B153%7D669a965aaa695a9a595556aaa5595a5a5599a90%5B14%3A12%3A49%5D&c=%7B153%7D669a965aaa695a9a595556aaa5595a5a5599a90%5B14%3A12%3A49%5D&c=%7B153%7D669a965aaa695a9a595556aa9a59655a5565660%5B14%3A12%3A55%5D&c=%7B153%7D669a965aaa695a9a595556aa9a59655a5565660%5B14%3A12%3A55%5D&c=%7B153%7D669a965aaa695a9a595556aa9a59655a5565660%5B14%3A13%3A00%5D&c=%7B153%7D669a965aaa695a9a595556aa9a59655a5565660%5B14%3A13%3A00%5D&c=%7B153%7D669a965aaa695a9a595556aa99595a5a5595998%5B14%3A13%3A05%5D&c=%7B153%7D669a965aaa695a9a595556aa99595a5a5595998%5B14%3A13%3A05%5D&c=%7B153%7D669a965aaa695a9a595556aa9659655a5566968%5B14%3A13%3A11%5D&c=%7B153%7D669a965aaa695a9a595556aa9659655a5566968%5B14%3A13%3A11%5D&c=%7B153%7D669a965aaa695a9a595556aa95595a5a5596698%5B14%3A13%3A16%5D&c=%7B153%7D669a965aaa695a9a595556aa95595a5a5596698%5B14%3A13%3A16%5D&c=%7B153%7D669a965aaa695a9a595556aa6a59655a5555a68%5B14%3A13%3A22%5D&c=%7B153%7D669a965aaa695a9a595556aa6a59655a5555a60%5B14%3A13%3A22%5D&c=%7B132%7Da9655a55a956aa9959955a9655965aa9b%5B14%3A13%3A26%5D&c=%7B132%7Da9655a55a956aa9959955a96559aaa554%5B14%3A13%3A26%5D&c=%7B132%7Da9655a55a956aa9959955a96559a5565b%5B14%3A13%3A26%5D&c=%7B153%7D669a965aaa695a9a595556aa6959655a55555a0%5B14%3A13%3A27%5D&c=%7B153%7D669a965aaa695a9a595556aa6959655a55555a8%5B14%3A13%3A27%5D&c=%7B153%7D669a965aaa695a9a595556aa5a59655a69a66a0%5B14%3A13%3A49%5D&c=%7B153%7D669a965aaa695a9a595556aa55595a5a6955650%5B1 [... cut ...] 7B153%7D669a965aaa695a9a595556aa96595a5a5596958%5B14%3A15%3A34%5D&c=%7B153%7D669a965aaa695a9a595556aa96595a5a696a998%5B14%3A15%3A43%5D&c=%5BStopped%20air%20leakage%5D&c=%5B35%20C%2021%20PSI%5D&c=%7B153%7D669a965aaa695a9a595556aa9659655a699a9a0%5B14%3A16%3A04%5D&c=%7B153%7D669a965aaa695a9a595556aa9659655a699a9a0%5B14%3A16%3A26%5D&c=%7B153%7D669a965aaa695a9a595556aa96595a5a696a998%5B14%3A16%3A47%5D&c=%7B153%7D669a965aaa695a9a595556aa9659655a699a9a0%5B14%3A17%3A09%5D&f=8h&z=1&d=MC&cw=4) you posted.

zuckschwerdt, 2024-06-27:
Here is [a BitBench](https://triq.net/bitbench#c=%5B29%20C%2024%20PSI%5D&c=669a965aaa695a9a599556569556a65a595695%5B14%3A01%3A52%5D&c=669a965aaa695a9a599556569556a55a5599aa%5B14%3A01%3A54%5D&c=669a965aaa695a9a599556569656a65a556595%5B14%3A01%3A57%5D&c=669a965aaa695a9a599556569656a65a556595%5B14%3A01%3A59%5D&c=%5B35%20C%2024%20PSI%5D&c=669a965aaa695a9a599556569556a65a556569%5B14%3A02%3A01%5D&c=669a965aaa695a9a599556569556a95a955566%5B14%3A02%3A27%5D&c=669a965aaa695a9a599556569559555a95a56a%5B14%3A03%3A31%5D&c=669a965aaa695a9a599556569659565a955955%5B14%3A04%3A35%5D&c=669a965aaa695a9a599556569559595a955a99%5B14%3A05%3A40%5D&c=669a965aaa695a9a599556569659595a955a65%5B14%3A06%3A44%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A07%3A48%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A08%3A53%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A09%3A57%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A11%3A01%5D&c=%5BLetting%20Air%20out%20slowly%5D&c=%5B36%20C%2022%2C%2021%2C%2020%20PSI%5D&c=669a965aaa695a9a599556565a59655a956a55%5B14%3A12%3A06%5D&c=669a965aaa695a9a599556555959655a656a66%5B14%3A12%3A11%5D&c=669a965aaa695a9a599556555959655a55a596%5B14%3A12%3A17%5D&c=%7B132%7Da9655a55a956aa996aa95a5aaa6a99a54%5B14%3A12%3A18%5D&c=%7B132%7Da9655a55a956aa996aa95a5aaa6a6695b%5B14%3A12%3A19%5D&c=%7B132%7Da9655a55a956aa996aa95a5aaa696566b%5B14%3A12%3A19%5D&c=669a965aaa695a9a5995565555595a5a555665%5B14%3A12%3A22%5D&c=669a965aaa695a9a595556aaaa595a5a559aa5%5B14%3A12%3A27%5D&c=669a965aaa695a9a595556aaa959655a556a5a%5B14%3A12%3A33%5D&c=669a965aaa695a9a595556aaa659655a556956%5B14%3A12%3A38%5D&c=669a965aaa695a9a595556aaa5595a5a5599a9%5B14%3A12%3A44%5D&c=669a965aaa695a9a595556aaa5595a5a5599a9%5B14%3A12%3A49%5D&c=669a965aaa695a9a595556aa9a59655a556566%5B14%3A12%3A55%5D&c=669a965aaa695a9a595556aa9a59655a556566%5B14%3A13%3A00%5D&c=669a965aaa695a9a595556aa99595a5a559599%5B14%3A13%3A05%5D&c=669a965aaa695a9a595556aa9659655a556696%5B14%3A13%3A11%5D&c=669a965aaa695a9a595556aa95595a5a559669%5B14%3A13%3A16%5D&c=669a965aaa695a9a595556aa6a59655a5555a6%5B14%3A13%3A22%5D&c=%7B132%7Da9655a55a956aa9959955a9655965aa9b%5B14%3A13%3A26%5D&c=%7B132%7Da9655a55a956aa9959955a96559aaa554%5B14%3A13%3A26%5D&c=%7B132%7Da9655a55a956aa9959955a96559a5565b%5B14%3A13%3A26%5D&c=669a965aaa695a9a595556aa6959655a55555a%5B14%3A13%3A27%5D&c=669a965aaa695a9a595556aa5a59655a69a66a%5B14%3A13%3A49%5D&c=669a965aaa695a9a595556aa55595a5a695565%5B14%3A14%3A10%5D&c=669a965aaa695a9a595556a9a659655a69695a%5B14%3A14%3A32%5D&c=%7B132%7Da9655a55a956aa9959a95a96559956a5a%5B14%3A14%3A33%5D&c=%7B132%7Da9655a55a956aa9959a95a965599a9954%5B14%3A14%3A33%5D&c=%7B132%7Da9655a55a956aa9959a95a9655aa95564%5B14%3A14%3A33%5D&c=669a965aaa695a9a595556a999595a5a699595%5B14%3A14%3A53%5D&c=669a965aaa695a9a595556a96a595a5a69a5a9%5B14%3A15%3A14%5D&c=669a965aaa695a9a595556aa95595a5a59a595%5B14%3A15%3A25%5D&c=669a965aaa695a9a595556aa96595a5a559695%5B14%3A15%3A27%5D&c=669a965aaa695a9a595556aa95595a5a559669%5B14%3A15%3A30%5D&c=669a965aaa695a9a595556aa96595a5a559695%5B14%3A15%3A32%5D&c=669a965aaa695a9a595556aa96595a5a559695%5B14%3A15%3A34%5D&c=669a965aaa695a9a595556aa96595a5a696a99%5B14%3A15%3A43%5D&c=%5BStopped%20air%20leakage%5D&c=%5B35%20C%2021%20PSI%5D&c=669a965aaa695a9a595556aa9659655a699a9a%5B14%3A16%3A04%5D&c=669a965aaa695a9a595556aa9659655a699a9a%5B14%3A16%3A26%5D&c=669a965aaa695a9a595556aa96595a5a696a99%5B14%3A16%3A47%5D&c=669a965aaa695a9a595556aa9659655a699a9a%5B14%3A17%3A09%5D&f=4h%20ID%3F8h8h8h8h%20PRES%3F4h12h%20TEMP%3F8d8h%20CRC%3F8h&z=1&d=MC&cw=4) with the data cleaned up, deduped and with some wild guesses.

There is a CRC-8, poly 0x2f, init 0xaa that checks out for all packets. And 2f poly is one of the AUTOSAR named ones.

cyberFIRM, 2024-06-27:
looks interesting. Been working on decoding pressure data. Any hints on that one?

data doesn't seem to be in kPa...

zuckschwerdt, 2024-06-27:
The most puzzling thing with the suspected pressure data is the placement of the upper bits. It looks like it's counting down nicely but above 1ff it should be 200 but we see 8100
[image]

The byte between temp and crc could be status/flags.

ProfBoc75, 2024-06-28:
@zuckschwerdt @cyberFIRM:

Pressure sounds to be 9 bits, PSI, offset 40, scale 10. So value can be from 0 PSI to 47.1 PSI ( 0 kPa to 324.7 kPa)

like this [bitbench](https://triq.net/bitbench#c=%5B29%20C%2024%20PSI%5D&c=669a965aaa695a9a599556569556a65a595695%5B14%3A01%3A52%20%23%2024%20PSI%2029%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569556a55a5599aa%5B14%3A01%3A54%20%23%2024%20PSI%2028%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569656a65a556595%5B14%3A01%3A57%20%23%2024.1%20PSI%2029%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569656a65a556595%5B14%3A01%3A59%20%23%2024.1%20PSI%2029%20%C2%B0C%20%5D&c=%5B35%20C%2024%20PSI%5D&c=669a965aaa695a9a599556569556a65a556569%5B14%3A02%3A01%20%23%2024%20PSI%2029%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569556a95a955566%5B14%3A02%3A27%20%23%2024%20PSI%2030%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569559555a95a56a%5B14%3A03%3A31%20%23%2024%20PSI%2032%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569659565a955955%5B14%3A04%3A35%20%23%2024.1%20PSI%2033%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569559595a955a99%5B14%3A05%3A40%20%23%2024%20PSI%2034%20%C2%B0C%20%5D&c=669a965aaa695a9a599556569659595a955a65%5B14%3A06%3A44%20%23%2024.1%20PSI%2034%20%C2%B0C%20%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A07%3A48%20%23%2024%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A08%3A53%20%23%2024%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A09%3A57%20%23%2024%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a5995565695595a5a95a65a%5B14%3A11%3A01%20%23%2024%20PSI%2035%20%C2%B0C%20%5D&c=%5BLetting%20Air%20out%20slowly%5D&c=%5B36%20C%2022%2C%2021%2C%2020%20PSI%5D&c=669a965aaa695a9a599556565a59655a956a55%5B14%3A12%3A06%20%23%2023.5%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a599556555959655a656a66%5B14%3A12%3A11%20%23%2021.8%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a599556555959655a55a596%5B14%3A12%3A17%20%23%2021.8%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a5995565555595a5a555665%5B14%3A12%3A22%20%23%2021.6%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aaaa595a5a559aa5%5B14%3A12%3A27%20%23%2021.5%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aaa959655a556a5a%5B14%3A12%3A33%20%23%2021.4%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aaa659655a556956%5B14%3A12%3A38%20%23%2021.3%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aaa5595a5a5599a9%5B14%3A12%3A44%20%23%2021.2%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aaa5595a5a5599a9%5B14%3A12%3A49%20%23%2021.2%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa9a59655a556566%5B14%3A12%3A55%20%23%2021.1%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa9a59655a556566%5B14%3A13%3A00%20%23%2021.1%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa99595a5a559599%5B14%3A13%3A05%20%23%2021%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa9659655a556696%5B14%3A13%3A11%20%23%2020.9%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa95595a5a559669%5B14%3A13%3A16%20%23%2020.8%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa6a59655a5555a6%5B14%3A13%3A22%20%23%2020.7%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa6959655a55555a%5B14%3A13%3A27%20%23%2020.6%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa5a59655a69a66a%5B14%3A13%3A49%20%23%2020.3%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa55595a5a695565%5B14%3A14%3A10%20%23%2020%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556a9a659655a69695a%5B14%3A14%3A32%20%23%2019.7%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556a999595a5a699595%5B14%3A14%3A53%20%23%2019.4%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556a96a595a5a69a5a9%5B14%3A15%3A14%20%23%2019.1%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa95595a5a59a595%5B14%3A15%3A25%20%23%2020.8%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa96595a5a559695%5B14%3A15%3A27%20%23%2020.9%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa95595a5a559669%5B14%3A15%3A30%20%23%2020.8%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa96595a5a559695%5B14%3A15%3A32%20%23%2020.9%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa96595a5a559695%5B14%3A15%3A34%20%23%2020.9%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa96595a5a696a99%5B14%3A15%3A43%20%23%2020.9%20PSI%2035%20%C2%B0C%20%5D&c=%5BStopped%20air%20leakage%5D&c=%5B35%20C%2021%20PSI%5D&c=669a965aaa695a9a595556aa9659655a699a9a%5B14%3A16%3A04%20%23%2020.9%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa9659655a699a9a%5B14%3A16%3A26%20%23%2020.9%20PSI%2036%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa96595a5a696a99%5B14%3A16%3A47%20%23%2020.9%20PSI%2035%20%C2%B0C%20%5D&c=669a965aaa695a9a595556aa9659655a699a9a%5B14%3A17%3A09%20%23%2020.9%20PSI%2036%20%C2%B0C%20%5D&f=4h%20ID%3F8h8h8h8h%20MSB_PRESS%201b%20FLAGS%3A7b%20LSB_PRESS%208d%20TEMP%208d%20FLAGS%3F%208b%20CRC%3F8h&z=1&d=MC&cw=4)

I removed some data with the {132} where bits are missing and shifted.

ProfBoc75, 2024-06-29:
I reviewed the rf decode part, it works fine a with FSK_MC_ZEROBIT , pulse is 54 µs and preamble = 0xfffff5, try the sample with:

    rtl_433 -X "n=Chinese_TPMS,m=FSK_MC_ZEROBIT,s=54,l=0,r=200,preamble=fffff5" -Y autolevel -Y magest -Y minlevel=-45 20240627_433.92M_1000k.cs8

We have packet payload: 9 bytes.

    Byte Position     0  1  2  3  4   5  6  7  8
    Sample      {72} b9 3f 63 b2 01  f9 24 36 bb
    Layout           II II II II MF1 LP TT F2 CC

- II {32} Sensor ID
- M   {1} MSB Pressure,
-  F  {7} Flags1, unknown ? always 0x1
- LP  {8} LSB Pressure, --> Pressure {9} MSB | LSB, offset 40, scale 10, PSI
- TT  {8} Temperature °C
- F2  {8} Flags 2, unknown ? upper nibble always 0x3, lower nibble is variable
- CC  {8} CRC, poly 0x2f, init 0xaa.

I drafted a decoder based on these above findings, ready to PR, same it works with the sample.

FYI: I checked with -A and the pulse width is more 54 µs than 50 µs.

klohner, 2024-06-30:
URH sees a whole lot of data in that same sample file.  I find that the symbol width is almost exactly 50 µs.

Some of the data seems to be from another device.  Here is what I found with URH:

[BitBench](https://triq.net/bitbench#c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659655a699a9a%20%5BPause%3A%2026760770%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659655a699a9a%20%5BPause%3A%2021446139%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659655a699a9a%20%5BPause%3A%2021446365%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659655a699a9a%20%5BPause%3A%2021446133%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659655a699a9a%20%5BPause%3A%2020058675%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659655a699a9a%20%5BPause%3A%2019371058%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659665a696659%20%5BPause%3A%2033853430%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa9659655a699a9a%20%5BPause%3A%2019611628%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa5959655a65956a%20%5BPause%3A%204093%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa5959655a65956a%20%5BPause%3A%204931967%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa5959655a555a9a%20%5BPause%3A%204092%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa5959655a555a9a%20%5BPause%3A%204896763%20samples%5D&c=55555554aaad52cab4ab52ad5532b2d4b52cab54aab28%20%5BPause%3A%2085490%20samples%5D&c=555554aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab334d4b2d5534ad4d2caaab552aacb2ad2aacb50%20%5BPause%3A%204094%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556aa5559655a55596a%20%5BPause%3A%2043768%20samples%5D&c=55555554aaad52cab4ab52ad5532b2d4b52cab52ad548%20%5BPause%3A%20201812%20samples%5D&c=55555554aaad52cab4ab52ad5532b2d4b52cab5353354%20%5BPause%3A%204669007%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9aa59655a5596a6%20%5BPause%3A%204092%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9aa59655a5596a6%20%5BPause%3A%204915087%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a959655a55965a%20%5BPause%3A%204093%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a959655a55965a%20%5BPause%3A%204980543%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a659655a559556%20%5BPause%3A%204093%20samples%5D&c=55555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a659655a559556%20%5BPause%3A%204935845%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a559655a5595aa%20%5BPause%3A%204094%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a559655a5595aa%20%5BPause%3A%204894150%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a559655a5595aa%20%5BPause%3A%204093%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a9a559655a5595aa%20%5BPause%3A%204939732%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a99959655a55999a%20%5BPause%3A%204093%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a99959655a668%20%5BPause%3A%204981543%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a99959655a55999a%20%5BPause%3A%204098%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a99959655a55999a%20%5BPause%3A%204873627%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a99559655a559a6a%20%5BPause%3A%204092%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a99559655a559a6a%20%5BPause%3A%204833503%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595752cb2ad2ad4d30%20%5BPause%3A%204093%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a96a59655a55a9a6%20%5BPause%3A%205047243%20samples%5D&c=5555555555555555555555555555555555555555555555555555555555555555555556669a965aaa695a9a595556a96a59655a55a9a6%20%5BPause%3A%204093%20samples%5D&c=55555555555555555555555555555aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaccd352cb554d2b534b2aaad52d4b2cab4ab534c%20%5BPause%3A%205030686%20samples%5D&c=aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa [... cut ...] t in the decode and it slips into the real decoding.  This is why FSK_MC_ZEROBIT finds a preamble of "fffff5", when the decoded MC data is actually "000015"

I had similar success using this FSK_PCM flex spec to find the same small subset of the data in that sample file.
```
rtl_433 -X "n=Chinese_TPMS,m=FSK_PCM,s=50,l=50,r=200,preamble=5556,symbol_zero={2}4,symbol_one={2}8" -Y autolevel -Y magest -Y minlevel=-45 RTL-SDR-20240627_143405-433_92MHz-1MSps-1MHz.complex16s
{76}5b93f63b201f92436bb

```

ProfBoc75, 2024-07-01:
@cyberFIRM: Yes, from the 9 bits we have the pressure_raw.

Pressure_PSI = ( pressure_raw - 40 ) * 0.1f   ( same as (pressure_raw - 40 ) / 10  or (pressure*0.1f) - 4)  

And the reverse, to get the 9 bits value,  (pressure_PSI * 10 ) + 40 = pressure_raw.

cyberFIRM, 2024-07-02:
[tpms_chinese.zip](https://github.com/user-attachments/files/16069773/tpms_chinese.zip)
 drafted the decoder based on above finding by @ProfBoc75 . Using [RTL_433_ESP](https://github.com/NorthernMan54/rtl_433_ESP/tree/main) library, I was able to get the codes after some tuning of CC1101.

`pulse_slicer_manchester_zerobit: Unbranded Chinese TPMS codes [{214}7ffffffffffffffffffffffffffffffffd6e4fd8ec80628e1c2ec0]`
`pulse_slicer_manchester_zerobit: Unbranded Chinese TPMS codes [{209}000000000001ffffffffffffffffffffeb727ec764037470e1758]`

Stuck on decoding part... in decoding section there is nothing to process. For more clarification: 
[#line547](https://github.com/NorthernMan54/rtl_433_ESP/blob/b8ea0e2d8bada6c6cde31640481c80fe674294a2/src/signalDecoder.cpp#L546)

klohner, 2024-07-02:
I'd like to suggest that we switch to a non-nation-based name for this device.  Is there any kind of identifying markings on the device we could use?  Or perhaps something like Unbranded-TPMS-202406?

gdt, 2024-07-02:
Perhaps those who have the device could try to describe everything about it that has any kind of markings as well as packaging, and how they bought it.   From context I am guessing this is something labeled tpms from aliexpress, without better information.

klohner, 2024-07-03:
Sounds reasonable to me.

"[Kylin](https://archive.shine.cn/sunday/now-and-then/Mythical-tales-of-kylin-lives-on/shdaily.shtml)" might be a transliteration of "[麒麟](https://en.wiktionary.org/wiki/%E9%BA%92%E9%BA%9F)", "a propitious beast in ancient Chinese mythology, with the shape of a deer, tail of an ox, a single horn and scales all over its body" and likely used in various company names.  Here's a reference to a product titled "[EKYLIN Wireless TPMS](https://www.amazon.sa/dp/B07KVHRCR7)" which may or may not have anything to do with what we're analyzing, nor even be manufactured by that company.

ProfBoc75, 2024-07-03:
Or this one : https://web.archive.org/web/20180826092330/http://kylin-tec.com/p_lr.asp?id=2756&cl_na=%E8%83%8E%E5%8E%8B%E7%9B%91%E6%B5%8B%E7%B3%BB%E7%BB%9F%E5%BC%80%E5%8F%91&cl_id=556&sid=0

From Kylin-Tec (site is hacked / not working properly), they used NXP pressure module. Another link to this site : https://web.archive.org/web/20180901193313/http://new.zlgmcu.com/Item/304046.aspx

Today link : https://www.zlgmcu.com/carbcontrol/carbcontrol/article/id/7.html

And yes, look like a Chinese product.

Notice some useful information about the product, like pressure range, frequency, MCU model and so on...

Aony-Live, 2024-08-29:
> [tpms_chinese.zip](https://github.com/user-attachments/files/16069773/tpms_chinese.zip) drafted the decoder based on above finding by @ProfBoc75 . Using [RTL_433_ESP](https://github.com/NorthernMan54/rtl_433_ESP/tree/main) library, I was able to get the codes after some tuning of CC1101.
> 
> `pulse_slicer_manchester_zerobit: Unbranded Chinese TPMS codes [{214}7ffffffffffffffffffffffffffffffffd6e4fd8ec80628e1c2ec0]` `pulse_slicer_manchester_zerobit: Unbranded Chinese TPMS codes [{209}000000000001ffffffffffffffffffffeb727ec764037470e1758]`
> 
> Stuck on decoding part... in decoding section there is nothing to process. For more clarification: [#line547](https://github.com/NorthernMan54/rtl_433_ESP/blob/b8ea0e2d8bada6c6cde31640481c80fe674294a2/src/signalDecoder.cpp#L546)
> 
> How to parse the codes into temp, pressure, id values ?? Thanks for the help

@cyberFIRM Is it possible for you to post the Register Keys for the CC1101?