merbanan/rtl_433 #2989: Switching between software using Plutosdr

imca-potato, 2024-07-05:
Greetings!

I have been trying to implement a weather monitoring system that switches among the different stations that are transmitting. The idea is to switch to a different software for the different stations. For a while, when I terminate rtl_433 then run the other software, it works just fine. However, I had to reinstall the rtl_433 software (I am not sure why but it somehow got deleted, perhaps an error on my part), and now the software is not functioning properly anymore.

To add a few details that may help, I am running this on a RaspberryPi 4B with Ubuntu OS and I am using the ADALM-Pluto SDR. Just to clarify as well, both software works, but only independently, i.e., if I only run the other software, it is able to decode signals. However, if I call rtl_433 before the other, only then will it not "work" (it is capable of detecting and "using" the device, but no signals are being detected). For reference, the other software is SDRangel by f4exb.

I was thinking that it may have something to do with how the software is still interfaced with the SDR despite the software being terminated, which was not an issue or just did not happen before for some reason. I have also raised this issue in the other forum and will update any information I gain from there.

Thank you in advance for those who can help!

zuckschwerdt, 2024-07-05:
It might be related to Pluto settings that are not reset between switching software.

Can you clarify: "call rtl_433 before the SDRangel, only then will it not work" -- does this mean you need to run SDRangel to somehow get the Pluto in the right state for rtl_433 to work properly?

Can you run `iio_info` before and after running a software and compare the important settings?

imca-potato, 2024-07-05:
> Can you clarify: "call rtl_433 before the SDRangel, only then will it not work" -- does this mean you need to run SDRangel to somehow get the Pluto in the right state for rtl_433 to work properly?

Apologies for this, to clarify, only SDRangel is the "malfunctioning" software. Rtl_433 runs regardless of what happens.

SDRangel works if it is the only software called.

SDRangel does not work if rtl_433 is called before it. (rtl_433 still runs and functions properly even after this happens)

> Can you run iio_info before and after running a software and compare the important settings?

Are there any specific tags? It seems mine does not output anything.

`sdr@SDR-group:~/Desktop$ iio_info
Library version: 0.21 (git tag: 565bf68)
Compiled with backends: local xml ip usb serial
Unable to create Local IIO context : No such file or directory`

It says here no IIO context was created, would that be a problem? The Pluto runs the rtl_433 even with this error.

imca-potato, 2024-07-05:
According to the following posts, the `Unable to create Local IIO Context : No such file or directory` can be ignored.

https://github.com/analogdevicesinc/libiio/issues/916
https://ez.analog.com/linux-software-drivers/f/q-a/543322/unable-to-create-local-iio-context-no-such-file-or-directory-2-on-ubuntu-20-4

zuckschwerdt, 2024-07-05:
You might have to select the backend to use, e.g. `iio_info --network 192.168.2.1`
Example output is seen here https://wiki.analog.com/resources/tools-software/linux-software/libiio/iio_info

imca-potato, 2024-07-05:
I was able to produce an output similar to what is shown, what device or parameters should I look out for?

Note: The message `No trigger on this device` shows for all devices. This is even before I call on any software.

(Please let me know if you need the output. I left it out for now because it is too long.)

zuckschwerdt, 2024-07-05:
Idk. Maybe filter, gain, bandwidth and such? Perhaps run `diff -U0 out1.txt out2.txt` and see if something interesting pops up in the changes?

imca-potato, 2024-07-05:
Ok, I will try that.

imca-potato, 2024-07-05:
Greetings!!

I have finally found the solution thanks to your suggestion! Turns out there is some parameter of PlutoSDR that isn't being "reset" when I terminate rtl_433. I simply had to reset that manually via iio_attr commands.

The specific parameter is the `powerdown value` under device `ad9361-phy`, channel `altvoltage0`.

Thank you so much for the assist!

closed 2024-07-05 by imca-potato

zuckschwerdt, 2024-07-05:
Interesting! That's an explicit setting added 4 years ago to SopayPlutoSDR to fix leaving the SDR running after streams stopped.
https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Streaming.cpp#L149
https://github.com/pothosware/SoapyPlutoSDR/commit/2cf04f6

I'd cautiously venture to say that SDRangel is at fault for not managing that value :) You might want to relay that issue.