merbanan/rtl_433 #2991: MQTT Connect error: Unknown Error

Bogdan50, 2024-07-07:
Trying to test the mqtt messaging with the  Mosquito test server. I am issuing this command:

rtl_433 -F mqtt:test.mosquitto.org -F log

I am receiving this:

rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR
MQTT: Publishing MQTT data to test.mosquitto.org port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/DESKTOP-M9DIDRD/devices[/type][/model][/subtype][/channel][/id]".
MQTT: Publishing events info to MQTT topic "rtl_433/DESKTOP-M9DIDRD/events".
MQTT: Publishing states info to MQTT topic "rtl_433/DESKTOP-M9DIDRD/states".
Found Elonics E4000 tuner
SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: , "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
MQTT: MQTT connect error: Unknown error

I tested the messaging with mosquitto_pub and sub and I can send and receive messages no problems.
Am I doing something wrong?

Bogdan50, 2024-07-07:
Using the IP address directly works, I am finding out.

rtl_433 -F mqtt:91.121.93.94 -F log

I can now receive messages from the server.
So use only IP addresses?

gdt, 2024-07-08:
As usual:
  - describe mqtt version, os, os version, cpu arch
  - update to current git master because 23.11 is old
  - explain your name resolution plan
  - explain how you tested name resolution

Bogdan50, 2024-07-08:
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR
Windows 10 Pro ver 22H2, 64-bit operating system, x64-based processor, AMD FX(tm)-9590 Eight-Core Processor              4.72 GHz

Not sure what do you mean by name resolution plan. 
I'm using the free server provided by Mosquitto here:

https://[test.mosquitto.org](https://test.mosquitto.org/)/

The name resolution I've tested with mosquitto_pub and mosquitto_sub using the address "test.mosquitto.org" without quotes.

gdt, 2024-07-08:
23.11 is quite old.  I have no idea if there are relevant fixes since then.    I personally do not engage in debugging old code.  Yes, I know that it might be the latest formal release.....

By name resolution, I mean how the system converts names to IP addresses.   There is regular DNS, with a configured resolver, /etc/hosts on Unix, and multicast dns.   Various systems have different configs.   It seems likely that either something is odd or there is something wrong in the mqtt code in rtl_433 that is not resolving the name correctly.   It will likely require using tools like dig and other things that resolve names, and a debugger on rtl_433 to figure it out.

Bogdan50, 2024-07-08:
Are there any tests you want me to do on my PC?

Bogdan50, 2024-07-08:
I can ping test.mosquitto.org just fine:

╰─ﬀ ping test.mosquitto.org

Pinging test.mosquitto.org [91.121.93.94] with 32 bytes of data:
Reply from 91.121.93.94: bytes=32 time=114ms TTL=50
Reply from 91.121.93.94: bytes=32 time=129ms TTL=50
Reply from 91.121.93.94: bytes=32 time=113ms TTL=50
Reply from 91.121.93.94: bytes=32 time=117ms TTL=50

Ping statistics for 91.121.93.94:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 113ms, Maximum = 129ms, Average = 118ms

cmgrayb, 2024-09-07:
I built from master yesterday, and I also show 23.11 in the version number:
rtl_433 version 23.11-160-gf0ba1538 branch master at 202408270014 inputs file rtl_tcp RTL-SDR

I am experiencing the exact same behavior on a brand new Raspberry Pi running Pi OS running its default DNS setup, purpose built for this function.  Outside of rtl_433, DNS works as expected, including with mosquitto_pub.

In rtl_433, I can only use the IP address in my MQTT string or I get the same Unknown error -1, using configuration file or passed as a -F command line parameter.

The preferred mqtt connection string in the configuration file is:
```
output mqtt://homeassistant.local:1883,user=[redacted],pass=[redacted],retain=1
```

But the only string that will work is:
```
output mqtt://192.168.1.40:1883,user=[redacted],pass=[redacted],retain=1
```

gdt, 2024-09-07:
That tag, 160 commits past 23.11, looks right.

Something is wrong but we can't tell what.  It could be a bug in the sources and it could be something odd about your machine.   You say "DNS", but mdns is separate and there is usually a resolver configuration that switches among host files, DNS, mDNS, and NIS.

I would suggest reading the code and adding printf of args and return codes and perhaps using gdb, to find out what step is going astray.

My guess, way out on a limb, is that rtl_433 is using an interface that is DNS only, not dns-or-mdns-or-?.   For example, what does "dig homeassitant.local" say?   For example, on my computer, even though I can ping/ssh/web to foo.local, "dig foo.local" returns NXDOMAIN.

zuckschwerdt, 2024-09-07:
> rtl_433 is using an interface that is DNS only, not dns-or-mdns-or-?.

Yes, we recently located that problem in #2695. There is no support for mDNS.

gdt, 2024-09-07:
Dup of #2695 indeed.

closed 2024-09-07 by gdt