merbanan/rtl_433 #2992: no messages when -F is log or influx

atkinson, 2024-07-08:
If I simply run, `rtl_433` - it detects messages and displays them in the terminal. My reciever is working correctly and rtl_433 is working fine with it.

However, when I try to store the messages in influxdb (v2.7) using the following command:

`rtl_433 -F influx://192.168.68.50:8086/api/v2/write?org=home&bucket=sensors,token=my_token_is_here -M time:unix:usec:utc`

It seems to work, but...

```
 rtl_433 version 23.11-142-g7e4cef28 branch master at 202406281317 inputs file rtl_tcp RTL-SDR with TLS
InfluxDB: Publishing data to InfluxDB (http://192.168.68.50:8086/api/v2/write?org=home)
Use "-F log" if you want any messages, warnings, and errors in the console.
Detached kernel driver
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
Exact sample rate is: 250000.000414 Hz
```

...it does not report any messages, does not store anything in influxdb and does not report any errors.

I've tried -vvv, but there is no additional information on this.

I've tried wrapping the influx URL in quotes, in both cases rtl_433 accepts the URL `(InfluxDB: Publishing data to InfluxDB (http://192.168.68.50:8086/api/v2/write?org=home)`)

I've tried `rtl_433 -F log` (as suggested by the ouput), and that behaves the same as influx (no messages, no response)

I've tried `rtl_433 -F kv` - and that works correctly.

How should I debug this?

many thanks

atkinson, 2024-07-08:
Another "clue", when rtl_433 is printing messages to the console, I can CTRL-C, and rtl_433 reports, `^CSignal caught, exiting!`.

When I try to write to influxdb (and the program seemingly hangs), and I CTRL-C, the programme doesn't exit properly. I must find the process id using ps and then kill it, upon which it finally says, `Signal caught, exiting! Reattached kernel driver`.

I'm not sure if this helps...

zuckschwerdt, 2024-07-08:
You can not have unquoted `&` on the command line.
Use multiple `-F` outputs, e.g. `-F kv -F influx…`. The `kv` includes log, `log` is just error messages, no data.

atkinson, 2024-07-09:
Thanks for the suggstions @zuckschwerdt .

I added back the quotes and tried adding -F kv displays the messages, but they still don't go into influx, and there is no feedback as to why (no error message).

`rtl_433 -F kv -F "influx://192.168.68.50:8086/api/v2/write?org=home&bucket=sensors,token=this_is_where_my_token_goes" -M time:unix:usec:utc`

gdt, 2024-08-04:
Please ask on the list or open a discussion; this is not a valid bug as it stands.  (And the issue tracker is overwhelmed.)

closed 2024-08-04 by gdt