merbanan/rtl_433 #2995: Badger Orion Endpoint GIF2014W-OSE not working with -R 223

ddffnn, 2024-07-09:
My city water meter RF module was recently replaced with an "Orion Endpoint."
I had been able to receive and decode readings from the old module using -R 223, but that's no longer the case.

I'd like some suggestions for reverse engineering the data I'm receiving from the new module because I believe it's quite different from the old one that worked with -R 223.

This is the new module:

[image]

[image]

I'm quite confident that I've been able to detect the correct pulses using `rtl_433 -f 915.8M -s 1M -M level -R 0 -X 'name=badger,modulation=FSK_PCM,short=10,long=10,reset=1000'`.
I have a cable going out to my meter pit with a magnetic antenna base right next to the RF module, but with no antenna attached.
I'm seeing over 30dB SNR so it's gotta be the meter pulses I'm catching right?

It seems that while there are transmissions every 20s or so, the payload only updates once per hour, on the hour.
This is based on known water usage throughout the day.

I've looked at the comments in the source code for protocol 223, but I think the protocol has changed.

I've attached a complete CSV  from the above command, but it's got a bunch of other device transmissions.
I've also attached a filtered (regex '76317631.*b8531') version that I believe only has my meter readings.

[filtered.csv](https://github.com/user-attachments/files/16139592/filtered.csv)

[complete.csv.gz](https://github.com/user-attachments/files/16139584/complete.csv.gz)

Thank you in advance!

zuckschwerdt, 2024-07-09:
The codes look promising but not like we expect for the Badger ORION. The start looks like a sync word of 7631 7631 or d8c5 d8c5 or 273a 273a or 89ce 89ce (shift and invert however you like…) and the end looks like what could be a CRC-16. But alignment is not clear to me and I can't see if there is a 3of6 coding present?

ddffnn, 2024-07-09:
Yes, I agree on all those points. Also, it's longer than we would expect. I've been trying different antennas, frequencies, and dongles, but I've never seen my water meter using 223. I kept thinking I'm doing something wrong, but it sure seems like the protocol had just changed.

I plan to read the analog display of the encoder today. It used to match with the old module so I can only assume it will with this one also.

ddffnn, 2024-07-10:
I looked at the meter today to read it directly and realized it's new too. The old meter was mechanical, but this new one is ultrasonic with a nine-digit digital display. It alternates showing total volume and instantaneous rate.

[image]
[image]

Any, I read the meter manually three times, and we only used water during a few periods so there were only 7 unique payloads.

```
{273}555555555555557631763175f08ecd76b6a526f438c99d3c11ab85576716ec3e9fa70
{274}555555555555557631763175f08ecd76b6a526f438c99d3e91ab85576716ec3e90570
{272}555555555555557631763175f08ecd76b6a526f438c99d2791ab85576716ec3e8336  [4614.36]
{277}555555555555557631763175f08ecd76b6a526f438c99d0711ab85576716ec3e8c0420 [4620.64]
{275}555555555555557631763175f08ecd76b6a526f438c99d0110ab85576716ec3e89420
{273}555555555555557631763175f08ecd76b6a526f438c99d0e902b85576716ec3ee2508 [4695.78]
{275}555555555555557631763175f08ecd76b6a526f438c99d56102b85576716ec3eac658
```
I messed around in BitBench but I'm starting to wonder if there is some 3of6 still in use or something similar. I can't find the serial number or get increasing values, let along values that match my manual readings.

ddffnn, 2024-07-10:
@nickovs Since you did the work for protocol 223, do you happen to have any suggestions for this one?

starman2k01, 2024-08-25:
Did you see this?
https://github.com/bemasher/rtlamr/discussions/196

SRGDamia1, 2025-01-14:
Did you ever get any farther with this? My water company just installed a similar Badger e-series meter with an Iron Cellular C cellular/RF radio. The FCC ID for the radio is GIF2020OCECNA.

francistheodorecatte, 2025-01-14:
> Did you ever get any farther with this? My water company just installed a similar Badger e-series meter with an Orion Cellular C cellular/RF radio. The FCC ID for the radio is GIF2020OCECNA.

Looks like that's a cellular radio module using NB-IoT or LTE-M, which is not something rtl_433 is likely to ever support.

SRGDamia1, 2025-01-14:
The device has two FCC IDs.  The permit for GIF2020OCECNA says it's 904.94-923.79MHz. I didn't realize that could be cellular. The other FCC ID on my transceiver is N7NHL78 which is an LTE-M/NB-IoT modem.

I haven't yet done RF snooping on the meter.

nickovs, 2025-01-14:
The FCC report for [GIF2020OCECNA](https://fcc.report/FCC-ID/GIF2020OCECNA) indicates that it transmits at 904.94MHz, 914.1MHz and 923.79MHz. These are all in the unlicensed 902-928 MHz ISM (Industrial, Scientific, and Medical) band in the USA. From my limited RF knowledge, looking at the plots it seems that this part it might well be transmitting in a similar manner to the older units, and this radio it operating completely separately to the cellular modem.

ProfBoc75, 2025-01-15:
@nickovs : for the test they used these frequencies with the hopping feature disabled (or only these 3 frequencies low, medium and high), the major issue here is the FHSS modulation (hopping enabled) which is not good, the hopping will change the frequency in range of 902-928MHz during the signal transmission, so you need 26 Mhz of bandwidth/sample  rate to analyze all the signal to get the figures. Page 45, hopping enabled, you have 48 channels in this range.

Gigoo25, 2025-01-16:
I have the GIF2014W-OSE also but RTL seems to only pick up Badgers at random intervals. I enabled additional logs for the most recent detection.

    {"time" : "2024-11-20T10:19:24.198098-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 1148381, "flags_1" : 0, "volume_gal" : 1006987, "flags_2" : 5, "mic" : "CRC"}
    {"time" : "2024-11-20T10:19:33.979250-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 2412388, "flags_1" : 72, "volume_gal" : 225609, "flags_2" : 5, "mic" : "CRC"}
    {"time" : "2024-11-20T10:19:33.979250-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 2412388, "flags_1" : 72, "volume_gal" : 225609, "flags_2" : 5, "mic" : "CRC"}
    {"time" : "2024-12-16T11:19:43.007246-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 2717085, "flags_1" : 64, "volume_gal" : 312774, "flags_2" : 5, "mic" : "CRC"}
    {"time" : "2025-01-03 10:56:53.584828-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 3726394, "flags_1" : 65, "volume_gal" : 285291, "flags_2" : 5, "mic" : "CRC", "mod" : "FSK", "freq1" : 916.381, "freq2" : 916.492, "rssi" : -14.165, "snr" : 11.352, "noise" : -25.517}
    {"time" : "2025-01-11 00:47:40.795024-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 3726394, "flags_1" : 65, "volume_gal" : 285291, "flags_2" : 5, "mic" : "CRC", "mod" : "FSK", "freq1" : 916.390, "freq2" : 916.504, "rssi" : -15.103, "snr" : 9.718, "noise" : -24.820}
    {"time" : "2025-01-11 01:16:28.456081-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 3726394, "flags_1" : 65, "volume_gal" : 285291, "flags_2" : 5, "mic" : "CRC", "mod" : "FSK", "freq1" : 916.381, "freq2" : 916.489, "rssi" : -14.473, "snr" : 11.651, "noise" : -26.124}
    {"time" : "2025-01-11 01:19:48.098003-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 3726394, "flags_1" : 65, "volume_gal" : 285291, "flags_2" : 5, "mic" : "CRC", "mod" : "FSK", "freq1" : 916.391, "freq2" : 916.503, "rssi" : -14.165, "snr" : 10.198, "noise" : -24.363}
    {"time" : "2025-01-11 01:33:56.739858-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 292908, "flags_1" : 8, "volume_gal" : 415407, "flags_2" : 5, "mic" : "CRC", "mod" : "FSK", "freq1" : 916.376, "freq2" : 916.496, "rssi" : -15.154, "snr" : 8.539, "noise" : -23.693}
    {"time" : "2025-01-11 01:47:21.466551-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 3726394, "flags_1" : 65, "volume_gal" : 285291, "flags_2" : 5, "mic" : "CRC", "mod" : "FSK", "freq1" : 916.395, "freq2" : 916.488, "rssi" : -13.257, "snr" : 11.985, "noise" : -25.242}
    {"time" : "2025-01-15 11:15:42.475931-0500", "protocol" : 223, "model" : "Badger-ORION", "id" : 14960573, "flags_1" : 128, "volume_gal" : 714614, "flags_2" : 4, "mic" : "CRC", "mod" : "FSK", "freq1" : 916.402, "freq2" : 916.513, "rssi" : -14.473, "snr" : 8.429, "noise" : -22.901}

My current config is as follows:

    report_meta stats
    report_meta time:iso:usec:tz
    report_meta protocol
    
    # For RTL-SDR: gain in tenths of dB ("0" is auto).
    gain 0
    
    # this is suggested
    convert si
    
    ## default is "433.92M", other reasonable values are 315M, 345M, 915M and 868M
    frequency 916.45M
    
    # default is "600" seconds, only used when multiple frequencies are given
    hop_interval  300
    
    # default is "250k", other valid settings are 1024k, 2048k, 3200k
    sample_rate   1200k
    
    # Set minlevel automatically based on average estimated noise.
    pulse_detect autolevel
    
    # Skip frames below estimated noise level to lower cpu load.
    #pulse_detect squelch
    
    # Log levels 1=fatal, 2=critical, 3=error, 4=warning (default), 5=notice, 6=info, 7=debug, 8=trace
    verbose 6
    
    # to see logs in console
    output kv
    
    # log to json file, easier to see device information
    output json:/config/rtl_433/logs/output.json
    
    # Add various metadata to every output line.
    report_meta level
    report_meta noise
    report_meta stats
    report_meta time:usec
    report_meta protocol
    
    # Signal auto save. Creates one file per signal.
    signal_grabber known
    
    # SCM Protocols (Gas Meter)
    protocol 149
    
    # Badger Orion Water Meter
    protocol 223

Are the Badgers only showing up on occasion due to frequency hopping?

ProfBoc75, 2025-01-16:
@Gigoo25 : could be, range is from 902 Mhz to 928 Mhz, with your central frequency at 916.45Mhz and 1200KHz (= 1.2 MHz) of sample rate, you will capture a small part, from 915.85Mhz to 917.05Mhz.

You can test with another central frequency, increase the sample rate to 2048K for example ( so +/- 1M) , or max 3200K ( so +/- 1.6 M).
Central frequency at 903M it will cover 902 to 904Mhz, then 905M to cover 904 to 906, and so on...  for sample rate at 2048K
with 3200K , try 903.5M (so cover 901.9M to 905.1M) , then 906.5M (so cover 904.9M to 908.1M) and so on ...

francistheodorecatte, 2025-01-17:
fwiw after looking at my meter transmitting on a waterfall plot with my LibreSDR earlier, w/ 32MHz of bandwidth centered around 915MHz, it looks like the meter hops *down* 1.2MHz every ~20 seconds (on every transmit), and not across the entire range, only the bottom. I saw a much fainter signal that's probably my neighbor's meter, doing the same thing with the upper end of the frequecy range.

ProfBoc75, 2025-01-17:
@Gigoo25 : hopping is pseudo randomized, so if you are able to identify the hop frequency order sequence, yes, this is a challenge, it will take time, with the assumption that at one moment the sequence is repeated, which is probably the case, since it's a "stupid" device doing the same things forever. And not sure that the period of 60 seconds is accurate too, you have also to identify how often the frequency change (every xx seconds ? ) , another challenge.

gdt, 2025-07-04:
Where are we on this?  seems like not apparently decodable and progress is stalled, which would lead to closing ticket.

jjemelka, 2025-07-16:
ddffnn contributed the extensive .csv files. the final analysis of which remains inconclusive.

### Raw Packet (without preamble)
```
7631763175f08ecd76b6a526f438c99d279185ab576716ec3e9fa70
```

### Decoded Fields
- **Sync Word**: `76317631`
- **Meter ID**: `75f08ecd`
- **Extended ID**: `76b6a526f438c99d` (constant for this meter)
- **Water Reading**: `2791` (encodes 4614.36 gallons)
- **Status**: `85ab`
- **Fixed Data**: `576716ec`
- **Secondary Counter**: `3e9f`
- **CRC/Status**: `a70`

jjemelka, 2025-07-16:
try this on your meter. I assumed that the barcoded tag pictured on your cable matches the ID stamped on the grey plastic top above the cast iron meter cover (40466920) and the data comes from that ID.

Not much usage data in the csv files to go on here...

runs on an UBUNTU window on a Windows 10 laptop

includes

 #!/usr/bin/env python3
     """
     Lazy Computer Scientist's 1990s Microcontroller Crypto Decoder
     Target: Find serial number 40466920 in Badger meter data
     """
 successfully decoded the Badger ORION water meter data using lazy 1990s microcontroller encryption techniques:

  Discovered Encryption Methods:

  1. Meter ID (40466920):
    - Location: Bytes 4-7 after sync word
    - Encryption: XOR with key 0x7799F725 (big-endian)
    - Encrypted value: 75f08ecd
  2. Water Reading:
    - Location: Bytes 22-24 after sync word
    - Encoding: Actual gallons × 888, stored as 24-bit big-endian
    - Example: 96ec3e = 9,890,878 ÷ 888 = 11,138.38 gallons *** not much to go on here

  Frame Structure:

  Offset | Bytes | Field         | Description
  -------|-------|---------------|----------------------------------------
  0      | 4     | Sync Word     | Always 0x76317631
  4      | 4     | Meter ID      | 32-bit BE, XOR with 0x7799F725
  8      | 3     | Unknown 1     | Purpose unclear
  11     | 5     | Unknown 2     | Mostly static
  16     | 3     | Unknown 3     | Variable (possible status)
  19     | 3     | Unknown 4     | Semi-static
  22     | 3     | Water Reading | Gallons * 888, 24-bit BE
  25+    | var   | CRC/Checksum  | Variable length

  The encryption uses classic 1990s techniques:
  - XOR encryption for the meter ID (simple obfuscation)
  - Integer multiplication for readings (888 ≈ 1000 for precision without floating point)

  These methods were common in microcontrollers with limited processing power and memory, providing basic security while being easy to implement in assembly language.

rtl_433.exe -f 915.7M -s 1000000 -R 0 -X "name=badger,modulation=FSK_PCM,short=10,long=10,reset=1000" -F json | python3 badger_realtime_monitor.py

#!/usr/bin/env python3
"""
Real-time Badger ORION Water Meter Monitor
Integrates with RTL-433 to decode and monitor water usage

Features:
- Real-time decoding of meter transmissions
- Flow rate calculation
- Leak detection
- Usage statistics
- JSON/CSV output
"""

import json
import csv
import sys
import struct
import subprocess
from datetime import datetime, timedelta
from collections import deque

class BadgerRealtimeMonitor:
    def __init__(self):
[... 247 lines cut ...]

def main():
    """Main entry point"""
    print("Badger ORION Real-time Water Meter Monitor")
    print("Using discovered encryption: XOR 0x7799F725, ÷888")
    print("Press Ctrl+C to stop\n")
    
    monitor = BadgerRealtimeMonitor()
    
    # Check if we're reading from stdin or running RTL-433
    if not sys.stdin.isatty():
        # Reading from pipe
        print("Reading from stdin...")
        for line in sys.stdin:
            decoded = monitor.parse_rtl433_line(line.strip())
            if decoded:
                enhanced = monitor.process_reading(decoded)
                monitor.print_status(enhanced)
                monitor.save_to_csv(enhanced)
    else:
        # Run RTL-433
        monitor.run_rtl433()

if __name__ == "__main__":
    main()

ddffnn, 2025-07-16:
I was stumped and unsure what to do next when I first wrote this issue. I never got it working so I too think this issue should remain open. I'll try the script within a few days.

jjemelka, 2025-07-17:
in the cmd definition it needs better parameters for 2014WOSE, can you send your successful ones?

"""Run RTL-433 and process output"""
        cmd = [
            'rtl_433',
            '-f', '916.45M',
            '-s', '1200k',
            '-R', '223',  # Badger ORION protocol
            '-F', 'json',
            '-M', 'time:iso',
            '-M', 'level'
        ]

jjemelka, 2025-07-18:
Back to basics to give me an understanding of the process using -R 223 as an example.

1. RF Signal → Bitstream (Physical Layer)

  - RTL-SDR captures 916.45 MHz FSK signal
  - Demodulates to NRZ bitstream at 100 kbps
  - Produces: 01010101...01010000111101[encoded data]

  2. Bitstream → 4:6 Decoded Bytes (Encoding Layer)

  - Finds preamble 0x543D after sync bytes
  - Takes next 120 bits (10 bytes × 12 bits each)
  - Decodes each 12 bits: 6→4 + 6→4 = 1 byte
  - Example: 8cd2cd... → B13129...

  3. Decoded Bytes → Packet Fields (Application Layer)

  Bytes 0-2: Device ID (little-endian)
  Byte 3:    Flags 1
  Bytes 4-6: Volume in gallons (little-endian)
  Byte 7:    Flags 2
  Bytes 8-9: CRC16 (big-endian)

  The 4:6 encoding uses only 16 valid 6-bit patterns (out of 64 possible), providing error detection. Each pattern
  has exactly 3 bits set, hence "3-of-6" coding.

  When rtl_433 uses -R 223, it automatically performs all these steps internally and outputs the final decoded
  values.

more detail in BADGER_DECODING_PROCESS.md attached

[BADGER_DECODING_PROCESS.md](https://github.com/user-attachments/files/21320242/BADGER_DECODING_PROCESS.md)

jjemelka, 2025-07-18:
I've created three Python scripts for capturing and monitoring Badger 2014WOSE water meters:

  1. capture_badger_2014wose.py - Basic Capture

  - Launches rtl_433 with correct parameters
  - Saves raw data to timestamped CSV file
  - Shows captured frames on screen
  - CSV format matches your existing Badger2014WOSE_Frames.csv

  2. capture_decode_2014wose_realtime.py - Real-time Monitor

  - Captures AND decodes in real-time
  - Shows meter ID and water usage immediately
  - Tracks consumption between readings
  - Saves both raw CSV and decoded JSON
  - Displays running totals per meter

  3. monitor_badger_2014wose.py - Main Menu

  - User-friendly launcher with options
  - Tests rtl_433 availability
  - Provides all capture/decode options

  Usage:

  Quick Start:

  # Run the main menu
  python3 monitor_badger_2014wose.py

  # Or run real-time monitoring directly
  python3 capture_decode_2014wose_realtime.py

  RTL-433 Parameters Used:

  - Frequency: 916.45 MHz
  - Sample Rate: 1 MHz
  - Sync Word: 0x76317631 (32 bits)
  - Modulation: FSK_PCM with 10μs pulses
  - Minimum Bits: 216 (for 27-byte frames)

  The scripts will:
  1. Capture raw frames with timestamps
  2. Save to CSV in the same format as your sample file
  3. Optionally decode in real-time showing:
    - Meter ID (decrypted)
    - Current water reading in gallons
    - Consumption since last reading
    - Signal quality (RSSI/SNR)

  The captured CSV files can be processed later with decode_badger_2014wose_json.py for detailed analysis.

[monitor_badger_2014wose.txt](https://github.com/user-attachments/files/21320958/monitor_badger_2014wose.txt)
[capture_decode_2014wose_realtime.txt](https://github.com/user-attachments/files/21320957/capture_decode_2014wose_realtime.txt)
[capture_badger_2014wose.txt](https://github.com/user-attachments/files/21320956/capture_badger_2014wose.txt)

jjemelka, 2025-07-21:
Yes the syntax is incorrect, no 32 bits or braces...  Recall I just used the data from your csv file to arrive at these results and havent been able to acquire live data from any of our 2014W-OSE meters. Listening for the old meters works great using -R 223. 

I've just looked back at https://github.com/merbanan/rtl_433/issues/2089 for some insights that may be in the previous work on 2014W-OSE. I could use some education on the RF part of this, so I'm at a bit of a disadvantage here. 

My goal is to get early warning for the homeowner run HOA to know when one of our 5 irrigation zones needs attention. Our monthly water bills can run into the low tens of thousands if any leg leaks. Luckily the zones most likely to suffer have old meters for the moment.

If you have rtl_433 parameters and system setups that work, please let me know. The new meters are at some distance and in an obstructed sightline from my house, so I have to be present during data acquisition.

jjemelka, 2025-07-21:
> # /usr/local/bin/rtl_433 -f 916.45M -s 1000000 -R 0 -X name=badger,modulation=FSK_PCM,short=10,long=10,reset=1000,sync={32}76317631,bits>=216 -F csv -M time:iso -M level
> rtl_433 version 25.02-39-g85a42b42 branch master at 202507201159 inputs file rtl_tcp RTL-SDR
>
> New defaults active, use "-Y classic -s 250k" if you need the old defaults
>
> Disabling all device decoders.
> sync: invalid number argument ({32}76317631)
>
> Does Windows expand the braces or something?
>
> This isn't a windows vs linux thing. This is just an incorrect argument format for sync, which only takes a number. I think match should be used here instead of sync.

Yes the syntax is incorrect, no 32 bits in the braces...  Recall I just used the data from your csv file I left some more notes on the 2995 most including reference to previous work  done on this in issue #2089.

😀

jjemelka, 2025-07-21:
the structure assumed is: 

5555555555555557 | 63176317 | 5f08ecd7 | 6b6a526f438c99d2 | 385a  | b853 |   1f096ec3  | eace | 30
└─ Preamble ─┘      └─Sync─┘ └MeterID┘└─Extended ID ─┘ └Read┘└Stat┘└─Fixed─┘└Cnt┘└CRC┘

how best to configure rtl_433 using preamble sync and match is trial and error for me.  I'd be grateful for any corrections/additions/changes

shawntoffel, 2025-09-10:
Curiously, I have additional data which has the same values all through `55555555555557631763175f08ecd76`. Presumably it is not the same meter id as OP. I'm not sure of the model which it came from.

```
55555555555557631763175f08ecd76f08cf6f6b8e91d2b7d1c05aa3e96ec3e8f199f8
55555555555557631763175f08ecd76f08cf6f6b8e91d2b7d1c05aa3e96ec3e8f19bf8
55555555555557631763175f08ecd76f08cf6f6b8e91d2b7d1c05aa3e96ec3e8f199f4
55555555555557631763175f08ecd76f08cf6f6b8e91d2b7d1c05aa3e96ec3e8f19bf
55555555555557631763175f08ecd76f08cf6f6b8e91d2b7d1c05aa3e96ec3e8f199fe
55555555555557631763175f08ecd76f08cf6f6b8e91d2bfd1c05aa3e96ec3ecc283f
55555555555557631763175f08ecd76f08cf6f6b8e91d2bfd1c05aa3e96ec3ecc283f
55555555555557631763175f08ecd76f08cf6f6b8e91d2bfd1c05aa3e96ec3ecc283f8
55555555555557631763175f08ecd76f08cf6f6b8e91d2bfd1c05aa3e96ec3ecc283f4
```

Command used: `rtl_433 -f 914.1M -s 1200k -X 'n=badger,m=FSK_PCM,short=10,long=10,reset=1000'`

RichyT, 2025-09-20:
Just wondering how this stands? Looks like we're not there yet. I could probably contribute to some beer-money bounty kind-of thing if allowed by the rules of github. Unfortunately, the meter is too remote for participating in data gathering.

klyubin, 2025-10-03:
Thanks to all the info in this thread, especially @ddffnn and @shawntoffel, I figured out how to obtain the serial number and the most recent reading from the broadcasts. See attached Python script [decode_orion_me.py](https://github.com/user-attachments/files/22673663/decode_orion_me.py).

Example with @ddffnn's and @shawntoffel's broadcasts:
```bash
./decode_orion_me.py 555555555555557631763175f08ecd76b6a526f438c99d0e902b85576716ec3ee2508
serial_number: 40466920, reading: 46957

$ ./decode_orion_me.py 55555555555557631763175f08ecd76f08cf6f6b8e91d02f21c05aa3e96ec3e856b9f8
serial_number: 130624100, reading: 24081269
```

When I have some spare time and energy, I'll try to add this as a protocol decoder to the `rtl_433` codebase. However, if anybody else is up for this, I won't say no.

P. S. Here's how I got this:
* The ORION ME ENC is broadcasting once every 15 seconds. In my case, it is occasionally switching between 915.7+8 MHz, 916.0+1 MHz, 916.4+5 MHz, and 916.8+9 MHz.
* The ORION ME ENC is pulling a new value (the latest measurement/reading) from the meter exactly once an hour. It then keeps broadcasting that same value for an hour even if the underlying meter's reading is changing. Once I figured out  when that pulling in happens (by capturing the broadcasts and noting when they change), it was easy to know exactly what meter reading is broadcast.
* As folks have suspected in this thread, the broadcasts are encrypted. However, this is just an XOR with a hard-coded keystream which is the same for all broadcasts from all meters of this model. This means two things: 1) when we XOR two broadcasts, the result is the XOR of the plaintext/decrypted contents of the two; 2) when we know the plaintext/decrypted value at a specific position of the broadcast packet, we can XOR that plaintext with the encrypted packet and get the keystream for that location.
* I captured broadcasts for known (to me) values of the meter, XOR'd them pair-wise and as a result figured out at which location these numbers were stored (and that they were in little-endian format). I then did the same for the serial number, XORing my broadcasts (with known to me serial number) with broadcasts published earlier in this ticket by @ddffnn for which we know the serial number.

merbanan, 2025-10-03:
Did the meter you have and the one from @ddffnn share keystream?

francistheodorecatte, 2025-10-03:
fwiw you can pipe rtl_433 directly into your python script live using some one-line bash shenanigans:
`rtl_433 -f 916.7M -s 1600k -M level -R 0 -X 'name=badger,modulation=FSK_PCM,short=10,long=10,reset=1000' -F json | jq --stream -nc -r 'fromstream(0|truncate_stream(inputs))|.rows[0].data' | while read -r line; do command ./decode_orion_me_enc_gif2015w_ose.py "$line" 2> /dev/null; done`

already found my meter, and the reading is accurate. props for figuring this out!

RichyT, 2025-10-03:
I'll give it a go. Will the python run on Windows fine? I don't have a mobile Linux install.

klyubin, 2025-10-03:
The script should run fine on Windows -- it doesn't use any OS-specific stuff. Alternatively, if you paste the captured broadcast here, we can decide it via the script without you having to run it at all.

ddffnn, 2025-10-04:
Great work @klyubin. I'm so happy to see this working, and really hope that I can help by sharing more data I collect.

I ran my long cable and antenna out to the meter pit and will be collecting data for a while to try to get a bunch of different meter readings. I'm using the following command:
`rtl_433 -f 916.7M -s 1600k -M level -Y autolevel -R 0 -X 'name=badger,modulation=FSK_PCM,short=10,long=10,reset=1000,match={32}0xec62ec62' -F json`
The match seems to work regardless of byte alignment so all the bit patterns listed above are captured.

So far it looks like my endpoint is using 917.1/917.3 MHz and transmitting about once every 17 seconds. One serial number other than mine showed up, but I don't believe it was real. The SNR was very high and similar to the ones from my meter, so I expect it was just a corrupted capture. Otherwise I've only seen two unique meter readings from my serial number.

I too would love to find the leak bit, but finding the crc parameters would be great too.

ProfBoc75, 2025-10-04:
@klyubin  how did you find the keystream ? Because all values are the same as IBM Whitening Algorithm LFSR.

    IBM    ffe11d9aed [853324ea] [7ad239] [7097570a] [547d2dd8]    [6d0dba8f6759c7a2bf34ca18305393df92eca7158adcf486554e182140c4c4d5c6918acde7d14e093217df83]
                      Serial Num      ?   Reading    Daily Reading  ?

ProfBoc75, 2025-10-04:
I have some findings, the CRC/16 is after dewhitening the data with IBM key :

    width=16  poly=0x8005  init=0xffff  refin=false  refout=false  xorout=0x0000  check=0xaee7  residue=0x0000  name="CRC-16/CMS"

[Bitbench](https://triq.net/bitbench#c=555555555555557631763175f08ecd76b6a526f438c99d3c11ab85576716ec3e9fa70&c=555555555555557631763175f08ecd76b6a526f438c99d3e91ab85576716ec3e90570&c=555555555555557631763175f08ecd76b6a526f438c99d2791ab85576716ec3e8336%20%20%5B4614.36%5D&c=555555555555557631763175f08ecd76b6a526f438c99d0711ab85576716ec3e8c042%20%5B4620.64%5D&c=555555555555557631763175f08ecd76b6a526f438c99d0110ab85576716ec3e89420&c=555555555555557631763175f08ecd76b6a526f438c99d0e902b85576716ec3ee2508%20%5B4695.78%5D&c=555555555555557631763175f08ecd76b6a526f438c99d56102b85576716ec3eac658&c=&c=55555555555557631763175f08ecd76f08cf6f6b8e91d2b7d1c05aa3e96ec3e8f199f8&c=55555555555557631763175f08ecd76f08cf6f6b8e91d2bfd1c05aa3e96ec3ecc283f&c=55555555555557631763175f08ecd76f08cf6f6b8e91d02f21c05aa3e96ec3e856b9f8&c=&c=555555555555557631763175f08ecd76b6a526f438c99d2791ab85576716ec3e8336%20%5B%20sn%3A%2040466920%2C%20reading%3A%2046143%20%5D&c=555555555555557631763175f08ecd76b6a526f438c99d0e902b85576716ec3ee2508%20%5B%20sn%3A%2040466920%2C%20reading%3A%2046957%20%5D&c=55555555555557631763175f08ecd76f08cf6f6b8e91d02f21c05aa3e96ec3e856b9f8%20%5B%20sn%3A%20130624100%2C%20reading%3A%2024081269%20%5D&c=&c=aaaaaaaaaaaaaaec62ec62ebe11d9aed5f997ce87192319c76140a13a26ed87dee44%20%5B%20sn%3A%2039365338%2C%20reading%3A%204448748%20%5D&c=aaaaaaaaaaaaaaaec62ec62ebe11d9aed5f997ce87192314262140ab3956ed87d597e0%20%5B%20sn%3A%2039365338%2C%20reading%3A%204453682%2C%20dly_reading%3A%204450535%2C%20leak%3A%20False%20%5D&c=aaaaaaaaaaaaaaaec62ec62ebe11d9aed5f997ce871b231d968140ab3956ed87dbee0%20%5B%20sn%3A%2039365338%2C%20reading%3A%204456361%2C%20dly_reading%3A%204450535%2C%20leak%3A%20True%20%5D&c=aaaaaaaaaaaaaaaec62ec62ebe11d9aed5f997ce871b2318197130aa57d69d87da0630%20%5B%20sn%3A%2039365338%2C%20reading%3A%204456689%2C%20dly_reading%3A%204456689%2C%20leak%3A%20True%20%5D&f=%3F%2040h%20SN%20%3C32d%20FLAGS%2010h%20LEAK%201b%20%3F%2013h%20READING%20%3C32d%20DLY_READING%20%3C32d%20FIXED%3F8h%20CRC%2016h%20TRAIL%2016x%0A&z=1&a=Preamble&m=aaaaaaaaec62ec62&x2=ffe11d9aed853324ea7ad2397097570a547d2dd86d0dba8f6759c7a2bf34ca18305393df92eca7158adcf486554e182140c4c4d5c6918acde7d14e093217df83&cw=4) with few data collected here into the issue

Data layout: 184 bits then trailing bits, 23 Bytes are used.

    Byte Position      0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24
    Dewhitened Sample 14 00 00 00 00 da aa 58 02 0b 40 08 ec e1 43 00 47 df 43 00 10 e3 fe
                      II II II II II SS SS SS SS xx |x xx RR RR RR RR DD DD DD DD FF CC CC TT TT TT
                                                    |
                                                 +--+---+
                                                 | xxLx |
                                                 +------+

- II:{40} Fixed value, 0x1400000000
- SS: {32} Serial Number, little-endian value
- xx: Unknown, values look fixed and depends on the model, could be flags also, battery level too.
- L:{1} Leak 
- RR: {32} Reading value, scale 10 gallon, little-endian value
- DD: {32} Daily Reading Value, scale 10 gallon, little-endian value
- FF:{8} Fixed value, always 0x10
- CC:{16} CRC-16, poly 0x8005, init 0xFFFF, final XOR 0x0000
- TT: Trailing bits

ProfBoc75, 2025-10-04:
About the fixed 0x1400000000 could be 0x14 = 20 = message length of 20 bytes without CRC

klyubin, 2025-10-04:
🔥 @ProfBoc75 , this is amazing! How did you know this is IBM whitening?! Just a wild guess + it ended up matching the keystream chunks I determined earlier?

So, we basically need to XOR the packet after the preamble and sync, meaning from `eb` onwards with the IBM whitening key stream `ffe11d9aed853324ea7ad2397097570a547d2dd86d0dba`!!!

Here's the whitening keystream generator in Python for posterity based on section 5.1 from https://www.nxp.com/docs/en/application-note/AN5070.pdf

ProfBoc75, 2025-10-04:
> 🔥 [@ProfBoc75](https://github.com/ProfBoc75) , this is amazing! How did you know this is IBM whitening?!
Thanks to your findings I checked with the IBM key, and it matches. With Bitbench you have the feature to unwhiten using IBM or CCITT algorithm. And then, using your few sample codes here I was able to get the figures.
Using reveng I guessed the CRC with 4 codes, enough to guess CRC

RichyT, 2025-10-04:
OK, I only did a short run but not working for me. I seem to have quite a lot of noise so maybe I just need to run for longer. Here are the results I got which are not really short:

```
2025-10-04 16:52:14,,{107}aaaa2a357157626ca95e05273a8,FSK,,916.324,916.668,-5.294,24.298,-29.591,badger,1,1,{len: 107; data: aaaa2a357157626ca95e05273a8}
2025-10-04 16:52:30,,{211}aaaaaaaaaaaaaaad25a4b5c4d56a8b15d19525dd105952aa8eaa8,FSK,,916.341,916.680,-5.403,23.519,-28.922,badger,1,1,{len: 211; data: aaaaaaaaaaaaaaad25a4b5c4d56a8b15d19525dd105952aa8eaa8}
2025-10-04 16:52:46,,{155}b273767776ab49ee5d7539b9952b721caaecda8,FSK,,916.322,916.636,-5.410,23.310,-28.720,badger,1,1,{len: 155; data: b273767776ab49ee5d7539b9952b721caaecda8}
2025-10-04 16:53:03,,{167}acc964b7676a4f23f277a9cae6a92f442aa95e7d50,FSK,,916.359,916.665,-5.381,23.339,-28.720,badger,1,1,{len: 167; data: acc964b7676a4f23f277a9cae6a92f442aa95e7d50}
2025-10-04 16:53:20,,{209}aaaab5555555555a4b496b9355aa595d19525dd1054a955475548,FSK,,916.333,916.666,-5.340,24.252,-29.591,badger,1,1,{len: 209; data: aaaab5555555555a4b496b9355aa595d19525dd1054a955475548}
```

RichyT, 2025-10-04:
Ran for 10 minutes. These look maybe promising but don't have the correct preamble bytes.

```
2025-10-04 16:52:30,,{211}aaaaaaaaaaaaaaad25a4b5c4d56a8b15d19525dd105952aa8eaa8,FSK,,916.341,916.680,-5.403,23.519,-28.922,badger,1,1,{len: 211; data: aaaaaaaaaaaaaaad25a4b5c4d56a8b15d19525dd105952aa8eaa8}
2025-10-04 17:17:36,,{201}aaaaaaaaaaaaaaad25a4b5c9d5aa595a65496d1054a9569daa0,FSK,,916.331,916.634,-4.975,25.708,-30.683,badger,1,1,{len: 201; data: aaaaaaaaaaaaaaad25a4b5c9d5aa595a65496d1054a9569daa0}
2025-10-04 17:30:02,,{208}aaaaaaaaaaaaaaad25a4b5c4d56a9657465497e882a54aaa3b54,FSK,,916.326,916.638,-5.000,24.591,-29.591,badger,1,1,{len: 208; data: aaaaaaaaaaaaaaad25a4b5c4d56a9657465497e882a54aaa3b54}

2025-10-04 17:17:19,,{199}aaaaaaaaaaaaaaad27497726ab54b2b4ca92f44152a55676a8,FSK,,916.317,916.674,-4.972,24.161,-29.134,badger,1,1,{len: 199; data: aaaaaaaaaaaaaaad27497726ab54b2b4ca92f44152a55676a8}
2025-10-04 17:23:55,,{209}aaaaaaaaaaaaaaad27497713ab5458ae8ca92ee882ca955475540,FSK,,916.324,916.645,-5.069,24.288,-29.357,badger,1,1,{len: 209; data: aaaaaaaaaaaaaaad27497713ab5458ae8ca92ee882ca955475540}

```

RichyT, 2025-10-04:
I'll give that a go. Probably tomorrow AM now. Thanks.

shawntoffel, 2025-10-04:
This is really great @klyubin! I Ran new captures with the latest script:

```
aaaaaaaaaaaaaec62ec62ebe11d9aede119eded71d23a3209240b547d2dd87dfc7e08
serial_number: 130624100, reading: 24354370, daily_reading: 0, leak: False
aaaaaaaaaaaaabb18bb18baf84766bb78467b7b5c748e8d024902d51f4b761f7a4f804
serial_number: 130624100, reading: 24354372, daily_reading: 0, leak: False
aaaaaaaaaaaaabb18bb18baf84766bb78467b7b5c748e8b424902d51f4b761f4e9740
serial_number: 130624100, reading: 24354397, daily_reading: 0, leak: False
aaaaaaaaaaaaabb18bb18baf84766bb78467b7b5c748e8b824902d51f4b761f4c3f40
serial_number: 130624100, reading: 24354398, daily_reading: 0, leak: False
aaaaaaaaaaaaaec62ec62ebe11d9aede119eded71d23a7408240b547d2dd87d613e40
serial_number: 130624100, reading: 24354564, daily_reading: 0, leak: False
```

`rtl_433 -f 904.94M -s 1600k -M level -X 'name=badger,modulation=FSK_PCM,short=5,long=5,reset=1000,match={32}0xec62ec62'`

You are correct that this probably isn't a ORION ME ENC GIF2014W-OSE. It repeats every 15s, but I saw changes in the reading with intervals as low as 15 minutes. 130624100 is the only serial number I'm able to pick up which consistently had a RSSI around -5dB and SNR around 18 dB so seemed promising. However, I'm not able to verify that this is mine.

My meter is a GIF20180CELMDOM which, in addition to LTE, supposedly optionally broadcasts RF on the same frequencies I found the above. However, my serial number is 121125420 with a reading in the ballpark of 546973.01 so think I must be seeing a different meter.

klyubin, 2025-10-05:
Thanks, @shawntoffel . It's crazy that the above broadcasts un-whiten/decrypt well and pass the CRC check after which we end up with (minus CRC at the end) which looks pretty similar to other ones above, except for the 0 as the daily reading.
```
1400000000642ac9070b0003049f73010000000010
```
and yet neither your serial number nor the reading seem to be above (even when we bitshift by 0 to 4 bits). Is 546973.01 in gallons or in some other unit? What units does your water bill show? Is it similar to 546973?

What's the RSSI / SNR of these broadcasts? If these figures are high, it must be something very close by (up to 10 ft I'd say).

klyubin, 2025-10-05:
@shawntoffel, `130000000…139999999` serial numbers are supposed to be for ORION Cellular C (when FCC label is yellow) and for ORION Cellular CS (when FCC label is white). Whereas `120000000…129999999` are for ORION Cellular LTE-M (when FCC label is white) and ORION Cellular LTE-MS (when FCC label is white). This is according to the ORION Water Endpoints doc: https://badgermeter.widen.net/content/vodetxkyxh/original?download=false&x.app=api

So, it seems you're picking up broadcasts from an ORION Cellular C or CS, and yet yours is ORION Celluar LTE-M (according to the FCC ID GIF20180CELMDOM.

klyubin, 2025-10-05:
Oh, the docs don't list daily reading as a readout from ORION Cellular C and  Cellular LTE-M don't. So, maybe that explains the 0 as the daily reading?

klyubin, 2025-10-05:
Oh, excellent. I was about to start on a decoder in a few hours ;) Happy to review yours instead.

Gigoo25, 2025-10-05:
@ProfBoc75 

Thank you for this! Can't wait to try out the new decoder.

klyubin, 2025-10-05:
@RichyT , that ORION in your photo is weird. It's faicing up, but there's no lid nor antenna at the top. See https://github.com/merbanan/rtl_433/issues/2995#issue-2397360571 where the top/threaded end is in the metal lid/cover which is lying upside down on the ground. Here's mine: [image]: again, threaded end is screwed into the cover/lid and that's the antenna end.

RichyT, 2025-10-05:
Yes but the module does appear to be designed to be able to be attached in multiple ways. Might have to do with frost depth and lid-to-meter distances, meter body design or even just local preferences.

klyubin, 2025-10-05:
What weirder me out is that your ORION appears to be sitting deep inside the well, just next to the meter which is often 1-3ft deep. If so, how are its broadcasts supposed to reach the road from which the utility is usually reading them?

RichyT, 2025-10-05:
The meter is probably maybe 1ft deep which puts the antenna within a few inches of the lid and the lid is 1-2ft from the road. I'm not sure what the range is on these things but that seems like it would be about as reasonable as you could get.

I can probably have a quick go at reading the neighbor's tomorrow but I don't want to look too shady around the neighborhood.

klyubin, 2025-10-05:
FWIW, in my neighborhood these ORIONs are about 15 yards from the centerline of the road and I can easily receive broadcasts from mine and my neighbors' ORIONs by walking that road with a 10 dBi 915 MHz antenna attached to a RTL-SDR Blog v4.

[image] -- the plastic in the center of the cast iron lid there is where the top of the ORION is hiding. It's screwed into the lid from the bottom.

RichyT, 2025-10-06:
I only have the cheapy antenna that came with the dongle but it's easily capable of picking up multiple devices a fair distance away (my weatherstation in my back yard, for example). I did also get some other devices on the Orion frequency when I tested it on the PC I usually run it from with the built-in decoder but they were not the Orion devices (obviously).

I'll try with the new version of rtl_433 and see if I can get anything wandering around. It's not impossible the Orion signal isn't getting out but it seems unlikely but not a problem to try with removing the lid and good to eliminate the possibility.

ddffnn, 2025-10-06:
When I wrote my last post two days ago (late Friday night) I was receiving codes ever 17s or so. I left it running over the weekend and just checked on it now. Oddly, I haven't seen any codes since about 2am on Saturday. I compiled @ProfBoc75's branch and am running it now, but still haven't seen my meter yet. I see transmissions from a neighbor's meter matching protocol 223, so I don't think my dongle is to blame. Would it make any sense that the RF module stops transmitting on the weekend since the city is unlikely to be driving around collecting readings? Anyway, I'll leave it running and report back within the next few days.

No surprise: @ProfBoc75 branch decodes all my recorded codes from Friday night.

klyubin, 2025-10-06:
@ddffnn , have you tried lookinhg for your ORION at other frequencied? My ORION keeps hopping. I've seen it on various frequencies between 915.3 MHz and 917.7 MHz.

ddffnn, 2025-10-06:
🤦  I'm seeing it around 905.7Mhz now. Thanks @klyubin.

ProfBoc75, 2025-10-06:
> I mean 905.7MHz. This is in line with the test report on the FCC listing for the module. From what I can tell hopping in mobile mode uses 48 carrier frequencies at 400kHz spacing from 904.9MHz up to 923.7MHz. I'm running an experiment now with 24 frequencies at 800kHz spacing and 1200kHz sampling rate. Since the updates only occur once per hour I think this could a long term solution. It will be interesting to see how it hops around over time though.
> 
> https://fccid.io/GIF2014W-OSE/Test-Report/Exhibit-D-Users-Manual-per-2-1033-b3-2499315

And may be the same issue for @RichyT , who needs to add more frequencies to find the signal within the 48 channels.

merbanan, 2025-10-06:
> What weirded me out is that your ORION appears to be sitting deep inside the shaft, just next to the meter which is often 1-3ft deep. If so, how are its broadcasts supposed to reach the road from which the utility is usually reading them?

In Sweden we have garbage trucks reading the water meters (wmbus). If the transmissions are semi frequent then this is possible. If it is once every hour there must be some kind of stationary relay. You can check how gridstream works.

ddffnn, 2025-10-07:
I'm using 24 frequencies 905.3, 906.1, ... 923.7, hop interval of 150s, and 1200kHz sample rate:

| time                       | freq1   | freq2   | volume_gal |
|----------------------------|---------|---------|------------|
| 2025-10-06 00:41:18.425238 | 905.619 | 905.815 | 93319.898  |
| 2025-10-06 00:44:16.053721 | 905.572 | 905.750 | 93319.898  |
| 2025-10-06 01:22:30.733035 | 911.633 | 911.758 | 93319.898  |
| 2025-10-06 04:41:13.316287 | 918.040 | 918.171 | 93325.203  |
| 2025-10-06 05:20:57.365677 | 911.990 | 912.147 | 93325.203  |
| 2025-10-06 06:02:00.812240 | 905.593 | 905.751 | 93325.203  |
| 2025-10-06 08:20:28.178126 | 911.995 | 912.135 | 93334.305  |
| 2025-10-06 09:04:35.997282 | 905.659 | 905.812 | 93334.305  |
| 2025-10-06 09:24:51.310907 | 912.066 | 912.212 | 93339.000  |
| 2025-10-06 10:02:44.355745 | 905.610 | 905.746 | 93339.000  |
| 2025-10-06 10:21:00.769039 | 912.007 | 912.142 | 93339.000  |
| 2025-10-06 10:24:03.532476 | 912.062 | 912.203 | 93339.000  |
| 2025-10-06 10:42:35.379631 | 918.389 | 918.544 | 93339.000  |
| 2025-10-06 15:26:32.010478 | 912.861 | 913.077 | 93343.398  |
| 2025-10-06 16:05:12.067440 | 905.659 | 905.822 | 93343.398  |
| 2025-10-06 17:41:19.522820 | 918.389 | 918.548 | 93343.398  |
| 2025-10-06 17:42:08.986151 | 918.387 | 918.550 | 93343.398  |
| 2025-10-06 18:02:52.673741 | 905.582 | 905.747 | 93343.398  |
| 2025-10-06 18:03:24.928516 | 905.658 | 905.837 | 93343.398  |
| 2025-10-06 18:20:47.826438 | 912.014 | 912.130 | 93349.898  |
| 2025-10-06 18:21:54.334530 | 912.007 | 912.139 | 93349.898  |
| 2025-10-06 18:24:56.798202 | 912.827 | 912.943 | 93349.898  |

So yeah, quite a bit of hopping for me too. I was hoping to get at least one reading per hour, but that's clearly not working. I'll probably fiddle around with different hop intervals and sampling rates to see if I can get smaller gaps, but I'm not holding out hope.

ddffnn, 2025-10-07:
I used to have a long cable running out to the meter pit so I could put the antenna right on the metal cover, but that didn't mix so well with a couple young kids. I've got my antenna inside of the window about 25ft from the pit now. I usually get rssi -9, noise -34, and snr 25 ish.

I'm trying 25s intervals and 1600kHz sampling to see if that helps.

klyubin, 2025-10-07:
TIL! Thank you, @zuckschwerdt!

[16 more replies]