merbanan/rtl_433 #2999: WH46 sensor readings. How can incorporate it into rtl_433?.

joanma747, 2024-07-17:
I have bought a Ecowitt WH46: https://www.ecowitt.com/shop/goodsDetail/296

I see that is not initially supported in RTL_433 but WH45 is: https://github.com/merbanan/rtl_433/issues/2099, https://github.com/merbanan/rtl_433/pull/2101

I have reused this line:
rtl_433 -c 0 -R 0 -X "n=DP250,m=FSK_PCM,s=56,l=56,r=1500,preamble=0x2dd4"
to get some samples of the data extracted and comparing with the WH45 it seems very similar, so this is my guess on the interpretation based on the WH45:

```
YY IIIIII 0TTT HH Bppp BPPP CCCC qqqq QQQQ ???? XX AA
46 0027f1 02b4 33 4022 402d 0325 0019 002a 0190 97 a9
46 0027f1 02b4 33 4024 402d 0320 001c 002a 0190 73 85
46 0027f1 02b4 33 401f 4020 031c 001d 0020 0190 95 88
46 0027f1 02b4 33 401d 401e 033d 001b 001e 0190 43 4f
46 0027f1 02b4 33 4026 4029 0334 0022 0028 0190 fe 26
46 0027f1 02b4 33 4024 4027 0330 0020 0026 0190 24 40
46 0027f1 02b4 33 401a 401a 0316 0017 001a 0190 d7 ad
46 0027f1 02b4 33 401f 4023 0315 001b 0022 0190 49 38
46 0027f1 02b5 33 4025 402a 0313 001f 0029 0190 12 18
46 0027f1 02b5 33 4028 402e 0310 0021 002c 0190 79 88
46 0027f1 02b4 33 402c 4034 0321 0025 0031 0190 3a 6c
46 0027f1 02b4 33 4029 402d 0326 0023 002c 0190 a9 cf
46 0027f1 02b5 33 4028 402a 0324 0023 0029 0190 60 7e
46 0027f1 02b4 33 4025 402a 031b 0020 0028 0190 91 9e
46 0027f1 02b5 33 401f 4025 0319 0018 0023 0190 75 69
46 0027f1 02b5 33 402a 4034 0318 0020 0031 0190 ad d0
46 0027f1 02b5 33 402e 4039 0316 0024 0035 0190 c5 f7
46 0027f1 02b5 33 402c 4034 0314 0023 0031 0190 f7 1b
46 0027f1 02b5 33 4035 4042 0312 0029 003d 0190 97 e2
46 0027f1 02b5 33 402f 4038 030b 0025 0035 0190 0b 33
46 0027f1 02b5 33 4028 402c 030c 0023 002a 0190 b7 c0
46 0027f1 02b5 33 4032 4039 030b 002a 0036 0190 e4 16
46 0027f1 02b8 3c 403a 403e 13cb 0034 003d 0190 61 8d <-- blow into the sensor. 
46 0027f1 02bb 37 4042 4047 0b83 003a 0045 0190 3a 33
46 0027f1 02bb 34 4037 403d 04dd 0030 003b 0190 d3 f3
```

- Y: 8 bit fixed sensor type 0x45
- I: 24 bit device ID
- T: 11 bit temperature, offset 40, scale 10
- H: 8 bit humidity
- B: 1 bit MSB of battery bars out of 5 (a value of 6 indicates external power via USB)
- p: 14 bit PM2.5 reading in ug/m3 * 10
- B: 2 bits LSBs of battery bars out of 5
- P: 14 bit PM10 reading in ug/m3 * 10
- C: 16 bit CO2 reading in ppm
- q: 14 bit PM1 or PM4 reading in ug/m3 * 10
- Q: 14 bit PM1 or PM4 reading in ug/m3 * 10
- ?: Constant value
- X: 8 bit CRC
- A: 8 bit checksum

but this is just a guess, as I do not have any "receiver" to validate this readings.

The questions are: 
Has anyone else tried to incorporate this sensor? 
Can anyone verify the meaning of p, P, q, Q and '?'  ?
Is it possible to incorporate this sensor to the RTL433? 
Can I help?

Thanks

ProfBoc75, 2024-07-19:
Hi @joanma747 

The last ?? constant value 0x0190  is probably the firmware version 1.9.0 or 1.4.4, like with WS85 / WS90.

As I didn't find any firmware information for WH46, here my assumption:

The firmware is code like that, for example:

- {8} 0x7B = 123 = version is 1.2.3, here you have the last byte = 0x90 = 144 = version 1.4.4 ?
- Or, if last 2 bytes is the firmware and BCD, you have {16} 0x0190 = version is 1.9.0 ?  

This is purely my assumption to be confirmed with a gateway ...

joanma747, 2024-07-27:
Please consider my pull request implementing what we know so far:

https://github.com/merbanan/rtl_433/pull/3010

gdt, 2024-07-29:
Issue closed in favor of PR #3010 (which is great progress, not a complaint!).

closed 2024-07-29 by gdt