merbanan/rtl_433 #3000: Frequency hopping issues

agc346, 2024-07-17:
I have tried frequency hopping from 433 to 915 with no luck. Whatever frequency is specified first in the command works, but the second does not. I have tried two SDRs with mixed success, however I am unable to get two SDRs to feed information into weewx. A couple commands I have tried are:

rtl_433 -M utc -F json -f 433.92M -f 914.98M -H 35 -R 40 -R 78
rtl_433 -d 0 -M utc -F json -R 40 & rtl_433 -d 1 -M utc -F json -f 914980000 -R 78

These work, although I am unable to combine them to receive data.
rtl_433 -d 0 -M utc -F json -R 40
rtl_433 -d 1 -M utc -F json -f 914980000 -R 78

I would like to have frequency hopping working so I only need one sdr.

zuckschwerdt, 2024-07-17:
Do you see the notice about hopping every 35 seconds?
With current rtl_433 you will need to add the kv or log output to get messages when using another output like json.

Maybe try just `rtl_433 -v -f 433.92M -f 914.98M -H 35 -R 40 -R 78 -M utc` (note that order can be important and input should go first, outputs last)

gdt, 2024-07-29:
This got a hint and has gone dormant.  Please feel free to ask on the list or open a discussion for further help, but this doesn't seem like something that belongs in an issue.

closed 2024-07-29 by gdt