merbanan/rtl_433 #3011: JSONL is printed in multiple chunks of data.

adminy, 2024-07-29:
I have rtl_433 wrapped inside a node process and use node streams from a spawned process to parse the json coming out of rtl_433, however I noticed that in the latest version, jsonl is chunking across multiple stream chunks / lines, which breaks `JSONStream` package that parses jsonl.

adminy, 2024-07-29:
I wasn't too sure whether I should open this issue or not, just thought maybe the last few git pushes have something, because all I did was update the hash of the source of rtl_433, then everything broke moving from `34c1e67d8c4d34faee704274688058d6a164bdc2` to `db2db908f373cf3c48ef663b75ea7d0c71d69740`. I don't need help with anything, also I don't know, maybe this is just intended that the stream doesn't conform with jsonl specifications, but wanted to confirm that was the case.

gdt, 2024-07-29:
If you are able to really explain what you mean and post a reproducer I suspect someone would look at it.    But I can't understand at all what you mean (still), and that sort of discussion doesn't really belong in issues.  I'm not really aware of rtl_433 doing jsonl, and you didn't give example output that you can argue is noncompliant.  So this is just too vague for anyone to understand, so far.

closed 2024-07-29 by gdt

zuckschwerdt, 2024-07-29:
> jsonl is chunking across multiple stream chunks / lines

The `-F json` output is JSONL. Are you hinting that there are newlines embedded inside the JSON lines? How to reproduce?