merbanan/rtl_433 #3012: Support for Apator Metra E-RM water meter

carpalex, 2024-07-29:
Hello,

I'm trying to decode the telegrams coming from some Apator Metra water meters, equipped with the [E-RM 30](https://metra-su.cz/en/product/electronic-radio-module-e-rm-30-for-residential-water-meters-apator-powogaz) radio module.

They transmit on 868.9M and I managed to capture some samples at 1M sample rate. The modulation seems to be FSK, but I wasn't able to figure out the data encoding. Here is an archive with samples captured from 6 different meters: [samples.zip](https://github.com/user-attachments/files/16412214/samples.zip). The folders in the archive have the following format for their names: `SERIALNUMBER_METERVALUE` (e.g. `80121968_00002875` contains samples from a water meter with `80121968` serial number and displaying `00002.875 m3` value.

I attempted to use the following flex decoder: `-X "n=test,m=FSK_PCM,s=25,l=25,r=5000"`, which outputs plausible bitstrings.

For one of the meters, I captured samples at consecutive meter values, and there are changes in the bitstrings only at a certain offset, which may mean that the data is not encrypted. For example (meter values are in square brackets):

```
{239}555555555554d335d89b1d36cadb9c49d4f5a472e12eae14f43bcf7161f4 [00002873]
{239}555555555554d335d89b1d36cad79c49d4f5a472e12eae14f43bcf716ddc [00002874]
{239}555555555554d335d89b1d36ca7b9c49d4f5a472e12eae14f43bcf71c234 [00002875]
{239}555555555554d335d89b1d36ca779c49d4f5a472e12eae14f43bcf71ce1c [00002876]
{239}555555555554d335d89b1d36ca5b9c49d4f5a472e12eae14f43bceabe628 [00002877]
{239}555555555554d335d89b1d36ca579c49d4f5a473732eae14fc3bcea46f18 [00002878]
{239}555555555554d335d89b1d36cbbb9c49d4f5a473732eae14fc3bcea58770 [00002879]
{239}555555555554d335d89b1d36cbb79c49d4f5a473732eae14fc3bcea58b58 [00002880]
{239}555555555554d335d89b1d36cb9b9c49d4f5a473732eae14fc3bcf7fa36c [00002881]
{239}555555555554d335d89b1d36cb979c49d4f5a473732eae14fc3bcf7faf44 [00002882]
```

I can provide additional samples if needed.

Thanks in advance!

zuckschwerdt, 2024-07-29:
Aligned to `34cd` the checksum works as CRC-16 poly 0x8005  init 0x1aaa.
Strange however that only two fields in that alignment change, and not in a linear way.
See [this BitBench](https://triq.net/bitbench#c=%7B239%7D555555555554d335d89b1d36cadb9c49d4f5a472e12eae14f43bcf7161f4%20%5B00002873%5D&c=%7B239%7D555555555554d335d89b1d36cad79c49d4f5a472e12eae14f43bcf716ddc%20%5B00002874%5D&c=%7B239%7D555555555554d335d89b1d36ca7b9c49d4f5a472e12eae14f43bcf71c234%20%5B00002875%5D&c=%7B239%7D555555555554d335d89b1d36ca779c49d4f5a472e12eae14f43bcf71ce1c%20%5B00002876%5D&c=%7B239%7D555555555554d335d89b1d36ca5b9c49d4f5a472e12eae14f43bceabe628%20%5B00002877%5D&c=%7B239%7D555555555554d335d89b1d36ca579c49d4f5a473732eae14fc3bcea46f18%20%5B00002878%5D&c=%7B239%7D555555555554d335d89b1d36cbbb9c49d4f5a473732eae14fc3bcea58770%20%5B00002879%5D&c=%7B239%7D555555555554d335d89b1d36cbb79c49d4f5a473732eae14fc3bcea58b58%20%5B00002880%5D&c=%7B239%7D555555555554d335d89b1d36cb9b9c49d4f5a473732eae14fc3bcf7fa36c%20%5B00002881%5D&c=%7B239%7D555555555554d335d89b1d36cb979c49d4f5a473732eae14fc3bcf7faf44%20%5B00002882%5D&f=8h8h%208h8h%208h8h8h%20%3F8d%208h8h%208h8h%208h8h%20%3F8h8h%208h8h%20%3F8h%208h8h%20%3F8d%20CRC%3A16h%208x&z=1&a=Sync&m=34cd&cw=4).

carpalex, 2024-07-29:
Applying the same alignment to readings from a meter in which the value is constant, it seems that the second field that changes can be some sort of timestamp / sequence number.
(bitbench [here](https://triq.net/bitbench#c=%7B239%7D555555555554d335d89b1d36cadb9c49d4f5a472e12eae14f43bcf7161f4%20%5B00002873%5D&c=%7B239%7D555555555554d335d89b1d36cad79c49d4f5a472e12eae14f43bcf716ddc%20%5B00002874%5D&c=%7B239%7D555555555554d335d89b1d36ca7b9c49d4f5a472e12eae14f43bcf71c234%20%5B00002875%5D&c=%7B239%7D555555555554d335d89b1d36ca779c49d4f5a472e12eae14f43bcf71ce1c%20%5B00002876%5D&c=%7B239%7D555555555554d335d89b1d36ca5b9c49d4f5a472e12eae14f43bceabe628%20%5B00002877%5D&c=%7B239%7D555555555554d335d89b1d36ca579c49d4f5a473732eae14fc3bcea46f18%20%5B00002878%5D&c=%7B239%7D555555555554d335d89b1d36cbbb9c49d4f5a473732eae14fc3bcea58770%20%5B00002879%5D&c=%7B239%7D555555555554d335d89b1d36cbb79c49d4f5a473732eae14fc3bcea58b58%20%5B00002880%5D&c=%7B239%7D555555555554d335d89b1d36cb9b9c49d4f5a473732eae14fc3bcf7fa36c%20%5B00002881%5D&c=%7B239%7D555555555554d335d89b1d36cb979c49d4f5a473732eae14fc3bcf7faf44%20%5B00002882%5D&f=8h8h%208h8h%208h8h8h%20%3F8d%208h8h%208h8h%208h8h%20%3F8h8h%208h8h%20%3F8h%208h8h%20%3F8d%20CRC%3A16h%208x&z=1&a=Sync&m=34cd&cw=4) - serial: `79909924`, meter value: `00000055`)

Most likely the first variable field participates in the encoding of the meter value, but in a non-trivial way.

zuckschwerdt, 2024-07-29:
The BitBench link is just the old data. Your findings of a sequence/timestamp at the end are plausible.
There might be a simple key like xor to preserve privacy. A plain text search for the id and counter could work.

carpalex, 2024-07-29:
Updated the bitbench link, in my previous comment.

ProfBoc75, 2024-07-30:
Hi, this water meter was already part of this [isssue/discussion](https://github.com/wmbusmeters/wmbusmeters/issues/368#issuecomment-999417215)

Sounds like a data coding is present and need to be decoded to get the figures ...

carpalex, 2024-07-31:
> Hi, this water meter was already part of this [isssue/discussion](https://github.com/wmbusmeters/wmbusmeters/issues/368#issuecomment-999417215)
> 
> Sounds like a data coding is present and need to be decoded to get the figures ...

Yes, but in the discussion you mentioned there were no samples provided, nor attempts to analyze the bit strings.
I think it's worth taking another look to try to analyze the message format and identify if/what fields are encrypted or not.

@zuckschwerdt already identified the CRC, I'm trying to figure out where the length can be, but with no success so far.
@merbanan Any ideas on your side?

zuckschwerdt, 2024-07-31:
What I mean is: if we look at the first variable field and watch the carry then we can adjust the bits so only a set bit does carry.
E.g. try `8h8h 8h8h 8h8h8h ?2b.2b.2b.2b 8h8h 8h8h 8h8h ?8h8h 8h8h ?8h 8h8h ?8d CRC:16h 8x`
then use `2a` as XOR and the values are ascending.

But the steps greater than 1 in that value, is your reading in gallons or m3? Are there values in between, i.e. different messages for the same reading?

Or there could be a mixing of bit-positions. Looking at that value's bits we see:
both last columns could the 1-bit (toggle with every row), the first and 6th column are MSBs (fixed value), the 4th column is the 2-bit (toggle every 2nd row), the 3rd column is the 4-bit (as in `0x40`) and the 2nd column maybe the 8-bit.

carpalex, 2024-07-31:
Thanks for elaborating. I'll capture more samples for consecutive values and analyze them as you suggested. I'll also post the bit strings.

Regarding your questions, the readings are in m3, with 3 decimal values, i.e possible values from 00000.000 to 99999.999. So I suppose that the encoded value is first multiplied by 1000, i.e from 0 to 99999999, which should require 32 bits.
There are no values in between. If I keep a meter with a constant value, it always transmits the same telegram (apart from the CRC and the field towards the end, which we suspect to be a sort of sequence, and also repeats every few messages). 

Related to the samples: can we be sure that the flex decoder parameters are correct (`m=FSK_PCM,s=25,l=25,r=5000`)? Should we try with different ones?

zuckschwerdt, 2024-07-31:
The samples are FSK and PCM. The bit width looks close enough to 25 µs. rtl_433 will auto-adjust on the preamble (0x5555). And the CRC checks out. All good there. Alignment is fixed by the CRC. The bits could be inverted though.

Also the poly 0x8005 is well known. The init 0x1aaa seems random but perhaps we need to start somewhere else or include some other value to get a nice number, random init's are used though.

carpalex, 2024-08-02:
Ok, so I managed to collect samples for 65 consecutive meter values, and I think I identified a pattern. Here is [the bitbench](https://triq.net/bitbench#c=%7B239%7D555555555554d335d89b1d36cb1a1a49d4f5a473732eae14b023cf6b566a%20%5B00004964%5D&c=%7B239%7D555555555554d335d89b1d36cb161a49d4f5a473732eae14b023cf705a18%20%5B00004965%5D&c=%7B239%7D555555555554d335d89b1d36ca9a1a49d4f5a473732eae14b023ceb0d7b6%20%5B00004966%5D&c=%7B239%7D555555555554d335d89b1d36ca961a49d4f5a473732eae14b023cf71df18%20%5B00004967%5D&c=%7B239%7D555555555554d335d89b1d36ca1a1a49d4f5a473732eae14b023ceb054b6%20%5B00004968%5D&c=%7B239%7D555555555554d335d89b1d36ca161a49d4f5a473732eae14b023ceab58c4%20%5B00004969%5D&c=%7B238%7D555555555554d335d89b1d36cbda1a49d4f5a473732eae14b023cf6b94e8%20%5B00004970%5D&c=%7B239%7D555555555554d335d89b1d36cbd61a49d4f5a473732eae14b023ceb19c1e%20%5B00004971%5D&c=%7B237%7D555555555554d335d89b1d36cbfa1a49d4f5a473732eae14b023ceb1b0f0%20%5B00004972%5D&c=%7B239%7D555555555554d335d89b1d36cbf61a49d4f5a473732eae14b023cf70b858%20%5B00004973%5D&c=%7B239%7D555555555554d335d89b1d36cafa1a49d4f5a473732eae14b023ceabb6ac%20%5B00004974%5D&c=%7B240%7D555555555554d335d89b1d36caf61a49d4f5a473732eae14b023ceb0badf%20%5B00004975%5D&c=%7B239%7D555555555554d335d89b1d36ca3a1a49d4f5a473732eae14b023ceab742c%20%5B00004976%5D&c=%7B235%7D555555555554d335d89b1d36ca361a49d4f5a473732eae14b023ceb0784%20%20%5B00004977%5D&c=%7B239%7D555555555554d335d89b1d36cb3a1a49d4f5a473732eae14b023cf7076f0%20%5B00004978%5D&c=%7B240%7D555555555554d335d89b1d36cb361a49d4f5a473732eae14b023cf6b7a83%20%5B00004979%5D&c=%7B239%7D555555555554d335d89b1d36caba1a49d4f5a473732eae14b023cf71f3f0%20%5B00004980%5D&c=%7B239%7D555555555554d335d89b1d36cab61a49d4f5a473732eae14b023cf71ffd8%20%5B00004981%5D&c=%7B239%7D555555555554d335d89b1d36cb7a1a49d4f5a473732eae14b023ceb133f6%20%5B00004982%5D&c=%7B239%7D555555555554d335d89b1d36cb761a49d4f5a473732eae14b023ceaa3f84%20%5B00004983%5D&c=%7B239%7D555555555554d335d89b1d36cada1a49d4f5a473732eae14b023cf6a92ea%20%5B00004984%5D&c=%7B240%7D555555555554d335d89b1d36cad61a49d4f5a473732eae14b023ceb09a1e%20%5B00004985%5D&c=%7B237%7D555555555554d335d89b1d36ca7a1a49d4f5a473732eae14b023ceb035f0%20%5B00004986%5D&c=%7B240%7D555555555554d335d89b1d36ca761a49d4f5a473732eae14b023cf713d58%20%5B00004987%5D&c=%7B238%7D555555555554d335d89b1d36ca5a1a49d4f5a473732eae14b023cf6a11e8%20%5B00004988%5D&c=%7B239%7D555555555554d335d89b1d36ca561a49d4f5a473732eae14b023ceb0191e%20%5B00004989%5D&c=%7B239%7D555555555554d335d89b1d36cbba1a49d4f5a473732eae14b023cf70f5f0%20%5B00004990%5D&c=%7B235%7D555555555554d335d89b1d36cbb61a49d4f5a473732eae14b023ceb1fd4%20%20%5B00004991%5D&c=%7B239%7D555555555554d335d89b1d36cb9a1a49d4f5a473732eae14b023cf6bd56a%20%5B00004992%5D&c=%7B238%7D555555555554d335d89b1d36cb961a49d4f5a473732eae14b023cf6bd940%20%5B00004993%5D&c=%7B239%7D555555555554d335d89b1d36cb5a1a49d4f5a473732eae14b023ceaa136c%20%5B00004994%5D&c=%7B240%7D555555555554d335d89b1d36cb561a49d4f5a473732eae14b023ceaa1f44%20%5B00004995%5D&c=%7B238%7D555555555554d335d89b1d36cb1a1049d4f5a473732eae14b023cf705c38%20%5B00004996%5D&c=%7B239%7D555555555554d335d89b1d36cb161049d4f5a473732eae14b023cf6b5048%20%5B00004997%5D&c=%7B239%7D555555555554d335d89b1d36ca9a1049d4f5a473732eae14b023cf6ad960%20%5B00004998%5D&c=%7B239%7D555555555554d335d89b1d36ca961049d4f5a473732eae14b023ceb0d194%20%5B00004999%5D&c=%7B239%7D555555555554d335d89b1d36ca1a1049d4f5a473732eae14b023ceb05ebc%20%5B00005000%5D&c=%7B238%7D555555555554d335d89b1d36ca161049d4f5a473732eae14b023cf715610%20%5B00005001%5D&c=%7B238%7D555555555554d335d89b1d36cbda1049d4f5a473732eae14b023cf709eb8%20%5B00005002%5D&c=%7B239%7D555555555554d335d89b1d36cbd61049d4f5a473732eae14b023ceb19614%20%5B00005003%5D&c=%7B239%7D555555555554d335d89b1d36cbfa1049d4f5a473732eae14b823cf711e7c%20%5B00005004%5D&c=%7B239%7D555555555554d335d89b1d36cbf61049d4f5a473732eae14b823ceab1688%20%5B00005005%5D&c=%7B239%7D555555555554d335d89b1d36cafa1049d4f5a473732eae14b823ceaa1ca0%20%5B00005006%5D&c=%7B238%7D555555555554d335d89b1d36caf61049d4f5a473732eae14b823ceb110d0%20%5B00005007%5D&c=%7B239%7D555555555554d335d89b1d36ca3a1049d4f5a473732eae14b823ceaade20%20%5B00005008%5D&c=%7B238%7D555555555554d335d89b1d36ca361049d4f5a473732eae14b823ceb1d250%20%5B00005009%5D&c=%7B239%7D555555555554d335d89b1d36cb3a1049d4f5a473732eae14b823ceabd820%20%5B00005010%5D&c=%7B238%7D555555555554d335d89b1d36cb361049d4f5a473732eae14b823ceb0d450%20%5B00005011%5D&c=%7B239%7D555555555554d335d89b1d36caba1049d4f5a473732eae14b823cf7059fc%20%5B00005012%5D&c=%7B236%7D555555555554d335d89b1d36cab61049d4f5a473732eae14b823cf6b558%20%20%5B00005013%5D&c=%7B238%7D555555555554d335d89b1d36cb7a1049d4f5a473732eae14b823ceb099f8%20%5B00005014%5D&c=%7B239%7D555555555554d335d89b1d36cb761049d4f5a473732eae14b823ceb095d2%20%5B00005015%5D&c=%7B239%7D555555555554d335d89b1d36cada1049d4f5a473732eae14b823cf6b38e6%20%5B00005016%5D&c=%7B239%7D555555555554d335d89b1d36cad61049d4f5a473732eae14b823cf703494%20%5B00005017%5D&c=%7B238%7D555555555554d335d89b1d36ca7a1049d4f5a473732eae14b823ceb19ff8%20%5B00005018%5D&c=%7B239%7D555555555554d335d89b1d36ca761049d4f5a473732eae14b823cf709754%20%5B00005019%5D&c=%7B238%7D555555555554d335d89b1d36ca5a1049d4f5a473732eae14b823ceb1bf38%20%5B00005020%5D&c=%7B240%7D555555555554d335d89b1d36ca561049d4f5a473732eae14b823ceaab348%20%5B00005021%5D&c=%7B237%7D555555555554d335d89b1d36cbba1049d4f5a473732eae14b823cf6a5fa0%20%5B00005022%5D&c=%7B239%7D555555555554d335d89b1d36cbb61049d4f5a473732eae14b823ceb05752%20%5B00005023%5D& [... cut ...]
Which means that the values for this field should wrap around after 32 iterations, which does happen (it can be seen that they repeat in the same order). However, I couldn't identify a simple transformation between the last 5 LSBs of the counter value and the 5 bits in the field. Maybe it's just a hardcoded lookup table? Any ideas here?

Now, after 32 increments, the carry seems to be in the next 2 bytes, so now I'm collecting samples for increments of 32 of the meter value, maybe I can notice another pattern there.

Do my assumptions seem correct for now?

zuckschwerdt, 2024-08-02:
Yes, perfect. Seeing a (or multiple) full range of the 5 LSBs we can try a table or some clever shuffling.

The overflow clearly marks the first and last value, strange that if we adjust the first value to 0's the last is not all 1's -- not a simple xor then. At least the positions of the bits seem in a fixed order, which I would not expect with a table.

It will be interesting to see if this "5-of-8" coding for the lower byte is then the same in the upper byte.

merbanan, 2024-08-03:
I think it is a 6in8 coding with some xor mask.

[0]0011[0]01 [0]1111[0]11

The 1st and 6th columns are 0 but the 8th and 8th bits is just something related to the lsb (both bits).

If we take byte 8 and 9 we get 12 bits which is somewhat large enough to encode the meter value.

zuckschwerdt, 2024-08-03:
Plausible. There is however a slight chance that a 16-bit value is encoded and the 1st and 6th column are MSBs?

merbanan, 2024-08-03:
If there is a linear relation we can solve this by matrix inversion. But going from 4994 to 4995 changes 2 bytes (byte 9 and byte 10 thus we could even have 3 bytes). 4975 to 4976 changes 2 bits, this implies some kind of extra operation ie xor because of bit carry.

zuckschwerdt, 2024-08-03:
Noticed that too, value steps sometimes affect a variable number of coded bits. It's very much not linear. But also bit columns seem fixed, not random like an S-Box.

merbanan, 2024-08-03:
@carpalex I think we need more values.

carpalex, 2024-08-04:
> If there is a linear relation we can solve this by matrix inversion. But going from 4994 to 4995 changes 2 bytes (byte 9 and byte 10 thus we could even have 3 bytes). 4975 to 4976 changes 2 bits, this implies some kind of extra operation ie xor because of bit carry.

At a value step, I noticed that at least 2 coded bits are changing. Could it be that the 8th column is just a parity bit?

> @carpalex I think we need more values.

I captured more samples [here](https://triq.net/bitbench#c=%7B239%7D555555555554d335d89b1d36cb1a1a49d4f5a473732eae14b023cf6b566a%20%5B00004964%5D&c=%7B238%7D555555555554d335d89b1d36cb1a1049d4f5a473732eae14b023cf705c38%20%5B00004996%5D&c=%7B238%7D555555555554d335d89b1d36cb1a1249d4f5a473732eae14b823cf6bfe60%20%5B00005028%5D&c=%7B238%7D555555555554d335d89b1d36cb1a0c49d4f5a473732eae14b823ceaae4f8%20%5B00005060%5D&c=%7B239%7D555555555554d335d89b1d36cb1a0e49d4f5a473732eae14b823ceb0e6a4%20%5B00005092%5D&c=%7B239%7D555555555554d335d89b1d36cb1a0249d4f5a473732eae14b823ceb0eaa8%20%5B00005124%5D&c=%7B238%7D555555555554d335d89b1d36cb1ba649d4f5a473732eae14a423ceb1ff08%20%5B00005156%5D&c=%7B239%7D555555555554d335d89b1d36cb1bbe49d4f5a473732eae14a423cf70e394%20%5B00005188%5D&c=%7B237%7D555555555554d335d89b1d36cb1bb649d4f5a473732eae14a423ceabef40%20%5B00005220%5D&c=%7B239%7D555555555554d335d89b1d36cb1baa49d4f5a473732eae14a423cf70f780%20%5B00005252%5D&c=%7B238%7D555555555554d335d89b1d36cb1ba849d4f5a473732eae14a423ceb0f100%20%5B00005284%5D&c=%7B239%7D555555555554d335d89b1d36cb1bb849d4f5a473732eae14a423cf6be5c8%20%5B00005316%5D&c=%7B239%7D555555555554d335d89b1d36cb1bb449d4f5a473732eae14a423ceb1ed18%20%5B00005348%5D&c=%7B239%7D555555555554d335d89b1d36cb1ba449d4f5a473732eae14a423ceabfd54%20%5B00005380%5D&c=%7B239%7D555555555554d335d89b1d36cb1bbc49d4f5a473732eae14a423cf71e190%20%5B00005412%5D&c=%7B238%7D555555555554d335d89b1d36cb1ba049d4f5a473732eae14a423cf70fd88%20%5B00005444%5D&c=%7B239%7D555555555554d335d89b1d36cb1bba49d4f5a473732eae14a623cf70cf90%20%5B00005476%5D&f=8h8h%208h8h%208h8h8h%20%3F1b.4b.1b.2b%20%3F8b.8b%208h8h%208h8h%20%3F8h8h%208h8h%20%3F8h%208h8h%20%3F8d%20CRC%3A16h%208x&z=1&a=Sync&m=34cd&i=true&cw=4), with increments of 32 of the meter value, so that the 5 LSBs of the meter are value are constant, and we can see how the higher order bits are changing. So far, no clear pattern is seen , but I'll continue capturing samples (I only have 17 now, hoping to see a repetition after 32 or 64).

carpalex, 2024-08-06:
I gathered enough [samples](https://triq.net/bitbench#c=%7B239%7D555555555554d335d89b1d36cb1a1a49d4f5a473732eae14b023cf6b566a%20%5B00004964%5D&c=%7B238%7D555555555554d335d89b1d36cb1a1049d4f5a473732eae14b023cf705c38%20%5B00004996%5D&c=%7B238%7D555555555554d335d89b1d36cb1a1249d4f5a473732eae14b823cf6bfe60%20%5B00005028%5D&c=%7B238%7D555555555554d335d89b1d36cb1a0c49d4f5a473732eae14b823ceaae4f8%20%5B00005060%5D&c=%7B239%7D555555555554d335d89b1d36cb1a0e49d4f5a473732eae14b823ceb0e6a4%20%5B00005092%5D&c=%7B239%7D555555555554d335d89b1d36cb1a0249d4f5a473732eae14b823ceb0eaa8%20%5B00005124%5D&c=%7B238%7D555555555554d335d89b1d36cb1ba649d4f5a473732eae14a423ceb1ff08%20%5B00005156%5D&c=%7B239%7D555555555554d335d89b1d36cb1bbe49d4f5a473732eae14a423cf70e394%20%5B00005188%5D&c=%7B237%7D555555555554d335d89b1d36cb1bb649d4f5a473732eae14a423ceabef40%20%5B00005220%5D&c=%7B239%7D555555555554d335d89b1d36cb1baa49d4f5a473732eae14a423cf70f780%20%5B00005252%5D&c=%7B238%7D555555555554d335d89b1d36cb1ba849d4f5a473732eae14a423ceb0f100%20%5B00005284%5D&c=%7B239%7D555555555554d335d89b1d36cb1bb849d4f5a473732eae14a423cf6be5c8%20%5B00005316%5D&c=%7B239%7D555555555554d335d89b1d36cb1bb449d4f5a473732eae14a423ceb1ed18%20%5B00005348%5D&c=%7B239%7D555555555554d335d89b1d36cb1ba449d4f5a473732eae14a423ceabfd54%20%5B00005380%5D&c=%7B239%7D555555555554d335d89b1d36cb1bbc49d4f5a473732eae14a423cf71e190%20%5B00005412%5D&c=%7B238%7D555555555554d335d89b1d36cb1ba049d4f5a473732eae14a423cf70fd88%20%5B00005444%5D&c=%7B239%7D555555555554d335d89b1d36cb1bba49d4f5a473732eae14a623cf70cf90%20%5B00005476%5D&c=%7B239%7D555555555554d335d89b1d36cb1bb049d4f5a473732eae14a623cf6bc5c0%20%5B00005508%5D&c=%7B239%7D555555555554d335d89b1d36cb1bb249d4f5a473732eae14a623ceb0c318%20%5B00005540%5D&c=%7B239%7D555555555554d335d89b1d36cb1bac49d4f5a473732eae14a623cf71d980%20%5B00005572%5D&c=%7B240%7D555555555554d335d89b1d36cb1bae49d4f5a473732eae14a623ceaadf58%20%5B00005604%5D&c=%7B239%7D555555555554d335d89b1d36cb1ba249d4f5a473732eae14a623ceaadf7c%20%5B00005636%5D&c=%7B238%7D555555555554d335d89b1d36cb1b0649d4f5a473732eae14a623cf717328%20%5B00005668%5D&c=%7B239%7D555555555554d335d89b1d36cb1b1e49d4f5a473732eae14a623ceb06fb4%20%5B00005700%5D&c=%7B239%7D555555555554d335d89b1d36cb1b1649d4f5a473732eae14b623cf6b2360%20%5B00005732%5D&c=%7B238%7D555555555554d335d89b1d36cb1b0a49d4f5a473732eae14b623cf6bba78%20%5B00005764%5D&c=%7B240%7D555555555554d335d89b1d36cb1b0849d4f5a473732eae14b623ceaa39f8%20%5B00005796%5D&c=%7B239%7D555555555554d335d89b1d36cb1b1849d4f5a473732eae14b623cf6a2d68%20%5B00005828%5D&c=%7B240%7D555555555554d335d89b1d36cb1b1449d4f5a473732eae14b623ceb025b8%20%5B00005860%5D&c=%7B239%7D555555555554d335d89b1d36cb1b0449d4f5a473732eae14ba23cf6bc172%20%5B00005892%5D&c=%7B239%7D555555555554d335d89b1d36cb1b1c49d4f5a473732eae14ba23ceaaddec%20%5B00005924%5D&c=%7B239%7D555555555554d335d89b1d36cb1b0049d4f5a473732eae14ba23cf6ac570%20%5B00005956%5D&c=%7B240%7D555555555554d335d89b1d36cb1b1a49d4f5a473732eae14ba23cf6bdf6c%20%5B00005988%5D&c=%7B238%7D555555555554d335d89b1d36cb1b1049d4f5a473732eae14ba23ceb1d1b8%20%5B00006020%5D&c=%7B239%7D555555555554d335d89b1d36cb1b1249d4f5a473732eae14ba23cf71d738%20%5B00006052%5D&c=%7B238%7D555555555554d335d89b1d36cb1b0c49d4f5a473732eae14ba23ceabcdf8%20%5B00006084%5D&c=%7B239%7D555555555554d335d89b1d36cb1b0e49d4f5a473732eae14ba23cf6bcb78%20%5B00006116%5D&c=%7B237%7D555555555554d335d89b1d36cb1b0249d4f5a473732eae14ba23cf714e00%20%5B00006148%5D&c=%7B239%7D555555555554d335d89b1d36cb1b2649d4f5a473732eae14ba23cf70e30a%20%5B00006180%5D&f=8h8h%208h8h%208h8h8h%20%3F8h%20%3F5b.4b.7b%208h8h%208h8h%20%3F8h8h%208h8h%20%3F8h%208h8h%20%3F8d%20CRC%3A16h%208x&z=1&a=Sync&m=34cd&i=true&cw=4) (with increments of 32) to observe a pattern. There is a 4-bit group that changes at each line, and they start repeating after 16 such iterations (captured the repetition twice for confirmation).

So most probably the 4-bit group encodes the next 4 LSBs of the meter value. But the transformation is again strange, maybe another lookup table?

Now for observing what happens further, we would need samples with increments of 512 of the meter value. I will try to capture them, but it would take a longer time. For now, we see that they overflow at the left of the 4-bit group.

With the info we have so far, (the LSB in clear text, but at a strange offset, then the next 2 groups of 4 LSBs each encoded with different lookup tables, and again at strange offsets), does this start to look like something? Or is it just arbitrary?

zuckschwerdt, 2024-08-06:
We have not seen any similar scheme where bits are somewhat randomly encrypted but at fixed positions. It's oddly weak. Something like AES with a fixed key or plain XOR would look very different. Seems like rather easy to map out and break, unless the position or key is per device.

Csaba94, 2025-02-05:
Does anyone still trying to solve this encryption or has anyone solved it?

mihai44, 2025-02-17:
I would also be interested to know if any development has been made, these water meters were just installed in my building and I'd like to be able to read them.

carpalex, 2025-02-17:
Hey, so I've managed to partially solve this. I opened a PR (#3203) which supports only the meter ID, Volume and Date. I know the radio module transmits other fields as well, but these three should be a good start for most common use cases.

So while the PR is still open, you can check out the source branch and compile the project and then try to run it. I could successfully capture and decode meters by using `rtl_433 -f 868.9M -s 1000k -g 0 -Y minmax`.

Regarding the decryption of the ID, Volume and Date fields, I'm pretty confident that it's correct, I've tested with around ~2M samples from water meters in my building.

Csaba94, 2025-02-19:
Hello, I can confirm it's also working for me. Thanks for the work!

K1ngC4ndy, 2025-03-12:
Hey, thanks for info. Do you have any news? thx

closed 2025-06-03 by zuckschwerdt