merbanan/rtl_433 #3015: repo has random branches, for which it is not clear there is a reason to remain

gdt, 2024-07-30:
This ticket is about  causing branches to either be deleted or have a ticket open describing the point and mentioning them.  My guess is that they are mostly just leftover and not useful, but that's wild/default guess.   Some are obviously valid.  Even if we closed 75% that would help a lot in terms of tidyness.   The current list of branches:

```
  remotes/upstream/feat-acurite899
  remotes/upstream/feat-arexxml
  remotes/upstream/feat-beepdetect
  remotes/upstream/feat-bresser_soil_moisture
  remotes/upstream/feat-cm130
  remotes/upstream/feat-dbgpd
  remotes/upstream/feat-dup_check
  remotes/upstream/feat-fsk_demod_new
  remotes/upstream/feat-fsk_demod_new_test
  remotes/upstream/feat-httpevs
  remotes/upstream/feat-lessdup
  remotes/upstream/feat-mfree
  remotes/upstream/feat-new_fsk_pulse_detector
  remotes/upstream/feat-openinbrowser
  remotes/upstream/feat-qwalkby
  remotes/upstream/feat-rfxmeter
  remotes/upstream/feat-sigmf
  remotes/upstream/feat-stlib
  remotes/upstream/feat-testpattern
  remotes/upstream/feat-watermark
  remotes/upstream/fix-SMD3MA4
  remotes/upstream/fix-mqttqos
  remotes/upstream/smdl
  remotes/upstream/switchdoclabs
  remotes/upstream/wip-bbdbg
  remotes/upstream/wip-tpms
  remotes/upstream/wip-trace
```

zuckschwerdt, 2024-07-31:
Most look like mine. The majority should have PRs, I'll take a look.
E.g. switchdoclabs was just to get a diff to a diverged fork. I'll see if that can be deleted now.

ProfBoc75, 2024-07-31:
feat-bresser_soil_moisture is mine, related to pr #3005 recently created but their is still some work on it.

gdt, 2024-07-31:
Thanks.   indeed the bresser soil moisture one jumped out at me as live.

(I opened this because I know it's hard to clean up and without effort/complaints things just tend to accumulate.)

inonoob, 2024-10-06:
[feat-arexxml] is fully working and now need to be merged with master