merbanan/rtl_433 #3019: Formally deprecate python2.7, enabling gcing support

gdt, 2024-08-03:
Right now, changes to examples/*.py are tested under relatively recent python 3.x and we are ignoring 2.7.  At least I am.
And, we say (I said):
```
Generally, python scripts should work with relatively recent Python 3.
(While they might work with 2.7, no one is paying attention to that.)
```

I hereby run up the flagpole that under examples/ we don't care about 2.7 to the point that we should
  - document that example scripts are for python3 only even harder than the current text.
  - greenlight removal of 2.7 accomodations

At this point it is hard to conceive of anyone not being able to have python3, and these are examples anyway.   (People running LTS that doesn't have python3 and not wanting to build it and prereqs themselves is in my view something we should not cater to, 4 years after python2.7 extended EOL, and "LTS considered troublesome" anyway.)

This would just be an edit to examples/README.md.

This might or might not actually be a breaking change because we don't know if any scripts work with 2.7, because... nobody cares if they do.

zuckschwerdt, 2024-08-03:
Agreed. Perhaps start with changing to a shebang of `#!/usr/bin/env python3` in all scripts.

gdt, 2024-08-03:
I'll let this sit a week for others to comment, and the dedup changes land, and then open a PR to change docs and #! change.   Then we can after another while start ripping out from future.

closed 2025-07-04 by gdt