merbanan/rtl_433 #3020: Support for VEVOR Wireless Weather Station 7-in-1

cipriancu, 2024-08-05:
Hello,
I'm trying to decode the strings coming from a Vevor 7 in 1 yt60231 Weather Stations

https://www.amazon.com/VEVOR-Wireless-Stations-Forecast-Temperature/dp/B0CHFF77JS
I’m capturing the data with: rtl_433 -s 1000k -Y classic -f 868.3M -X "n=Vevor,m=FSK_PCM,s=87,l=87,r=89088"
And so far I have managed to figure out some measurements 

The captured data looks like this :

time      : 2024-08-05 09:02:56
model     : Vevor        count     : 1             num_rows  : 1             rows      :
len       : 963          data      : fff8000aaaaaaaaaacaca54aa00f8f79d02e332010e03020b013802397a86e08721856ad008dafaab2f644ae30000fffff000155555555559594a95401f1ef3a05c664021c06041602700472f50dc10e430ad5a011b5f5565ec895c6000000000000000000000000000000000000000000000000000000000
codes     : {963}fff8000aaaaaaaaaacaca54aa00f8f79d02e332010e03020b013802397a86e08721856ad008dafaab2f644ae30000fffff000155555555559594a95401f1ef3a05c664021c06041602700472f50dc10e430ad5a011b5f5565ec895c6000000000000000000000000000000000000000000000000000000000

breaking up the data I figure out this 
fff8000aaaaaaaaaacaca54aa00f8f7 - this piece never changes 
9d - this looks to have something to do with the battery - if the battery is low is 9d when it is good the value is 1d
02e3 - this is temperature, the formula to calculate it is Value in hex converted in decimals /10 minus 50
32 - humidity - Value in hex converted in decimals  
010e - Unknown
030 - wind speed but I couldn't figure out the formula 
20b - wind direction measured in degrees - Value in hex converted in decimals minus 257 
01 - Unknown
38 - rain, the sensor is a tipping bucket rain gauge and it counts each tipping in hex i have no clue how it is converted in values displayed 
02 - UVI index - the formula seems to be hex converted in decimals minus 1 
397 - lux measurement Here i have a formula that works on small values but not o high-value  
a86e087 - Unknown

21856ad008dafaab2f644ae30000fffff000155555555559594a95401f1e - this piece never changes 
f - when the battery is 9d is f when battery is 1d is e
3a05c664021c06041602700472f50dc10e-Unknown
430ad5a011b5f5565ec895c6000000000000000000000000000000000000000000000000000000000 - this piece never changes

i can give more sample date

zuckschwerdt, 2024-08-05:
Are you sure that you want `-Y classic` with FSK? Also 87 µs is a very uncommon value. Note that 8908 µs reset is far too long, that's why you see lots of 0's at the end. Reset should be maybe 20x (how many consecutive zeros you expect in the message) the bit length.
Best to look at a .cu8 sample.

cipriancu, 2024-08-05:
Hi Christian,

I chose -Y classic because the beginning of the code resembles the Universal Radio Hacker result. Since I managed to decode some measurements, it seemed like a good starting point.

The values 87 and 8908 came from rtl_433 guidance: -A g001_868.3M_1000k.cu8 "Use a flex decoder with -X 'n=name,m=FSK_PCM,s=87,l=87,r=89088'".

Here's the .cu8 sample:
[g001_868.3M_1000k.zip](https://github.com/user-attachments/files/16499846/g001_868.3M_1000k.zip)

Bruno I'll try your guidance 

Thanks both for your help!

ProfBoc75, 2024-08-05:
@cipriancu : thanks, based on your cu8 sample, reset value is 1400 to get the 2 messages.

    rtl_433 -M level -X "n=Vevor,m=FSK_PCM,s=90,l=90,r=1400,preamble=caca54aa" g001_868.3M_1000k.cu8

[image]

So try with:

     rtl_433 -s 1000k -f 868.3M -M level -X "n=Vevor,m=FSK_PCM,s=90,l=90,r=1400,preamble=caca54aa"

[bitbench updated](https://triq.net/bitbench#c=fff8000aaaaaaaaaacaca54aa00f8f79d02e332010e03020b013802397a86e08721856ad008dafaab2f644ae30000&c=fffff000155555555559594a95401f1ef3a05c664021c06041602700472f50dc10e430ad5a011b5f5565ec895c6000&c=%7B371%7Dfff8000aaaaaaaaaacaca54aa00f8f71d0386110108010263011a079e25d579d621856ad008dafaab2f644ae30000&c=%7B378%7Dfffff000155555555559594a95401f1ee3a070c2202100204c602340f3c4baaf3ac430ad5a011b5f5565ec895c60000&f=TYPE%204h%20CHANNEL%204h%20ID%2016h%20FLAGS_BAT%208h%20TEMP_C%2016d%20HUM%208d%20WIND_GUST%3F%2016d%20WIND_SPEED%2012d%20WIND_DIR%2012d%20RAIN%2016d%20UVI%208h%20LUX_MSB%20b%20LUX_LSB%2015d%20CTR%208h%20CHECK_SUM%208h%20CTR%2BONE%208h%20%3F%20112h&z=1&a=Preamble&m=caca54aa&cw=4)

cipriancu, 2024-08-05:
Bruno 
sometimes times it breaks the data into two pieces other times dont 
[image]
here's my Excel with sample data for wind speed and formulas for temperature humidity wind direction  and UVI 
[Decode.xlsx](https://github.com/user-attachments/files/16500723/Decode.xlsx)

ProfBoc75, 2024-08-05:
Thanks for the excel file ! I'll check and back later.

So reduce the reset to 1000 or 900 not less to get the 2 messages.

ProfBoc75, 2024-08-06:
Hi @cipriancu 

I drafted a decoder but I need more samples to decode properly the values, the ratio between the raw hexa value and the real values.

I confirm that same behavior as with Emax decoder, some data are coded with 0x0101 = 0  (offset 257) or 0x01 = 0 (offset 1)

Please test and share samples with the values you have on the weather display station.

cipriancu, 2024-08-07:
Bruno, I think i made some progress
1 Breaking into pieces didn’t work despite adjusting the reset. Using -Y classic was the only way to get consistent results. Now, my command is: - Y classic so now the command that I'm use is:  rtl_433 -s 1000k -Y classic -M level -f 868M -X "n=Vevor,m=FSK_PCM,s=90,l=90,r=1400,preamble=caca54aa" The results now break into two equal pieces.
2 I changed my Excel to read the new format 
3. I made some adjustments in bitbench https://triq.net/bitbench#c=00f8f71d02f5370101000210015d010101a902aa21856ad008dafaab2f644ae30000&f=TYPE%204h%20CHANNEL%204h%20ID%2016h%20FLAGS_BAT%208h%20TEMP_C%2016d%20HUM%208d%20%20WIND_SPEED%3F%2016d%20WIND_GUST%2012d%20WIND_DIR%2012d%20RAIN%2016d%20UVI%20d%20LUX_MSB%20b%20LUX_LSB%2015d%20CTR%208h%20CHECK_SUM%208h%20CTR%2BONE%208h%20%3F%20112h&z=1&cw=4
4. The LUX formula for low values seems to be ROUND(((BitBench decimal - 257)/1007), 2), but I need to test with higher values.
5. For wind, there are two measurements: wind speed (strings 15-18) and wind gust (strings 19-21).
6. Wind speed formula: =ROUND(((HEX2DEC(strings 15,16,17,18) / 34) - 7.63), 1).
7. Wind gust formula: =ROUND((HEX2DEC(strings 19,20,21) / 86) - 0.06, 1).

I need higher values to verify the formulas, but up to 2 m/s they seem accurate (except for 0).

For rain, I will test with 100 ml poured at once to see the indicator. The display resets to 0 after inactivity, but the string retains the same value.

Thanks for your help!

[Decode.xlsx](https://github.com/user-attachments/files/16534374/Decode.xlsx)

ProfBoc75, 2024-08-08:
Hi @cipriancu 

Thanks a lot, with your last decode.xlsx I found useful information that confirm some assumption and I need to correct few things in my drafted decoder, but I'm very close to the solution.

1- Ok, and in fact, for the decoder, I keep one row and looking the 2 preambles into the row, so not an issue if you still have one row instead of two, with bitbench we can easily split it and add the caca54aa preamble, but I understand for your excel file, it's easier to get directly the 2 rows.
2- Ok, good.
3- My last update with your codes : [bitbench](https://triq.net/bitbench#c=00f8f71d02f5370101000210015d010101a902aa21856ad008dafaab2f644ae30000&c=00f8f79d030d2a01070201fb015d01057406540721856ad008dafaab2f644ae30000&c=00f8f79d030d2a01010201f8015d01056b2d692e21856ad008dafaab2f644ae30000&c=00f8f79d030d2a01150501f9015d01055b54985521856ad008dafaab2f644ae30000&c=00f8f79d030d2a010f030203015d0105537bba7c21856ad008dafaab2f644ae30000&c=00f8f79d030d2a01280801f6015d01053ac9ffca21856ad008dafaab2f644ae30000&c=00f8f79d030c2a012f0701fe015d01052bf024f121856ad008dafaab2f644ae30000&c=00f8f79d030c2a01330b01f4015d01051e173c1821856ad008dafaab2f644ae30000&c=00f8f79d030c2a012c090201015d0105173e613f21856ad008dafaab2f644ae30000&c=00f8f79d030c2b012d07020f015d01050565846621856ad008dafaab2f644ae30000&c=00f8f79d030b2c01460b0100015d0104fa8cac8d21856ad008dafaab2f644ae30000&c=00f8f79d030b2c01320e01f0015d0104f0b3a8b421856ad008dafaab2f644ae30000&c=00f8f71d02fc3401220801f8015d010101127f1321856ad008dafaab2f644ae30000&c=00f8f71d02fb3401140601fd015d010101399a3a21856ad008dafaab2f644ae30000&c=00f8f71d02fb340118030201015d01010160c76121856ad008dafaab2f644ae30000&c=00f8f71d02fb34010f0501fb015d01010187e08821856ad008dafaab2f644ae30000&c=00f8f71d02fb3401160401fe015d010101ae10af21856ad008dafaab2f644ae30000&c=00f8f71d02fb3401190401fe015d010101d53ad621856ad008dafaab2f644ae30000&c=00f8f71d02f6360101000204015d0101014f9c5021856ad008dafaab2f644ae30000&c=00f8f71d02f5370101000210015d010101a902aa21856ad008dafaab2f644ae30000&f=ZEROS%208h%20ID%2016h%20BAT_LOW%201b%20%3F%207h%20TEMP_C%2016d%20HUM%208d%20%20WIND_SPEED%2016h%20WIND_GUST%208h%20ZERO%204h%20WIND_DIR%2012h%20RAIN%2016h%20UVI%20d%20LUX_BY_TEN%20b%20LUX%2015h%20CTR%208h%20CHECK_SUM%208h%20CTR%2BONE%208h%20%3F%20112h&z=1&cw=4)

4- LUX value is based on 16 bits (4 nibbles, 31/32/33/34), and first bit + 15 bits, in fact you just need to offset 257 to get LUX value, you don't need to convert the result in KLUX value, so you don't need to divide by 1007, this let us have a better precision than the weather display in KLUX, if the first bit is = 1 (so the 16 bits > 33025) , then x 10 the 15 bits decimal value .  ( 33025 = 32768 + 257)

    =IF(HEX2DEC(LUX_16bits)>33025;(HEX2DEC(LUX_16bits)-33025)*10;HEX2DEC(LUX_16bits)-257)
    Result is in LUX (not KLUX)

5- Ok for wind speed, not ok for wind gust, the nibble 21 is always = 0, so you just need 19 & 20 for wind gust.

6- Wind speed, 1 kmh = 3.6 x 1 m/s

    m/s formula =ROUND(((HEX2DEC(strings 15,16,17,18) - 257) / 36), 1). 
    kmh formula =ROUND(((HEX2DEC(strings 15,16,17,18) - 257) / 10), 1).

7- Wind gust

    m/s formula: =ROUND((HEX2DEC(strings 19,20) / 6), 1).
    kmh formula: =ROUND((HEX2DEC(strings 19,20) * 0.6), 1).

Notice that we prefer to get km_h as it is handle by the `-C` option to convert to customary mi_h, the m/s or m_h is not taking account, this is the subject of another opened issue, but I put here the both formula.

I will update your excel file accordingly tomorrow.

cipriancu, 2024-08-08:
Bruno 
I’ve confirmed the LUX formula works for higher values:

00f8f79d031d2a0113050201015d04987378817921856ad008dafaab2f644ae30000 (LUX value = 60)
00f8f79d03282e0101020205016d059b810c350d21856ad008dafaab2f644ae30000 (LUX value = 67.8)
00f8f71d03182f012a080203017e069d5d5c135d21856ad008dafaab2f644ae30000 (LUX value = 72.6)

I also tested the rain sensor by pouring 100 ml of water:

15 tips: Rain rate = 20.6 mm/h
16 tips: Rain rate = 21.9 mm/h
17 tips: Rain rate = 23.3 mm/h
Each tip adds 1.3 mm ish (I think is something more than 1.3)  to the rain measurement. The rain rate remains steady for 10 minutes and then resets to 0 if no more rain is detected.
Thanks

ProfBoc75, 2024-08-08:
Good.

For the rain value, you should change the weather display to cumulative or total value and not rain rate mm/h, because this one is an average and each hour the value is reset to 0, so not possible to get the figures, i.e how many mm / tips.

And about the rain test, you must pour slowly drip by drip, like the real rain. What is the diameter of the rain gauge ?

Here your excel updated, I keep 1 tips = 1 mm for the moment into the formula. (with the offset 257 too)

[Decode_ProfBoc75_reviewed.xlsx](https://github.com/user-attachments/files/16549072/Decode_ProfBoc75_reviewed.xlsx)

cipriancu, 2024-08-08:
ok 
Today's cumulative rain is 14.8 mm with 65 tips, with a total 400 ml of water. This gives about 6 ml per tip. The rain gauge has a diameter of 13.5 cm, which means the area is approximately 143.1 cm².
400ml = 400 cm3 in theory measured quanity shoud be (400/143.38 )*10 = 27.94490084 mm 

This is significantly different from the 14.8 mm displayed. 
Any thoughts on this discrepancy? but to be honest this is the last thing that i'm interested in if i know that each tip is 6 ml i'm more than happy 

My goal is to read this vevor wheather station with and ESP32 that have OpenMQTTGateway on it and to push measurents in home assistant

ProfBoc75, 2024-08-08:
I'm answering to myself, as I found the FCC ID of the device (915MHz version) [2AQBD-R56](https://fccid.io/2AQBD-R56). 

The original manufacturer is Fujian Youtong Industries Co., Ltd. rebrand under Vevor name.

If I'm not wrong the diameter is around 15 cm [external photos](https://fccid.io/2AQBD-R56/External-Photos/External-Photos-6589936)

From your test, I understood a total of 17 tips for 0.1 L glass ( 100ml )

The result is 0.33 mm/tips

Diameter cm | Radius cm | Surface cm² | Ratio m² / Surface | Glass Liter | Glass mm/m² | Glass Num of tips | mm / tips
-- | -- | -- | -- | -- | -- | -- | --
15.00 | 7.50 | 176.71 | 56.59 | 0.10 | 5.66 | 17.00 | 0.33

The result may be refined if you test 2 or 3 times more, and if you confirm the funnel diameter (the widest internal part at the top of the funnel ).

In another hand, from the user guide, the Rain range is from 0-12999 mm/m². And based on 2 bytes values, we have 65536 tips which give : 12999 / 65536 = 0.198 mm/tips, could be round to 0.2 mm , so not exactly 0.33 mm, the reason why it needs to be refined/confirmed. Notice that for other Emax Weather station using the same protocol, it's 0.2 mm/tips.

ProfBoc75, 2024-08-08:
@cipriancu 

I just updated the decoder accordingly, can you try to test the rtl_433 version from branch `feat-vevor_7in1` , see PR #3023 

My test with your last lux code (just added the preamble in front of the code to get a result):

    rtl_433 -y aaaacaca54aa00f8f71d03182f012a080203017e069d5d5c135d21856ad008dafaab2f644ae30000
[image]

cipriancu, 2024-08-09:
Bruno 
I installed the decoder, and the results look great! I adjusted the display to show km/h to match the readings from rtl_433.

Observations:

A simple rtl_433 -f 868M command works, but some measurements aren’t captured. When I use -Y classic, all measurements are captured correctly. It seems that without -Y classic, the string isn’t properly split into two pieces.
Wind max speed: The display value is usually 0.1 km/h higher than the readings. I'll try to get more readings and to see if I can refine the formula 
Rain total: The display shows 78 mm, while the readings show a total rainfall of 67.5 mm.

Thank you for your time and help with this

ProfBoc75, 2024-08-09:
Hi @cipriancu:

> the string isn’t properly split into two pieces

Into the decoder, we don't care as I keep one row, (reset = 9000), and within a loop, I'm searching the first preamble, then extract the first 21 bytes, decode the data, if it's ok, then show the values, if not ok, search for the second preamble, extract the next 21 bytes and decode and so on.

May be add `-M level` option to get the central frequency of the signal and adjust to it may help also, and you will see the signal quality too, if you have a bad reception or a bad antenna, check rssi / snr / noise dB values.

> Wind max speed: The display value is usually 0.1 km/h higher than the readings
Probably a round value, we can add 0.1f to final value.

    float gust_kmh    = 0.1f + gust_raw / 1.5f ;

>Rain total: The display shows 78 mm, while the readings show a total rainfall of 67.5 mm.
We need to change the 0.43 ratio to 0.5 (line 112):

    67.5 / 0.43 =  157   --> + 257 = 0x19E in the original message
    78 / 157 = 0.5 (rounded value)

    float rain_mm     = rain_raw * 0.5f; // calculation is 0.43f ratio

cipriancu, 2024-08-10:
Bruno.
I ran a few more tests and think I’ve nailed it. I reset both the external and internal units to start from 0 with the rain measurements, manually tipping the sensor to generate values. Here’s what I discovered:

float rain_mm = rain_raw * 0.233f;
I compiled the build with this new formula, and it’s working perfectly!

The only issue I’m noticing is with rounding. The display doesn’t round numbers; it just takes digits as they come, while rtl_433 rounds the values. This affects the conversion for lux to klux and wind gust as well.

Thanks again for your time and help with this!

ProfBoc75, 2024-08-10:
@cipriancu 

Good job ! As you reset all, did you notice if the sensor changed ID ?

ProfBoc75, 2024-08-10:
Ciprian

I merged the PR with your updates. Enjoy !

Thanks for your contribution. If there is any error, do not hesitate and share here your findings, we'll update accordingly.

cipriancu, 2024-08-10:
> @cipriancu
> 
> Good job ! As you reset all, did you notice if the sensor changed ID ?

nope nothing was changed after reset other than string for rain which went to 0101  
the ID f8f7 is still the same 
and the end of the string too 21856ad008dafaab2f644ae30000

closed 2024-08-12 by cipriancu

JV16Bar, 2024-08-15:
Hi, I'm successfully running rtl_433 with the new decoder, and it gives me good results. My unit is a wifi version (YT60234), but it's almost sure that the outdoor units are identical, hence the protocol is the same. Do you guys know, if the pressure sensor values can be decoded?

zuckschwerdt, 2024-08-15:
Ambient pressure sensors are always mounted in the display units. There is no pressure difference to outside ;)

JV16Bar, 2024-08-15:
You're absolutely right :)
OT: does anybody accidentally knows if the outdoor unit is available somewhere for sale separately? The outdoor unit with rtl_433 would be a good weather station and could be integrated in home automation/nodered/etc.

cipriancu, 2024-08-15:
I asked that already buy they dont 
[image]

ProfBoc75, 2024-08-15:
> Hi, I'm successfully running rtl_433 with the new decoder, and it gives me good results. My unit is a wifi version (YT60234), but it's almost sure that the outdoor units are identical, hence the protocol is the same. Do you guys know, if the pressure sensor values can be decoded?

Hi @JV16Bar , since it's a wifi version, this means that the data is probably sent to the cloud if you have this information in the application and here you may be able to hack and get the figures. If your are able to capture the wifi frame with Wireshark you may be able to get the data. This needs some knowledge and work ...

vlovmx, 2024-08-28:
> > Hi, I'm successfully running rtl_433 with the new decoder, and it gives me good results. My unit is a wifi version (YT60234), but it's almost sure that the outdoor units are identical, hence the protocol is the same. Do you guys know, if the pressure sensor values can be decoded?
> 
> Hi @JV16Bar , since it's a wifi version, this means that the data is probably sent to the cloud if you have this information in the application and here you may be able to analyze and get the figures. If your are able to capture the wifi frame with Wireshark you may be able to get the data. This needs some knowledge and work ...
> 
> Edit: from https://fccid.io/2AQBD-60234 internal photos show a nice Ai-Thinker-WB2-12F, another possibility here to deep dive the weather station.

I did this and currently have Home Assistant ingesting sensor values. I bought the wifi version and didn't want to dedicate a full SDR dongle to just capture this information all of the time. 

I performed a packet capture and found the URL the station calls containing sensor values.    I then used that to write a PHP script that parses sensor data, it then formats a JSON payload that ultimately gets sent to Home Assistant via the Hassio REST API. 

Luckily, the base station sends the data unencrypted to weather cloud services. This is what I initially suspected and it was what drove me to even attempt this. I thought such an affordable device would not have an advanced enough SoC to deal with cryptographic handshakes like SSL/TLS. 

The way I redirected station traffic to my script was via DNS record. I run Pi-Hole as DNS server in my IoT vLAN. I just created a DNS record for rtupdate.wunderground.com to resolve to my local docker container which is where I have an NGINX and PHP stack running along with my custom PHP page. If anyone is interested I can share it here as long it's allowed in this forum since it's not rtl_443 related. 

[image]

zuckschwerdt, 2024-08-28:
> I can share it here as long it's allowed in this forum since it's not rtl_443 related.

Any knowledge, discussion or hacks interesting to rtl_433 users or the the wider community is fine with us.

vlovmx, 2024-08-28:
Thank you @zuckschwerdt.

This is in "work in progress" but it works so far. 

This script redirects weather information uploaded to Weather Underground by the VEVOR 7-in-1 Wi-Fi Solar Self-Charging Weather Station (Model YT60234) into Home Assistant. 
The base station sends weather data to Weather Underground service as the URL format below. 
x represent values

http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?ID=XXXXX&PASSWORD=XXXXX&dateutc=xxxx-x-xx&baromin=x&tempf=x&humidity=x&dewptf=x&rainin=x&dailyrainin=x&winddir=x&windspeedmph=x&windgustmph=x&UV=x&solarRadiation=x

How to deploy

1. You will need a local web server in your network, you can create an instance of apache, nginx or any other web server that is configured to interpret PHP. Using docker should make it quick and painless to set it up, make sure the server runs on port 80. Once installed, create a directory called "weatherstation" under /var/www/html and copy this script to it. The full path should look something like this. /var/www/html/weatherstation/updateweatherstation.php . You can now start this web server instance, take note of the IP address.

2. We can redirect the base station's traffic to the web server by creating a DNS record in your router if supported but an easier way to do this is by using Pi-Hole. The record should resolve as shown below.

rtupdate.wunderground.com = (web server IP address)

3. Activate pairing mode in the base station by pressing the BARO key for 5 seconds, connect to the WiFi network called WWS_XXXXX, navigate to http://192.168.1.1, under the WeatherUnderground section enter any value for Station ID and Station Key, then enable the "Upload wunderground.com" checkbox. Click Apply and wait for it to connect to your previously configured WiFi network. 

4. Replace the relevant details in this script such as Home Assistant IP, token and time zone. Allow 1 to 2 minutes for the base station to upload sensor data, you should now have multiple sensors with entity prefix "sensor.weather_station_..."

```
<?php
//Author: @vlovmx
//This script redirects weather information uploaded to Weather Underground by the VEVOR 7-in-1 Wi-Fi Solar Self-Charging Weather Station (Model YT60234) into Home Assistant. 
// The base station sends weather data to Weather Underground service as the URL format below. 
// x represent values

//http://rtupdate.wunderground.com/weatherstation/updateweatherstation.php?ID=XXXXX&PASSWORD=XXXXX&dateutc=xxxx-x-xx&baromin=x&tempf=x&humidity=x&dewptf=x&rainin=x&dailyrainin=x&winddir=x&windspeedmph=x&windgustmph=x&UV=x&solarRadiation=x

// How to deploy

// 1. You will need a local web server in your network, you can create an instance of apache, nginx or any other web server that is configured to interpret PHP. Using docker should make it quick and painless to set it up, make sure the server runs on port 80. Once installed, create a directory called "weatherstation" under /var/www/html and copy this script to it. The full path should look something like this. /var/www/html/weatherstation/updateweatherstation.php . You can now start this web server instance, take note of the IP address. 

// 2. We can redirect the base station's traffic to the web server by creating a DNS record in your router if supported but an easier way to do this is by using Pi-Hole. The record should resolve as shown below. 
// rtupdate.wunderground.com = (web server IP address)

// 3. Activate pairing mode in the base station by pressing the BARO key for 5 seconds, connect to the WiFi network called WWS_XXXXX, navigate to http://192.168.1.1, under the WeatherUnderground section enter any value for Station ID and Station Key, then enable the "Upload wunderground.com" checkbox. Click Apply and wait for it to connect to your previously configured WiFi network. 

// 4. Replace the relevant details in this script such as Home Assistant IP, token and time zone. Allow 1 to 2 minutes for the base station to upload sensor datwe, you should now have multiple sensors with entity prefix "sensor.weather_station_..."

$attributes = array(
    "Barometric Pressure" => array(
        "value" => isset($_GET['baromin']) ? $_GET['baromin'] : '',
        "unit" => "inHg",
        "device_class" => "atmospheric_pressure" 
    ),
    "Temperature" => array(
        "value" => isset($_GET['tempf']) ? $_GET['tempf'] : '',
        "unit" => "°F",
        "device_class" => "temperature" 
    ),
    "Humidity" => array(
        "value" => isset($_GET['humidity']) ? $_GET['humidity'] : '',
        "unit" => "%",
        "device_class" => "humidity" 
    ),
    "Dew Point" => array(
        "value" => isset($_GET['dewptf']) ? $_GET['dewptf'] : '',
        "unit" => "°F",
        "device_class" => "temperature" 
    ),
    "Rainfall" => array(
        "value" => isset($_GET['rainin']) ? $_GET['rainin'] : '',
        "unit" => "in",
        "device_class" => "precipitation" 
    ),
    "Daily Rainfall" => array(
        "value" => isset($_GET['dailyrainin']) ? $_GET['dailyrainin'] : '',
[... 52 lines cut ...]
        )
    );

    $json_data = json_encode($data);
    $home_assistant_url = $home_assistant_url_base . $sensor_name;
    $ch = curl_init($home_assistant_url);
    curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);
    curl_setopt($ch, CURLOPT_HTTPHEADER, array(
        'Authorization: Bearer ' . $access_token,
        'Content-Type: application/json'
    ));
    curl_setopt($ch, CURLOPT_POST, true);
    curl_setopt($ch, CURLOPT_POSTFIELDS, $json_data);
    $response = curl_exec($ch);
    if ($response === false) {
        echo 'cURL Error: ' . curl_error($ch);
    }
}

// The VEVOR base station expects a "success" string after submitting sensor values. If this is not returned the base station performs 2 more attempts consecutively. 
echo "success";

curl_close($ch);
?>
```

zenguru84, 2024-09-30:
> ne. Allow 1 to 2 minutes for the base station to upload sensor datwe, you should now have multiple sensors

hey @vlovmx!
thanks for sharing!
I'm about to buy a Vevor weather station, but I'm curious: how often do the sensors update?

JV16Bar, 2024-09-30:
If you're asking the external sensor unit, it sends update to the base station every 20 seconds.

zenguru84, 2024-09-30:
> If you're asking the external sensor unit, it sends update to the base station every 20 seconds.

good to know, thanks, but I was asking about the request that sends the data to rtupdate.wunderground.com.

vlovmx, 2024-09-30:
@JV16Bar, the base station pushes sensor data to rtupdate.wunderground.com every 60 seconds.

hash6iron, 2024-10-29:
> Thank you @zuckschwerdt.
>
> This is in "work in progress" but it works so far.
>
> This script redirects weather information uploaded to Weather Underground by the VEVOR 7-in-1 Wi-Fi Solar Self-Charging [...]

Hi. 

I'd like to use your script. Could you help me to run it?

drew442, 2024-11-21:
for compatibility with home assistant using what's discussed above, should I buy YT60234 or YT60231?

I'm confused because Vevor says yt60231 doesn't have wifi but @cipriancu says that's what they're using?

zenguru84, 2024-11-22:
@drew442 
both models use RF to send data to indoor display. 
the one with wifi only sends data to weather underground. you want this if you gonna use @vlovmx method.
you can use any of them with rtl.

sergicastellsague, 2024-12-25:
if it helps anyone, I partially translated the PHP code above to NodeJS. I simply store the values received from the device into the program memory, thanks to the DNS hack, and return it through another endpoint in the API.

```
// WeatherData.ts

export class WeatherData {
    barometricPressure: { value: string, unit: string } = { value: '', unit: 'inHg' };
    temperature: { value: string, unit: string } = { value: '', unit: '°F' };
    humidity: { value: string, unit: string } = { value: '', unit: '%' };
    dewPoint: { value: string, unit: string } = { value: '', unit: '°F' };
    rainfall: { value: string, unit: string } = { value: '', unit: 'in' };
    dailyRainfall: { value: string, unit: string } = { value: '', unit: 'in' };
    windDirection: { value: string, unit: string } = { value: '', unit: '°' };
    windSpeed: { value: string, unit: string } = { value: '', unit: 'mph' };
    windGustSpeed: { value: string, unit: string } = { value: '', unit: 'mph' };
    uvIndex: { value: string, unit: string } = { value: '', unit: 'index' };
    solarRadiation: { value: string, unit: string } = { value: '', unit: 'W/m²' };

    public convertUnits() {
        const data = this;
        return {
            barometricPressure: {
                value: (parseFloat(data.barometricPressure.value) * 33.8639).toFixed(2),
                unit: 'hPa'
            },
            temperature: {
                value: ((parseFloat(data.temperature.value) - 32) * 5 / 9).toFixed(2),
                unit: '°C'
            },
            humidity: data.humidity,
            dewPoint: {
                value: ((parseFloat(data.dewPoint.value) - 32) * 5 / 9).toFixed(2),
                unit: '°C'
            },
            rainfall: {
                value: (parseFloat(data.rainfall.value) * 25.4).toFixed(2),
                unit: 'mm'
            },
            dailyRainfall: {
                value: (parseFloat(data.dailyRainfall.value) * 25.4).toFixed(2),
                unit: 'mm'
            },
            windDirection: data.windDirection,
            windSpeed: {
                value: (parseFloat(data.windSpeed.value) * 1.60934).toFixed(2),
                unit: 'km/h'
            },
            windGustSpeed: {
                value: (parseFloat(data.windGustSpeed.value) * 1.60934).toFixed(2),
                unit: 'km/h'
            },
            uvIndex: data.uvIndex,
            solarRadiation: {
                value: (parseFloat(data.solarRadiation.value) * 0.0079).toFixed(2),
                unit: 'klumens'
            }
        };
    }
}
```

```
// api.ts
private weatherData = new WeatherData();

app.get("/weatherstation/updateweatherstation.php", async (req, res) => {
      this.weatherData.barometricPressure.value = req.query.baromin as string || '';
      this.weatherData.temperature.value = req.query.tempf as string || '';
      this.weatherData.humidity.value = req.query.humidity as string || '';
      this.weatherData.dewPoint.value = req.query.dewptf as string || '';
      this.weatherData.rainfall.value = req.query.rainin as string || '';
      this.weatherData.dailyRainfall.value = req.query.dailyrainin as string || '';
      this.weatherData.windDirection.value = req.query.winddir as string || '';
      this.weatherData.windSpeed.value = req.query.windspeedmph as string || '';
      this.weatherData.windGustSpeed.value = req.query.windgustmph as string || '';
      this.weatherData.uvIndex.value = req.query.UV as string || '';
      this.weatherData.solarRadiation.value = req.query.solarRadiation as string || '';

      res.send("success");
});

app.get("/weather", (req, res) => {
  res.json(this.weatherData);
  // call convertUnits method in case you want different units to be used before returning the info.
});
```

umrath, 2025-03-28:
Any good idea on how to convert the sensor measurements to non-US units?

C9H13NO3-dev, 2025-06-09:
I converted this to python and made it available via docker.
https://github.com/C9H13NO3-dev/VevorWeatherbridge
Has the option to switch between US and regular units ;)

Be aware that some routers have a DNS rebind protection. Check before being frustrated because your pihole config isnt working ;)

jcv-pt, 2025-08-05:
Worked pretty well @vlovmx! Thank you all

jcv-pt, 2025-08-05:
> Any good idea on how to convert the sensor measurements to non-US units?

You need to create an "helper" in HA check this [https://community.home-assistant.io/t/configuration-template-how-to-convert-f-to-c-on-temperature/189003/36](https://community.home-assistant.io/t/configuration-template-how-to-convert-f-to-c-on-temperature/189003/36)

hendrik607, 2025-08-05:
Hi, is there also a way to get the value of the ligtht? I would like to use this in HA to turn on the lights. Thank you.

howwww, 2025-11-04:
I’m having the same issue with the Vevor 7-in-1 weather station and data spikes. (However, to be clear, I'm not sure if is an rtl_433 issue or OpenMQTTGateway issue).

To handle this, I’ve implemented a workaround in Node-RED, a function node sanitizes the incoming data before reposting them to MQTT. 
For example, if there’s a spike — say more than 10 °C in temperature or similarly unrealistic spikes in other sensors , the data is filtered out.
Instead, the previous valid value is used once. 
When the next reading arrives (**usually after about a minute or more**), it’s checked again, and if it looks normal, it’s published to MQTT for my Home Assistant as the new “clean” value.

This approach works ok as a temporary solution, but I’d like to ask:

Is there already a proper fix in place for these spikes? Should we expect an update to address this behavior, or are there limitations that mean we’ll need us to keep working with workarounds like this?

Thanking you in advance.