merbanan/rtl_433 #3022: Add documentation for `-F http` rxui web user-interface on port `localhost:8443`

norpol, 2024-08-05:
I knew it's possible to start rtl_433 with a web-ui. Neither `README.md`, `rtl_433 -h` or `rtl_433 -F help` will tell me about how to start with a web-server, which is `rtl_433 -F http`.

Is there any particular reason why I can't find documentation on that feature?

zuckschwerdt, 2024-08-05:
There wasn't too much interest in the HTTP API or the web UI so far. It's experimental, not well tested and features are missing. Happy to change that though! Did it work well for you?

norpol, 2024-08-06:
Hi. To me it was a very valuable addition, it makes navigating results much easier than just on the console.

Some parts are a bit confusing, like

* "center frequency" let's me select a different frequency, but it takes a while until the scanning switches to the selected frequency
* "Settings" tab is greyed out, and it's unclear why that is the case
* "Log" and "Sys" tab is behaving differently than "Monitor" "Stats" "Protocols" "Settings", but there is no visual indicator that this is the case
* In "Options" Enabling "DB Level" will start showing a diagram once there are results that contain signal strength/quality
* "Stats" and "Protocol" tabs will show a screen loading endlessly, no indicator what I should do differently
* There is a second button that will activate the logs screen, I'm not sure whether this is needed

Besides all of this, I think it is quite useful already - gives me a good idea of what signals are in my area and whether there would be a benefit for setting up a Prometheus exporter for example.

Thank you

zuckschwerdt, 2024-08-06:
Thanks for the detailed report! Good pointers for a cleanup.