merbanan/rtl_433 #3024: rtl-433 can't find librtl.sdr.so.0 or libSoapySDR.so.0.7 after upgrading from Linux Mint 20.3 to Linux Mint 22

emfi2r, 2024-08-06:
Feel free to edit this post or make a HOWTO from it.

rtl-433 can't find librtl.sdr.so.0 or libSoapySDR.so.0.7 after upgrading from Linux Mint 20.3 to Linux Mint 22, possibly other versions higher than Mint 20.3. 

The resulting error is (paraphrased) "no such file or directory"

The old libraries get deleted and newer libraries get install, effectively killing rtl-433 without manual changes shown below.

The first time I ran it after upgrading Linux, it errored out with the librtlsdr file. 

After installing anything RTL-SDR related, that fixed the librtlsdr file problem, but then it couldn't find the libSoapySDR file.

These are the main files I have now:
librtl.sdr.so.0.6.0
librtlsdr.so.2.0.1 
libSoapySDR.so.0.8.1

I noticed there was also a shortcut (link) to librtlsdr.so.2.0.1 that was named 

I finally got rtl_433 to work in terminal again by creating to shortcuts (link) listed below in the same folder as the libraries (in my case it's "lib/x86_64-linux-gnu"). I had to open a file manager as root and go tousr/local/bin

1 - You need to create a shortcut in the same folder named "librtlsdr.so.0" (without the quotation marks) pointing to librtlsdr.so.0.6.0 (or newer than version librtlsdr.so.0) if it doesn't already exist . I had one after installing all the other programs, so that's where I got the idea.

2- You need to create a shortcut t in the same folder named "libSoapySDR.so.0" (without the quotation marks) pointing to libSoapySDR.so.0.8.1 (or whatever is newer than 0.7). I already had a shortcut named libSoapySDR.so.0.8 so I made a copy of that shortcut and renamed it to libSoapySDR.so.0.7.

Now rtl-433 works again in Linux Mint 22 (Wilma)!

Norman, KD6AAJ

I'm a little rusty with Linux so I don't know how to automate this stuff right now.

gdt, 2024-08-06:
This is not a bug in rtl_433.  Basically you need to either:
  - use rtl_433 as packaged by your OS/packaging system (distribution in Linux speak)
  - build rtl_433 yourself

The libraries have different major versions because they have had ABI changes.  Or at least that's what major versions mean.  So it is unsound to symlink them back, and I don't think it is reasonable to recommend that.

I don't understand what you are doing, unless it was "build rtl_433 yourself, but don't rebuild it after updating the OS".

emfi2r, 2024-08-06:
replying to gdt:

Thank you for the reply.

Other Linux Mint users have experienced the same problem as I after upgrading Linux Mint from version 20.3. I just figured out the fix on my own.

I got rtl-433 from the official package manager for Linux Mint, I didn't compile it myself. I don't know why they didn't make links to the newer versions of the libraries, but I also understand Linux is mostly unpaid people volunteering to create and fix stuff.

I believe it is standard practice to create the links for backwards comparability, that's why there are so many other links in the same directory pointing to newer versions of different libraries in the foilder.

The last time I installed a working version of RTL-433 was also from the official Linux Mint Package Manager. Somehow someone forgot to create symbolic links to the newer libraries to be compatible with the older versions. I don't know if it was the Linux Mint team, or the people who provided the newer libraries. The behind the scenes workings are beyond my knowledge.

gdt, 2024-08-07:
If rtl_433 from Mint doesn't work with libraries from Mint then that is definitely a Mint bug and you should file that in their tracker and ask them to fix it.    They absolutely should not be making cross-shlib-version symlinks.    It is not normal.  What you will see is libfoo.so.22 -> libfoo.so.22.0.1, and that is about saying which 22.x is the implementation for the 22 interface.   I'm going to declare your question answered (that you should address Mint bugs to Mint).   If you need more help please feel free to ask on the mailinglist or open a Discussion.

closed 2024-08-07 by gdt