merbanan/rtl_433 #3027: Optimizing Rainpoint Soil Moisture Sensor

kami83, 2024-08-10:
Hi, 

after trying to implement a Bresser Soil Moisture Sensor with FSK, i connect the distributor from Rainpoint and got 2 new sensor which are a little bit different to the old Rainpoint sensor.

But they are just working in Single Way and with correct 433.92M OOK. 

I think the decoder needs a little bit tuning. The Sensor is sending after reboot the message but no all the time the decoder can capture it. It should be a problem with the recieving. The decoding of the data works correct.

So how could the decoder recieving adjusted???

Thanks a lot.

BR kami

@ProfBoc75 and @zuckschwerdt would be really great if you can help here. This device is much better in communication and is the same type as the Bresser. But there is now dual communication. Thx a lot.

zuckschwerdt, 2024-08-10:
Did you find out if it is a problem receiving the transmission, or if it is a problem with demodulation?
Can you pick up the transmission with `-S unknown`? Then receiving works.
When you test the received .cu8 you can try different demodulation adjustments like `-Y autolevel -Y magest`.

kami83, 2024-08-10:
Hi, sorry i did some more testing. And for now it is okay. I will close the issue. It is working right now.

BR kami

closed 2024-08-10 by kami83