merbanan/rtl_433 #3028: Good outdoor Antenna for RTL-SDR

kami83, 2024-08-11:
Hi,

can someone tell me a good not to expensive Antenna for the RTL-SDR Stick?

Thx a lot.

BR kami

ProfBoc75, 2024-08-11:
Hi @kami83 

Depends on lot of criteria:
- Frequency / purpose
- full wave length or 1/4 model
- the distance between stick and sensors 
- stick model so the kind of plug (SMA , MCX ...)
- are you able to diy ? This will the cheaper and you will be able to design it for your needs, but you needs to calculate the length, the impedance, the position... Search on the net you will have lots of samples.

kami83, 2024-08-12:
> Hi @kami83
>
> Depends on lot of criteria:
>
>     * Frequency / purpose
>
>     * full wave length or 1/4 model
>
>     * the distance between stick and sensors
>
>     * stick model so the kind of plug (SMA , [...]

Hi,

for me the 433.92 is really interesting and i can use an adapter for the Antenna. I need only something like 1-2m to go outside. 

I can do it on my own with a little bit copper, but i also can spend some money for a manufactured version. 

Can you tell me something. In the Internet you find a lot of distributor but i want to by a product where someone has some experience with.

BR kami

ProfBoc75, 2024-08-13:
@kami83 

I don't have experience from a vendor version but I can share my knowledge about antenna. Notice that you may have some answers in the [discussion](https://github.com/merbanan/rtl_433/discussions) section as it's a better place for this topic than a rtl_433 issue.

So for an outside antenna at 433.92 MHz , I guess you would like to mount it on the wall like this model

[image]

This is a simple model , omnidirectionnel,  1/4 wave length (17.25 cm) and 3.5 dBi maximum gain.

Or this good model , omnidirectionnel, I did it for a rfxcom 433 to receive and to transmit rf signal. I can receive signals from the garden at 30 m , my antenna is inside.
[image]
[image]

So any antenna like above examples should be ok , with max gain > = 3 dBi is fine, best is +6 dBi but you'll have more noise too.

If your rtl sdr stick is biastee capable (the official one is capable) you can add a [LNA amplifier](https://fr.aliexpress.com/item/33011424874.html?pdp_npi=4%40dis%21EUR%21%E2%82%AC%2022%2C84%21%E2%82%AC%2022%2C84%21%21%2124.40%2124.40%21%40211b617a17235623410594996eefd4%2167183464017%21sh%21FR%212263185571%21X&spm=a2g0o.store_pc_allItems_or_groupList.new_all_items_2007550666144.33011424874&gatewayAdapt=glo2fra) between antenna and stick to increase the gain.

ProfBoc75, 2024-08-14:
I'm converting this issue into Q&A discussion.