merbanan/rtl_433 #3031: LaCrosse TX141-BV3 4 lines not recognized

brstone89, 2024-08-15:
Commit [ade6355](https://github.com/merbanan/rtl_433/commit/ade63558fdc85e2571de115c666cf300fd903951) added support for a version of the TX141-BV3 that only sends 4 packets instead of the normal expected 5. It would appear that [5c17e4e](https://github.com/merbanan/rtl_433/commit/5c17e4e36c2e9df071925c89d67f25a2a89316cb) may have undone that change. For whatever reason, the messages are not working unless repeated a 5th time. @zuckschwerdt

zuckschwerdt, 2024-08-15:
A very detailed report, thanks! 5c17e4e (from #2088) made the check more lenient, 5 or 3 packets are needed, 5 only when there are 6 or more available (`bitbuffer->num_rows > 5 ? 5 : 3,`).

Test it out like this: `rtl_433 -R 73:v -y '{33}de6d796f00{33}de6d796f00{33}de6d796f00'`

Has to something else, try `rtl_433 -R 73:v` to check what codes you really get.

brstone89, 2024-08-21:
This can be cancelled. Apparently I had a bug in my config and reinstalling rtl_433 cleared it out. Thanks for the response.

closed 2024-08-21 by brstone89