merbanan/rtl_433 #3033: Fanimation Fan Remote

roblandry, 2024-08-15:
I have decoded and figured out the remote codes. 
[bitbench](https://triq.net/bitbench#c=%5BDip%20switch%20set%20to%2000011%5D&c=&c=%5BFan%20Control%5D&c=%7B25%7D0x8001800%20%5B0x8001800%5D%5BFan%20On%2FOff%5D&c=%7B25%7D0x8001a08%20%5B0x8001a08%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a18%20%5B0x8001a18%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a28%20%5B0x8001a28%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a38%20%5B0x8001a38%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a48%20%5B0x8001a48%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a58%20%5B0x8001a58%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a68%20%5B0x8001a68%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a78%20%5B0x8001a78%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a88%20%5B0x8001a88%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001a98%20%5B0x8001a98%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001aa8%20%5B0x8001aa8%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001ab8%20%5B0x8001ab8%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001ac8%20%5B0x8001ac8%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001ad8%20%5B0x8001ad8%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001ae8%20%5B0x8001ae8%5D%5BFan%20Up%2FDown%5D&c=%7B25%7D0x8001af8%20%5B0x8001af8%5D%5BFan%20Up%2FDown%5D&c=&c=%5BLight%20Control%5D&c=%7B25%7D0x8001c00%20%5B0x8001c00%5D%5BLight%20Off%5D&c=%7B25%7D0x8001d90%20%5B0x8001d90%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d88%20%5B0x8001d88%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d80%20%5B0x8001d80%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d78%20%5B0x8001d78%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d70%20%5B0x8001d70%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d68%20%5B0x8001d68%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d60%20%5B0x8001d60%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d58%20%5B0x8001d58%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d50%20%5B0x8001d50%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d48%20%5B0x8001d48%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d40%20%5B0x8001d40%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d38%20%5B0x8001d38%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d30%20%5B0x8001d30%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d28%20%5B0x8001d28%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d20%20%5B0x8001d20%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d18%20%5B0x8001d18%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d10%20%5B0x8001d10%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d08%20%5B0x8001d08%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001d00%20%5B0x8001d00%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cf8%20%5B0x8001cf8%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cf0%20%5B0x8001cf0%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001ce8%20%5B0x8001ce8%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001ce0%20%5B0x8001ce0%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cd8%20%5B0x8001cd8%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cd0%20%5B0x8001cd0%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cc8%20%5B0x8001cc8%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cc0%20%5B0x8001cc0%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cb8%20%5B0x8001cb8%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001cb0%20%5B0x8001cb0%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001ca8%20%5B0x8001ca8%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001ca0%20%5B0x8001ca0%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c98%20%5B0x8001c98%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c90%20%5B0x8001c90%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c88%20%5B0x8001c88%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c80%20%5B0x8001c80%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c78%20%5B0x8001c78%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c70%20%5B0x8001c70%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c68%20%5B0x8001c68%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c60%20%5B0x8001c60%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c58%20%5B0x8001c58%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c50%20%5B0x8001c50%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c48%20%5B0x8001c48%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c40%20%5B0x8001c40%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001c38%20%5B0x8001c38%5D%5BLight%20Brightness%5D&c=%7B25%7D0x8001b30%20%5B0x8001b30%5D%5BLight%20ColorTemp%5D&c=&c=%5BTimer%20Control%5D&c=%7B25%7D0x8001908%20%5B0x8001908%5D%5B1HR%20Timer%5D&c=%7B25%7D0x8001910%20%5B0x8001910%5D%5B3HR%20Timer%5D&c=%7B25%7D0x8001918%20%5B0x8001918%5D%5B6HR%20Timer%5D&c=&c=%5BMode%20Control%5D&c=%7B25%7D0x8001b20%20%5B0x8001b20%5D%5BAway%20Simulation%5D&c=%7B25%7D0x80019d0%20%5B0x80019d0%5D%5BNatural%20Breeze%5D&c=%7B25%7D0x8001b28%20%5B0x8001b28%5D%5B1%20Min%20Timer%5D&c=&c=%5BFan%20Direction%5D&c=%7B25%7D0x8001b18%20%5B0x8001b18%5D%5BWinter%20-%20Counter%20Clockwise%5D&c=%7B25%7D0x8001b10%20%5B0x8001b10%5D%5BSummer%20-%20Clockwise%5D&f=12b%205b%208b&z=1&cw=4)

[fanimation_fan.zip](https://github.com/user-attachments/files/16628583/fanimation_fan.zip)

I am down to figuring out how to create the decoder and am stumped as to the format. 
```EditorConfig
# Decoder for Fanimation fan remotes.

# https://fcc.report/FCC-ID/KUJCE10711/4100428
# rtl_433 -f 304.25M -R 0 -X 'n=name,m=OOK_PWM,s=400,l=800,g=1000,r=12000,bits=25'

# Dipswitch:
# up = off(0), down = on(1)
# 12b 5b[dip switch] 8b[command]
# 100000000000 11111 00000000 (0x800f800)
# 100000000000 00011 00000000 (0x8001800)
# 100000000000 00000 00000000 (0x8000000)

# g001_304.25M_250k = Fan On/Off
# {25}0x8001800

# g002_304.25M_250k = Fan Down
# g003_304.25M_250k = Fan Up
# {25}0x8001a08, 18, 28, 38, 48, 58, 68, 78, 88, 98, a8, b8, c8, d8, e8, f8

# g004_304.25M_250k = Light On/Off
# {25}8001d90 (brightness value), 0x8001c00 (off)

# g005_304.25M_250k = Light Brightness
# {25}8001d90, d88, d80, d78, d70, d68, d60, d58, d50, d48, d40, d38, d30, 
#              d28, d20, d18, d10, d08, d00, cf8, cf0, ce8, ce0, cd8, cd0, 
#              cc8, cc0, cb8, cb0, ca8, ca0, c98, c90, c88, c80, c78, c70,
#              c68, c60, c58, c50, c48, c40, c38

# g006_304.25M_250k = Light ColorTemp
# {25}8001b30

# g007_304.25M_250k = 1HR Timer
# {25}8001908

# g008_304.25M_250k = 3HR Timer
# {25}8001910

# g009_304.25M_250k = 6HR Timer
# {25}8001918

# g0010_304.25M_250k = Away (Auto on/off simulation)
# {25}8001b20

# g0011_304.25M_250k = Natural Breeze
# {25}80019d0

# g0012_304.25M_250k = Person (1 min turn off)
# {25}8001b28

# g0013_304.25M_250k = Winter (counter clockwise)
# {25}8001b18

# g0014_304.25M_250k = Summer (clockwise)
# {25}8001b10

frequency 304.25M

decoder {
    name        = Fanimation-Fan,
    modulation  = OOK_PPM,
    bits        = 25,
    short       = 400,
    long        = 800,
    gain        = 1000,
    reset       = 12000,
    # this is preexisting from the template used
    # get         = id:@0:{16},
    # get         = button:@16:{5}:[16:fan_low 8:fan_med 24:fan_hi 2:fan_off 5:light_off_delayed 26:light_on_off 14:light_dim_brighten 21:learn_mode],
    # get         = dimmable:@23:{1}:[0:no 1:yes]
}

```
Additionally I would like to know if it is possible to send commands, as that's my ultimate goal (with homeassistant). And im not looking for homeassistant support here.

zuckschwerdt, 2024-08-15:
That's an EV1527 code. The first bit will be the sync (of the previous code).
The get-commands should work, the args are:
name `:` @ start-bit `:` { bit-length } `:` [ optional value mapping, decimal number to text ]

roblandry, 2024-08-15:
So would it be like this?

```EditorConfig
decoder {
    name        = Fanimation-Fan,
    modulation  = OOK_PWM,
    bits        = 25,
    short       = 400,
    long        = 800,
    gain        = 1000,
    reset       = 12000,
    get         = id:@13:{5},  # use the dipswitch as the id
    get         = fan_on_off:@18:{8}:[0:fan_on_off],
    get         = fan_up_down:@18:{8}:[
                    65:fan_speed_1
                    67:fan_speed_5
                    69:fan_speed_10
                    71:fan_speed_15
                    73:fan_speed_20
                    75:fan_speed_25
                    77:fan_speed_30
                    79:fan_speed_40
                    81:fan_speed_50
                    82:fan_speed_60
                    83:fan_speed_70
                    85:fan_speed_75
                    87:fan_speed_80
                    89:fan_speed_85
                    91:fan_speed_90
                    93:fan_speed_95
                    95:fan_speed_100],
    get         = light_off:@18:{8}:[128:light_off],
    get         = light_on_brightness:@18:{8}:[
                    135:light_brightness_1
                    136:light_brightness_2
                    137:light_brightness_4
                    138:light_brightness_6
                    139:light_brightness_8
                    140:light_brightness_10
                    141:light_brightness_12
                    142:light_brightness_14
                    143:light_brightness_16
                    144:light_brightness_18
                    145:light_brightness_20
                    146:light_brightness_22
                    147:light_brightness_24
                    148:light_brightness_26
                    149:light_brightness_28
                    150:light_brightness_30
                    151:light_brightness_32
                    152:light_brightness_34
                    153:light_brightness_36
                    154:light_brightness_38
                    155:light_brightness_40
                    156:light_brightness_44
                    157:light_brightness_48
                    158:light_brightness_52
                    159:light_brightness_56
                    160:light_brightness_60
                    161:light_brightness_64
                    162:light_brightness_68
                    163:light_brightness_70
                    164:light_brightness_72
                    165:light_brightness_74
                    166:light_brightness_76
                    167:light_brightness_78
                    168:light_brightness_80
                    169:light_brightness_82
                    170:light_brightness_84
                    171:light_brightness_86
                    172:light_brightness_88
                    173:light_brightness_90
                    174:light_brightness_92
                    175:light_brightness_94
                    176:light_brightness_96
                    177:light_brightness_98
                    178:light_brightness_100],
    get         = light_color_temp:@18:{8}:[102:light_change_color_temp],
}
```

zuckschwerdt, 2024-08-15:
Should be `@17:{8}` I'd say, first index is `0`.

roblandry, 2024-08-15:
So i have some of it functional, but it looks like you cannot have very many mappings.
16 is fine. 
17: [1]    3180535 segmentation fault (core dumped)  rtl_433 -g 1000 -c fanimation_fan.conf

Am i missing something?

zuckschwerdt, 2024-08-15:
Ah yes, you are right: 12 getters and a maximum of 16 mappings each.
You could raise that here if needed: https://github.com/merbanan/rtl_433/blob/master/src/devices/flex.c#L66

roblandry, 2024-08-15:
Id really like to not mess with core code. Based on the limitations, this is the best i could come up with:

```EditorConfig
# Decoder for Fanimation fan remotes.

# https://fcc.report/FCC-ID/KUJCE10711/4100428
# rtl_433 -f 304.25M -R 0 -X 'n=name,m=OOK_PWM,s=400,l=800,g=1000,r=12000,bits=25'

# Dipswitch:
# up = off(0), down = on(1)
# 12b 5b[dip switch] 8b[command]
# 100000000000 11111 00000000 (0x800f800)
# 100000000000 00011 00000000 (0x8001800)
# 100000000000 00000 00000000 (0x8000000)

# g001_304.25M_250k = Fan On/Off
# {25}0x8001800 [dec 0]

# g002_304.25M_250k = Fan Down
# g003_304.25M_250k = Fan Up
# {25}0x8001a08, 18, 28, 38, 48, 58, 68, 78, 88, 98, a8, b8, c8, d8, e8, f8 [dec(odd numbers) 65 - 95]

# g004_304.25M_250k = Light On/Off
# {25}8001d90 (brightness value), 0x8001c00 (off) [dec 128]

# g005_304.25M_250k = Light Brightness
# {25}8001d90, d88, d80, d78, d70, d68, d60, d58, d50, d48, d40, d38, d30, 
#              d28, d20, d18, d10, d08, d00, cf8, cf0, ce8, ce0, cd8, cd0, 
#              cc8, cc0, cb8, cb0, ca8, ca0, c98, c90, c88, c80, c78, c70,
#              c68, c60, c58, c50, c48, c40, c38 [dec 135 - 178]

# g006_304.25M_250k = Light ColorTemp
# {25}8001b30 [dec 102]

# g007_304.25M_250k = 1HR Timer
# {25}8001908 [dec 33]

# g008_304.25M_250k = 3HR Timer
# {25}8001910 [dec 34]

# g009_304.25M_250k = 6HR Timer
# {25}8001918 [dec 35]

# g0010_304.25M_250k = Away (Auton on/off simulation)
# {25}8001b20 [dec 100]

# g0011_304.25M_250k = Natural Breeze
# {25}80019d0 [dec 58]

# g0012_304.25M_250k = Person (1 min turn off)
# {25}8001b28 [dec 101]

# g0013_304.25M_250k = Winter (counter clockwise)
# {25}8001b18 [dec 99]

# g0014_304.25M_250k = Summer (clockwise)
# {25}8001b10 [dec 98]

frequency 304.25M

decoder {
    name        = Fanimation-Fan,
    modulation  = OOK_PWM,
    bits        = 25,
    short       = 400,
    long        = 800,
    gap         = 1000,
    reset       = 12000,
    get         = id:@12:{5},
    get         = modes:@17:{8}:[
                    33:timer_1hr
                    34:timer_3hr
                    35:timer_6hr
                    100:fan_away
                    101:fan_person
                    58:fan_breeze
                    98:summer
                    99:winter
                    0:fan_on_off
                ],
    get         = fan_speed_low:@17:{8}:[
                    1:fan_speed_5
                    3:fan_speed_10
                    5:fan_speed_15
                    7:fan_speed_20
                    9:fan_speed_25
                    11:fan_speed_30
                    13:fan_speed_40
                    15:fan_speed_50
                ],
    get         = fan_speed_high:@17:{8}:[
                    17:fan_speed_60
                    19:fan_speed_70
                    21:fan_speed_75
                    23:fan_speed_80
                    25:fan_speed_85
                    27:fan_speed_90
                    29:fan_speed_95
                    31:fan_speed_100
                ],
    get         = light_control:@17:{8}:[
                    128:light_off
                    102:light_change_color_temp
                ],
    get         = light_bright_low:@17:{8}:[
                    135:light_brightness_1
                    136:light_brightness_2
                    137:light_brightness_4
                    138:light_brightness_6
                    139:light_brightness_8
                    140:light_brightness_10
                    141:light_brightness_12
                    142:light_brightness_14
                    143:light_brightness_16
                    144:light_brightness_18
                    145:light_brightness_20
                    146:light_brightness_22
                    147:light_brightness_24
                    148:light_brightness_26
                ],
    get         = light_bright_mid:@17:{8}:[
                    149:light_brightness_28
                    150:light_brightness_30
                    151:light_brightness_32
                    152:light_brightness_34
                    153:light_brightness_36
                    154:light_brightness_38
                    155:light_brightness_40
                    156:light_brightness_44
                    157:light_brightness_48
                    158:light_brightness_52
                    159:light_brightness_56
                    160:light_brightness_60
                    161:light_brightness_64
                    162:light_brightness_68
                    163:light_brightness_70
                ],
    get         = light_bright_high:@17:{8}:[
                    164:light_brightness_72
                    165:light_brightness_74
                    166:light_brightness_76
                    167:light_brightness_78
                    168:light_brightness_80
                    169:light_brightness_82
                    170:light_brightness_84
                    171:light_brightness_86
                    172:light_brightness_88
                    173:light_brightness_90
                    174:light_brightness_92
                    175:light_brightness_94
                    176:light_brightness_96
                    177:light_brightness_98
                    178:light_brightness_100
                ],
}
```

zuckschwerdt, 2024-08-16:
Very nice, clever solution! PR the conf file if you like.

gdt, 2024-10-28:
@roblandry We are awaiting a PR to store the config in the sources.   (Leaving issues open long term is not ok.)