merbanan/rtl_433 #3036: Need support for simple Temp and Humidity Sensor

kajoty, 2024-08-16:
1. Sorry for my bad english.
2.  I want to receive the temp and Humidity of a EN8829TX Sensor. This sensor is available for ~7€ on amazon. 

Im a very very beginner of this kind of analysis and decoding. What I get so far is the following:

` 
rtl_433 -g 30 -s 1M -f 434M -A
`
gives my

[00] {43} 00 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000000 000
[01] { 2} 00                : 00
[02] {43} 00 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000000 000
[03] { 2} 00                : 00
[04] {43} 00 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000000 000
[05] { 2} 00                : 00
[06] {43} 00 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000000 000
[07] { 2} 00                : 00
[08] {43} 00 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000000 000
[09] { 3} 20                : 001

But 

`
rtl_433 -g 30 -s 1M -f 434M -a 4
`
gives me

*** signal_start = 4540001, signal_end = 5340104, signal_len = 800103, pulses_found = 226
Iteration 1. t: 396    min: 303 (1)    max: 489 (225)    delta 1225
Iteration 2. t: 396    min: 303 (1)    max: 489 (225)    delta 0
Distance coding: Pulse length 396

Short distance: 1951, long distance: 3905, packet distance: 7811

p_limit: 396
bitbuffer:: Number of rows: 10
[00] {42} a8 41 0d 7e 0e 40 : 10101000 01000001 00001101 01111110 00001110 01
[01] { 1} 00                : 0
[02] {42} a8 41 0d 7e 0e 40 : 10101000 01000001 00001101 01111110 00001110 01
[03] { 1} 00                : 0
[04] {42} a8 41 0d 7e 0e 40 : 10101000 01000001 00001101 01111110 00001110 01
[05] { 1} 00                : 0
[06] {42} a8 41 0d 7e 0e 40 : 10101000 01000001 00001101 01111110 00001110 01
[07] { 1} 00                : 0
[08] {42} a8 41 0d 7e 0e 40 : 10101000 01000001 00001101 01111110 00001110 01
[09] { 2} 40                : 01

The triq bitbench gives me already the correct temp. But what to do now? I appriciate your help very much to understand the process of decoding this

ProfBoc75, 2024-08-16:
Hi @kajoty 

Your sensor is already handle by rtl_433 and recognize as `Vauno-EN8822C`, looks like you have an old version of rtl_433. ( option  `-a 4` is no more present). Do your temp and humidity match the following values ?

    rtl_433 -y {42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40,{42}a8410d7e0e40
[image]

So, you need to update your rtl_433 to the last version, then if you have gap between your values and the Vauno-EN8822C result, update this issue accordingly, we will update the decoder. If the values match, then you can confirm and we will update the decoder with this compatible model.

kajoty, 2024-08-16:
I updated to 
`
version 23.11-159-gd0142bb0 branch master at 202408141413 inputs file rtl_tcp RTL-SDR SoapySDR
`

Now it works like charme. All the Channels and the shown infos on LCD are the same as rtl_433 gives me. Thanks a lot !

closed 2024-08-16 by kajoty