merbanan/rtl_433 #3038: variant of the somfy-rts protocol?

xjazzor, 2024-08-21:
Hi there,
I have some Roller Shades at home, which seem to be using the Somfy-RTS protocoll. I sniffed some packets with the rtl_433-plugin for SDR# as seen here:
[image]
You can see, everything is recognised except Lines 8 and up. These are the lines, where I want to program a new remote onto the shade.
If I press the same "buttons" on the virtual remote from the great EspSomfy (https://github.com/rstrouse/ESPSomfy-RTS) I get the following:
[image]
Are my shades on a different protocol, or do I need to edit something? I can provide more info if needed and want to solve this riddle :D

zuckschwerdt, 2024-08-21:
If you can try listening with current rtl_433. Try `-R 167:vv` to get the raw packets also.
There are some senders with extra data or slightly different meaning, see https://github.com/merbanan/rtl_433/blob/master/src/devices/somfy_rts.c

xjazzor, 2024-08-21:
Thank you for your help!
I pulled the data from rtl_433 and attached the readout as a file. As an excerpt, here are some readouts from down, stop, up and then program: (Attention: the examples below are not directly one after the other. The counter increments in single steps)

`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-08-21 20:00:55
model     : Somfy-RTS    id        : C5154A
Control   : Down (8)     Counter   : 1086          Retransmission: 1         Integrity : CHECKSUM
[pulse_slicer_pcm] Somfy RTS
codes     : {179}f0f0f0f0f0f0f0fe696a965a966a9996a95aaa695a5a0
[pulse_slicer_pcm] Somfy RTS
codes     : {139}f0f0fe696a965996699996a95aaa695a5a0
[somfy_rts_decode] seed=0x98, chksum=0x5
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-08-21 20:00:58
model     : Somfy-RTS    id        : C5154A
Control   : Stop (5)     Counter   : 1088          Retransmission: 1         Integrity : CHECKSUM
[pulse_slicer_pcm] Somfy RTS
codes     : {180}f0f0f0f0f0f0f0fe6999969996a9a6a99665955665650
[somfy_rts_decode] seed=0x95, chksum=0x0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-08-21 20:01:00
model     : Somfy-RTS    id        : C5154A
Control   : Up (6)       Counter   : 1089          Retransmission: 1         Integrity : CHECKSUM
[pulse_slicer_pcm] Somfy RTS
codes     : {180}f0f0f0f0f0f0f0fe6996969a96aaa6a99665955665650
[somfy_rts_decode] seed=0x96, chksum=0x2
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-08-21 20:01:16
model     : Somfy-RTS    id        : C5154A
Control   : ? (12)       Counter   : 1091          Retransmission: 1         Integrity : CHECKSUM
[pulse_slicer_pcm] Somfy RTS
codes     : {180}f0f0f0f0f0f0f0fe695a969696a6a6a99665955665650
[somfy_rts_decode] seed=0x9c, chksum=0xa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-08-21 20:01:16
model     : Somfy-RTS    id        : C5154A
Control   : ? (12)       Counter   : 1091          Retransmission: 1         Integrity : CHECKSUM
[pulse_slicer_pcm] Somfy RTS
codes     : {180}f0f0f0f0f0f0f0fe695a969696a6a6a99665955665650
[somfy_rts_decode] seed=0x9c, chksum=0xa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
`

From my - somewhat limited perspective - it looks, like the remote just spams its id with another seed code - Before it was 0x95 now its 0x9c. I'm just gessing here. ;)

[somfy-like-packets.txt](https://github.com/user-attachments/files/16696310/somfy-like-packets.txt)

zuckschwerdt, 2024-08-21:
Those codes are great to analyze this. Maybe @matthiasseemoo (original author in #1496) and @Velocet (good info in #2356) can help?

xjazzor, 2024-08-26:
Should I provide more packets/ recordings? Can I help in any way?
My second remote died and I cant't find any shop to buy a new one. Therefore I want this to work. ;)

Velocet, 2024-09-22:
There are many RTS variants (at least 5 .. RTS, RTD, RTDS, RTW, RTN..). In case of RTS there are two different protocol modes which differ in length but are compatible to one and another.

You can find my notes i did for [somfy-rts](https://github.com/rstrouse/ESPSomfy-RTS) here:
[RTx: Somfy RTS and its different versions](https://github.com/Velocet/iown-homecontrol/blob/main/docs/firmware/other/README.md#rtx-somfy-rts-and-its-different-versions)

xjazzor, 2024-09-24:
> There are many RTS variants (at least 5 .. RTS, RTD, RTDS, RTW, RTN..). In case of RTS there are two different protocol modes which differ in length but are compatible to one and another.

I am with you that they are nearly compatible. Sadly I can't get espsomfy to be paired with my blinds because that parts seems to differ. Maybe in this protocol the pairing sequence is only the remote-id repated 7times?

xjazzor, 2024-10-28:
I would love to see a PR to improve rtl_433, as I believe it could benefit not only me but others as well.

Crsarmv7l, 2024-12-14:
I noticed in lines 8 and up the counter is the same, which means a long (held) press. With somfy long presses the beginning preamble will be different from the repeats, repeats are shorter preambles. 

Also as you noticed, the seed does change with each button press (something I noticed when building a decoder/encoder for the Yardstick One), but only the second nibble and the change isn't really checked by the receiver. I was able to use a constant value for the second nibble.

Something that was a big help to me was using both rtl_433 and protocol info from here: https://github.com/DarkFlippers/unleashed-firmware/blob/306b7285879abe05b1e3e7bfdca6442c86161b94/lib/subghz/protocols/somfy_telis.c

Slightly different approaches between the two, but I was able to get enough information with both sources to get rx and tx working.

I would need to go back and look at my notes for more specifics, its been a while since I touched somfy