merbanan/rtl_433 #3042: New device - GN Resound / AB Transistor - Lynx Doorbell

kimballen, 2024-08-26:
New to this but can't get it right there's something I'm missing, very simple device just a doorbell.

433MHz 

[https://github.com/kimballen/rtl_433-Recorded-Signals/tree/main/GN%20Resound%20-%20AB%20Transistor%20Lynx%20Doorbell%20(433MHz)](https://github.com/kimballen/rtl_433-Recorded-Signals/tree/main/GN%20Resound%20-%20AB%20Transistor%20Lynx%20Doorbell%20(433MHz))

zuckschwerdt, 2024-08-27:
Try a sample on https://triq.org/pdv/ -- looks like 13 bits of (leading gap) PWM with 1000µs / 2000 µs pulses, bit-width 3000 µs. I.e. this should get your data: `rtl_433 -R 0 -X 'n=lynx,m=OOK_PWM,s=1000,l=2000,r=3000'`
Also best not to hit the transmission frequency dead on, default 433.92M should be fine.

klohner, 2024-09-02:
The "code" value seems to have its bits reversed.  This seems to work for me:

```
decoder {
    name          = Lynx_Doorbell,
    modulation    = OOK_PCM,
    short         = 1016,
    long          = 1016,
    reset         = 3048,
    bits         >= 40,
    bits         <= 41,
    preamble      = {1}8,
    symbol_zero   = {3}2,
    symbol_one    = {3}6,
    get           = id:@4:{8}:%d,
    get           = button:@0:{4}:[0:0 1:8 2:4 3:C 4:2 5:A 6:6 7:E 8:1 9:9 10:5 11:D 12:3 13:B 14:7 15:F],
}
```

klohner, 2024-09-02:
Just found the product manual for the [Lynx Door Transmitter](https://cdn.abicart.com/shop/ws15/43915/art55/28693455-c3a9ad-Lynx_Door_Transmitter_Manual_En_De_Fr_It_Ne_No_Sv_2014-06.pdf).  Perhaps, if what I thought was the ID is actually a static product code for all of these devices, this might be more discriminating:

```
decoder {
    name          = Lynx_Doorbell,
    modulation    = OOK_PCM,
    short         = 1016,
    long          = 1016,
    reset         = 3048,
    bits         >= 40,
    bits         <= 41,
    match         = {24}24b659,
    preamble      = {1}8,
    symbol_zero   = {3}2,
    symbol_one    = {3}6,
    get           = id:@4:{8}:%d,
    get           = button:@0:{4}:[0:0 1:8 2:4 3:C 4:2 5:A 6:6 7:E 8:1 9:9 10:5 11:D 12:3 13:B 14:7 15:F],
}
```

zuckschwerdt, 2024-09-02:
@klohner do suspect it's really 40 bits PCM? I'm pretty sure it's  13 bits PWM with leading gaps. I.e. symbols of 1000 µs gap + 2000 µs pulse and 2000 µs gap + 1000 pulse.
E.g. "Code A" g188_433.884M_250k.cu8:

[image]

klohner, 2024-09-02:
You're right, but my decoder is doing OOK_PCM, discarding the first pulse symbol, then doing essentially "gap width modulation" on each group of three symbols with the symbol_zero and symbol_one patterns.  The "bits" value seems to operate before the "symbol_*" decodings.

1 (discarded preamble symbol) + 12 * (3 symbols per logical bit) + 3 (trailing gap symbols before reset detected) = 40 symbols.

zuckschwerdt, 2024-09-02:
Oops, completely missed the symbol_zero/symbol_one in your decoder, my bad.