merbanan/rtl_433 #3045: No data from HackRF, but data from RTLSDR next to it

phoniclynx, 2024-09-04:
I'm running a HackRF and an RTLSDR side by side with two different antennas. RTL_433 is running on both simultaneously and says it picks up both devices, but I can't get any readings out of the HackRF. 

I tried turning the gain up on the HackRF but that didn't have an effect (and tried it without that command too):
[image]

```
SoapySDRUtil --find                                                                                                                                                                  ✔ 
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 15b062dc2d4f37cf
  part_id = a000cb3c006a4768
  serial = 000000000000000015b062dc2d4f37cf
  version = v2.0.1
```
The driver reports that is is fine

zuckschwerdt, 2024-09-04:
Try using `-Y autolevel -M level -M noise` to check the levels.
(Also the misspelling`-d druver=hackrf` is a no-op, you effectivly use `-d ""`)

phoniclynx, 2024-09-04:
Opps.. my bad I did do that earlier and fix it, must have done it again:

But that didn't seem to help:
[image]

zuckschwerdt, 2024-09-04:
The signal on the HackRF seems low, but is not quiet. Perhaps use the pass-through `-F rtl_tcp` output and view it on an SDR UI.

phoniclynx, 2024-09-05:
Just to try and make things a little easier I took a video shot of the area.. You can see RTL_433 running on the RTLSDR and the HackRF behind running in SDR++

https://youtu.be/v7uPJRi9B4k

You can see there is quite a bit of activity around that area, but I have no clue as to what it all is

zuckschwerdt, 2024-09-05:
It looks too busy and the band is much wider with the HackRF than the default 250k. Maybe the band is too wide and signals overlap as apparent interference.
Perhaps try to shift the center frequency (`-f 433.6M`). Also you can try to see if there is anything received with `-A`

phoniclynx, 2024-09-05:
This produces quite a bit of information here:

``` rtl_433 -d driver=hackrf -f 433.6M  -A                                                                                          ✔ 
rtl_433 version -128-NOTFOUND branch master at 202311281827 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Using device HackRF One: clock source=internal part id=a000cb3c006a4768 serial=000000000000000015b062dc2d4f37cf version=v2.0.1 
Found 1 antenna(s): TX/RX 
Found 3 gain(s): LNA 0 - 116 (step 0) AMP 0 - 116 (step 0) VGA 0 - 116 (step 0) 
Found 1 frequencies: RF 
Found 1 frequency range(s): 0 - 7250000000 (step 0) 
Found 20 sample rate range(s): 1000000 2000000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000 11000000 12000000 13000000 14000000 15000000 16000000 17000000 18000000 19000000 20000000 
Found 16 bandwidth range(s): 1750000 - 1750000 (step 0) 2500000 - 2500000 (step 0) 3500000 - 3500000 (step 0) 5000000 - 5000000 (step 0) 5500000 - 5500000 (step 0) 6000000 - 6000000 (step 0) 7000000 - 7000000 (step 0) 8000000 - 8000000 (step 0) 9000000 - 9000000 (step 0) 10000000 - 10000000 (step 0) 12000000 - 12000000 (step 0) 14000000 - 14000000 (step 0) 15000000 - 15000000 (step 0) 20000000 - 20000000 (step 0) 24000000 - 24000000 (step 0) 28000000 - 28000000 (step 0) 
Found current bandwidth 0
Found 4 stream format(s): CS8 CS16 CF32 CF64 
Found native stream format: CS8 (full scale: 128.0)
Detected FSK package	2024-09-05 16:58:28
Analyzing pulses...
Total count:   22,  width: 11.74 ms		( 2934 S)
Pulse width distribution:
 [ 0] count:    5,  width:  104 us [100;112]	(  26 S)
 [ 1] count:    6,  width:  252 us [212;320]	(  63 S)
 [ 2] count:    4,  width:  164 us [140;180]	(  41 S)
 [ 3] count:    2,  width:   60 us [60;60]	(  15 S)
 [ 4] count:    3,  width:  468 us [420;504]	( 117 S)
 [ 5] count:    1,  width:  336 us [336;336]	(  84 S)
 [ 6] count:    1,  width:   48 us [48;48]	(  12 S)
Gap width distribution:
 [ 0] count:    2,  width:  648 us [604;696]	( 162 S)
 [ 1] count:    3,  width:  992 us [920;1124]	( 248 S)
 [ 2] count:    9,  width:  248 us [208;300]	(  62 S)
 [ 3] count:    3,  width:   64 us [60;76]	(  16 S)
 [ 4] count:    1,  width:   48 us [48;48]	(  12 S)
 [ 5] count:    2,  width:   88 us [88;92]	(  22 S)
 [ 6] count:    1,  width:  148 us [148;148]	(  37 S)
Pulse period distribution:
 [ 0] count:    3,  width:  756 us [716;796]	( 189 S)
 [ 1] count:    2,  width: 1392 us [1364;1424]	( 348 S)
 [ 2] count:    9,  width:  376 us [308;420]	(  94 S)
 [ 3] count:    2,  width:  252 us [228;276]	(  63 S)
 [ 4] count:    1,  width: 1000 us [1000;1000]	( 250 S)
 [ 5] count:    3,  width:  476 us [456;500]	( 119 S)
 [ 6] count:    1,  width:  136 us [136;136]	(  34 S)
Pulse timing distribution:
 [ 0] count:    7,  width:  100 us [88;112]	(  25 S)
 [ 1] count:   15,  width:  252 us [208;320]	(  63 S)
 [ 2] count:    5,  width:  160 us [140;180]	(  40 S)
 [ 3] count:    4,  width:   60 us [60;64]	(  15 S)
 [ 4] count:    3,  width:  468 us [420;504]	( 117 S)
 [ 5] count:    1,  width:  336 us [336;336]	(  84 S)
 [ 6] count:    2,  width:   48 us [48;48]	(  12 S)
 [ 7] count:    2,  width:  648 us [604;696]	( 162 S)
 [ 8] count:    3,  width:  992 us [920;1124]	( 248 S)
 [ 9] count:    1,  width:   76 us [76;76]	(  19 S)
 [10] count:    1,  width:   36 us [36;36]	(   9 S)
Level estimates [high, low]:   2773,    488
RSSI: -15.4 dB SNR: 15.1 dB Noise: -30.5 dB
Frequency offsets [F1, F2]:   15565,  -9793	(+59.4 kHz, -37.4 kHz)
Guessing modulation: No clue...

Detected OOK package	2024-09-05 16:58:33
Analyzing pulses...
Total count:    1,  width: 11.76 ms		( 2940 S)
Pulse width distribution:
 [ 0] count:    1,  width: 11760 us [11760;11760]	(2940 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width: 11760 us [11760;11760]	(2940 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]	(25001 S)
Level estimates [high, low]:   4751,    468
RSSI: -10.8 dB SNR: 20.1 dB Noise: -30.9 dB
[... 891 lines cut ...]
 [ 2] count:    3,  width:  408 us [352;440]	( 102 S)
 [ 3] count:    7,  width:  700 us [600;800]	( 175 S)
 [ 4] count:    3,  width:  192 us [164;220]	(  48 S)
 [ 5] count:    7,  width:  292 us [252;352]	(  73 S)
Pulse timing distribution:
 [ 0] count:    6,  width:  108 us [100;124]	(  27 S)
 [ 1] count:    2,  width:  720 us [712;728]	( 180 S)
 [ 2] count:    6,  width:  320 us [292;348]	(  80 S)
 [ 3] count:    4,  width:  416 us [396;428]	( 104 S)
 [ 4] count:   12,  width:  204 us [172;240]	(  51 S)
 [ 5] count:    4,  width:  148 us [140;160]	(  37 S)
 [ 6] count:    4,  width:   60 us [56;64]	(  15 S)
 [ 7] count:    1,  width:   80 us [80;80]	(  20 S)
 [ 8] count:    3,  width:   44 us [40;48]	(  11 S)
 [ 9] count:    1,  width: 1420 us [1420;1420]	( 355 S)
 [10] count:    2,  width:  544 us [520;568]	( 136 S)
 [11] count:    1,  width:   32 us [32;32]	(   8 S)
Level estimates [high, low]:   2785,    461
RSSI: -15.4 dB SNR: 15.6 dB Noise: -31.0 dB
Frequency offsets [F1, F2]:   16039,  -9472	(+61.2 kHz, -36.1 kHz)
Guessing modulation: No clue...

^CSignal caught, exiting!

```

zuckschwerdt, 2024-09-05:
Things are received (gain is not a problem) but it all looks like noise. You would look for less pulse and gap bins (ideally 2 or 3) with much more repeats (count should roughly be the number of bits, so 40 or such is expected).

Shift nearly out of the band or use a bandwidth filter (does the HackRF support that?)

phoniclynx, 2024-09-05:
haha wouldn't have the foggiest idea

zuckschwerdt, 2024-09-05:
Ok, it does have a bandwidth filter which can be used from rtl_433 with e.g. `-t "bandwidth=4.5M"`
Though we would like 250k or such low value which does not seem possible: https://hackrf.readthedocs.io/en/latest/sampling_rate.html ("minimum bandwidth of 1.75MHz")

phoniclynx, 2024-09-05:
I tried to put it to it's lowest and as expected it didn't make any difference.. but having said that the RTLSDR should be able to go higher than that as a default too

zuckschwerdt, 2024-09-05:
With an RTL-SDR rtl_433 will by default at 433.92M use only 250k.

phoniclynx, 2024-09-06:
How come the HackRF driver can not do the same thing? If I look at the RTL-SDR on SDR++ it shows the exact same data.

zuckschwerdt, 2024-09-06:
Note that visually you perceive all different frequencies (the waterfall is a FFT). rtl_433 has no true FFT and will see only the dominant frequency -- so there needs to be one clear dominant frequency, easier the smaller the band is. There is a SigRok PulseView output (`-r mysample.cu8 -w myoutput.sr`) to visualize what rtl_433 "sees".

gdt, 2024-10-27:
question more or less answerd.    Please use mailinglist or discussions if you need further help.

closed 2024-10-27 by gdt