merbanan/rtl_433 #3046: (Possible?) WH40 message detection problems?

nitrogen76, 2024-09-04:
I have a WH40 rain gauge.  It seems that rtl_433 stops detecting signals from it.
The panel that comes with it is still recieving signals (the little wireless display on it shows a recieved signal)

While watching in debug mode, I get messages like this:

ambientweather_whx_decode] WH31E/WH31B/WH40 detected, buffer is 352 bits length
[ambientweather_whx_decode] unknown message type 90 (expected 0x30/0x40/0x68)
[ambientweather_whx_decode] WH31E/WH31B/WH40 detected, buffer is 351 bits length
[ambientweather_whx_decode] unknown message type 90 (expected 0x30/0x40/0x68)
[ambientweather_whx_decode] WH31E/WH31B/WH40 detected, buffer is 197 bits length

I'm not sure how to get the "codes" for these broken messages though, but can easily grab them if instructed how

nitrogen76, 2024-09-04:
[g003_915M_1000k.cu8.zip](https://github.com/user-attachments/files/16881818/g003_915M_1000k.cu8.zip)
This MIGHT be what i need, i hope?

nitrogen76, 2024-09-04:
[signals.zip](https://github.com/user-attachments/files/16881879/signals.zip)
Here are random signals around the time this happens.  (I also have a WS90 and WS68 that are probably in there)

zuckschwerdt, 2024-09-05:
The WH40 is protocol 113. Use `-R 113:v` to get verbose messages and raw codes.
Note that message type matches the product number, e.g. 90 is your WS90.

Excluding empty files and the WS90, WS68 only g004 & g005 could contain WH40 data -- but both seem too noisy.
It might be a problem that those signals are dead center (915.0M) perhaps try tuning a little lower or higher, e.g. `-f 915.05M`

nitrogen76, 2024-09-05:
I Tried moving the frequency up and down a bit, and it didn't really help.

I had to do -R 113:vvv to get good info, but here's what I got:

```
[pulse_slicer_pcm] Ambient Weather WH31E Thermo-Hygrometer Sensor, EcoWitt WH40 rain gauge, WS68 weather station
codes     : {523}93329b333332766934cc933fff0000329e73d83333f3bf79e7819fff4ff9fff3333e3fe79cffd0feff9eff872e0e4ffc7ffe3b7f318f32f1b3f8ce65e1879fb78e6
[ambientweather_whx_decode] WH31E/WH31B/WH40 detected, buffer is 348 bits length
[ambientweather_whx_decode] unknown message type 90 (expected 0x30/0x40/0x68)
[pulse_slicer_pcm] Ambient Weather WH31E Thermo-Hygrometer Sensor, EcoWitt WH40 rain gauge, WS68 weather station
codes     : {348}aaaaaaaaaaaa2dd490004c791542a4a2984e1463141e3fffa74b681c8b35a84dffbffb00008ae3100000000
[pulse_slicer_pcm] Ambient Weather WH31E Thermo-Hygrometer Sensor, EcoWitt WH40 rain gauge, WS68 weather station
```

I actually bought a second one to troubleshoot, and both do it

The rain gauge is located maybe 50-60 feet on the other side of a window with my reciever.  Iv'e even changed antennas and it still happens.

./rtl_433 -f 915.05M -R 113:vvv > output.txt
[output.txt](https://github.com/user-attachments/files/16897535/output.txt)

nitrogen76, 2024-09-10:
Thanks for the help.  For all i know, it might just be that the transmitter on these things is junk, as this is the 2nd one of these i've had.

closed 2024-09-10 by nitrogen76