merbanan/rtl_433 #3047: support to switchdoc SwitchDoc Labs F016TH Thermo-Hygrometer

tim-oe, 2024-09-05:
they forked this and added an implementation for it,.
https://github.com/switchdoclabs/rtl_433/blob/master/src/devices/switchdoclabs_F016TH.c
not sure if this already exists by another name.
it there a way to get this added to this repo?
switchdoc is now defunct in thehobiest space and the source hasn't been updated in a couple of years.

tim-oe, 2024-09-05:
i think this is the sainlogic version of the same device, the part # is pretty similar  WS016TH
https://www.sainlogic.com/english/additional-temperature-and-humidity-sensor-for-sainlogic-weather-station-ft0300.html

tim-oe, 2024-09-05:
a little RTM... it's supported by device 20.

closed 2024-09-05 by tim-oe

zuckschwerdt, 2024-09-05:
Yes ;) We did clean up and merge interesting parts of that fork.
https://github.com/merbanan/rtl_433/blob/f0ba1538213a2fcb7487862fa1cfe7a1969a174a/src/devices/ambient_weather.c#L14-L21