merbanan/rtl_433 #3051: WARNING: Undeclared field "temperature (zone 0)" in [185] "Honeywell CM921 Wireless Programmable Room Thermostat"

nijel, 2024-09-09:
I'm getting the following warning:

```
WARNING: Undeclared field "temperature (zone 0)" in [185] "Honeywell CM921 Wireless Programmable Room Thermostat"
```

I guess this is because the field name is dynamic and not defined in the static field list. Is this something to worry about?

zuckschwerdt, 2024-09-09:
Spot on! We don't have declared field names for the dynamicly generated keys.
Declared field names are only used for column names in the CSV output (as we will need to output a header before any actual data from decoders is seen).
Just know that CSV will not work properly with dynamic field names. JSON is recommended over CSV anyway ;)

gdt, 2024-10-28:
sounds like 'not a bug'

closed 2024-10-28 by gdt