merbanan/rtl_433 #3054: Eeztire / Continental TPMS not decoding with test repo file

Crsarmv7l, 2024-09-15:
I wanted to look into Continental TPMS a bit and noticed it wasn't listed in the devices, a little further digging took me to this issue https://github.com/merbanan/rtl_433/issues/2920 which clued me in that it is Eeztire.

I grabbed this file from the test repo: https://github.com/merbanan/rtl_433_tests/blob/master/tests/continental_tpms/01/g001_433.92M_250k.cu8 

Ran it through rtl_433 and...nothing:
[image]

I realize my build is a bit outdated (but built after initial support for the decoder was added Feb 2023), so I built and installed a new version just now. Still nothing.
[image]

I thought maybe the decoder was disabled by default, but it isn't.

zuckschwerdt, 2024-09-15:
You need to force the `-Y minmax` discriminator. It's a default for 868M and above and recommended for FSK.

Crsarmv7l, 2024-09-15:
> You need to force the `-Y minmax` discriminator. It's a default for 868M and above and recommended for FSK.

Ah thanks. Seems to work. Interesting that Continental is BMW?

[image]

closed 2024-09-15 by Crsarmv7l

ProfBoc75, 2024-09-15:
@Crsarmv7l : Yes, depends on the brand id:

- 0x00 = Audi Pressure Alert,
- 0x03 = HUF Gen 5/Beru, 
- 0x23 = 35 = Schrader/Sensata, 
- 0x80 = 128 = Continental, 
- 0x88 = 136 = Audi

Crsarmv7l, 2024-09-15:
> @Crsarmv7l : Yes, depends on the brand id:
> 
> * 0x00 = Audi Pressure Alert,
> * 0x03 = HUF Gen 5/Beru,
> * 0x23 = 35 = Schrader/Sensata,
> * 0x80 = 128 = Continental,
> * 0x88 = 136 = Audi

Gotcha. Is the short/long width really 25? That seems really fast for TPMS, most of the others I have played with and created synthetics for are 9600 baud -  ~20400 (elantra 2012 with short/long width 49 and before now the fastest I have seen)