merbanan/rtl_433 #3060: Sensor sometimes detected as Kedsum-TH, sometimes as Conrad-S3318P

V10lator, 2024-09-29:
So this is the same sensor:
```
time      : 2024-09-29 11:19:56                                                                                                                                                               
model     : Conrad-S3318P ID        : 117
Channel   : 1            Battery   : 1             Temperature: 23.17 C      Humidity  : 43 %          Button    : 0             Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.2 dB      SNR       : 6.6 dB        Noise     : -6.8 dB
```
```
time      : 2024-09-29 11:20:46                                                                                                                                                               
model     : Kedsum-TH    ID        : 117
Channel   : 1            Battery   : 0             Flags2    : 1             Temperature: 23.11 C      Humidity  : 43 %          Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 7.4 dB        Noise     : -7.5 dB
```

As you see both has CRC integritty but different battery state. What's going on here and how to fix this?

This is btw a cheap no-name china weather station from ebay:
[image]

zuckschwerdt, 2024-09-29:
The protocol is almost identical and a small error in reception can lead to identification as either of the protocols. You can disable the ones you don't want to see with `-R`
https://github.com/merbanan/rtl_433/blob/master/src/devices/s3318p.c
https://github.com/merbanan/rtl_433/blob/master/src/devices/esperanza_ews.c
https://github.com/merbanan/rtl_433/blob/master/src/devices/kedsum.c

closed 2024-09-29 by zuckschwerdt

V10lator, 2024-09-29:
`$ rtl_433 -C si -M time -M level -f 433900000 -R -57 -R -47` doesn't receive anything from the sensors.
`$ rtl_433 -C si -M time -M level -f 433900000 -R -57` receives nothing since some minutes (sensors are sending all `50 + ((--ID) * 3)` seconds (IIRC)).

So I guess the correct one would be `$ rtl_433 -C si -M time -M level -f 433900000 -R -47` but I need to change source codes to read the battery state correctly? Cause batteries are new and there is no low bat warning for any of the sensors on the weather station nor the sensors itself. Or am I interpreting something wrong here: `B: Battery state 10 = Ok, 01 = weak, 00 = bad` ? For me that would be (converted from binary to integer) : `2 = OK, 1 = weak, 0 = bad` but I see this (as shown above) : `Battery   : 0`

Or is this maybe an unknown protocol?

zuckschwerdt, 2024-09-29:
We only have the information shown in the source files. You should be able to choose one of the protocols based on which one shows the correct battery value. With a fresh battery you are looking for `Battery: 1` (in JSON the field is named `battery_ok`).

If you have a recent version of rtl_433 then you can enable raw bytes output with e.g. `-R 47:v` and inspect the flags, maybe use a known bad battery or a variable power supply to gleam what your flags do.