merbanan/rtl_433 #3061: One concept Beaufort weather station (wh1080 clone)

roberto1984, 2024-09-30:
Hi all,

I have a one concept Beaufort weather station,another clone of the fineoffset wh1080 5 in 1 weather station.

Unfortunately,even if I try to decide it using protocol number 155
rtl_433 -f 868.3M -R 155
data is not received.

I can seek clearly data arrives if I run rtl_433 with -A 
,every 48 second,that is the interval of the data transmission by the station.
But except this,the protocol 155 is not decoding this data.
I don't know how to integrate it so if some of you can help me on doing this I can provide all you need.
Thank you very much in advance!

ProfBoc75, 2024-09-30:
Hi @roberto1984

You have also device protocol 32 with different modulation OOK.
Try these commands to get the figures and share your result here to help understand the issue:

    rtl_433 -f 868M -R 155:v -R 32:v -M level
    rtl_433 -f 868M -R 155 -R 32 -A -M level

You can play with the verbose options to have more information,  -v -vv -vvv -vvvv, same at protocol level -R 155:vv -R 32:vv
Depends on the frequency result, you may adjust the frequency, at 868M the sample rate is 1Mhz so the range is from 867.5M to 868.5M.

gdt, 2025-07-04:
No reply for 9 months.