merbanan/rtl_433 #3064: Nooelec nano 3 - Strange gain behaviour

phrxmd, 2024-10-03:
I have a Nooelec Nano 3 that I'm trying to use for 433 MHz signals. I notice strange behaviour with autolevels:

    > rtl_433 -d :2 -f 433.95M -s 1024k -C si -M level -Y autolevel -F json -F log
    rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR
    Found Rafael Micro R820T tuner
    SDR: Using device 0: NooElec, NESDR Nano 3, SN: 2, "Generic RTL2832U OEM"
    Enabled direct sampling mode, input 2
    [R82XX] PLL not locked!
    Disabled direct sampling mode
    Allocating 15 zero-copy buffers
    Auto Level: Estimated noise level is -17.9 dB, adjusting minimum detection level to -14.9 dB
    Auto Level: Estimated noise level is -20.4 dB, adjusting minimum detection level to -17.4 dB
    Auto Level: Estimated noise level is -22.5 dB, adjusting minimum detection level to -19.5 dB
    Auto Level: Estimated noise level is -24.4 dB, adjusting minimum detection level to -21.4 dB
    Auto Level: Estimated noise level is -26.1 dB, adjusting minimum detection level to -23.1 dB
    Auto Level: Estimated noise level is -27.5 dB, adjusting minimum detection level to -24.5 dB
    Auto Level: Estimated noise level is -28.8 dB, adjusting minimum detection level to -25.8 dB
    Auto Level: Estimated noise level is -29.9 dB, adjusting minimum detection level to -26.9 dB
    Auto Level: Estimated noise level is -31.4 dB, adjusting minimum detection level to -28.4 dB
    Auto Level: Estimated noise level is -32.9 dB, adjusting minimum detection level to -29.9 dB
    Auto Level: Estimated noise level is -34.0 dB, adjusting minimum detection level to -31.0 dB
    Auto Level: Estimated noise level is -35.2 dB, adjusting minimum detection level to -32.2 dB
    {"time" : "2024-10-03 12:11:01", "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.778, "humidity" : 85, "rain_mm" : 29.200, "wind_dir_deg" : 324, "wind_avg_m_s" : 2.700, "wind_max_m_s" : 3.600, "light_lux" : 18552, "uv" : 13, "mic" : "CRC", "mod" : "ASK", "freq" : 433.909, "rssi" : -24.151, "snr" : 17.993, "noise" : -42.144}
    Auto Level: Estimated noise level is -36.3 dB, adjusting minimum detection level to -33.3 dB
    Auto Level: Estimated noise level is -37.3 dB, adjusting minimum detection level to -34.3 dB
    {"time" : "2024-10-03 12:11:11", "model" : "Rubicson-Temperature", "id" : 148, "channel" : 1, "battery_ok" : 1, "temperature_C" : 14.600, "mic" : "CRC", "mod" : "ASK", "freq" : 434.013, "rssi" : -19.403, "snr" : 22.742, "noise" : -42.144}
    {"time" : "2024-10-03 12:11:11", "model" : "Solight-TE44", "id" : 148, "channel" : 1, "temperature_C" : 14.600, "mic" : "CRC", "mod" : "ASK", "freq" : 434.013, "rssi" : -19.403, "snr" : 22.742, "noise" : -42.144}
    Auto Level: Estimated noise level is -35.3 dB, adjusting minimum detection level to -32.3 dB
    Auto Level: Estimated noise level is -36.5 dB, adjusting minimum detection level to -33.5 dB
    Auto Level: Estimated noise level is -37.5 dB, adjusting minimum detection level to -34.5 dB
    {"time" : "2024-10-03 12:11:17", "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.778, "humidity" : 85, "rain_mm" : 29.200, "wind_dir_deg" : 295, "wind_avg_m_s" : 3.300, "wind_max_m_s" : 4.000, "light_lux" : 18365, "uv" : 15, "mic" : "CRC", "mod" : "ASK", "freq" : 433.908, "rssi" : -24.151, "snr" : 17.993, "noise" : -42.144}

I've tried to compare this with running the same Nano3 not directly, but through `rtl_tcp`, and with an RTL-Blog v4. Here is the same when I run it through `rtl_tcp`:

    > rtl_433 -d rtl_tcp -f 433.95M -s 1024k -C si -M level -Y autolevel -F json -F log
    rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR
    SDR: rtl_tcp input from localhost port 1234
    connect: Connection refused
    SDR: rtl_tcp connected to localhost:1234 (Tuner: R820T)
    {"time" : "2024-10-03 12:09:25", "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.778, "humidity" : 85, "rain_mm" : 29.200, "wind_dir_deg" : 308, "wind_avg_m_s" : 2.100, "wind_max_m_s" : 3.000, "mic" : "CRC", "mod" : "ASK", "freq" : 433.908, "rssi" : -0.104, "snr" : 7.124, "noise" : -7.228}
    {"time" : "2024-10-03 12:09:41", "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.833, "humidity" : 85, "rain_mm" : 29.200, "wind_dir_deg" : 301, "wind_avg_m_s" : 2.000, "wind_max_m_s" : 2.600, "light_lux" : 23001, "uv" : 13, "mic" : "CRC", "mod" : "ASK", "freq" : 433.911, "rssi" : -0.095, "snr" : 7.126, "noise" : -7.221}
    {"time" : "2024-10-03 12:09:41", "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.833, "humidity" : 85, "rain_mm" : 29.200, "wind_dir_deg" : 301, "wind_avg_m_s" : 2.000, "wind_max_m_s" : 2.600, "light_lux" : 23001, "uv" : 13, "mic" : "CRC", "mod" : "ASK", "freq" : 433.911, "rssi" : -0.095, "snr" : 7.126, "noise" : -7.221}

For reference here is the output of `rtl_tcp` during that time, nothing special:

    > rtl_tcp
    Found 1 device(s):
      0:  NooElec, NESDR Nano 3, SN: 2
    
    Using device 0: Generic RTL2832U OEM
    Found Rafael Micro R820T tuner
    Enabled direct sampling mode, input 2
    [R82XX] PLL not locked!
    Disabled direct sampling mode
    Tuned to 100000000 Hz.
    listening...
    Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
    to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
    client accepted! localhost 58874
    Allocating 15 zero-copy buffers
    set sample rate 1024000
    set gain mode 0
    set freq 433950000
    worker socket bye
    Signal caught, exiting!

zuckschwerdt, 2024-10-03:
Noise of -40 dB is dead silence for an rtlsdr and normally -8 db is a jammed band, rtl_433 usually will trigger above -12 dB and needs at least 9 dB SNR.

Seems like there might be problems setting the gain for the Nano 3 (with that rtlsdr lib version). Running a second rtl_433 over rtl_tcp will again set some parqameters. You can observe that running the rtl_tcp-server rtl_433 with `-R 0 -vvvv`.

Only settings of freq and rate are passed through. You can issue those with http requests to the api also (use `-F kv -F http`) or by adjusting things in the http UI. Example commands:
```
xh :8433/cmd cmd==center_frequency val==433920123
xh :8433/cmd cmd==sample_rate val==250000
xh :8433/cmd cmd==gain arg==10
```

Try those calls on the http api when running only one rtl_433 and watch how the levels change. If it matches what you saw from the rtl_tcp experiment then we know what that apparently fixed.

phrxmd, 2024-10-03:
Thank you. Here is what happens when I feed various parameters into the HTTP interface with `xh` - first your suggestions (center frequency 433920123, sample rate 250k, gain 10), then a bit more playing with the gain settings. You do notice that at higher gain settings the autolevel messages disappear:

    > rtl_433 -d :2 -f 433.95M -s 1024k -C si -M level -Y autolevel -F json -F log -F http
    rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR
    HTTP server: Starting HTTP server at 0.0.0.0 port 8433
    HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
    Detached kernel driver
    Found Rafael Micro R820T tuner
    SDR: Using device 1: NooElec, NESDR Nano 3, SN: 2, "Generic RTL2832U OEM"
    [R82XX] PLL not locked!
    Allocating 15 zero-copy buffers
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    {"time" : "2024-10-03 13:37:57", "model" : "Renault", "type" : "TPMS", "id" : "500bc0", "flags" : "36", "pressure_kPa" : 213.000, "temperature_C" : 14.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.988, "freq2" : 433.872, "rssi" : -0.236, "snr" : 4.631, "noise" : -4.867}
    {"time" : "2024-10-03 13:37:57", "model" : "Renault", "type" : "TPMS", "id" : "500bc0", "flags" : "36", "pressure_kPa" : 213.000, "temperature_C" : 14.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 434.042, "freq2" : 433.845, "rssi" : -0.534, "snr" : 3.896, "noise" : -4.430}
    POST Got center_frequency, arg , val 433920123 (433920123)
    {"time" : "2024-10-03 13:38:33", "model" : "Renault", "type" : "TPMS", "id" : "62dc1b", "flags" : "36", "pressure_kPa" : 202.500, "temperature_C" : 18.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.996, "freq2" : 433.864, "rssi" : -0.093, "snr" : 5.631, "noise" : -5.724}
    {"time" : "2024-10-03 13:38:33", "model" : "Renault", "type" : "TPMS", "id" : "62dc1b", "flags" : "36", "pressure_kPa" : 202.500, "temperature_C" : 18.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.965, "freq2" : 433.830, "rssi" : -1.338, "snr" : 4.164, "noise" : -5.502}
    {"time" : "2024-10-03 13:38:34", "model" : "Renault", "type" : "TPMS", "id" : "065a4c", "flags" : "36", "pressure_kPa" : 227.250, "temperature_C" : 17.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.989, "freq2" : 433.850, "rssi" : -0.099, "snr" : 4.798, "noise" : -4.897}
    {"time" : "2024-10-03 13:38:34", "model" : "Renault", "type" : "TPMS", "id" : "065a4c", "flags" : "36", "pressure_kPa" : 227.250, "temperature_C" : 17.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.991, "freq2" : 433.868, "rssi" : -0.158, "snr" : 5.499, "noise" : -5.657}
    POST Got sample_rate, arg , val 250000 (250000)
    Exact sample rate is: 250000.000414 Hz
    {"time" : "2024-10-03 13:39:05", "model" : "Toyota", "type" : "TPMS", "id" : "f42f7a85", "status" : 128, "pressure_kPa" : 137.895, "temperature_C" : 22.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.934, "freq2" : 433.881, "rssi" : -0.097, "snr" : 9.867, "noise" : -9.964}
    POST Got gain, arg 10, val  (0)
    Auto Level: Estimated noise level is -18.4 dB, adjusting minimum detection level to -15.4 dB
    Auto Level: Estimated noise level is -21.4 dB, adjusting minimum detection level to -18.4 dB
    Auto Level: Estimated noise level is -24.0 dB, adjusting minimum detection level to -21.0 dB
    Auto Level: Estimated noise level is -26.3 dB, adjusting minimum detection level to -23.3 dB
    Auto Level: Estimated noise level is -28.2 dB, adjusting minimum detection level to -25.2 dB
    Auto Level: Estimated noise level is -30.0 dB, adjusting minimum detection level to -27.0 dB
    Auto Level: Estimated noise level is -31.5 dB, adjusting minimum detection level to -28.5 dB
    Auto Level: Estimated noise level is -32.8 dB, adjusting minimum detection level to -29.8 dB
    Auto Level: Estimated noise level is -34.0 dB, adjusting minimum detection level to -31.0 dB
    Auto Level: Estimated noise level is -35.0 dB, adjusting minimum detection level to -32.0 dB
    Auto Level: Estimated noise level is -36.7 dB, adjusting minimum detection level to -33.7 dB
    Auto Level: Estimated noise level is -38.0 dB, adjusting minimum detection level to -35.0 dB
    Auto Level: Estimated noise level is -39.3 dB, adjusting minimum detection level to -36.3 dB
    Auto Level: Estimated noise level is -40.5 dB, adjusting minimum detection level to -37.5 dB
    Auto Level: Estimated noise level is -41.6 dB, adjusting minimum detection level to -38.6 dB
    POST Got gain, arg 49.6, val  (0)
    {"time" : "2024-10-03 13:39:49", "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 13.444, "humidity" : 85, "rain_mm" : 29.200, "wind_dir_deg" : 308, "wind_avg_m_s" : 2.100, "wind_max_m_s" : 2.600, "mic" : "CRC", "mod" : "ASK", "freq" : 433.914, "rssi" : -12.175, "snr" : 7.441, "noise" : -19.616}
    {"time" : "2024-10-03 13:39:49", "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 13.444, "humidity" : 85, "rain_mm" : 29.200, "wind_dir_deg" : 308, "wind_avg_m_s" : 2.100, "wind_max_m_s" : 2.600, "mic" : "CRC", "mod" : "ASK", "freq" : 433.914, "rssi" : -12.175, "snr" : 7.441, "noise" : -19.616}
    POST Got gain, arg 22.5, val  (0)
    Auto Level: Estimated noise level is -28.3 dB, adjusting minimum detection level to -25.3 dB
[... cut ...]
    rtl-433/now 23.11-2 armhf [installed,local]
    rtl-sdr/stable,now 0.6.0-4 armhf [installed]

The `rtl-433` package I use is a backport from Debian unstable. The `rtl-433` in the base distribution had the same issue. In backporting I following the usual steps (with the source repository `deb-src https://deb.debian.org/debian unstable main contrib non-free` in my `sources.list` as the only change from stock): `sudo apt build-dep rtl-433`, `apt -b source rtl-433`, `sudo apt install rtl-433_23.11-2_armhf.deb`).

zuckschwerdt, 2024-10-03:
Note that you can also omit the arg for the gain call (or use an empty string) and that enables automatic gain in the rtlsdr.

> I'm not using rtl_433 as the rtl_tcp server (if I try that, nothing changes).

Good clue, as the rtl_433 will disregard the rtl_tcp gain command.

> The rtl-433 package I use is a backport from Debian unstable.

It might be easier to just use one of our binary releases. Those are just executables with very few deps needed.

I'd guess that setting automatic gain once things are running will then work. I.e. does this work: start rtl_433 as you normally would but add the http api (`-F kv -F http`). Observe bad levels, then `xh :8433/cmd cmd==gain` and now things should be working?

phrxmd, 2024-10-03:
Not sure whether passing an empty argument to the gain command has really done much. Currently autolevel is quiet. But look at the Cotech weatherstation here: initially it's seen it at -37db noise, then something happens and half a minute later it's seen again at -14db; then I issue the empty gain command and 15 seconds later it's again seen at -14 db. 

    > rtl_433 -d :2 -f 433.95M -Y autolevel -Y minmax -Y magest -C si -M level -M protocol -F json -F log -F http
    rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR
    HTTP server: Starting HTTP server at 0.0.0.0 port 8433
    HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
    Detached kernel driver
    Found Rafael Micro R820T tuner
    SDR: Using device 1: NooElec, NESDR Nano 3, SN: 2, "Generic RTL2832U OEM"
    Exact sample rate is: 250000.000414 Hz
    [R82XX] PLL not locked!
    Allocating 15 zero-copy buffers
    {"time" : "2024-10-03 15:43:33", "protocol" : 153, "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.389, "humidity" : 89, "rain_mm" : 29.200, "wind_dir_deg" : 329, "wind_avg_m_s" : 2.800, "wind_max_m_s" : 4.000, "light_lux" : 20058, "uv" : 12, "mic" : "CRC", "mod" : "ASK", "freq" : 433.926, "rssi" : -1.315, "snr" : 35.930, "noise" : -37.245}
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    {"time" : "2024-10-03 15:43:53", "protocol" : 20, "model" : "Ambientweather-F007TH", "id" : 62, "channel" : 1, "battery_ok" : 1, "temperature_C" : 14.333, "humidity" : 77, "mic" : "CRC", "mod" : "ASK", "freq" : 433.922, "rssi" : -3.220, "snr" : 10.664, "noise" : -13.884}
    {"time" : "2024-10-03 15:43:59", "protocol" : 2, "model" : "Rubicson-Temperature", "id" : 148, "channel" : 1, "battery_ok" : 1, "temperature_C" : 15.000, "mic" : "CRC", "mod" : "ASK", "freq" : 434.003, "rssi" : -0.233, "snr" : 14.569, "noise" : -14.801}
    {"time" : "2024-10-03 15:43:59", "protocol" : 85, "model" : "Solight-TE44", "id" : 148, "channel" : 1, "temperature_C" : 15.000, "mic" : "CRC", "mod" : "ASK", "freq" : 434.003, "rssi" : -0.233, "snr" : 14.569, "noise" : -14.801}
    {"time" : "2024-10-03 15:44:05", "protocol" : 153, "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.389, "humidity" : 90, "rain_mm" : 29.200, "wind_dir_deg" : 306, "wind_avg_m_s" : 3.100, "wind_max_m_s" : 4.000, "light_lux" : 20907, "uv" : 10, "mic" : "CRC", "mod" : "ASK", "freq" : 433.917, "rssi" : -1.836, "snr" : 12.520, "noise" : -14.355}
    {"time" : "2024-10-03 15:44:05", "protocol" : 153, "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.389, "humidity" : 90, "rain_mm" : 29.200, "wind_dir_deg" : 306, "wind_avg_m_s" : 3.100, "wind_max_m_s" : 4.000, "light_lux" : 20907, "uv" : 10, "mic" : "CRC", "mod" : "ASK", "freq" : 433.917, "rssi" : -1.836, "snr" : 12.520, "noise" : -14.355}
    POST Got gain, arg , val  (0)
    {"time" : "2024-10-03 15:44:18", "protocol" : 90, "model" : "Renault", "type" : "TPMS", "id" : "92686e", "flags" : "34", "pressure_kPa" : 204.000, "temperature_C" : 18.000, "mic" : "CRC", "mod" : "FSK", "freq1" : 433.890, "freq2" : 433.957, "rssi" : -7.077, "snr" : 6.794, "noise" : -13.871}
    {"time" : "2024-10-03 15:44:21", "protocol" : 153, "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.389, "humidity" : 90, "rain_mm" : 29.200, "wind_dir_deg" : 353, "wind_avg_m_s" : 3.200, "wind_max_m_s" : 4.300, "mic" : "CRC", "mod" : "ASK", "freq" : 433.918, "rssi" : -1.114, "snr" : 13.311, "noise" : -14.425}

I'm sorry if this is not really helpful. If I knew more about SDR and radio in general, I could probably give more qualified answers.

zuckschwerdt, 2024-10-03:
The autolevel will be quiet if there are no changes. You can force the level output, e.g. every second (default is 10), with `-M noise:1`

If just running rtl_433 reads around -40 dB will setting manual gain, then back to auto gain get better levels?
`xh :8433/cmd cmd==gain arg=10` then `xh :8433/cmd cmd==gain arg=`

I.e. if starting with no gain selected or `-g 0` won't work, will it work if you reset it alter on?

phrxmd, 2024-10-03:
Something definitely happens, but to be honest I'm not sure it's the gain settings causing it? 

Here we see two devices (a CoTech weatherstation and a Rubicson/Solight thermometer). We start off with no `-g` parameter, then we set gain via HTTP.
At startup we see the Rubicson/Solight with a noise floor of -33.5 dB, the CoTech with noise floor -40.1 dB.
Setting gain to 10, then to 0 does not change the SNR on the Rubicson/Solight (stays at -30-33 dB) or the CoTech (stays at -37-40 dB). Seems almost too good to me.
Setting gain to 49.6 makes the SNR on the Rubicson/Solight and on the CoTech drop to -14 dB.
Setting gain back to 0 makes the SNR on the CoTech go back up again, but only a bit (from -14 up to -22 dB).

    > /usr/bin/rtl_433 -d :2 -f 434M -Y autolevel -Y minmax -Y magest -C si -M level -M protocol -M time:iso:tz:local -F json -F log -F http -M noise:10
    rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR
    HTTP server: Starting HTTP server at 0.0.0.0 port 8433
    HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
    Detached kernel driver
    Found Rafael Micro R820T tuner
    SDR: Using device 1: NooElec, NESDR Nano 3, SN: 2, "Generic RTL2832U OEM"
    Exact sample rate is: 250000.000414 Hz
    [R82XX] PLL not locked!
    Allocating 15 zero-copy buffers
    Auto Level: Current noise level -11.2 dB, estimated noise -12.2 dB
    {"time" : "2024-10-03T18:15:01+0200", "protocol" : 153, "model" : "Cotech-367959", "id" : 186, "battery_ok" : 1, "temperature_C" : 12.056, "humidity" : 89, "rain_mm" : 29.200, "wind_dir_deg" : 42, "wind_avg_m_s" : 1.700, "wind_max_m_s" : 2.000, "light_lux" : 1980, "uv" : 5, "mic" : "CRC", "mod" : "ASK", "freq" : 433.925, "rssi" : -0.386, "snr" : 39.712, "noise" : -40.098}
    {"time" : "2024-10-03T18:15:02+0200", "protocol" : 2, "model" : "Rubicson-Temperature", "id" : 148, "channel" : 1, "battery_ok" : 1, "temperature_C" : 14.000, "mic" : "CRC", "mod" : "ASK", "freq" : 434.011, "rssi" : -0.849, "snr" : 32.658, "noise" : -33.507}
    {"time" : "2024-10-03T18:15:02+0200", "protocol" : 85, "model" : "Solight-TE44", "id" : 148, "channel" : 1, "temperature_C" : 14.000, "mic" : "CRC", "mod" : "ASK", "freq" : 434.011, "rssi" : -0.849, "snr" : 32.658, "noise" : -33.507}
    Auto Level: Current noise level -11.8 dB, estimated noise -11.9 dB
    POST Got gain, arg 10, val  (0)
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    Auto Level: Estimated noise level is -19.0 dB, adjusting minimum detection level to -16.0 dB
    Auto Level: Estimated noise level is -21.8 dB, adjusting minimum detection level to -18.8 dB
    Auto Level: Estimated noise level is -24.3 dB, adjusting minimum detection level to -21.3 dB
    Auto Level: Estimated noise level is -26.5 dB, adjusting minimum detection level to -23.5 dB
    POST Got gain, arg , val  (0)
    Auto Level: Current signal level -10.7 dB, estimated noise -25.7 dB
    Auto Level: Current signal level -12.4 dB, estimated noise -19.7 dB
    Auto Level: Current signal level -12.3 dB, estimated noise -16.2 dB
    Auto Level: Current signal level -9.4 dB, estimated noise -12.9 dB
    Auto Level: Current signal level -6.1 dB, estimated noise -11.9 dB
    {"time" : "2024-10-03T18:16:00+0200", "protocol" : 2, "model" : "Rubicson-Temperature", "id" : 148, "channel" : 1, "battery_ok" : 1, "temperature_C" : 14.000, "mic" : "CRC", "mod" : "ASK", "freq" : 434.007, "rssi" : -0.329, "snr" : 26.996, "noise" : -27.325}
    {"time" : "2024-10-03T18:16:00+0200", "protocol" : 85, "model" : "Solight-TE44", "id" : 148, "channel" : 1, "temperature_C" : 14.000, "mic" : "CRC", "mod" : "ASK", "freq" : 434.007, "rssi" : -0.329, "snr" : 26.996, "noise" : -27.325}
    POST Got gain, arg 10, val  (0)
    Auto Level: Estimated noise level is -15.9 dB, adjusting minimum detection level to -12.9 dB
    Auto Level: Estimated noise level is -19.1 dB, adjusting minimum detection level to -16.1 dB
    Auto Level: Estimated noise level is -21.9 dB, adjusting minimum detection level to -18.9 dB
    Auto Level: Estimated noise level is -24.4 dB, adjusting minimum detection level to -21.4 dB
    Auto Level: Estimated noise level is -26.6 dB, adjusting minimum detection level to -23.6 dB
    Auto Level: Estimated noise level is -28.5 dB, adjusting minimum detection level to -25.5 dB
    Auto Level: Estimated noise level is -30.1 dB, adjusting minimum detection level to -27.1 dB
    Auto Level: Estimated noise level is -31.5 dB, adjusting minimum detection level to -28.5 dB
    Auto Level: Estimated noise level is -32.8 dB, adjusting minimum detection level to -29.8 dB
    Auto Level: Estimated noise level is -33.9 dB, adjusting minimum detection level to -30.9 dB
    Auto Level: Estimated noise level is -35.7 dB, adjusting minimum detection level to -32.7 dB
    Auto Level: Estimated noise level is -37.1 dB, adjusting minimum detection level to -34.1 dB
    Auto Level: Estimated noise level is -38.2 dB, adjusting minimum detection level to -35.2 dB
    Auto Level: Current noise level -41.8 dB, estimated noise -38.2 dB
    Auto Level: Estimated noise level is -39.3 dB, adjusting minimum detection level to -36.3 dB
    POST Got gain, arg , val  (0)
[... cut ...] " : 2.300, "wind_max_m_s" : 3.000, "light_lux" : 1727, "uv" : 3, "mic" : "CRC", "mod" : "ASK", "freq" : 433.926, "rssi" : -0.418, "snr" : 22.150, "noise" : -22.568}
    Auto Level: Current signal level -10.8 dB, estimated noise -16.5 dB
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached
    bitbuffer_add_bit: Warning: row count limit (50 rows) reached

zuckschwerdt, 2024-10-03:
Something is really wrong with how the auto gain on the Nano 3 behaves. Can't tell you what though.
On a clear band with working auto gain we expect noise levels around -20 dB (very roughly, dependent on the antenna) and reception level (rssi) should in most cases be near -0 dB. The Nano 3 seems to not raise the automatic gain enough, and also only after nudging it with a few gain sets.

phrxmd, 2024-10-03:
It's very frustrating. As a newbie, when you don't know what to expect, you never know if it's user error.

In order to exclude driver issues I've tried again with the precompiled armhf binaries from here, the behaviour is the same.

I'm a bit out of ideas. What would be a way to go forward?

zuckschwerdt, 2024-10-03:
It's not an error on your part. The SDR is behaving strangely. AFAIK the Nano 3 uses a R820T2 tuner and we don't expect any of that weird gain behaviour.
It might help to update librtlsdr or to try another one. E.g. the RTL-SDR Blog V4 needs an updated rtlsdr lib. Swapping libs is not an easy task though.

I would really dig into the rf spectrum presented. I.e. run rtl_433 with http and rtl_tcp output, run an SDR UI to observe the rtl_tcp waterfall, then issue gain commands and visually check the changes in the waterfall.
Compare runs with e.g. -g0 (automatic, default), -g10, -g30 given to rtl_433 and what happens when you issue gain commands. Esp. is the waterfall identical after a round-trip (back to the initial -g value)?

Maybe even time the gain-set events with some script (using e.g. `sleep 1`'s) and record the output (`rtl_433 -w myfile_433.92M_250k.cu8`) -- you can share sample files a zip here.

phrxmd, 2024-10-05:
OK, so here are a few waterfalls from SDR++ connected to `rtl_433` with different gain values.  

I made a little script that cycles gain values at 1 second intervals so that they're visually reproducible, with a frequency change in the middle to see if that has an impact:

```sh
#!/bin/sh
SLEEP=1

# Supported gain values (29): 
# 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 
# 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 
# 40.2 42.1 43.4 43.9 44.5 48.0 49.6

for i in 10 0 0.9 12.5 20.7 32.8 40.2 49.6 0 12.5
do
  echo Setting gain $i
  xh -q 10.0.0.6:8433/cmd cmd==gain arg==$i
  sleep $SLEEP
done;

sleep 3

echo Setting frequency 434 MHz
xh -q 10.0.0.6:8433/cmd cmd==center_frequency val==434000000

for i in 10 0 0.9 12.5 20.7 32.8 40.2 49.6 0
do
  echo Setting gain $i
  xh -q 10.0.0.6:8433/cmd cmd==gain arg==$i
  sleep $SLEEP
done;
```

Here's the output from `rtl_433`:

    ./rtl_433 -d :2 -f 433.92M -Y autolevel -Y minmax -Y magest -M noise:1 -F log -F http -F rtl_tcp:0.0.0.0:10433 -vvvv -R 0
    rtl_433 version -128-NOTFOUND branch  at 202408270014 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
    HTTP server: Starting HTTP server at 0.0.0.0 port 8433
    HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
    rtl_tcp server: Starting rtl_tcp server at 0.0.0.0 port 10433
    rtl_tcp server: Serving rtl_tcp on address 0.0.0.0 10433
    Disabling all device decoders.
    Protocols: Registered 0 out of 263 device decoding protocols
    Input: The internals of input handling changed, read about and report problems on PR #1978
    SDR: Found 2 device(s)
    SDR: trying device 1: NooElec, NESDR Nano 3, SN: 2
    Detached kernel driver
    Found Rafael Micro R820T tuner
    SDR: Using device 1: NooElec, NESDR Nano 3, SN: 2, "Generic RTL2832U OEM"
    Exact sample rate is: 250000.000414 Hz
    [R82XX] PLL not locked!
    SDR: Sample rate set to 250000 S/s.
    Input: Bit detection level set to 0.0 (Auto).
    SDR: Tuner gain set to Auto.
    Input: Reading samples in async mode...
    SDR: rtlsdr_set_center_freq 433920000 = 0
    SDR: Tuned to 433.920MHz.
    acquire_thread: acquire_thread enter...
    Allocating 15 zero-copy buffers
    Auto Level: Current signal level -7.7 dB, estimated noise -14.9 dB
    Auto Level: Current signal level -7.7 dB, estimated noise -14.5 dB
    rtl_tcp: client connected from 10.0.0.10 port 39826
    rtl_tcp: received command SET_FREQ with 433920000
    rtl_tcp: received command SET_SAMPLE_RATE with 250000
    rtl_tcp: received command SET_FREQ_CORRECTION with 4294967295
    rtl_tcp: received command SET_DIRECT_SAMPLING with 0
    rtl_tcp: received command SET_AGC_MODE with 0
    rtl_tcp: received command SET_BIAS_TEE with 0
    rtl_tcp: received command SET_OFFSET_TUNING with 0
    rtl_tcp: received command SET_GAIN_MODE with 1
[... 68 lines cut ...]
    Auto Level: Current signal level -7.6 dB, estimated noise -31.8 dB
    Auto Level: Current signal level -7.5 dB, estimated noise -30.3 dB
    Auto Level: Current signal level -7.5 dB, estimated noise -29.6 dB
    Auto Level: Current signal level -7.4 dB, estimated noise -28.3 dB
    Auto Level: Current signal level -7.6 dB, estimated noise -27.0 dB
    Auto Level: Current signal level -7.6 dB, estimated noise -25.8 dB
    ^CSignal caught, exiting!

Here's the waterfall (with no `-g` parameter): 

[image]

Here's the waterfall with `-g 0.9`:

[image]

Here's the waterfall with `-g 22.7`:

[image]

Here's the waterfall with `-g 49.6`:

[image]

This was a bit tedious to produce because the `rtl_tcp` connection tends to drop off a lot randomly (with a log message `rtl_tcp: send not ready for write?`, client disconnection and subsequently a hanging socket), but the behaviour in the waterfalls look quite reproducible.

gdt, 2025-07-04:
@phrxmd Where are we on this?  Seems like "with nooelec nano3, something is wrong, maybe dongle, maybe librtlsdr, maybe rtl_433" -- but I don't see a path forward.