merbanan/rtl_433 #3065: Output format for pulse analyzer (-A) data

KonradIT, 2024-10-04:
Trying to use a RTL-SDR V3 as a basic spectrum analyzer.

Using:

```bash
$ rtl_433 -A -f 915M -M level -s 1024k -R 0 
```

yields some data in the command line, most importantly the rssi in dBm format, but the rtl_433 is not obeying the "-F json:data.json" option in CLI, anyone knows how to get the pulses saved to a file or better, sent via syslog?

ProfBoc75, 2024-10-05:
@KonradIT 

-F option is applied only to the decoded outputs not to other options like -A

But for your needs, try this: (-w file.ook for new file, it will failed if a file already exist, or -W file.ook to overwrite an existing file)

    rtl_433 -f 915M -s 1024k -R 0 -W pulse_test.ook

All you need is into the ook file, clear text, like this for fsk sample:

```
;pulse data
;version 1
;timescale 1us
;created 2024-10-05 10:56:16.127376+0200
;received 2024-10-05 10:56:32.535456+0200
;fsk 29 pulses
;freq1 867960832
;freq2 868051264
;centerfreq 868000000 Hz
;samplerate 1024000 Hz
;sampledepth 8 bits
;range 42.1 dB
;rssi -12.1 dB
;snr 10.7 dB
;noise -22.8 dB
0 2
5 2
6 4
3 2
2 1
2 2
11 1
4 101
101 100
101 100
101 100
101 100
100 101
100 100
100 101
101 100
101 100
101 100
100 100
100 101
101 100
101 100
100 100
101 101
100 100
101 100
101 100
101 100
101 100
;end
```

or ook sample:

```
;received 2024-10-05 10:56:32.791321+0200
;ook 16 pulses
;freq1 867965440
;centerfreq 868000000 Hz
;samplerate 1024000 Hz
;sampledepth 8 bits
;range 42.1 dB
;rssi -12.1 dB
;snr 9.6 dB
;noise -21.7 dB
3279 34
1635 13
6388 11
6014 7937
6272 14
1996 17
13488 7938
5561 34
1513 13
3060 15
3036 21
196 21
207 11
515 44
6528 30
1006 100001
;end
```

KonradIT, 2024-10-06:
This fixes my issue, thanks.

closed 2024-10-06 by KonradIT