merbanan/rtl_433 #3067: Analysizing data from Aston Martin 433MHz Smartire TPMS

MatthewRuddy, 2024-10-11:
I have been trying to reverse decode the TPMS sensors for a 2008 Aston Martin, which uses a Smartire system with 433MHz TPMS sensors. I have no doubt they're similar to something already decoded with rtl_433, but can't quite figure out which sensors they are most like. I'm also pretty new to this!

I've been able to capture some data when running `rtl_433 -S unknown`, which I've attached. The sensors I'm using isn't inside a tyre, so it will be reading 0/1kpa, FYI. Temperature should work. I have an Autel TS408 TPMS programming device, which can activate the sensor remotely, which I'm doing to capture the data here with a Nooelec NESDR Smart V5 and an antenna.

Can anyone help point me in the right direction? I'm sure this sensor is similar to something already figured out here.

Thanks.

[g012_433.92M_250k.cu8.zip](https://github.com/user-attachments/files/17340489/g012_433.92M_250k.cu8.zip)
[g013_433.92M_250k.cu8.zip](https://github.com/user-attachments/files/17340491/g013_433.92M_250k.cu8.zip)

zuckschwerdt, 2024-10-11:
Try the sample file on https://triq.org/pdv/ -- it's quiet slow (for a TPMS) at 167 µs a half-bit, 6 kBps. And it uses OOK -- we normally see FSK at 50 µs.
Use a decoding of MC with 167 / 167. Note that there is a 3 half-bit desync at the start.

Try a flex decoder like this and record some changes in temperature or pressure to spot those data fields.
`rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=167,l=167,r=600'`
The data from e.g. your first sample is `{56}518c001e187f4b`.

MatthewRuddy, 2024-10-11:
Thanks for your reply @zuckschwerdt. I'm afraid I'm out of my depth here so I will have to figure out what most of that means 😅  I would make sense for the system to be slow - it was developed in the early 2000's and is known to be a pretty mediocre.

I ran `rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=167,l=167,r=600'` and recorded two scenarios. One with the sensor just sitting on a table, and another with me spinning it around in a circular motion (to simulate a wheel). When spinning it, upon reading with the TPMS programming device it showed 40kpa pressure. It seems moving the sensor around triggers it into providing a KPa value other than 0.

On table (no movement):
[image]

Spinning in circular motion (40kpa):
[image]

MatthewRuddy, 2024-10-12:
Another test later this evening. A few 0KPa reads, then a 70KPa read. All triggered via the Autel TS408 TPMS reader, and the later 70KPa reading by me again moving the sensor around quickly.

Before this latest test, I reprogrammed the sensor, but it appears to have the same ID as reported by the Autel reader. It reports the ID as `17489`.

I can see a bit of a pattern - the code/data repeats with `518c001e187f4b` every 1, 3, 5, 7, 9 times, etc. This never changes. That's as far as I've been able to figure out with my limited knowledge.

ProfBoc75, 2024-10-12:
@MatthewRuddy 

Decimal ID = 17489 => 0x4451 = Hexadecimal ID, if you try the bellow command I found 0x4451 in the middle of the data from your cu8 files:

     rtl_433 -R 0 -X 'n=Aston-Martin-Smartire,m=OOK_PCM,s=167,l=167,r=600,preamble=32,decode_dm,get=id:@20:{16}'

from g012_433.92M_250k.cu8: 5 x times these values

    {51}52800445103ba --> Fixed value, even with the other flex decoder and your text files, always same value
    {51}53f404451011a

from g013_433.92M_250k.cu8: 5 x times these values

    {51}52800445103ba --> Fixed value, even with the other flex decoder and your text files, always same value
    {51}54040445100ea --> Few values changed, same from your text files, second message values changed a little.

For each transmit, you have 10 messages. ( Message A then B x 5 times )
Now, I'm not sure about the preamble, I can't find any CRC, so probably a checksum from these data to guess.

Try other TPMS sensors to see if you can identify the hexa values into the codes with my command.

MatthewRuddy, 2024-10-14:
Thanks @ProfBoc75 that's really helpful and made some good progress. ID is correct - I've been able to reprogram a spare sensor with custom IDs and each time it's been correct. I've also figured out the temperature reading has an offset of 40, so figured out this much:

```
5 31 58 96 7f 02 e2 [ID 1611391, 0x31 = 49 - 40 = 9c, put in fridge for 5 mins]
5 14 58 96 7f 03 e2 [ID 1611391, 0x14 = 20 - 40 = -20c, freezer]
5 27 58 96 7f 01 7a [ID 1611391, 0x27 = 39 - 40 = -1c, out of freezer warmed in hand 2 mins later]
5 2f 58 96 7f 00 02 [ID 1611391, 0x2f = 47 - 40 = 7c, still warming up]
5 3a 58 96 7f 00 3a [ID 1611391, 0x3a = 58 - 40 = 18c, warmed up in hand again]
```

I did notice that in the above strings, the ID parts are missing a leading 1? ID `1611391` = `18967F` but I'm only seeing `58967F` here. The max ID I can program is 9999999 (7 digits) FYI. 8 digit IDs are not supported (can't be programmed by the Autel reader).

I'm still unsure on pressure. Can't find a correlation. The chip in this TPMS is a Sensornor SP11A, but I can't find any datasheets/documentation about it. I have found the datasheet on a newer variant of the IC used in another TPMS sensor, which might give some clues (it gave me the -40c offset for temp). It uses the SP13 version of the chip, which is 6/7 years newer - https://www.alldatasheet.com/datasheet-pdf/view/208717/MICROCHIP/AN238.html.

ProfBoc75, 2024-10-14:
@MatthewRuddy 

I guessed the data layout, at least the first part, missing the checksum not yet decoded.

bitbench:

    ? 4h TEMP/PRESSURE 8d MESSAGE_TYPE 2b ID 22d FLAG ? 4b CHECKSUM ? 8h TRAILING ? 4h

Message_type is 2 bit flags, 00 or 01

The ID is 22 bit coded and it match with your new ID, just to confirm, did you try to set ID 9 999 999 ?
And not only able to program but also able to read the ID 9999999 from your Autel device.
As for me the max is 4194303 / 0x3FFFFF (22 bits)

About the data layout, could be:
Message A (type b00) = Pressure offset 40, Message B (type b01) = Temp, offset 40.

More details here: [Bitbench](https://triq.net/bitbench#c=%7B51%7D52800445103ba%20%5B%20message%20A%20from%20g012_433.92M_250k.cu8%2C%20ID%2017489%2C%20PRESSURE%200%20kPa%20%5D&c=&c=%7B51%7D53f404451011a%20%5B%20message%20B%20from%20g012_433.92M_250k.cu8%2C%20ID%2017489%20%20TEMP%2023C%5D&c=&c=&c=%7B51%7D52800445103ba%20%5B%20message%20A%20from%20g013_433.92M_250k.cu8%2C%20ID%2017489%2C%20PRESSURE%200%20kPa%20%5D&c=&c=%7B51%7D54040445100ea%20%5B%20message%20B%20from%20g013_433.92M_250k.cu8%2C%20ID%2017489%2C%20TEMP%2024C%20%5D&c=&c=5%2031%2058%2096%207f%2002%20e2%20%5BID%201611391%2C%200x31%20%3D%2049%20-%2040%20%3D%209c%2C%20put%20in%20fridge%20for%205%20mins%2C%200kpa%5D&c=5%2014%2058%2096%207f%2003%20e2%20%5BID%201611391%2C%200x14%20%3D%2020%20-%2040%20%3D%20-20c%2C%20put%20in%20freezer%2C%200kpa%5D&c=5%2027%2058%2096%207f%2001%207a%20%5BID%201611391%2C%200x27%20%3D%2039%20-%2040%20%3D%20-1c%2C%20out%20of%20freezer%20warmed%20in%20hand%202%20mins%20later%2C%200kpa%5D&c=5%202f%2058%2096%207f%2000%2002%20%5BID%201611391%2C%200x2f%20%3D%2047%20-%2040%20%3D%207c%2C%20still%20warming%20up%2C%200kpa%5D&c=5%203a%2058%2096%207f%2000%203a%20%5BID%201611391%2C%200x3a%20%3D%2058%20-%2040%20%3D%2018c%2C%20warmed%20up%20in%20hand%20again%2C%200kpa%5D&f=%3F%204h%20TEMP%2FPRESSURE%208d%20MESSAGE_TYPE%202b%20ID%2022d%20FLAG%20%3F%204b%20CHECKSUM%20%3F%208h%20TRAILING%20%3F%204h&z=1&cw=4)

MatthewRuddy, 2024-10-14:
Thanks @ProfBoc75 - you're correct. I was actually writing a response as you posted! :)

I put the sensor inside a garden pressure sprayer and was able to simulate pressures. Using the data below, I've figured the formula is `X - 40 = Y * 2.5 = Z`, Z being the KPa value. Similar to the datasheet above, but with an offset like the temperature of -40. Here is a couple of values I recorded:

```
5 37 18967f 03 d2 (0x37 = 55) ((55 - 40) * 2.5 = 38kpa)
5 42 58967f 01 c2 (0x42 = 66) ((66 - 40) = 26c)

548 18967f 02 22 (0x48 = 72) ((72 - 40) * 2.5 = 80kpa)
543 58967f 00 a2 ((66 - 40) = 27c)

5 56 18967f 00 c2 (0x56 = 86) ((86 - 40) * 2.5 = 115kpa)
5 43 58967f 00 a2 ((67 - 40) = 27c)

5 64 18967f 03 3a (0x64 = 100) ((100 - 40) * 2.5 = 150kpa))
5 41 58967f 02 62 ((65 - 40) = 25c)
```

Nearly there! Very promising, although unusual it comes as two messages.

I will try 9999999 as an ID - you're probably correct here.

ProfBoc75, 2024-10-14:
@MatthewRuddy :

Can you please share other samples, as I need more to guess the checksum, I'm close to the good one, 2 possibilities for the moment guessed with [revdgst](https://github.com/triq-org/revdgst), but I'm not sure about the exact position of the checksum, looks like it could be shifted a little, the flags before it could be less with 4 or 6 bits then the checksum (see the bitbench below)

    Done with g 0c k 0c final XOR ae using xor xor (100 %)
    Done with g 8a k 8a final XOR 28 using xor xor (100 %)
    Time elapsed in s: 0.03 for: Rev-Galois BYTE_REFLECT

Another point, we need more use cases, like: fast pressure increase, fast pressure decrease, low bat, and rolling wheel to identify some flags somewhere into the data layout.
Notice that a normal pressure around 250 kPa could also trigger flag.

[bitbench](https://triq.net/bitbench#c=%7B51%7D52800445103ba%20%5B%20message%20A%20from%20g012_433.92M_250k.cu8%2C%20ID%2017489%2C%20PRESSURE%200%20kPa%20%5D&c=%7B51%7D53f404451011a%20%5B%20message%20B%20from%20g012_433.92M_250k.cu8%2C%20ID%2017489%20%20TEMP%2023C%5D&c=&c=%7B51%7D52800445103ba%20%5B%20message%20A%20from%20g013_433.92M_250k.cu8%2C%20ID%2017489%2C%20PRESSURE%200%20kPa%20%5D&c=%7B51%7D54040445100ea%20%5B%20message%20B%20from%20g013_433.92M_250k.cu8%2C%20ID%2017489%2C%20TEMP%2024C%20%5D&c=&c=5%2031%2058%2096%207f%2002%20e2%20%5BID%201611391%2C%200x31%20%3D%2049%20-%2040%20%3D%209c%2C%20put%20in%20fridge%20for%205%20mins%2C%200kpa%5D&c=5%2014%2058%2096%207f%2003%20e2%20%5BID%201611391%2C%200x14%20%3D%2020%20-%2040%20%3D%20-20c%2C%20put%20in%20freezer%2C%200kpa%5D&c=5%2027%2058%2096%207f%2001%207a%20%5BID%201611391%2C%200x27%20%3D%2039%20-%2040%20%3D%20-1c%2C%20out%20of%20freezer%20warmed%20in%20hand%202%20mins%20later%2C%200kpa%5D&c=5%202f%2058%2096%207f%2000%2002%20%5BID%201611391%2C%200x2f%20%3D%2047%20-%2040%20%3D%207c%2C%20still%20warming%20up%2C%200kpa%5D&c=5%203a%2058%2096%207f%2000%203a%20%5BID%201611391%2C%200x3a%20%3D%2058%20-%2040%20%3D%2018c%2C%20warmed%20up%20in%20hand%20again%2C%200kpa%5D&c=&c=5%2037%2018967f%2003%20d2%20%5B0x37%20%3D%2055%2C%2055%20-%2040%2C%20*%202.5%20%3D%2038kpa%20%5D&c=5%2042%2058967f%2001%20c2%20%5B0x42%20%3D%2066%2C%2066%20-%2040%20%3D%2026c%20%5D&c=&c=5%2048%2018967f%2002%2022%20%5B0x48%20%3D%2072%2C%2072%20-%2040%2C%20*%202.5%20%3D%2080kpa%20%5D&c=5%2043%2058967f%2000%20a2%20%5B66%20-%2040%20%3D%2027c%20%5D&c=&c=5%2056%2018967f%2000%20c2%20%5B0x56%20%3D%2086%2C%2086%20-%2040%2C%20*%202.5%20%3D%20115kpa%20%5D&c=5%2043%2058967f%2000%20a2%20%5B67%20-%2040%20%3D%2027c%20%5D&c=&c=5%2064%2018967f%2003%203a%20%5B0x64%20%3D%20100%2C%20100%20-%2040%2C%20*%202.5%20%3D%20150kpa%5D&c=5%2041%2058967f%2002%2062%20%5B65%20-%2040%20%3D%2025c%20%5D&f=4x%20TEMP%2FPRESSURE%208d%20MESSAGE_TYPE%202b%20ID%2022d%20FLAG%20%3F%208b%20CHECKSUM%208h&z=1&cw=4) updated with your last findings

MatthewRuddy, 2024-10-18:
@ProfBoc75 I will try get some more in-car readings for you over the weekend. Might be difficult as only one sensor currently works (they're all 14 years old!). The sensor I've been testing with isn't installed.

I've been using a garden pressure spray bottle (for spraying weed killer etc) to simulate the increase and decrease in pressures. I'm not sure if this is useful. Here are the results. The highest I can get it to go is 235/240kpa.

[rapid-decrease.txt](https://github.com/user-attachments/files/17434454/rapid-decrease.txt)
[rapid-increase.txt](https://github.com/user-attachments/files/17434455/rapid-increase.txt)

I ran the commands with `-W` flag and can upload those files if also useful? They're quite large, 33mb and 66mb respectively.

ProfBoc75, 2024-10-18:
@MatthewRuddy: if you recorded files, you can zip them into single file and upload here as you did already. This will let me test them / replay the files.

zuckschwerdt, 2024-10-18:
> quite large, 33mb and 66mb respectively.

Perhaps best to upload such big files somewhere else, e.g. some free file hoster.

MatthewRuddy, 2024-10-19:
Here it is :) https://www.dropbox.com/scl/fi/sscc3tu37djqvvtocb72f/rtl_433_rapid_increase_decrease.zip?rlkey=0aloumt77t3z6j9aq7pdlvrc2&st=vo23uat1&dl=0

ProfBoc75, 2024-10-19:
@MatthewRuddy 

Some findings and good news, I guessed the CRC and found Fast Pressure Increase Flag position, but still needs some clarification.

- The message is based on 6 bytes, but missing the last bit, expected 48 bits, and we have 47 bits.
- Looking at the last byte and the 2 or 3 last bits (LSB), they are always xxxxx01 (7bits) or xxxxx010 (8bits), but I found a CRC-7 here, poly 0x45, init 0x6f , final xor 0x00.
- Looking before last byte, the first MSB bit is Fast Pressure Increase flag, the last 2 LSB bits are may be linked to CRC or Parity or XOR ???, here the issue, because of the 2 or 3 last bits of the last byte are always the same, these 2 last bits could be part of CRC (MSB/LSB ?), but in that case I'm not able to find the algorithm. And these 2 bits are not linked to a flag behavior at Temp nor Pressure level, but previous bits for sure, like a checksum.

About behavior, in case of fast pressure increasing a single message is sent with the flag to 1 and pressure value, following 10 messages with the new pressure, still flag to 1, and sent every 1.5 or 2 seconds. When the pressure is stable, the flag disappeared, and message sent less often.

I did not see flag when pressure decreased.

[Bitbench ](https://triq.net/bitbench#c=%7B51%7D52800445103ba%20%5B%20message%20A%20from%20g012_433.92M_250k.cu8%2C%20ID%2017489%2C%20PRESSURE%200%20kPa%20%5D&c=%7B51%7D53f404451011a%20%5B%20message%20B%20from%20g012_433.92M_250k.cu8%2C%20ID%2017489%20%20TEMP%2023C%5D&c=&c=%7B51%7D52800445103ba%20%5B%20message%20A%20from%20g013_433.92M_250k.cu8%2C%20ID%2017489%2C%20PRESSURE%200%20kPa%20%5D&c=%7B51%7D54040445100ea%20%5B%20message%20B%20from%20g013_433.92M_250k.cu8%2C%20ID%2017489%2C%20TEMP%2024C%20%5D&c=&c=5%2031%2058%2096%207f%2002%20e2%20%5BID%201611391%2C%200x31%20%3D%2049%20-%2040%20%3D%209c%2C%20put%20in%20fridge%20for%205%20mins%2C%200kpa%5D&c=5%2014%2058%2096%207f%2003%20e2%20%5BID%201611391%2C%200x14%20%3D%2020%20-%2040%20%3D%20-20c%2C%20put%20in%20freezer%2C%200kpa%5D&c=5%2027%2058%2096%207f%2001%207a%20%5BID%201611391%2C%200x27%20%3D%2039%20-%2040%20%3D%20-1c%2C%20out%20of%20freezer%20warmed%20in%20hand%202%20mins%20later%2C%200kpa%5D&c=5%202f%2058%2096%207f%2000%2002%20%5BID%201611391%2C%200x2f%20%3D%2047%20-%2040%20%3D%207c%2C%20still%20warming%20up%2C%200kpa%5D&c=5%203a%2058%2096%207f%2000%203a%20%5BID%201611391%2C%200x3a%20%3D%2058%20-%2040%20%3D%2018c%2C%20warmed%20up%20in%20hand%20again%2C%200kpa%5D&c=&c=5%2037%2018967f%2003%20d2%20%5B0x37%20%3D%2055%2C%2055%20-%2040%2C%20*%202.5%20%3D%2038kpa%20%5D&c=5%2042%2058967f%2001%20c2%20%5B0x42%20%3D%2066%2C%2066%20-%2040%20%3D%2026c%20%5D&c=&c=5%2048%2018967f%2002%2022%20%5B0x48%20%3D%2072%2C%2072%20-%2040%2C%20*%202.5%20%3D%2080kpa%20%5D&c=5%2043%2058967f%2000%20a2%20%5B66%20-%2040%20%3D%2027c%20%5D&c=&c=5%2056%2018967f%2000%20c2%20%5B0x56%20%3D%2086%2C%2086%20-%2040%2C%20*%202.5%20%3D%20115kpa%20%5D&c=5%2043%2058967f%2000%20a2%20%5B67%20-%2040%20%3D%2027c%20%5D&c=&c=5%2064%2018967f%2003%203a%20%5B0x64%20%3D%20100%2C%20100%20-%2040%2C%20*%202.5%20%3D%20150kpa%5D&c=5%2041%2058967f%2002%2062%20%5B65%20-%2040%20%3D%2025c%20%5D&c=&c=5833fffff029a%20&c=5427fffff016a&c=5633fffff039a&c=55e3fffff0112&c=5593fffff011a&c=5543fffff02aa&c=54f3fffff0282&c=54b3fffff012a&c=5473fffff03fa&c=5433fffff0052&c=5413fffff0292&c=53d3fffff00c2&c=53a3fffff00ca&c=5333fffff00d2&c=52e3fffff0192&c=52d3fffff0232&c=5283fffff00fa&c=5407fffff03aa&c=5403fffff03f2&c=5533fffff02a2&c=5417fffff02ca&c=5653fffff02f2&c=56f3fffff014a&c=56f3fffff837a&c=5723fffff823a&c=5417fffff80fa&c=5753fffff8232&c=5773fffff80f2&c=57a3fffff0172&c=5427fffff016a&c=5803fffff013a&c=5863fffff0052&c=5843fffff0292&f=4x%20TEMP%2FPRESSURE%208d%20MESSAGE_TYPE%202b%20ID%2022d%20FAST_INCREASE%201b%20FLAGS%20%3F%205b%20PARITY_XOR%20%3F%202b%20CRC_SEVEN%207b%201x&z=1&cw=4)updated

ProfBoc75, 2024-10-19:
@MatthewRuddy , I created the pr #3072 with a draft version of the decoder, which is working from my side with your sample files.

Please, test and let me know here or into the pr if you have any findings/improvement or correction.

MatthewRuddy, 2024-10-24:
Thank you @ProfBoc75, I will test this as soon as I can. Will share any findings.

I finally managed to get in-car driving data. This is from a 10 minute drive, with some traffic: [driving.txt](https://github.com/user-attachments/files/17507921/driving.txt). I forgot to run -W, my apologies.

I think it's actually plausible that two of the sensors are working - I'm unsure (IDs `113640` and `1119036`).

ProfBoc75, 2024-10-24:
@MatthewRuddy:

We have 2 new types into your last driving.txt file, across 3 IDs.

- Message type b10 ( = 2), with different values (0x82 = 130, 0x85 = 133, 0x80 = 128), need more sample, (speed/wheel position ??? ) as we have only one value for each sensor, 
- Message type b11 ( = 3), always with the same value 0xa8 = 168, whatever the sensor ID. optimal pressure / battery level / ??? This value was repeated several times, always before and after message type 2, see table here:

Sorted by Sensor ID then Date Time.
Date   Time | Codes | Value (dec) | Message | ID | Pressure kPa | Temp °C
-- | -- | -- | -- | -- | -- | --
24/10/2024   08:07:33 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:33 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:07:34 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:34 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:07:34 | {51}5a8c194d502da | 168 | 3 | 103637 |   |  
24/10/2024   08:07:34 | {51}5808194d50032 | 128 | 2 | 103637 |   |  
24/10/2024   08:07:34 | {51}5a8c194d502da | 168 | 3 | 103637 |   |  
24/10/2024   08:07:34 | {51}5a8c194d502da | 168 | 3 | 103637 |   |  
24/10/2024   08:07:52 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:52 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:52 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:52 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:52 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:52 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:07:52 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:52 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:07:52 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:07:52 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:08:10 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:10 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:11 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:11 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:11 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:11 | {51}5434194d501b2 | 67 | 1 | 103637 |   | 27
24/10/2024   08:08:11 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:11 | {51}5434194d501b2 | 67 | 1 | 103637 |   | 27
24/10/2024   08:08:11 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:11 | {51}5434194d501b2 | 67 | 1 | 103637 |   | 27
24/10/2024   08:08:29 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:29 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:29 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:29 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:29 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:08:29 | {51}5860194d501ea | 134 | 0 | 103637 | 235 |  
24/10/2024   08:08:29 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:08:29 | {51}5424194d500d2 | 66 | 1 | 103637 |   | 26
24/10/2024   08:03:27 | {51}58e01bbe803c2 | 142 | 0 | 113640 | 255 |  
24/10/2024   08:03:27 | {51}53f41bbe800b2 | 63 | 1 | 113640 |   | 23
24/10/2024   08:03:27 | {51}58e01bbe803c2 | 142 | 0 | 113640 | 255 |  
24/10/2024   08:03:27 | {51}53f41bbe800b2 | 63 | 1 | 113640 |   | 23
24/10/2024   08:03:27 | {51}58e01bbe803c2 | 142 | 0 | 113640 | 255 |  
24/10/2024   08:03:27 | {51}53f41bbe800b2 | 63 | 1 | 113640 |   | 23
24/10/2024   08:03:27 | {51}5a8c1bbe8018a | 168 | 3 | 113640 |   |  
24/10/2024   08:03:27 | {51}58581bbe801aa | 133 | 2 | 113640 |   |  
24/10/2024   08:03:27 | {51}5a8c1bbe8018a | 168 | 3 | 113640 |   |  
24/10/2024   08:03:45 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:03:45 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:03:45 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:03:45 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:03:45 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:03:46 | {51}53f41bbe800b2 | 63 | 1 | 113640 |   | 23
24/10/2024   08:03:46 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:03:46 | {51}53f41bbe800b2 | 63 | 1 | 113640 |   | 23
24/10/2024   08:03:46 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:03:46 | {51}53f41bbe800b2 | 63 | 1 | 113640 |   | 23
24/10/2024   08:04:04 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:04:04 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:04:04 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:04:04 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
24/10/2024   08:04:04 | {51}58c01bbe80102 | 140 | 0 | 113640 | 250 |  
[... 48 lines cut ...]
24/10/2024   08:04:16 | {51}58c11133c02ba | 140 | 0 | 1119036 | 250 |  
24/10/2024   08:04:17 | {51}58c11133c02ba | 140 | 0 | 1119036 | 250 |  
24/10/2024   08:04:17 | {51}58c11133c02ba | 140 | 0 | 1119036 | 250 |  
24/10/2024   08:04:17 | {51}58c11133c02ba | 140 | 0 | 1119036 | 250 |  
24/10/2024   08:04:17 | {51}58c11133c02ba | 140 | 0 | 1119036 | 250 |  
24/10/2024   08:04:17 | {51}54051133c02fa | 64 | 1 | 1119036 |   | 24
24/10/2024   08:04:17 | {51}58c11133c02ba | 140 | 0 | 1119036 | 250 |  
24/10/2024   08:04:17 | {51}54051133c02fa | 64 | 1 | 1119036 |   | 24
[... cut ...]
24/10/2024   08:07:30 | {51}54151133c039a | 65 | 1 | 1119036 |   | 25
24/10/2024   08:07:30 | {51}58c11133c02ba | 140 | 0 | 1119036 | 250 |  
24/10/2024   08:07:30 | {51}54151133c039a | 65 | 1 | 1119036 |   | 25

Looks like the message type is repeated in same order, in sequence.  0, 0, 0, 0, 0, 1, 0, 1, 0, 1, 3, 2, 3, 0, 0, 0, 0, .... 
or 0, 0, 0, 0, 0, 1, 0, 1, 0, 1, 0, 1, 0, 0, 0, 0, 0 ... strange ...

No sure if you were always driving during the capture, but the message 2 & 3 are very few compare to the 0 & 1 messages more frequent.

gdt, 2025-07-04:
@MatthewRuddy Where are we on this?  There's a PR to add support.  My question as ticket gardener: is there any useful information in the issue that is not in the PR?  Can you test the PR and comment on it?  I would like to see the PR merged and the ticket closed.

ProfBoc75, 2025-07-05:
@gdt , @MatthewRuddy 

The status on this decoder:
- 4 types of message are sent, 2 of them are guessed, the pressure and the temperature, and decoded.
- 2 other types, one is always a fixed value and the same between valves, the other is variable, these 2 types are sent only in motion, not guessed nor decoded --> I need to commit a little update to show these values and message type.
- One flag for pressure increasing is guessed also and decoded.
- 7 other flags not guessed but exposed / shown in the rtl_433 result.

I will try this weekend.