merbanan/rtl_433 #3068: RTL-SDR V4 does not Work on Windows 11

pmaine, 2024-10-12:
RTL-SDR V4 does not Work with rtl_433 on Windows 11. I downloaded the latest version fromGitHub - 23.11
My RTL-SDR V3 works well.

thank you

pmaine, 2024-10-13:
[image]

ProfBoc75, 2024-10-13:
Hi @pmaine 

May be try more recent version of rtl_433, August 27, from [Nightly releases](https://github.com/merbanan/rtl_433/releases) , as for release v23.11 you have several compiled versions for Windows x32/x64 /msvc x64 and with different libraries that could help ?

zuckschwerdt, 2024-10-13:
You might also need to replace the librtlsdr with a RTL-SDR Blog V4 supporting version.

pmaine, 2024-10-13:
I have downloaded the latest nightly of rtl_433. I next installed the Zadig V4 driver. I plug my RTL-SDR V3 in and start rtl_433 and I start seeing data immediately. So at this point I know my antenna is receiving and rtl_433 is properly decoding. Next I plug in my RTL-SDR V4 in and startup SDR++. The V4 is working fine with SDR++. I next run rtl_433 and it displays the following and hangs:C:\Users\pmain\Downloads\rtl_433-win-x64-nightly>rtl_433rtl_433 version -128-NOTFOUND branch  at 202408270014 inputs file rtl_tcp RTL-SDR SoapySDRFound Rafael Micro R828D tuner[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832UOEM"Exact sample rate is: 250000.000414 Hz
[R8  2XX] PLL not locked!
[R82XX] PLL not locked!
I hit ctl-c to exit
I next copy rtlsdr.dll,  msvcr100.dll and pthreadVC2.dll into the rtl_433 directory and run rtl_433. I see the following:
C:\Users\pmain\Downloads\rtl_433-win-x64-nightly>rtl_433
rtl_433 version -128-NOTFOUND branch  at 202408270014 inputs file rtl_tcp RTL-SDR SoapySDR
Found Rafael Micro R828D tuner
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[R82XX] PLL not locked!
It just hangs here.

pmaine, 2024-10-14:
Where do I get librtlsdr?
I have spent all day trying to get the Blog V4 to work on Windows 11 without success. I pluged the Blog V4 into a linux laptop running DragonOS and it immediately started working!

Dudestarfan, 2024-10-22:
Look here: https://github.com/hayguen/librtlsdr/releases

gdt, 2024-10-27:
question answered.   (and this is really not an rtl_433 bug.   Please use the mailinglist or discussions if you still need help.)

closed 2024-10-27 by gdt