merbanan/rtl_433 #3074: Device support for Technoline WS 9767

Sola85, 2024-10-21:
I would like to add device support for my [Technoline WS 9767](https://www.technoline-berlin.de/product/de/WS_9767) temperature sensor. I have recorded several signals and understand them to the point where I can extract the Sensor ID, Battery state, button state, temperature and I can reproduce the checksum. 

Upon replacing the battery in my sensor, I noticed that only then did my sensor get recognised as an Acurite-606TX (and the data provided by rtl_433 matches the data that I had already reconstructed using a flex-decoder). 

I emphasize, the sensor is only recognised by rtl_433 if the battery state bit is set to 1. Unfortunately the sensor spends 99% of its battery life in the self-reported low-battery state, so for 99% of the time, the sensor is not recognized.

Would you prefer if I attempt to make the existing Acurite-606TX decoder work with both Acurite and my Technoline sensor, or would you prefer a completely new decoder, despite the data and encoding being seemingly almost identical?

zuckschwerdt, 2024-10-21:
Did you find out what prevents the sensor from being decoded when the battery bit is unset?

Best case we'd fix or enhance the existing decoder. But if the battery bit is inverted then we would need a new decoder to override the Acurite-606TX -- and it would need to be default-disabled (only enabled by users who know that they have the sensor and want the battery state the right way around).

Sola85, 2024-10-22:
I'm currently guessing that the issue is [this check](https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1634) in the Acurite decoder. 
Both Technoline and Acurite sensors seem to send 33 bits, which get rounded up to 4 bytes. For Acurite, the last bit (=last byte) seems to be always zero, at least that is what the line above in the Acurite decoder sanity-checks. But for my Technoline sensor, the last bit seems to be the negated battery bit (i.e. the battery status is encoded in 2 bit positions, for whatever reason), hence the final-byte sanity check fails, depending on the battery state.

Here are two example datasets, obtained using `-X name=Technoline,modulation=OOK_PPM,s=1948,l=3900,g=4000,r=9000,priority=n,bits=32`

```
time      : 2024-10-22 17:16:01
model     : Technoline   count     : 1             num_rows  : 4             rows      :
len       : 33           data      : eb80d75e0,
len       : 33           data      : eb80d75e0,
len       : 33           data      : eb80d75e0,
len       : 32           data      : eb80d75e
codes     : {33}eb80d75e0, {33}eb80d75e0, {33}eb80d75e0, {32}eb80d75e
```
which gets correctly decoded as 
```
time      : 2024-10-22 17:16:01
model     : Acurite-606TX id        : 235
Battery   : 1            Temperature: 21.5 C       Integrity : CHECKSUM
```
(`0xeb=235` is the id, `0x8=0b1000` is the battery bit, button bit, and 2 channel bits, `0x0d7=21.5*10` is the temperature, `0x5e` is the checksum and the final zero is the bit i mentioned above.)

If I replace the batteries with lower voltage ones I get 
```
time      : 2024-10-22 17:21:02
model     : Technoline   count     : 1             num_rows  : 4             rows      :
len       : 33           data      : 0400d8fd8,
len       : 33           data      : 0400d8fd8,
len       : 33           data      : 0400d8fd8,
len       : 32           data      : 0400d8fd
codes     : {33}0400d8fd8, {33}0400d8fd8, {33}0400d8fd8, {32}0400d8fd
```
which does not get decoded. 
(Here the ID changed after battery replacement to `0x04`, the battery bit is unset, of course the temperature and checksum are slightly different, but crucially, the final bit is now set (I don't know why rtl_433 turns the final bit being set into `0x8`))

zuckschwerdt, 2024-10-22:
Bits are collected left to right, with left being MSB, thus a single set bit in rtl_433 parlance would be `{1}80` and not e.g. `{1}01`, this allows to drop a trailing unused nibble.

If you like PR your findings to the (sadly empty) doc comment for the decoder and implement the fix as needed.

Sola85, 2024-10-22:
Yup, I still need to build rtl_433 first and try out the fix (so far I only looked at code and data). Then I'll submit a PR.

"Fix" here would then probably mean removing the sanity check on the final bit.

AkaBoing, 2024-10-23:
hi Sola85,
one question:
you speak of "Both Technoline and Acurite" sensors
can u post a photo of the Technoline sensor?, (the backview with opened back cover.)
TX Button present ?

my data for 2 of them (Acurite with TX button)
<165>1 2024-10-23T08:28:52Z mx23Liqu rtl_433 - - - {"time":"2024-10-23 10:28:51","model":"Acurite-606TX","id":164,"channel":3,"battery_ok":1,"button":0,"temperature_C":9.2,"mic":"CHECKSUM","mod":"ASK","freq":433.96733,"rssi":-7.1295,"snr":28.025,"noise":-35.1545}
<165>1 2024-10-23T08:29:25Z mx23Liqu rtl_433 - - - {"time":"2024-10-23 10:29:24","model":"Acurite-606TX","id":164,"channel":3,"battery_ok":1,"button":0,"temperature_C":9.2,"mic":"CHECKSUM","mod":"ASK","freq":433.96275,"rssi":-8.6301,"snr":26.5244,"noise":-35.1545}

<165>1 2024-10-23T08:57:28Z mx23Liqu rtl_433 - - - {"time":"2024-10-23 10:57:27","model":"Acurite-606TX","id":88,"channel":2,"battery_ok":1,"button":0,"temperature_C":-18.5,"mic":"CHECKSUM","mod":"ASK","freq":433.96614,"rssi":-0.107883,"snr":32.49389,"noise":-32.6018}
<165>1 2024-10-23T09:58:57Z mx23Liqu rtl_433 - - - {"time":"2024-10-23 11:02:57","model":"Acurite-606TX","id":88,"channel":2,"battery_ok":1,"button":0,"temperature_C":-18.3,"mic":"CHECKSUM","mod":"ASK","freq":433.96717,"rssi":-1.72042,"snr":35.65257,"noise":-37.373}

battery_ok = 1 if VBatt is > 2.71V
battery_ok = 0 if VBatt is < 2.70V
no signal at all if VBatt is <2.66V

have fun
Boing

AkaBoing, 2024-10-23:
Just compiled without sanity Check b[4]
Acurite works.
Technoline -you have to ckeck yourself
The real battery_OK ckeck is in b[1]
Boing

Sola85, 2024-10-23:
> can u post a photo of the Technoline sensor?, (the backview with opened back cover.)
TX Button present ?

[image]
Yup, TX Button is present, along with 3-channel selector.
Upon pressing the tx button, a message is sent immediately and the button bit is set, apparently (from looking at the decoder) just like with the acurite sensor

AkaBoing, 2024-10-23:
Thats good,
Now compile your private acurite.c with deaktivated sanity check b[4].
should work then.
Good luck.
Boing

AkaBoing, 2024-10-23:
Sola:
And thx for the Photo.
Official, this Sensor (TX960) is for WS 6449 and WS 9490 - but NOT for WS 9767.
good to know.  lol

AkaBoing, 2024-10-23:
technoline devices - a big surprise bag
acurite TX606 = technoline TX960 = technoline TX960 with optional Humidity = technoline TXH960 with Humidity.

all the same hardware and m.b same software
i bet b[4] indicates the difference.
rtl_433 uses acurite606 for all of them m.b. this is the problem.
Boing

AkaBoing, 2024-10-25:
Sola:
my TX960 both with same behavior at low level: VBat:2.69V

Full Battery:
time      : 2024-10-24 23:44:01
model     : Acurite-606TX
Payload   : a4a05bde00
id        : 164
Channel   : 3
Battery   : 1
Button    : 0
Temperature: 9.1 C
Integrity : CHECKSUM
[106;96m[[30mpulse_slicer_ppm[96m][0m Acurite 606TX Temperature Sensor
codes     : {0}0, {33}a4a05bde0, {33}a4a05bde0, {33}a4a05bde0, {32}a4a05bde

Empty Battery:
time      : 2024-10-25 14:52:31
model     : Acurite-606TX Payload   : ac610e3380   id        : 172
Channel   : 3            Battery   : 0             Button    : 1             Temperature: 27.0 C       Integrity : CHECKSUM
[106;96m[[30mpulse_slicer_ppm[96m][0m Acurite 606TX Temperature Sensor
codes     : {0}0, {33}ac610e338, {33}ac610e338, {33}ac610e338, {32}ac610e33

time      : 2024-10-25 14:52:37
model     : Acurite-606TX Payload   : ac610e3380   id        : 172
Channel   : 3            Battery   : 0             Button    : 1             Temperature: 27.0 C       Integrity : CHECKSUM
[106;96m[[30mpulse_slicer_ppm[96m][0m Acurite 606TX Temperature Sensor
codes     : {0}0, {33}ac610e338, {33}ac610e338, {33}ac610e338, {32}ac610e33

good find!

Boing

Sola85, 2024-10-26:
@AkaBoing Thanks for confirming my observations. 

I have since compiled rtl_433 and confirmed that the sensor gets correctly decoded with unset battery bit if I remove the check on the final byte.
PR will follow shortly.

AkaBoing, 2024-10-26:
hi Sola,
as u found out:
this is my small working codechange:
634  // if (b[4] != 0)
635  // return DECODE_FAIL_SANITY;

Another problem with "acurite.c" and TX960:

if u put your TX960 very near to the antenna
it will often be interpreted as "Acurite-Rain" with strange rain value. lol
<165>1 2024-10-25T12:05:26Z mx23Liqu rtl_433 - - - {"time":"2024-10-25 14:05:25","model":"Acurite-Rain","id":98,"rain_mm":7513.0,"mod":"ASK","freq":433.95104,"rssi":-0.110065,"snr":37.26292,"noise":-37.373}
there is no CHECKSUM test , but thats another story  lol.

have fun
Boing

closed 2024-10-26 by zuckschwerdt