merbanan/rtl_433 #3079: release is overdue

gdt, 2024-10-27:
The latest release is 23.11,  and that's more than 6 months ago.  rtl_433 moves quickly, and we frequently get reports about  the latest release not working somehow.  While technically that's fair, in the land of rtl_433 one has to run git master.   Thus, the latest release should always be reasonably fresh   Arguably it should be once a month, but more than 6 is too much.

gdt, 2024-10-27:
Many packaging systems have doctrine that they only package releases.  This makes  a lot of sense, but does not play well with upstreams that don't have releases that they recommend users use.

alexmyczko, 2025-01-30:
i would like a release of 25xx within 1 month for debian 13 (trixie)….

zuckschwerdt, 2025-01-30:
Thanks! Yes, a new release will happen 1-2 weeks.

zuckschwerdt, 2025-02-18:
Let's release now. There are some decoder additions to catch up on, but the current version is tested stable.
I'll tag the release tomorrow.