merbanan/rtl_433 #3080: `[Input] async start failed (-1).`

dcousens, 2024-10-28:
Similar to https://github.com/merbanan/rtl_433/issues/2621, `git checkout 22.11` is OK.
This process is not operating with access to the network, if that's important.

```
[SDR] Found 1 device(s)
...
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 433920000 = 0
[SDR] Tuned to 433.920MHz.
[Input] async start failed (-1).
```

gdt, 2024-10-28:
Please test with current git master.   22.11 is ancient.    Also, please include all the needed information; I can't tell anything about your environment.   It is likely better if you open a discussion asking for help, vs a bug, if you aren't able to have this working on a different platform for comparison or aren't using to providing data for bug reports.   I marked this as feedback pending retest with up-to-date code and details.

dcousens, 2024-10-28:
@gdt I have updated my original post to clarify, the `22.11` tag is working.

Tags `23.11`, and `master`, are not.
What other information do you need? I upped the verbosity of the output snippet omitting only decoding protocols registrations.

gdt, 2024-10-28:
your problem seems rare and thus the path is you debugging it.  I would suggest
  - bisecting on commits from 22.11 to 23.11 to find the one that causes the problem, and then reading it
  - use ktrace/strace/ktruss or however it is spelled to look at system calls in working and not working

you still didn't explain what CPU type your system is, and what distribution it is running.  Also the precise model of your rtl-sdr dongle.   and if you have tried this on other computers.

zuckschwerdt, 2024-10-28:
The big thing that changed between those version is threading. The "async start failed" is the point where a new thread (for SDR input) is spawned. I suspect that your system has problems with threading.

zuckschwerdt, 2024-10-28:
Ah, wait, you don't have `sdr_start: error in pthread_create, rc: -1`, right? Then this is simply a case where you compiled without support for threads. Without threads rtl_433 can only only run in batch mode, convert files and such.

dcousens, 2024-10-28:
@zuckschwerdt good call, I did a fresh reset, a previous debugging step of `cmake -DENABLE_THREADS=OFF` had muddled my environment.

A threading / blocking problem remains, but the output is different.

```bash
rtl_433 version 23.11-171-g0b426692 branch master at 202410261548 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Registering protocol [142] "Fine Offset Electronics/ECOWITT WH51, SwitchDoc Labs SM23 Soil Moisture Sensor"
[Protocols] Registered 1 out of 266 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 433920000 = 0
[SDR] Tuned to 433.920MHz.
[acquire_thread] acquire_thread enter...
[Input] Input device start failed, exiting!
[sdr_stop] EXITING...
[sdr_stop] JOINING...
[rtlsdr_read_loop] rtlsdr_read_async done
[acquire_thread] acquire_thread async stop...
[acquire_thread] acquire_thread done...
[sdr_stop] EXITED.
[rtl_433] stopping...
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
(exits)
```

dcousens, 2024-10-28:
@zuckschwerdt seeing as you closed the issue, did you want me to make a new issue with the other output?

commit 96976c16 (2024-10-28): minor: Add message if threads missing (closes #3080)