merbanan/rtl_433 #3081: `[Input] Input device start failed, exiting!`

dcousens, 2024-10-28:
Followup to https://github.com/merbanan/rtl_433/issues/3080
Similar to https://github.com/merbanan/rtl_433/issues/2621, `git checkout 22.11` is OK.
`23.11` and `master` are producing the same error.

```bash
> ./src/rtl_433 -vvv -R 142
rtl_433 version 23.11-171-g0b426692 branch master at 202410261548 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
...
[Protocols] Registered 1 out of 266 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 433920000 = 0
[SDR] Tuned to 433.920MHz.
[acquire_thread] acquire_thread enter...
[Input] Input device start failed, exiting!
[sdr_stop] EXITING...
[sdr_stop] JOINING...
[rtlsdr_read_loop] rtlsdr_read_async done
[acquire_thread] acquire_thread async stop...
[acquire_thread] acquire_thread done...
[sdr_stop] EXITED.
[rtl_433] stopping...
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
(exits)
```

Compiled cleanly with
```bash
cmake .
make
```

dcousens, 2024-10-28:
@zuckschwerdt unblocking UDP outbound to loopback resolves this problem.
Is something in `rtl_433` trying to broadcast on UDP?

zuckschwerdt, 2024-10-28:
> unblocking UDP outbound to loopback resolves this problem.

Ah yes, actually the same error as in the referenced issue. The Mongoose network lib uses an internal socket to message across threads.

dcousens, 2024-10-28:
Ah, sorry I missed part of your comment in https://github.com/merbanan/rtl_433/issues/2621#issuecomment-1727776776 - the thread was quite dense and I missed it, my bad.

zuckschwerdt, 2024-10-28:
I wanted to refactor to just mutexes or conditions, but our (Mongoose) event loop is based on `select(2)` and thus needs to be triggered with an fd. As far as I can see we are stuck with Mongoose's broadcast mechanism.

dcousens, 2025-07-02:
@gdt if you are using something like `systemd` to start `rtl_433` as a service, and your service is starting from a restricted security profile, it isn't that unusual

gdt, 2025-07-02:
It sound like Linux packaging of rtl_433 should come with some example config files that allow networking, and perhaps some hints.   It is irregular on Unix to block localhost, even if systemd is into it :-)

It might be possible to use a unix-domain socket instead, but of course systemd might block that too.

A useful contribution would be a PR that adds better diagnostics for failed calls; it seems (without really investigating) that perhaps a bind call or sendto call is failing and not being checked, but the log above doesn't say that.