merbanan/rtl_433 #3082: Add support for ThermoPro TP828B

rbray89, 2024-11-01:
This device is similar enough to the ThermoPro TP829B that I was able to key off of the support implemented for it, but different enough that the existing encoder isn't working.

I have the following output with the first probe inserted, and the second unused, ending at 86F:

Temps:
Probe 1: Target=170, Actual=70's - 86
Probe 1: Target=165, Actual=None

```
rtl_433 -d rtl_tcp://radio-rtl-2.home:1234 -f 915M -F json -X "n=tp828b,m=FSK_PCM,s=102,l=102,r=2000,preamble=d2552dd4,bits>=100" | jq --unbuffered -r '.codes[0]'
rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Use "-F log" if you want any messages, warnings, and errors in the console.
{154}772c2eceaa4f3eddeaa4d7b2d2d2d2d2d200000
{164}772c2eceaa4f3eddeaa4d7b2d2d2d2d2d20000000
{154}772c5d9d549e3eddeaa4d7b2d2d2d2d2d200000
{154}772c2f8eaa4f3eddeaa4d7e8d2d2d2d2d200000
{182}772c369d549e7dbbd549af61a5a5a5a5a4000000000000
{166}772c33ceaa4f3eddeaa4d703d2d2d2d2d200000000
{170}772c341eaa4f3eddf5526bcfe969696969000000000
{154}772c335eaa4f3eddeaa4d7ccd2d2d2d2d200000
{156}772c331eaa4f3eddeaa4d70dd2d2d2d2d200000
{169}772c35fd549e7dbbd549ae39a5a5a5a5a4000000000
{154}7716194eaa4f3eddeaa4d759d2d2d2d2d200000
{154}772c31deaa4f3eddeaa4d778d2d2d2d2d200000
```

Let me know if you need anything else.

ProfBoc75, 2024-11-01:
Hi @rbray89 

The data layout is not exactly the same,, here more data / more information (temp,  target, unknown ) than tp829b which provides only temp from the 4 probes. Can you share more samples to confirm, please.

rbray89, 2024-11-02:
@ProfBoc75 

Here you are:
```
Temperatures in C:

<DATA>                                                                       <TARGET1/PROBE1> <TARGET2/PROBE2>
770ceddeaa1feeddeaa1fea0d2d2d2d2d20000000000 1/None   1/None
770ceddeaa20aeddeaa1fe43d2d2d2d2d2000000000  2/None   1/None
770ceddeaa215eddeaa1fef5d2d2d2d2d200000      3/None   1/None
770ceddeaa220eddeaa1fecad2d2d2d2d200000      4/None   1/None
770ceddeaa226eddeaa1fe03d2d2d2d2d200000      5/None   1/None
770ceddeaadaceddeaa1fef4d2d2d2d2d200000    300/None   1/None
770ceddeaadaceddeaa20a14d2d2d2d2d200000    300/None   2/None
770ceddeaadaceddeaa215add2d2d2d2d200000    300/None   3/None
770ceddeaadaceddeaa2202cd2e5a5a5a400000    300/None   4/None
770ceddeaadae76ef5511327e96969696900000    300/None   5/None

770cedd6086afeddeaadac2cd2d2d2d2d200000 BBQ 104-121/None 300/None
770cedd6086afeddeaadac2cd2d2d2d2d200000 BBQ 104-121/None 300/None
770cedd190daceddeaadac24d2d2d2d2d200000 BBQ -1-300/None 300/None
770cedd1f4daceddeaadac04d2d2d2d2d200000 BBQ 0-300/None 300/None
770cedd1f41feeddeaadac50d2d2d2d2d200000 BBQ 0-1/None 300/None
770cedd1f41feeddeaadac50d2d2d2d2d200000 BBQ 300/None 300/None
770ceddeaadacedd190dac7ed2d2d2d2d200000 BBQ 300/None -10-300/None
770ceddeaadacedd1eadacecd2d2d2d2d200000 BBQ 300/None -1-300/None
770ceddeaadacedd1f4dac3ed2d2d2d2d200000 BBQ 300/None 0-300/None
770ceddeaadacedd1f41fe41d2d2d2d2d200000 BBQ 300/None 0-1/None

770ceddeaadaceddeaadac8bd2d2d2d2d200000    300/None 300/None
770c302eaadaceddeaadacddd2d2d2d2d200000    300/27   300/None
770c320eaadaceddeaadac22d2d2d2d2d200000    300/30   300/None
770c32ceaadaceddeaadac50d2d2d2d5a400000    300/31   300/None
770c335eaadaceddeaadac04d2d2d2d2d200000    300/32   300/None
770c33aeaadaceddeaadacf2d2d2d2d2d200000    300/33   300/None
770ceddeaadac335eaadac4ad2d2d2d2d200000    300/None 300/32
770cdbbd55b5865dd55b5887a5a5a5a5a400000    300/None 300/31
770ceddeaadac323eaadac1dd2d2d2d2d200000    300/None 300/30
770ceddeaadac314eaadacbad2d2d2d2d200000    300/None 300/29
770ceddeaadac30feaadacb3d2d2d2d2d200000    300/None 300/28
```

Note that the "BBQ" samples have a high and low set-point.

rbray89, 2024-11-02:
@ProfBoc75 I think I may have gotten the format:

[bitbench](https://triq.net/bitbench#c=082f33733533c33771&c=082f33733633c33748&c=082f33733633b3370a&c=082f334331337333b4&c=082f33132e33332f87&c=082f32e32a33032d8e&c=082f32d32732e32a41&c=082f32b32432c3282d&c=082f32832232832522&c=082f327320327322db&c=082f32531e3263225f&c=082f32331c32432024&c=082f32131b32231d53&c=082f32031932031cb8&c=082f31e31831f31bc6&c=082f31b31631d319d1&c=082f31a31631c31893&c=082f31931431a316f3&c=082f3173123193157d&c=082f3173103183138d&c=082f31530f3173125f&c=082f31330e3163109a&c=082f31130c31530fc9&c=082f31030c31330d00&c=082f30f30a31230d53&c=082f30d30931030c27&c=082f30c30831030aeb&c=082f30b30730f3098b&c=085e61260a61a611d1&c=082f30830430c30603&c=082f30730230a3054d&c=082f30630130930328&c=082f304300307301c5&c=082f3032fe305300e8&c=082f3022fd3032fea5&c=082f3012fb3032fd96&c=082f3002fa3022fcc3&c=082f2ff2f93012fa39&f=ID%20%3F%3A%2016h%20TEMPONE%3A12d%20TEMPTWO%3A12d%20TEMPTHREE%3A12d%20TEMPFOUR%3A12d%20CHECKSUM%20%3F%3A8h&z=1&cw=4)

ProfBoc75, 2024-11-02:
@rbray89 

You shared the wrong bitbench with the 4 probes / TP829B, but I understand, and it's the same approach offset and scale as for the TP829B.

TP828B (2 probes) : ID {16} , 6 x Temp values {12}, Checksum {8}, 0xd2d2d2d2d2, and trailed zeros ... 
TP829B (4 probes) : ID {16} , 4 x Temp values {12}, Checksum {8}, 0xd2d2d2d2d2, and trailed zeros ... 

Now I'm thinking how to update the current decoder to add this model, I will add some checks around checksum and 0xd2 value position to get the model and then decode properly the values.

The ID ( 2 bytes) is also another possibility, but we need more feedback from other users to compare models and confirm that the model could be identify at ID level, for example, 0x08 = tp829b and 0x77 = tp828w ???  My guess, without other samples, we can't conclude.

ProfBoc75, 2024-11-02:
Notice that the checksum is not exactly the same key and final XOR:

    Done with g 98 k 16 final XOR ac using xor xor (100 %)
    Time elapsed in s: 0.16 for: Galois BIT_REFLECT BYTE_REFLECT

It's gen 0x98, key 0x55 and final XOR 0x00 for TP829B

rbray89, 2024-11-03:
@ProfBoc75 Oops, sorry: [correct Bitbench](https://triq.net/bitbench#c=770ceddeaa1feeddeaa1fea0&c=770ceddeaa20aeddeaa1fe43&c=770ceddeaa215eddeaa1fef5&c=770ceddeaa220eddeaa1feca&c=770ceddeaa226eddeaa1fe03&c=&c=770ceddeaadaceddeaa1fef4&c=770ceddeaadaceddeaa20a14&c=770ceddeaadaceddeaa215ad&c=770ceddeaadaceddeaa2202c&c=770ceddeaadae76ef5511327&c=&c=770cedd6086afeddeaadac2c&c=770cedd6086afeddeaadac2c&c=&c=770cedd190daceddeaadac24&c=770cedd1f4daceddeaadac04&c=&c=770cedd1f41feeddeaadac50&c=770cedd1f41feeddeaadac50&c=&c=770ceddeaadacedd190dac7e&c=770ceddeaadacedd1eadacec&c=770ceddeaadacedd1f4dac3e&c=770ceddeaadacedd1f41fe41&c=&c=770ceddeaadaceddeaadac8b%20&c=&c=770c302eaadaceddeaadacdd%20&c=770c320eaadaceddeaadac22%20&c=770c32ceaadaceddeaadac50%20&c=770c335eaadaceddeaadac04%20&c=770c33aeaadaceddeaadacf2%20&c=&c=770ceddeaadac335eaadac4a%20&c=770cdbbd55b5865dd55b5887%20&c=770ceddeaadac323eaadac1d%20&c=770ceddeaadac314eaadacba%20&c=770ceddeaadac30feaadacb3&f=ID%20%3F%3A%2016h%20TEMPONEP%3A12d%20TEMPONELO%3A12d%20TEMPONEHI%3A12d%20TEMPTWOP%3A12d%20TEMPTWOLO%3A12d%20TEMPTWOHI%3A12d%20CHECKSUM%20%3F%3A8h&z=1&cw=4)

Neat, Thanks for the help!

rbray89, 2024-11-07:
Probe1: target: 300 probe: 16-17C
Probe2: target: 1-300 Probe: None

Probe1: target: 74 probe: None
Probe2: target: 66 probe: 18
[cu8 samples.zip](https://github.com/user-attachments/files/17656010/cu8.samples.zip)

ProfBoc75, 2024-11-07:
@rbray89 , thanks for the samples, only g001 & g003 contain thermopro information, and for g003 I cropped the beginning to get the figures, both gave me the same values:

[image]

To share the good cu8 files, just replay them like this:

    rtl_433 -X "n=tp828b,m=FSK_PCM,s=102,l=102,r=2000,preamble=d2552dd4,bits>=100" g001_915M_1000k.cu8
[image]

if you have an answer, like above, it can be shared, do it for each cu8 file.

I'm busy, so I will update my decoder later as @zuckschwerdt shared some advices to improve / simplify it.

rbray89, 2024-11-08:
Ah I still had the fsk decoder in there, so I don't think the samples had any useful data...
Attempt #2:
[samples.zip](https://github.com/user-attachments/files/17673236/samples.zip)
I cycled through the various preset functions for each probe, starting with probe #1 then moving to probe #2, then cycling, and then finally warming up the probe from 20's to mid 30's.

ProfBoc75, 2024-11-08:
@rbray89 

Decoder is ready to merge. I reworked it to get 2 decoders but inside same file, easier to maintain.

some results with your last samples:

[image]

.../...

[image]

ProfBoc75, 2024-11-10:
@rbray89  merged into master rtl_433 , enjoy.

ProfBoc75, 2024-11-15:
@rbray89 , can you please confirm that the master rtl_433 version is working properly as expected with your TP828B ?
If yes, you can close with comment this issue.
If not, let us know and we will still work on it with your support.

Thx.

rbray89, 2024-11-28:
Looks like it's working! Thanks!

closed 2024-11-28 by rbray89