merbanan/rtl_433 #3083: Silent Gliss SG11490 blinds remote decode help

hamchapman, 2024-11-01:
I've got an electric blinds remote that I'm trying to decode and I'm struggling to work out how to make progress.

Some links about the remote:

* https://www.sg-s.co.uk/spares-parts/electrical-curtain-track-system-parts/5600-system-parts/11490?srsltid=AfmBOopMpzI4QPV2duq-pkS7fTJx6W-xJeDVfEFnO7E0nQz_4kt-Z3mq&dsort=sort_order
* https://www.discountelectriccurtains.co.uk/accessories/5-channel-wireless-wall-switch-11490/?srsltid=AfmBOor6ifFjxo5H4jX31uemS4POdMaD2S4ITrwGdznX0UmyBXGqiDyZ

A PDF with some info: 

[image]

I've captured some samples of pressing some different buttons and attached them as `sg-samples.zip`:

[sg-samples.zip](https://github.com/user-attachments/files/17603010/sg-samples.zip)

I captured them using the following command:

```
rtl_433 -f 868M -S unknown
```

Hopefully the filenames explain what each sample is but the summary is that there are 5 different types of sample, with 3 instances of each. The different "types" are:

* Channel 1, pressing up button
* Channel 1, pressing down button
* Channel 2, pressing up button
* Channel 2, pressing down button
* Channel 1, pressing stop button

Please let me know if you can suggest some ways for me to progress with getting to a decoder 🙏

zuckschwerdt, 2024-11-01:
It seems there is much noise and little of a sine waveform to clearly see the FSK.
(try https://triq.org/spectrogram-next/ to load a cu8 and mouse-scroll zoom in to see no real sine waves.)

Try a higher sample rate (`-s 2048k` ?) and different placement or antenna and such.
The first 20 or so bits in the decoding on https://triq.org/pdv/ should be a regular pattern like 101010101…

hamchapman, 2024-11-02:
Thanks for the suggestions. I've tried using a higher sample rate but things look pretty much the same, same with different placement. 

I'm using an RTL-SDR v4 with [this antenna](https://www.amazon.co.uk/dp/B08WJ9LRMB). Should that work as a combination to be able to get a good read on the signals? Any suggestions for other things I could try if not? Thanks!

zuckschwerdt, 2024-11-03:
It should work. Maybe try without an antenna and 20cm – 1m distance? You can compare with some FSK signals in https://github.com/merbanan/rtl_433_tests to get a feeling what you are looking for.

hamchapman, 2025-04-18:
I played around with the sample rate and some other flags and ran this:

```
rtl_433 -f 868.3M -s 2048k -g 28 -A -S all
```

and got what _looks_ like a better signal capture.

[image]

Here's a zip of the capture: [g003_868.3M_2048k.cu8.zip](https://github.com/user-attachments/files/19816548/g003_868.3M_2048k.cu8.zip)

Am I right and that _is_ a better capture? Should it be even better to try and work with it?

If it is good enough, how should I go about trying to decode things? I loaded it up in `urh` but couldn't ever get it to detect what was going on.

[image]

These are the settings I was using to try and get sense out of it, but whenever I tried to `Autodetect Parameters` I'd get an error.

zuckschwerdt, 2025-04-24:
The signal is better but still awefully noisy. Visibly there is quite the aliasing and if you zoom in you can't make out a decent carrier sine.

Load your sample in https://triq.org/pdv3/ and compare to e.g. https://github.com/merbanan/rtl_433_tests/blob/master/tests/bresser_6in1/01/g002_868.3M_1000k.cu8

There should be (FSK) two pronounced carrier lines, maybe some weak aliasing but not massive like in your screenshot.

[image]

Click the spectrogram then mouse-scroll zoom in, the waveform display should show something vaguely like sines (two, fast and slow for FSK).

[image]

Do you get any other clean signals to rule out a broken setup (antenna or such)? If the sender is really the problem then you could try to tune close to the signal (`-f 868.5M`) and use a narrow filter (`-Y filter 50000`).

zuckschwerdt, 2025-04-25:
Both look better. g007 above mostly looks like a proper FSK signal, you can nearly read the data by eye.
The filter is only used in the rtl_433 decoding and not visible here. Maybe I should add a filter option to PDV.

If you could get any FSK sender, preferably 868M, to compare that would be best. Maybe some (868M FSK) keyfob or a (868M FSK) temperature sensor.

Since the blinds come in 868M and 915M, maybe the antenna is not tuned or tuned to 915M and this noise is the result? Not sure. I'll try to visualize the filter option to see if we can reliably clean up a signal that way.

hamchapman, 2025-04-25:
I've ordered [a keyfob](https://cpc.farnell.com/rf-solutions/fobber-8tl1/keyfob-transmitter-868mhz-on-off/dp/RF00910?fi=wp) that should help me test at 868MHz.

In the meantime I've tested with that 434MHz Somfy remote I mentioned above and got this capture from it, which looks very clean (to my eyes at least!): 

[g006_433.4M_250k.cu8.zip](https://github.com/user-attachments/files/19908600/g006_433.4M_250k.cu8.zip)

[image]

and zoomed in on parts of it:

[image]

Does that imply anything about my setup or the remote I'm _actually_ trying to capture signals from (at 868MHz)?

hamchapman, 2025-04-25:
Okay yeah, it looks like things are a bit clearer at 869.5MHz. Here's a capture having run:

```
rtl_433 -f 869.5M -Y filter=50000 -s 2048k -M time -R 0 -S all -g 8
```

with me holding the remote about 1m away from the antenna.

[g008_869.5M_2048k.cu8.zip](https://github.com/user-attachments/files/19916071/g008_869.5M_2048k.cu8.zip)

This is what it looks like:

[image]

and a snippet zoomed in:

[image]

zuckschwerdt, 2025-04-26:
Very nice. So the sender is just at an odd freq with confusing shadows then at 868M. For a good signal try to reduce the clipping, either remove the antenna when you are only ~1m from the SDR or use more distance or reduce the gain -- you don't want color in the waveform.

hamchapman, 2025-04-26:
I tried without the antenna and couldn't seem to get any signal being recorded at all, even after cranking the gain up to 40.

I tried moving further away from the antenna (~2m and ~3m) and the signal then seemed to be too weak to even generate a "yellow"-y waveform when loaded up into the web UI.

So then I tried doing a test `g017` at ~1.5m and a second test at about 15cm and the screenshot below shows both of them (`g017` top, `g018` bottom)... and they look basically identical to me. Is that expected?

[image]

I then tried with reduced gain (6dB instead of 8dB) but still at ~1m and got a capture (`g022`) that looks like this:

[image]

Here are the mentioned captures: [17_18_22.zip](https://github.com/user-attachments/files/19925119/17_18_22.zip)

Am I getting anywhere with them? 

Maybe I've misunderstood what you mean by the clipping in terms of how it appears in the screenshots - do you mean that the waveform below the spectrogram isn't all one solid (yellow-y) colour?

zuckschwerdt, 2025-04-26:
Clipping is any color, yellow for soft clipping the orange and red for worse clipping. Ideally you want an all gray waveform.
You can also spot clipping by the "chopped of" waveform, it should be sines without flat tops or bottoms.

If you reduce the preselected gain in PDV from 6 dB to 0 dB then the amplitude bar (all red in your screenshots) also shows possible clipping, red means clipping at this level of (additional) gain.

The last screenshot seems to show nearly no clipping.

hamchapman, 2025-05-01:
So I tried again with lower gain (`-g 4` instead of `-g 6`) and I think I've managed to avoid all clipping and now have an all gray waveform.

Here's the capture.

[g037_869.5M_2048k.cu8.zip](https://github.com/user-attachments/files/20004591/g037_869.5M_2048k.cu8.zip)

And this is what it looks like (with the preselected gain also set down to 0 dB):

[image]

Is that the sort of thing I want to be aiming for?

If yes, what are the next steps I should be trying? Should I be get more captures of different operations and then load all the captures into URH and use if I can use that to figure out the messages being sent? 

Thanks again for all the help 🙏

zuckschwerdt, 2025-05-01:
Yes, the capture looks great for analysis. Drop FFT N to e.g. 64 to see better.
It's a very strange waveform though. Ideally we'd see a carrier of sine+cosine as I+Q, i.e. I and Q are the same sine only phase shifted by 90 degree. Then for OOK we'd see the carrier going on and off or for FSK we'd see a fast and a slow sine (two distinct freqs).
But here we see a capped off sine, i.e. a sine overlaid with a nearly matching faster signal (in your picture the first 4 waves in the Q waveform) then I guess we see it going out of sync, looking like a heartbeat curve for 4 wave cycles and then back in sync.
I think this is interference and not intentional. Could be an artifact of being too close to DC. Could you retry this capture with `-f 868.4` and `-f 868.3`?

hamchapman, 2025-07-31:
I managed to find a bit of time to play with this again. Below are the captures I managed to get using `-f 868.4` and `-f 868.3`.

To my untrained eye they look like a step in the wrong direction, but I honestly don't really know 😅  Part of that guess is because of the nice new UI below which seems to think that it might be OOK rather than FSK.

`-f 868.4`:

[image]

zoomed in a bit:

[image]

and `-f 868.3`:

[image]

zoomed in a bit:

[image]

[captures.zip](https://github.com/user-attachments/files/21537138/captures.zip)

And this is compared to a capture I then took with `-f 869.5` again:

[image]

zoomed in a bit:

[image]

Does anything stand out to you as something to investigate further or something to tweak?

zuckschwerdt, 2025-08-01:
Very nice, now we can see the waveform clearly as as sinesoid. Change the window to Flat-Top and FFT size to N=128 you can see the frequency as rtl_433 might see it. It's GFSK which is not well supported in rtl_433 but might work.

Looks like (quite fast) 13 µs PCM (75 kbps). After the preamble there seems to be a sync word of `a723 a723`.
But we'd need to see multiple codes and locate a checksum to be sure of sync word and alignment of the bytes.
If we align to that one trailing bit, i.e. shifting left 2 bits, we get a sync word of maybe `69c8 e9c8` which does not look as plausible.

hamchapman, 2025-08-01:
I captured [60 captures.zip](https://github.com/user-attachments/files/21554875/60captures.zip), 15 for each of channel 1, 2, 3, and the "all" channel (where the "all" channel is when all 5 channels are activated at the same time). I did 5 `UP` presses, then 5 `STOP`, then 5 `DOWN` for each channel, in sequence, consecutively. Then I got Claude Code to analyze the captures having provided it with some context about what I was trying to do and it produced this, which, to my uneducated eyes looks quite convincing, although the part about the rolling code (checksum) makes me think I might've hit a bit of a roadblock 🙃

### Signal Parameters
- **Frequency**: 869.4 MHz
- **Modulation**: GFSK (Gaussian Frequency Shift Keying)
- **Data Rate**: 75 kbps (13 µs per bit)
- **Encoding**: PCM (Pulse Code Modulation)

### Complete Packet Structure

```
[Preamble][Sync Word][Cmd Byte][Fixed Header][Variable Data][Rolling Code/Checksum]
aaaa...   b4e474e4   79XX      YZ22ba21XXYY0e4ad5ba8844e116d55XXb76   [varies per press]
```

### Detailed Field Analysis

#### 1. Preamble
- Fixed pattern: `aaaaaaaaaaaaaaaaaa` (18 bytes of alternating bits)
- Purpose: AGC settling and bit synchronization

#### 2. Sync Word
- Fixed: `b4e474e4` (likely `a723 a723` with proper alignment)
- Marks the start of data payload

#### 3. Command Byte Structure (2 bytes after sync)
The command bytes follow a clear pattern based on channel and action:

| Channel | UP Commands | STOP Commands | DOWN Commands |
|---------|-------------|---------------|---------------|
| 1       | 7922-7923   | 7923-7924     | 7924-7925     |
| 2       | 7918-791a   | 791b-791c     | 791c-791e     |
| 3       | 791f-7910   | 7911-7912     | 7912-7914     |
| ALL     | 7897-7888   | 7888-7889     | 788a-788b     |

Pattern observation:
- Each channel has a base value
- Commands increment through a range
- The increment appears to be a rolling counter (0-5 or 0-7)

#### 4. Fixed Header Analysis
After the command byte, there's a semi-fixed header that varies by channel:

| Channel | Fixed Header Pattern                    |
|---------|----------------------------------------|
| 1       | `[X]622ba2108a80e4ad5ba8844e116d55b0b76` |
| 2       | `[X]622ba2104680e4ad5ba8844e116d557cb76` |
| 3       | `[X]622ba210c280e4ad5ba8844e116d55f8b76` |
| ALL     | `[X]622ba2133280e4ad5ba8844e116d5db03f6db4` |

Where `[X]` is a single hex digit that cycles through values (appears to be a sub-counter).

Key differences by channel:
- Byte 6-7: `08a8` (ch1), `0468` (ch2), `0c28` (ch3), `1332` (all)
- Byte 20: `b0` (ch1), `7c` (ch2), `f8` (ch3), `db` (all)
- ALL channel has extended header: `...d5db03f6db4` instead of `...d55XXb76`

#### 5. Rolling Code Section
The variable section shows clear rolling code behavior:
- Changes with every button press
- Approximately 8-10 bytes of data
- No obvious sequential pattern (likely encrypted or hashed)
- Different for each press even with same command

### Command Byte Decoding Logic

The command byte appears to encode:
1. **Channel ID**: Base value determines channel
2. **Command Type**: Range determines UP/STOP/DOWN
3. **Sequence Counter**: Increments with each press (wraps around)

Pseudo-decode:
```
Channel 1: 0x792X range (X = 2-5)
Channel 2: 0x791X range (X = 8-E)  
Channel 3: 0x791X/0x790X range (X = 0-4, F)
ALL:       0x788X/0x789X range (X = 8-B, 7)
```

### Rolling Code Analysis

Each button press generates a unique rolling code:
- No repeated codes observed across 60 captures
- Code changes even for identical button/channel combinations
- Likely uses a cryptographic algorithm (KeeLoq, AES, or proprietary)
- Purpose: Prevent replay attacks

Example progression for Channel 1 UP:
```
Press 1: a991f7f73fb1efc48a0
Press 2: b6f5ed633739c8d1f0
Press 3: f1739b1564566358368
Press 4: e3dfa2dd4ae6770d60
Press 5: 940e651bcd1e3d4a918
```

zuckschwerdt, 2025-08-02:
That's some semi-plausible guesses, but not too useful.

If you take the whole body of codes an put them in [a BitBench](https://triq.org/bitbench/#c=%7B366%7D555555555555555555555555a723a723c912b115d10845407256add4422708b6aad85bb7f54c8fbfb9fd8f7e2450&c=%20%7B363%7D555555555555555555555555a723a723c910b115d10845407256add4422708b6aad85bb695b7af6b19b9ce468f8&c=%20%7B367%7D555555555555555555555555a723a723c916b115d10845407256add4422708b6aad85bb6c78b9cd8ab22b31ac1b4&c=%20%7B362%7D555555555555555555555555a723a723c914b115d10845407256add4422708b6aad85bb6e71efd16ea5733b86b0&c=%20%7B367%7D555555555555555555555555a723a723c91ab115d10845407256add4422708b6aad85bb6a4a07328de68f1ea548c&c=%20%7B365%7D555555555555555555555555a723a723c918b115d10845407256add4422708b6aad85bb7057972329f5050ae4620&c=%20%7B366%7D555555555555555555555555a723a723c91eb115d10845407256add4422708b6aad85bb7b9c5594fec35e5d6aa68&c=%20%7B366%7D555555555555555555555555a723a723c91cb115d10845407256add4422708b6aad85bb6606683fa14bac15cdc28&c=%20%7B366%7D555555555555555555555555a723a723c922b115d10845407256add4422708b6aad85bb7c035e3c43484216439f0&c=%20%7B367%7D555555555555555555555555a723a723c920b115d10845407256add4422708b6aad85bb75e118cfc539e0a7137b2&c=%20%7B367%7D555555555555555555555555a723a723c926b115d10845407256add4422708b6aad85bb77e490d8132e08ac21700&c=%20%7B366%7D555555555555555555555555a723a723c924b115d10845407256add4422708b6aad85bb78b2346a391cb5fac50b8&c=%20%7B367%7D555555555555555555555555a723a723c92ab115d10845407256add4422708b6aad85bb72adb665170137f93bb2c&c=%20%7B367%7D555555555555555555555555a723a723c928b115d10845407256add4422708b6aad85bb656ef201544c70d298ac8&c=%20%7B367%7D555555555555555555555555a723a723c92eb115d10845407256add4422708b6aad85bb6377c402dc54c6d34d460&c=%20%7B366%7D555555555555555555555555a723a723c8c4b115d10823407256add4422708b6aabe5bb7e95085bda9ff9c10e550&c=%20%7B367%7D555555555555555555555555a723a723c8cab115d10823407256add4422708b6aabe5bb689a9a56109a1dc624904&c=%20%7B363%7D555555555555555555555555a723a723c8ccb115d10823407256add4422708b6aabe5bb6b4ae6f34d866ef88c50&c=%20%7B367%7D555555555555555555555555a723a723c8d2b115d10823407256add4422708b6aabe5bb714dc2e4c980c4ff49eaa&c=%20%7B367%7D555555555555555555555555a723a723c8d6b115d10823407256add4422708b6aabe5bb6668cbc9e0b7cd3df74ec&c=%20%7B367%7D555555555555555555555555a723a723c8deb115d10823407256add4422708b6aabe5bb7818062c99529410a493c&c=%20%7B367%7D555555555555555555555555a723a723c8dcb115d10823407256add4422708b6aabe5bb72118e313745161afab14&c=%20%7B367%7D555555555555555555555555a723a723c8e2b115d10823407256add4422708b6aabe5bb65ea4cd2f52750b97c8aa&c=%20%7B367%7D555555555555555555555555a723a723c8e0b115d10823407256add4422708b6aabe5bb63f662c3bd35a6abc4626&c=%20%7B367%7D555555555555555555555555a723a723c8e6b115d10823407256add4422708b6aabe5bb60bd507447025bfc42780&c=%20%7B366%7D555555555555555555555555a723a723c8e4b115d10823407256add4422708b6aabe5bb7f67181fda4bf8d50b150&c=%20%7B367%7D555555555555555555555555a723a723c8eab115d10823407256add4422708b6aabe5bb69629e1c10481cd629d50&c=%20%7B367%7D555555555555555555555555a723a723c8eeb115d10823407256add4422708b6aabe5bb6ec5a3590e0f334d20ddc&c=%20%7B366%7D555555555555555555555555a723a723c8f0b115d10823407256add4422708b6aabe5bb65cb32b275c6f09e0b910&c=%20%7B367%7D555555555555555555555555a723a723c8f6b115d10823407256add4422708b6aabe5bb63d6c4a3fdd5c68bfd696&c=%20%7B365%7D555555555555555555555555a723a723c8feb115d10861407256add4422708b6aafc5bb6c20190e8a788b64bb9b0&c=%20%7B362%7D555555555555555555555555a723a723c8fcb115d10861407256add4422708b6aafc5bb6e258f192e6f1370fe20&c=%20%7B367%7D555555555555555555555555a723a723c882b115d10861407256add4422708b6aafc5bb6a92464aec9d4fdb74b5a&c=%20%7B364%7D555555555555555555555555a723a723c880b115d10861407256add4422708b6aafc5bb708c6245a881a5d9c3d0&c=%20%7B367%7D555555555555555555555555a723a723c884b115d10861407256add4422708b6aafc5bb67090b69c037eda91ccb6&c=%20%7B367%7D555555555555555555555555a723a723c88ab115d10861407256add4422708b6aafc5bb7d5e9ee6039002f438258&c=%20%7B366%7D555555555555555555555555a723a723c888b115d10861407256add4422708b6aafc5bb747439da24bea136d3920&c=%20%7B366%7D555555555555555555555555a723a723c88eb115d10861407256add4422708b6aafc5bb767bbbd712bb29252e938&c=%20%7B367%7D555555555555555555555555a723a723c890b115d10861407256add4422708b6aafc5bb65932d8a74dcf05a03cbc&c=%20%7B365%7D555555555555555555555555a723a723c896b115d10861407256add4422708b6aafc5bb638cc385fcc1c659e2f60&c=%20%7B367%7D555555555555555555555555a723a723c894b115d10861407256add4422708b6aafc5bb600f7a20a74d9a0a6b1d6&c=%20%7B367%7D555555555555555555555555a723a723c89ab115d10861407256add4422708b6aafc5bb7e08b8299b563809a5728&c=%20%7B367%7D555555555555555555555555a723a723c898b115d10861407256add4422708b6aafc5bb69fffec771317caf810b4&c=%20%7B367%7D555555555555555555555555a723a723c89cb115d10861407256add4422708b6aafc5bb6ebb82772f1b13e6f3fe0&c=%20%7B367%7D555555555555555555555555a723a723c8a2b115d10861407256add4422708b6aafc5bb6b784c0cec534ed179126&c=%20%7B380%7D555555555555555555555555a723a723c4b8b115d10999407256add4422708b6aed81fb6da1c66dabefb69c0ccf4622&c=%20%7B382%7D555555555555555555555555a723a723c4beb115d10999407256add4422708b6aed81fb6da1cff646c45035eea67d6d0&c=%20%7B383%7D555555555555555555555555a723a723c4bcb115d10999407256add4422708b6aed81fb6da1cdf3dac7e63666b230564&c=%20%7B383%7D555555555555555555555555a723a723c442b115d10999407256add4422708b6aed81fb6da1c3b009542d863bf7aec00&c=%20%7B383%7D555555555555555555555555a723a723c440b115d10999407256add4422708b6aed81fb6da1c9b43f436390d [... cut ...] d391&cw=4), then you can shift them around and find:
- the preamble is `aa..aa` actually
- there is a well-known sync word of `d391 d391`
- there is a length or type indication byte (e4 for the shorter messages, e2 for the longer ones.
- then a mostly-fixed portion, maybe src/dst ID and command.
- the some encrypted portion or a MAC
- there should be something like a CRC-16 somewhere, not sure yet.
- there could be some data-whitening scheme, would explain the somewhat random looking first bytes

zuckschwerdt, 2025-08-02:
If we pick just the codes with full 30 bytes (len 367) we get an CRC-16 poly 0x8005 init 0x41c6, which is somewhat plausible.
If we pick the codes with full 32 bytes (len 383) we get poly 0x8005 init 0x8277, same generator but indicates that something is off with the init, or that we still need data-whitening.

And yes, the (somewhat expected) IBM Whitening Algorithm works to produce more plausible fields (e.g. len and repeat bytes) and a stable CRC-16 of poly 0x8005 init 0xffff (i.e. "CRC-16/CMS")

The good codes are in [this BitBench](https://triq.org/bitbench/#c=%7B367%7D555555555555555555555555a723a723c916b115d10845407256add4422708b6aad85bb6c78b9cd8ab22b31ac1b4&c=%7B367%7D555555555555555555555555a723a723c91ab115d10845407256add4422708b6aad85bb6a4a07328de68f1ea548c&c=%7B367%7D555555555555555555555555a723a723c920b115d10845407256add4422708b6aad85bb75e118cfc539e0a7137b2&c=%7B367%7D555555555555555555555555a723a723c926b115d10845407256add4422708b6aad85bb77e490d8132e08ac21700&c=%7B367%7D555555555555555555555555a723a723c92ab115d10845407256add4422708b6aad85bb72adb665170137f93bb2c&c=%7B367%7D555555555555555555555555a723a723c928b115d10845407256add4422708b6aad85bb656ef201544c70d298ac8&c=%7B367%7D555555555555555555555555a723a723c92eb115d10845407256add4422708b6aad85bb6377c402dc54c6d34d460&c=%7B367%7D555555555555555555555555a723a723c8cab115d10823407256add4422708b6aabe5bb689a9a56109a1dc624904&c=%7B367%7D555555555555555555555555a723a723c8d2b115d10823407256add4422708b6aabe5bb714dc2e4c980c4ff49eaa&c=%7B367%7D555555555555555555555555a723a723c8d6b115d10823407256add4422708b6aabe5bb6668cbc9e0b7cd3df74ec&c=%7B367%7D555555555555555555555555a723a723c8deb115d10823407256add4422708b6aabe5bb7818062c99529410a493c&c=%7B367%7D555555555555555555555555a723a723c8dcb115d10823407256add4422708b6aabe5bb72118e313745161afab14&c=%7B367%7D555555555555555555555555a723a723c8e2b115d10823407256add4422708b6aabe5bb65ea4cd2f52750b97c8aa&c=%7B367%7D555555555555555555555555a723a723c8e0b115d10823407256add4422708b6aabe5bb63f662c3bd35a6abc4626&c=%7B367%7D555555555555555555555555a723a723c8e6b115d10823407256add4422708b6aabe5bb60bd507447025bfc42780&c=%7B367%7D555555555555555555555555a723a723c8eab115d10823407256add4422708b6aabe5bb69629e1c10481cd629d50&c=%7B367%7D555555555555555555555555a723a723c8eeb115d10823407256add4422708b6aabe5bb6ec5a3590e0f334d20ddc&c=%7B367%7D555555555555555555555555a723a723c8f6b115d10823407256add4422708b6aabe5bb63d6c4a3fdd5c68bfd696&c=%7B367%7D555555555555555555555555a723a723c882b115d10861407256add4422708b6aafc5bb6a92464aec9d4fdb74b5a&c=%7B367%7D555555555555555555555555a723a723c884b115d10861407256add4422708b6aafc5bb67090b69c037eda91ccb6&c=%7B367%7D555555555555555555555555a723a723c88ab115d10861407256add4422708b6aafc5bb7d5e9ee6039002f438258&c=%7B367%7D555555555555555555555555a723a723c890b115d10861407256add4422708b6aafc5bb65932d8a74dcf05a03cbc&c=%7B367%7D555555555555555555555555a723a723c894b115d10861407256add4422708b6aafc5bb600f7a20a74d9a0a6b1d6&c=%7B367%7D555555555555555555555555a723a723c89ab115d10861407256add4422708b6aafc5bb7e08b8299b563809a5728&c=%7B367%7D555555555555555555555555a723a723c898b115d10861407256add4422708b6aafc5bb69fffec771317caf810b4&c=%7B367%7D555555555555555555555555a723a723c89cb115d10861407256add4422708b6aafc5bb6ebb82772f1b13e6f3fe0&c=%7B367%7D555555555555555555555555a723a723c8a2b115d10861407256add4422708b6aafc5bb6b784c0cec534ed179126&c=%7B383%7D555555555555555555555555a723a723c4bcb115d10999407256add4422708b6aed81fb6da1cdf3dac7e63666b230564&c=%7B383%7D555555555555555555555555a723a723c442b115d10999407256add4422708b6aed81fb6da1c3b009542d863bf7aec00&c=%7B383%7D555555555555555555555555a723a723c440b115d10999407256add4422708b6aed81fb6da1c9b43f436390d9e319308&c=%7B383%7D555555555555555555555555a723a723c444b115d10999407256add4422708b6aed81fb6da1d89d587f08be882dcae86&c=%7B383%7D555555555555555555555555a723a723c44ab115d10999407256add4422708b6aed81fb6da1dabeca9ad3f37400e6f50&c=%7B383%7D555555555555555555555555a723a723c44eb115d10999407256add4422708b6aed81fb6da1d36ffa3dc4ce5347e43ac&c=%7B383%7D555555555555555555555555a723a723c44cb115d10999407256add4422708b6aed81fb6da1d6e4a7819f4105044a358&c=%7B383%7D555555555555555555555555a723a723c452b115d10999407256add4422708b6aed81fb6da1d4f399861b57bd058c116&c=%7B383%7D555555555555555555555555a723a723c450b115d10999407256add4422708b6aed81fb6da1d1117774b92781b6dfe72&c=%7B383%7D555555555555555555555555a723a723c456b115d10999407256add4422708b6aed81fb6da1cb1495633d30aba331846&c=%7B383%7D555555555555555555555555a723a723c454b115d10999407256add4422708b6aed81fb6da1c0433dd06b02f0f0b77ba&f=LEN%3A8h%20REPEAT%3A8h%208h8h%208h8h%20CHAN%3F8h%208h%208h8h%208h8h%208h8h%208h8h%20CMD%3F8h8h%20%3F8h8h%20MAC%3F8h8h%208h8h%208h8h%208h8h%20CRC%3F8h8h&z=1&a=Preamble&m=aaaad391d391&x=ffe11d9aed853324ea7ad2397097570a547d2dd86d0dba8f6759c7a2bf34ca18&cw=4). For the longer codes insert `8h8h` before "MAC".

hamchapman, 2025-08-02:
Amazing, thanks @zuckschwerdt!

Are there specific next steps I should take to try and get to the point where writing a decoder is possible? Ultimately I'd like to be able generate my own signals from my own transmitter, so it seems like figuring out what the deal is with the encrypted portion is the biggest unknown remaining. 

Should I be collecting tons more samples and then eyeballing them for patterns, or something else? 

Many thanks, as ever

hamchapman, 2026-09-04:
Just wanted to drop in and say thank you to all of you that helped with this ❤️ 

I've got a setup working at home now and it's fantastic!