merbanan/rtl_433 #3084: Help with debian bookworm install

Jocala, 2024-11-04:
I've been successfully running rtl_433 on a laptop running linux mint. I need to move my  usb radio to a new computer running debian bookworm, but it's not going well. The new computer is literally sitting right next to the old one. The antenna is unchanged. 

testing with:

rtl_433 -q  -Y autolevel -Y magest -M level -M noise  -f 433M -C customary

produces:

rtl_433 version 24.10-1-gf7c24ae6 branch master at 202410312030 inputs file rtl_tcp RTL-SDR
quiet option (-q) is default and deprecated. See -v to increase verbosity
Detached kernel driver
Found Rafael Micro R828D tuner
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[Auto Level] Current signal level -7.6 dB, estimated noise -13.9 dB

Here are the dmesgs on device insertion 

[linux mint dmesg: device works]
[395143.535788] usb 2-3: new high-speed USB device number 13 using xhci_hcd
[395143.696293] usb 2-3: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
[395143.696308] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[395143.696313] usb 2-3: Product: Blog V4
[395143.696318] usb 2-3: Manufacturer: RTLSDRBlog
[395143.696322] usb 2-3: SerialNumber: 00000001
[395143.704102] usb 2-3: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[395143.873919] usb 2-3: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[395143.873941] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
[395143.873950] usb 2-3: media controller created
[395143.874757] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[395143.877796] i2c i2c-10: Added multiplexed i2c bus 11
[395143.877800] rtl2832 10-0010: Realtek RTL2832 successfully attached
[395143.877812] usb 2-3: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[395143.877819] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
[395143.877868] r820t 11-003a: creating new instance
[395143.884894] r820t 11-003a: Rafael Micro r820t successfully identified
[395143.887175] rtl2832_sdr rtl2832_sdr.2.auto: Registered as swradio0
[395143.887178] rtl2832_sdr rtl2832_sdr.2.auto: Realtek RTL2832 SDR attached
[395143.887180] rtl2832_sdr rtl2832_sdr.2.auto: SDR API is still slightly experimental and functionality changes may follow
[395143.894015] Registered IR keymap rc-empty
[395143.894067] rc rc0: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb2/2-3/rc/rc0
[395143.894201] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[395143.894247] input: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb2/2-3/rc/rc0/input27
[395143.894550] usb 2-3: dvb_usb_v2: schedule remote query interval to 200 msecs
[395143.906904] usb 2-3: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected

[debian bookworm dmesg: not working]
[85501.714099] dvb_usb_v2: 'Realtek RTL2832U reference design:1-4.4' successfully deinitialized and disconnected
[86362.081283] usb 1-4.4: new high-speed USB device number 13 using xhci_hcd
[86362.218031] usb 1-4.4: New USB device found, idVendor=0bda, idProduct=2838, bcdDevice= 1.00
[86362.218044] usb 1-4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[86362.218048] usb 1-4.4: Product: Blog V4
[86362.218052] usb 1-4.4: Manufacturer: RTLSDRBlog
[86362.218054] usb 1-4.4: SerialNumber: 00000001
[86362.226387] usb 1-4.4: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[86362.395587] usb 1-4.4: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[86362.395606] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
[86362.395613] usb 1-4.4: media controller created
[86362.396299] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[86362.400290] i2c i2c-12: Added multiplexed i2c bus 13
[86362.400296] rtl2832 12-0010: Realtek RTL2832 successfully attached
[86362.400316] usb 1-4.4: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[86362.400352] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
[86362.400468] r820t 13-003a: creating new instance
[86362.407613] r820t 13-003a: Rafael Micro r820t successfully identified, chip type: R828D
[86362.410756] rtl2832_sdr rtl2832_sdr.1.auto: Registered as swradio0
[86362.410761] rtl2832_sdr rtl2832_sdr.1.auto: Realtek RTL2832 SDR attached
[86362.410764] rtl2832_sdr rtl2832_sdr.1.auto: SDR API is still slightly experimental and functionality changes may follow
[86362.418113] Registered IR keymap rc-empty
[86362.418207] rc rc0: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/rc/rc0
[86362.418425] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[86362.418507] input: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/rc/rc0/input173
[86362.418836] usb 1-4.4: dvb_usb_v2: schedule remote query interval to 200 msecs
[86362.431459] usb 1-4.4: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected

Package: librtlsdr0
Source: rtl-sdr
Version: 0.6.0-4

Is exactly the same on both machines. I initially just copied the rtl_433 binary from the old computer to the new, but afer that failed I recompiled a fresh copy from git. No joy.

zuckschwerdt, 2024-11-04:
Did you properly blacklist the kernel modules? I don't think there should be DVB messages and such.

Jocala, 2024-11-04:
Probably not. I'll check that out.  Thanks for the rapid response.

Jocala, 2024-11-05:
All fixed!  Thanks for your very kind assistance!

closed 2024-11-05 by zuckschwerdt