merbanan/rtl_433 #3086: Can get rtl_433.exe runing on Win 10?

kami83, 2024-11-08:
Hi,

i am using a RTL SDR v4 with rtl_433 under Linux no problem. Everything works fine. 
Under Windows i used Zadig to switch my Blog v4 device to WinUSB (v6.1.7600.16385).

But if i run rtl_433.exe under CMD i get this error.

C:\rtl_433>rtl_433.exe
rtl_433 version 24.10 branch  at 202410301452 inputs file rtl_tcp
[sdr_open] No input drivers (RTL-SDR or SoapySDR) compiled in.

SDRsharp etc is running fine.

BR kami

ProfBoc75, 2024-11-08:
@kami83 : look at this issue #3068 you could found some help regarding your issue, may be not the same, but, try also 32 and 64 bit versions, look at the dll too to get the last version of libraries...

zuckschwerdt, 2024-11-08:
Did you compile that version yourself? Or is this a release version from https://github.com/merbanan/rtl_433/releases ?

kami83, 2024-11-08:
Hi, i just used the compiled releases. After testing the 32 and 64 bit version. Both works but i am not recieving any captures from devices.

zuckschwerdt, 2024-11-08:
I guess the included README is confusing. The mentioned `rtl_433.exe` is without RTLSDR support. All MINGW64 builds have RTLSDR support, but for the MSVC builds only the ones with `rtlsdr` in the name have support.

kami83, 2024-11-08:
Ahh okay. First of all i only used MSVC. The other version are running, but no signals ;(

gdt, 2025-07-01:
This seems to have concluded as a support request.  Feel free to open a discussion to ask for help, or an issue if you can articulate a bug in the sources.

closed 2025-07-01 by gdt