merbanan/rtl_433 #3087: Broken hardware or mistake on my side?

Tobster77, 2024-11-08:
I try to receive a Honeywell doorbell type [DW915s](https://media.hornbach.de/hb/datasheet/as.46611649.pdf) ringing? The doorbell uses 868Mhz and a protocoll called ActivLink. I think found it in [this list ](https://github.com/merbanan/rtl_433?tab=readme-ov-file).

I bought a simple SDR USB stick (https://www.berrybase.de/dvb-t/dab/fm/sdr-usb-stick-mit-rtl2832u-chipsatz?number=125187) and tried first in Home Assistant, but I could get no readings. I also tried a 315M remote, but no readings as well

I also installed the stick on a Windows PC together with Zadig drivers. Here I could enable verbose mode: Looks like both the doorbell and the remote cause some receivable data, but seems not to be proper. Logs are attached.

Am I too silly to configure this, or is this a hardware issue?

Thanks for your help!
[rtl_433_868M.txt](https://github.com/user-attachments/files/17683612/rtl_433_868M.txt)
[rtl_433_315M.txt](https://github.com/user-attachments/files/17683613/rtl_433_315M.txt)

ProfBoc75, 2024-11-08:
@Tobster77 

Unfortunately your usb stick is not using a good tuner. Fitipower FC0012 is not known for being sensitive, quite the contrary, and especially at 868Mhz it's close to its limit.

Interference must be limited, use a good usb cable extension 50 cm / 1m , to reduce computer noise, a good antenna. Then play with -g gain level and -Y options could help.

Tobster77, 2024-11-09:
@ProfBoc75 Thanks for your advice!

I now used the following command line:
`rtl_433 -f 868M -g 0 -F kv -v -Y autolevel`

Then the magic happened:

```rtl_433 version 24.10 branch  at 202410301452 inputs file rtl_tcp RTL-SDR SoapySDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

[Protocols] Registered 230 out of 266 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-47 49-60 63 67-71 73-85 87-100 102-105 108-116 119-122 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-232 234-241 243-244 246-247 249-259 261-266
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Fitipower FC0012 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[SDR] Sample rate set to 1000000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] Tuned to 868.000MHz.
[Auto Level] Estimated noise level is -17.2 dB, adjusting minimum detection level to -14.2 dB
[Baseband] low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
[Auto Level] Estimated noise level is -19.1 dB, adjusting minimum detection level to -16.1 dB
[Auto Level] Estimated noise level is -20.7 dB, adjusting minimum detection level to -17.7 dB
[Auto Level] Estimated noise level is -22.1 dB, adjusting minimum detection level to -19.1 dB
[Auto Level] Estimated noise level is -23.4 dB, adjusting minimum detection level to -20.4 dB
[Auto Level] Estimated noise level is -24.4 dB, adjusting minimum detection level to -21.4 dB
[Auto Level] Estimated noise level is -26.2 dB, adjusting minimum detection level to -23.2 dB
[Auto Level] Estimated noise level is -27.6 dB, adjusting minimum detection level to -24.6 dB
[Auto Level] Estimated noise level is -28.6 dB, adjusting minimum detection level to -25.6 dB
[Auto Level] Estimated noise level is -29.8 dB, adjusting minimum detection level to -26.8 dB
[Auto Level] Estimated noise level is -30.9 dB, adjusting minimum detection level to -27.9 dB
[Auto Level] Estimated noise level is -31.9 dB, adjusting minimum detection level to -28.9 dB
[Auto Level] Estimated noise level is -30.8 dB, adjusting minimum detection level to -27.8 dB
[Auto Level] Estimated noise level is -31.8 dB, adjusting minimum detection level to -28.8 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-09 15:28:30
model     : Honeywell-ActivLink                    Class     : Doorbell      Id        : 74f14
Battery   : 1            Alert     : Normal        Secret Knock: 1           Relay     : 1             Integrity : PARITY
[Auto Level] Estimated noise level is -24.4 dB, adjusting minimum detection level to -21.4 dB
[Auto Level] Estimated noise level is -26.2 dB, adjusting minimum detection level to -23.2 dB
[Auto Level] Estimated noise level is -27.5 dB, adjusting minimum detection level to -24.5 dB
[Auto Level] Estimated noise level is -28.6 dB, adjusting minimum detection level to -25.6 dB
[Auto Level] Estimated noise level is -29.7 dB, adjusting minimum detection level to -26.7 dB
[Auto Level] Estimated noise level is -30.8 dB, adjusting minimum detection level to -27.8 dB
[Auto Level] Estimated noise level is -31.9 dB, adjusting minimum detection level to -28.9 dB
[Auto Level] Estimated noise level is -30.8 dB, adjusting minimum detection level to -27.8 dB
[Auto Level] Estimated noise level is -31.8 dB, adjusting minimum detection level to -28.8 dB
Signal caught, exiting!
```

I switched to the following conf file
```
frequency 868M
output kv
gain 0
verbose 5
pulse_detect autolevel
```
Here, I get neither `Auto Level` adjustments nor the doorbell. What am I doing wrong?

ProfBoc75, 2024-11-09:
@Tobster77 

Noise at -15 dB could be ok, depends on the signal noise ratio (snr) should be around 10 / 12 dB.

Try to get the figures with option `-M level -M noise:1`  ,  1 will give you a noise information every second

If you have a RSSI close to 0 dB , then SNR = 15 dB with Noise = -15 dB it's ok, very good signal.
But if you have a RSSI = -10 dB, then SNR = 5 dB with Noise = -15 dB, not ok, not enough, could work but less success.

Again, lot of attention around the antenna and nothing around the stick and antenna that could generate rf noise like a computer, a screen, any electronic device generate rf interference/noise.

Another option is `-Y filter=150` will filter all signals with pulse width less than 150 us, your device is little more so should help (try filter=100, filter=50 ... ), you have other -Y options like -Y magest (for FSK signal) or -Y ampest (for OOK signal) , -Y minmax (for FSK), those options could help too, you have to try one by one and find the good one for you, lot of possibility here.

Tobster77, 2024-11-10:
OK, this looks good. Conf file is:
```
frequency 868300000
output kv
gain 0
verbose 5
pulse_detect magest
report_meta noise
report_meta level
report_meta noise:1
pulse_detect filter=150
```
Output is:
```
[Auto Level] Current noise level -15.0 dB, estimated noise -15.0 dB
[Auto Level] Current signal level -0.9 dB, estimated noise -14.6 dB
[Auto Level] Current signal level 1.8 dB, estimated noise -11.3 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-11-10T14:14:28.013466+0100
model     : Honeywell-ActivLink                    Class     : Doorbell      Id        : 745bc
Battery   : 1            Alert     : Normal        Secret Knock: 1           Relay     : 1             Integrity : PARITY
Modulation: FSK          Freq1     : 868.2 MHz     Freq2     : 868.3 MHz
RSSI      : -0.7 dB      SNR       : 14.5 dB       Noise     : -15.2 dB
[Auto Level] Current noise level -15.1 dB, estimated noise -12.3 dB
[Auto Level] Current noise level -15.1 dB, estimated noise -13.7 dB
[Auto Level] Current noise level -15.0 dB, estimated noise -14.5 dB
```
If I got your point right, that is a quite usable signal, right?

Tobster77, 2024-11-13:
Thanks a lot! The cheap stick now works well. Your proposal of the filter setting was the game changer... :-)