merbanan/rtl_433 #3089: update broke wmbusmeters decryption

adamaltmejd, 2024-11-11:
I read my water meter with a RTL-SDR blog v4 using rtl_433, publish the data over MQTT and then read and decrypt it using https://github.com/wmbusmeters/wmbusmeters. The setup was working for a few weeks but broke once the latest rtl_433 version was released. When I skip rtl_433 completely and just use the built-in rtl_sdr and rtl_wmbus in wmbusmeters everything works. I'm guessing the problem is related to https://github.com/merbanan/rtl_433/pull/2749 since when I try to decrypt the data published by rtl_433 wmbusmeters produces a "CRC failed check" error.

Here is my config. I need to keep the rtl_433 in a config file because I couldn't get secrets to work properly with the rtl_433 docker image.

```
# docker-compose.yml
rtl433:
    image: hertzg/rtl_433:latest
    restart: always
    devices:
        - /dev/bus/usb
    environment:
      - TZ=${TIMEZONE}
    volumes:
      - ./rtl_433.conf:/root/rtl_433.conf

  # WMBUSMETERS
  wmbusmeters:
    image: wmbusmeters/wmbusmeters
    container_name: wmbusmeters
    secrets:
      - wmbus_mqtt_password
      - watermeter_encryption_key
    environment:
      - TZ=${TIMEZONE}
      - MQTT_USER=watermeter
      - MQTT_PASSWORD=/run/secrets/wmbus_mqtt_password
      - METER_KEY=/run/secrets/watermeter_encryption_key
    volumes:
      - ./start_wmbusmeters.sh:/start_wmbusmeters.sh
      - /etc/localtime:/etc/localtime:ro
      - ./mosquitto/config/isrgrootx1.pem:/isrgrootx1.pem
    entrypoint: /start_wmbusmeters.sh
```

rtl_433.conf
```
frequency 868.95M
sample_rate 1600000
output mqtts://mqtt.adamaltmejd.com:8883,retain=0,user=rtl433,pass=***,devices=rtl_433[/id]
```

start_wmbusmeters.sh
```
#!/bin/sh
/wmbusmeters/wmbusmeters \
    --donotprobe=all \
    --usestdoutforlogging \
    --ignoreduplicates=false \
    --shell='/usr/bin/mosquitto_pub -h mqtt.adamaltmejd.com -p 8883 --cafile /isrgrootx1.pem -u ${MQTT_USER} -P $(cat ${MQTT_PASSWORD}) --retain -t wmbusmeters/$METER_NAME -m "$METER_JSON"' \
    --alarmshell='/usr/bin/mosquitto_pub -h mqtt.adamaltmejd.com -p 8883 --cafile /isrgrootx1.pem -u ${MQTT_USER} -P $(cat ${MQTT_PASSWORD}) --retain -t wmbusmeters/$METER_NAME -m "$ALARM_TYPE $ALARM_MESSAGE"' \
    --alarmexpectedactivity="mon-sun(00-23)" \
    --alarmtimeout=3h \
    'hex:CMD(/usr/bin/mosquitto_sub -h mqtt.adamaltmejd.com -p 8883 --cafile /isrgrootx1.pem -u ${MQTT_USER} -P $(cat ${MQTT_PASSWORD}) -t rtl_433/57596857/data)' \
    BrevenTapWater multical21:c1 57596857 $(cat ${METER_KEY})
```

zuckschwerdt, 2024-11-11:
Can you share published mbus messages? We need to look at those to see if CRCs are there and if the length indication is correct.

adamaltmejd, 2024-11-11:
Encrypted but here are a couple data fields published to MQTT by rtl_433.

```
25442d2c576859571b168d2059b8225d2342d0eb0890d6275e89b81ae6081e9b0f83
25442d2c576859571b168d205adc225d2310f4ed990512f6aab71982adfe4287684c
25442d2c576859571b168d205be0225d233180c39be82d8f687b93a9413a4d7b6e35
30442d2c576859571b168d205c04235d236cd7ba3341283ecc27e680a980880b698de881ce2b555966a1e71414
```

When decrypting the last one in wmbusmeters using the key that worked two weeks ago i get

```
(wmbusmeters) version: 1.17.1-29-g3af7a4c
(config) using device: <30442d2c576859571b168d205c04235d236cd7ba3341283ecc27e680a980880b698de881ce2b555966a1e71414>:any 
(config) number of meters: 1
(config) template BrevenTapWater 57596857 multical21
(lookup) with file/hex "30442d2c576859571b168d205c04235d236cd7ba3341283ecc27e680a980880b698de881ce2b555966a1e71414" unknown
(lookup) driver: simulation hex
(main) opening <30442d2c576859571b168d205c04235d236cd7ba3341283ecc27e680a980880b698de881ce2b555966a1e71414>:any
Started config simulation listening on any
(simulation) in 
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "30442d2c576859571b168d205c04235d236cd7ba3341283ecc27e680a980880b698de881ce2b555966a1e71414"
(mbus) checkMBUSFrame
 "30442D2C576859571B168D205C04235D236CD7BA3341283ECC27E680A980880B698DE881CE2B555966A1E71414"
(wmbus) checkWMBUSFrame "30442D2C576859571B168D205C04235D236CD7BA3341283ECC27E680A980880B698DE881CE2B555966A1E71414"
(wmbus) not enough bytes, frame length byte changed from 48(30) to 44(2c)!
(simulator) is_mbus=false is_wmbus=true
(simulator) is wmbus telegram.
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 45
(wmbus) parseELL @10 35
(meter) BrevenTapWater: for me? 57596857.M=KAM.V=1b.T=16 in 57596857
(meter) BrevenTapWater: yes for me
(meter) created BrevenTapWater multical21 57596857 encrypted
(meter) used meter template BrevenTapWater 57596857 multical21 to match 57596857.M=KAM.V=1b.T=16
(meter) started meter 1 (BrevenTapWater 57596857 multical21) identity mode: id R57596857
(wmbus) parseDLL @0 45
(wmbus) parseELL @10 35
(meter) BrevenTapWater: for me? 57596857.M=KAM.V=1b.T=16 in 57596857
(meter) BrevenTapWater: yes for me
(meter) BrevenTapWater(1) multical21  handling telegram from 57596857.M=KAM.V=1b.T=16
(meter) BrevenTapWater 57596857.M=KAM.V=1b.T=16 "2C442D2C576859571B168D205C04235D236CD7BA3341283ECC27E680A980880B698DE881CE2B555966A1E71414"
(wmbus) parseDLL @0 45
(telegram) DLL L=2c C=44 (from meter SND_NR) M=2c2d (KAM) A=57596857 VER=1b TYPE=16 (Cold water meter) (driver multical21) DEV= RSSI=0
(wmbus) parseELL @10 35
(ELL) decrypting "6CD7BA3341283ECC27E680A980880B698DE881CE2B555966A1E71414"
(ELL) IV 2D2C576859571B162004235D23000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "845D7802FF20002004130F670C004413"
(ELL) block 1 block_size 12 offset 16
(ELL) decrypted "BA5C0C00A1015B098101E7FF"
(ELL) decrypted "845D7802FF20002004130F670C004413BA5C0C00A1015B098101E7FF"
(wmbus) WARNING! decrypted payload crc failed check, did you use the correct decryption key? 845d payload crc (calculated bc3e) Permanently ignoring telegrams from id: 57596857 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Cold water meter (0x16) ver: 0x1b
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=5c SN=04235d23 (AES_CTR session=4 time=3527216) CRC=845d
(meter) newly created meter (BrevenTapWater 57596857 multical21) did not handle telegram!
(wmbus) telegram from TODO ignored by all configured meters!
(serial) stopping manager
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted simulation
/wmbusmeters # 
```

If I instead listen directly using wmbusmeters
[... 186 lines cut ...]
(meters) using field info time_bursting(Text)[11] to extract 02FF20 at offset 24
(multical21) 000   : 23 length (35 bytes)
(multical21) 001   : 44 dll-c (from meter SND_NR)
(multical21) 002   : 2d2c dll-mfct (KAM)
(multical21) 004   : 57685957 dll-id (57596857)
(multical21) 008   : 1b dll-version
(multical21) 009   : 16 dll-type (Cold water meter)
(multical21) 010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical21) 011   : 20 ell-cc (slow_resp sync)
(multical21) 012   : 62 ell-acc
(multical21) 013   : 9c235d23 sn (AES_CTR)
(multical21) 017   : 1a6c payload crc (calculated 1a6c OK)
(multical21) 019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(multical21) 020   : eb61 format signature
(multical21) 022   : 9ac0 data crc
(multical21) 024 C!: 0020 ("status":"OK") ("current_status":"") ("time_dry":"") ("time_reversed":"") ("time_leaking":"") ("time_bursting":"1-8 hours")
(multical21) 026 C!: 10670C00 ("total_m3":812.816)
(multical21) 030 C!: BA5C0C00 ("target_m3":810.17)
(multical21) 034 C!: 09 ("flow_temperature_c":9)
(multical21) 035 C!: 15 ("external_temperature_c":21)
BrevenTapWater  57596857        812.816 m³      810.17 m³       null m³/h       9 °C    21 °C   OK      2024-11-11 18:59:21
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2024-11-11 17:59:21.000;129;128;57596857;0x23442d2c576859571b168d20629c235d23b33e89beb61f0658cc7b9c02a26b2ea10f78c2<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram leng=36.
```

zuckschwerdt, 2024-11-11:
Okay, that confirms that in your case rtl_433's output has a lenght indication that is 2 bytes too high (0x25 vs 0x23) and also the last two bytes seem missing. Exactly what the change undid in the code.
I did look rather closely at that change and I can't readily say why the input frame would be broken/bad in such a way that this fudging of the output is needed.

Which mode is this with, i.e. which decoder is active? Is is protocol 104, Mode C&T? (others are 105-107, 238). If needed you can get that info using `-M protocol` or `report_meta protocol` in a conf file.
Also can you grab the raw data going into the decoder using the verbose option, e.g. `-R 104:v` -- it won't output to MQTT though, but to the console only.

If I can playback a raw input then pretty sure we can find out where the processing breaks that length indication and trailing CRC.

adamaltmejd, 2024-11-11:
Thanks for the quick reply. Strange.

Protocol is 104. From MQTT topic:

```
time = 2024-11-11 20:33:03
mode = C
M = KAM
id = 57596857
version = 27
type_string = Cold Water
C = 68
data_length = 34
data = 25442d2c576859571b168d209e7c295d23ed1821ca1c0aa00637ad6fdd10ce24f74e
mic = CRC
min_volume_flow_min_0 = 0.000 m3/min
err_volume_flow_min_0 = 0.000 m3/min
min_power_w_0 = 0.000 kW
protocol = 104
mod = FSK
freq1 = 868.86586
freq2 = 868.9961
rssi = -0.101917
snr = 28.82009
noise = -28.922
```

Is this what you are after?

```
❯ docker run -i -t --privileged -v /dev/:/dev/ hertzg/rtl_433:master -R104:v -f 868.95M -s 1600000
rtl_433 version 24.10-4-g8900b188 branch master at 202411101418 inputs file rtl_tcp RTL-SDR with TLS

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
[m_bus_mode_c_t_callback] PREAMBLE_T: found at: 179
codes     : {886}aaaaaaa595b52a534b64a7590b72805ba5964aaaaaaaaa87aa87a4a885a58aed0b2ae362d1a4145b852ba47e655a8816ca876b0ee51f35b969d0ce5d3ed9647f31a0709aaf828b5a09b4946f97b00622a9af4a4bb92160d1a0ef2dc63f02d560d5cb263b724e6d801f291058167ac8
[m_bus_mode_c_t_callback] M-Bus: Mode C, Format B
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-11-11 19:39:21
model     : Wireless-MBus Mode      : C            Manufacturer: KAM         ID        : 57596857
Version   : 27           Device Type: 0x16         Device Type String: Cold Water                      Control   : 0x44          Data Length: 34
Data      : 25442d2c576859571b168d20a2dc295d23f32ad440b6543b587728f9adcb4e8672e9                       Integrity : CRC
Min Volume flow[0]: 0.000 m3/min
[pulse_slicer_pcm] Wireless M-Bus, Mode C&T, 100kbps (-f 868.95M -s 1200k)
codes     : {886}aaaaaaa595b52a534b64a7590b72805ba5964aaaaaaaaa87aa87a4a885a58aed0b2ae362d1a4145b852ba47e655a8816ca876b0ee51f35b969d0ce5d3ed9647f31a0709aaf828b5a09b4946f97b00622a9af4a4bb92160d1a0ef2dc63f02d560d5cb263b724e6d801f291058167ac8
```

adamaltmejd, 2024-11-11:
Not sure if it matters but noticed sample rate is set to 1600k (according to wmbusmeters instructions) but further down it says -s 1200k.

zuckschwerdt, 2024-11-11:
The `codes` line at the very bottom is the raw rf input to the decoder. I'll play around with that.

zuckschwerdt, 2024-11-12:
I think changing the next to last line in `m_bus_decode_format_b()` https://github.com/merbanan/rtl_433/blob/master/src/devices/m_bus.c#L855 would work.
Can you test to insert this
```diff
--- a/src/devices/m_bus.c
+++ b/src/devices/m_bus.c
@@ -852,6 +879,8 @@ static int m_bus_decode_format_b(r_device *decoder, const m_bus_data_t *in, m_bu

         out->length -= 2;   // Subtract the two extra CRC bytes
     }
+    // Include the final CRC
+    out->length += 2;
     return 1;
 }

```
I'm not sure if this would work with longer format B packets (and the length indication is still wrong but it's not needed).

adamaltmejd, 2024-11-13:
Just to double check there was nothing strange with the Docker image that I had been running before I built rtl_433 from source without your change first:

```
./rtl_433 -R104:v -f 868.95M -s 1600000
{1062}aaaaaaa595b52a534b66045cc797600165264aaaaaaaaa87aa87a60885a58aed0b2ae362d1a41692956ba47379bd95c7e29175cf96876c28c920f2bee2ee5ca394a861e5fc262b34c09f34a502f130eb6740cd27e82ba05c45be767e84f952bb69962d757e15874d23886a92789473b71bb543b8e84b6dce6b3e20f9ea7a19ec0635fceb68
{887}d5555552cada9529a5b253ac85b9402dd292455555555543d543d25442d2c576859571b168d20b5b8ab5d23e1111a9025e739e334e08a6632509267001514e7d97d9970f17c145ad04da49e2cbf63e93a0f13b42a92c08edd6b03b983e2c468c9ea4ff8f819195a479e197b8067764
```

Data strings fail crc check. And now with the change

```
./rtl_433 -R104:v -f 868.95M -s 1600000

{885}aaaaaaa595b52a534b64a7590b72805ba5272aaaaaaaaa87aa87a4a885a58aed0b2ae362d1a416fc156ba46e91fde99be9ecf2ec473472d9ec0da2bd194c07771150f96f6f828b5a09b4907773ca438b8094c02f798e19c8988c61333341f3788aef90c714781928e5995a534d3bc8

{887}d5555552cada9529a5b253ac85b9402dd292f55555555543d543d25442d2c576859571b168d20b804ac5d233b23afdb1ac6893a51c677cf3b9c64fd100a50f573a30c7b7a7c145ad04da4b622e9e64f0ce22f4c9d5307656ca1628730387aee6f4c90b3db522f7083392840f724b44
```

Inputted the data strings into wmbusmeters and they are successfully decoded!

PR #3091 (2024-11-13): Fix M-Bus Mode C Format B for wmbusmeters
With M-Bus Mode C Format B include the final CRC for wmbusmeters to verify decryption.
Partially rolls back #2749 change.

Fixes #3089
closed 2024-11-13 by zuckschwerdt