merbanan/rtl_433 #3090: RTL_433 & Alecto WS-5500 weather station

ExquisiteMetropolis, 2024-11-11:
I am attempting to read data from my Alecto WS-5500 weather station.
I assume it uses protocol 16, as other Alecto's are mentioned there.
The USB stick I use is a: Realtek Semiconductor Corp. RTL2838 DVB-T

My config file is very simple:
```
device 1
output mqtt://192.168.1.13:1883,retain=1
report_meta time:iso:usec:tz
frequency 868M
protocol 16

```

When starting the add-on I see only this in the log of the container:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433_868.conf...
[rtl_433_868] rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
[rtl_433_868] MQTT: Publishing MQTT data to 192.168.1.13 port 1883
[rtl_433_868] MQTT: Publishing device info to MQTT topic "rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id]".
[rtl_433_868] MQTT: Publishing events info to MQTT topic "rtl_433/9b13b3f4-rtl433/events".
[rtl_433_868] MQTT: Publishing states info to MQTT topic "rtl_433/9b13b3f4-rtl433/states".
[rtl_433_868] 
[rtl_433_868] New defaults active, use "-Y classic -s 250k" if you need the old defaults
[rtl_433_868] 
[rtl_433_868] Use "-F log" if you want any messages, warnings, and errors in the console.
[rtl_433_868] Found Fitipower FC0012 tuner
[rtl_433_868] Exact sample rate is: 1000000.026491 Hz
[rtl_433_868] Allocating 15 zero-copy buffers
```

No data is shown. 
The MQTT topic isn't appearing either.
Does anyone know what I am missing?

ProfBoc75, 2024-11-11:
Hi @ExquisiteMetropolis before going to mqtt, did you test with this simple rtl_433 command ?

    rtl_433 -f 868M

To see if your model is well decoded ?
Your model WS-5500 looks more recent with wifi than those from protocol 16, and I did not find any issue about it nor pull request, so it's probably not yet decoded by rtl_433.

Next steps, if you have some time and knowledge, you can contribute to decode it, we can drive you for that.

Try to get some samples with

    rtl_433 -f 868M -S unknown

This is the starting point to get cu8 files to play  / replay signals to get the figures.

ExquisiteMetropolis, 2024-11-11:
The external unit communicates normally with an indoor base station via 868M.
From there it uses wifi indeed. 
As don't use the base station, i want everything in Home Assistant eventually, i looking to get the data directly via the 868 freq.

I'll have a look (later this week) on what the simple command will give me for data. Gave it up for today.

ExquisiteMetropolis, 2024-11-12:
I'm using this Home Assistant addon to run RTL_433: https://community.home-assistant.io/t/home-assistant-add-on-rtl-433-with-mqtt-auto-discovery/260665 (without the MQTT part for now).

Changed the config file to:

```
device 1
#output mqtt://192.168.1.13:1883,retain=1
report_meta time:iso:usec:tz
frequency 868M
signal_grabber unknown
#protocol 16

```

This results in this log:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting rtl_433 with rtl_433_868.conf...
[rtl_433_868] rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
[rtl_433_868] 
[rtl_433_868] New defaults active, use "-Y classic -s 250k" if you need the old defaults
[rtl_433_868] 
[rtl_433_868] Found Fitipower FC0012 tuner
[rtl_433_868] Exact sample rate is: 1000000.026491 Hz
[rtl_433_868] [SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000868, "Generic RTL2832U OEM"
[rtl_433_868] Allocating 15 zero-copy buffers
```

Already re-seated the usb-dongle as well, no luck

ProfBoc75, 2024-11-12:
@ExquisiteMetropolis 

Are you able to open a terminal session / ssh to directly enter rtl_433 commands without using HA ?
Because it's more simple, nothing to configure only command options to enter and probably more verbose and flexible too.

During your test, did you have the weather sensor close to the usb dongle ? Our advice is to remove the dongle antenna and put sensors around 50 cm from it. Wait few minutes to capture few cu8 files, one transmit every 16 seconds, even if you will not use the base station display, you will have to note the values for each rf transmit to let decode properly the figures.

ExquisiteMetropolis, 2024-11-13:
Ok. Some lessons learned:

- I was testing the outside-unit inside, with an older battery installed.
- Battery died while waiting for signales. Replaced the battery with a new one.
- When moving it, the wind catcher & director were accidentally moved, and I received data. 
- Turns out that when nothing is registered by the unit, no 'ping' is send to the receiver either.
- Manually moving the wind director and wind catcher, resulted in data.

So it seems the Alteco WS-5500 outside unit via 868Mhz is supported:

```
[rtl_433_868] time      : 2024-11-13T20:31:03.970578+0100
[rtl_433_868] model     : Fineoffset-WH24
[rtl_433_868] ID        : 164
[rtl_433_868] Battery   : 1
[rtl_433_868] Temperature: 15.7 C
[rtl_433_868] Humidity  : 67 %
[rtl_433_868] Wind direction: 59
[rtl_433_868] Wind speed: 1.1 m/s
[rtl_433_868] Gust speed: 2.2 m/s
[rtl_433_868] Rainfall  : 0.0 mm
[rtl_433_868] UV        : 1
[rtl_433_868] UVI       : 0
[rtl_433_868] Light     : 0.0 lux
[rtl_433_868] Integrity : CRC
```

Looks good! 
Not sure which protocol to use, so I won't pickup any clutter later in time.
Would that be 164?

zuckschwerdt, 2024-11-13:
Use `-M protocol` to get the protocol number info, or search the `-R` output. it's `-R 78` here :)

ProfBoc75, 2024-11-15:
No more issue here, protocol identified as 78.

closed 2024-11-15 by ProfBoc75