merbanan/rtl_433 #3097: 2023 KIA EV6 TPMS not decodable

fresh4, 2024-11-18:
I've been testing TPMS RX via rtl_433 and an RTL SDR and have been able to detect most tested sensors except for one belonging to a 2023 KIA EV6. I was able to capture information with -A, and I did test the disabled by default -R decoders. Specifically, -R 123 returned an incorrectly decoded output. I'm fairly novice to rtl_433, so let me know if there's anything else I need to include or if I did something wrong.

Here is the analyzer output for the captured signal with this command: \
`rtl_433 -R 123 -r g005_433.92M_250k.cu8 -C customary -A`

```
> rtl_433 -R 123 -r g005_433.92M_250k.cu8 -C customary -A      
rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR SoapySDR
Reading conf from "/etc/rtl_433/rtl_433.conf".
[Input] Test mode active. Reading samples from file: g005_433.92M_250k.cu8
Detected FSK package	@0.164064s
Analyzing pulses...
Total count:   23,  width: 8.82 ms		( 2204 S)
Pulse width distribution:
 [ 0] count:   20,  width:  172 us [172;180]	(  43 S)
 [ 1] count:    3,  width:  364 us [348;404]	(  91 S)
Gap width distribution:
 [ 0] count:   20,  width:  172 us [172;180]	(  43 S)
 [ 1] count:    2,  width:  352 us [352;352]	(  88 S)
Pulse period distribution:
 [ 0] count:   19,  width:  348 us [348;360]	(  87 S)
 [ 1] count:    2,  width:  524 us [524;528]	( 131 S)
 [ 2] count:    1,  width:  700 us [700;700]	( 175 S)
Pulse timing distribution:
 [ 0] count:   40,  width:  172 us [172;180]	(  43 S)
 [ 1] count:    5,  width:  360 us [348;404]	(  90 S)
 [ 2] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  15986,     30
RSSI: -0.1 dB SNR: 27.3 dB Noise: -27.4 dB
Frequency offsets [F1, F2]:    7376,  -7950	(+28.1 kHz, -30.3 kHz)
Guessing modulation: Manchester coding
view at https://triq.org/pdv/#AAB10300AC01680000808080808080808080808080808080808080809081919255
Attempting demodulation... short_width: 172, long_width: 0, reset_limit: 356, sync_width: 0
Unsupported

Detected FSK package	@0.269712s
Analyzing pulses...
Total count:   52,  width: 21.42 ms		( 5355 S)
Pulse width distribution:
 [ 0] count:   42,  width:  172 us [172;180]	(  43 S)
 [ 1] count:   10,  width:  356 us [348;412]	(  89 S)
Gap width distribution:
 [ 0] count:   42,  width:  172 us [172;180]	(  43 S)
 [ 1] count:    9,  width:  348 us [348;352]	(  87 S)
Pulse period distribution:
 [ 0] count:   36,  width:  348 us [348;360]	(  87 S)
 [ 1] count:   12,  width:  524 us [524;528]	( 131 S)
 [ 2] count:    3,  width:  696 us [696;700]	( 174 S)
Pulse timing distribution:
 [ 0] count:   84,  width:  172 us [172;180]	(  43 S)
 [ 1] count:   19,  width:  352 us [348;412]	(  88 S)
 [ 2] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  15873,     23
RSSI: -0.1 dB SNR: 28.4 dB Noise: -28.5 dB
Frequency offsets [F1, F2]:    9580,  -8096	(+36.5 kHz, -30.9 kHz)
Guessing modulation: Manchester coding
view at https://triq.org/pdv/#AAB10300AC016000008080808080808080808080808080808080808090808190818090819080808180918080908190808191918080808080808080809255
Attempting demodulation... short_width: 172, long_width: 0, reset_limit: 356, sync_width: 0
Unsupported

Detected FSK package	@1.041708s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @1.041708s
model     : Jansite      type      : TPMS          id        : 867a058
flags     : 9            Pressure  : 17 PSI        Temperature: 172 F        code      : 867a058945805c
Analyzing pulses...
Total count:   71,  width: 9.43 ms		( 2358 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   50,  width:   52 us [48;64]	(  13 S)
[... 117 lines cut ...]
 [ 4] count:    1,  width:  392 us [392;392]	(  98 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:  101,  width:   52 us [44;64]	(  13 S)
 [ 2] count:   37,  width:  104 us [104;104]	(  26 S)
 [ 3] count:    1,  width:   32 us [32;32]	(   8 S)
 [ 4] count:    1,  width:  288 us [288;288]	(  72 S)
 [ 5] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  15924,     36
RSSI: -0.1 dB SNR: 26.5 dB Noise: -26.6 dB
Frequency offsets [F1, F2]:    9597, -11162	(+36.6 kHz, -42.6 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB0540601000000340068002001200000819191919191919191919191919191A192919191A19291A1919192A29191919191A2A1929191A291A2A29191A2A19291919191919191A2A19192919191A291A192A192A2A2A455+AAB00F0601000000340068002001200000B055
Attempting demodulation... short_width: 52, long_width: 104, reset_limit: 292, sync_width: 32
Use a flex decoder with -X 'n=name,m=FSK_PWM,s=52,l=104,r=292,g=0,t=0,y=32'
[pulse_slicer_pwm] Analyzer Device
codes     : {69}fffdeddf3a67f9f540, {0}0
```

THE ACTUAL DATA for this signal based on the TPMS sensor I used (as opposed to what the 123 decoder gives) is\
ID: 8679EDB4\
PSI: 34.5\
Temperature: 91F

Hopefully that helps with decoding; I'm not so sure how to do so myself.

zuckschwerdt, 2024-11-19:
The later signals look like TPMS, 50µs PCM (MC) is common.
I.e. this will work to get raw data: `-X 'n=name,m=FSK_PCM,s=52,l=52,r=150,bits>=60'`

Your examples have raw data of `5555555695696A99556695966566955566A55969A666` which MC decoded is `fffe 7985 fa76 ba7f a3d9 2a` or the inverse `0001 867a 0589 4580 5c26 d5` (which starts with the ID you have).
Try to find more raw codes and create [a BitBench](https://triq.org/bitbench/?c=5555555695696A99556695966566955566A55969A666%5B8679EDB4%20PSI%2034.5%2091F%5D&f=PRE%3A16h%20ID%3F16h%20%3F16h%20%3F16h%20%3F16h%20CHK%3F8h&i=true&d=MCI&cw=4) like this.

fresh4, 2024-11-20:
@zuckschwerdt you'll have to forgive me as signal processing is not my wheelhouse. I will try to collect more samples and see what I can do. How do you come up with the flex decoder that you're using? I've read the documentation but of course being pretty unfamiliar I'm not sure what each thing (l,s,y, etc) means exactly.

As an aside, I'm pretty sure (most?) TPMS signals use a 32 bit ID (after the preamble, I guess). There's a chance that the ID I provided was actually for a different tire on the same car, so 867a0589 might be the real ID. I will check when I'm able to get more samples.

zuckschwerdt, 2024-11-20:
It's just the minimum of options to get a working flex decoder: modulation (m) FSK with PCM slicer, short and long symbol (s, l) are the same "half-bit" length of 52 µs, and since we must never see any gap or pulse of more that two identical bits in Manchester coding the reset (r) of slightly longer that 2x 52 (and then demanding some minimum number of bits).

gdt, 2025-07-01:
Where are we on this?  Is this about adding a  decoder, not yet done, or is it a request for help, concluded?

fresh4, 2025-07-01:
Hi! I brought this issue up to hopefully add a new decoder, as it seemed this sensor was not recognized by rtl-433, but was detected when using a TPMS programming tool. 

Unfortunately I don't readily have access to this vehicle at the moment (it's a colleague's) but let me know if there's anything else needed to facilitate this.

gdt, 2025-07-02:
Someone needs to record a whole bunch of samples, figure out the crc, and what bits are what.  It doesn't seem like there is much not-generally-known wisdom in the ticket, so I'm going to close it as not outweighting the burden of open tickets.  It's been 9 months and nobody else is working on this, so I don't think leaving it open will help.  Feel free to open if you're going to be able to dig in.