merbanan/rtl_433 #3104: Problem with displaying results / not finding TPMS

TombusPL, 2024-12-01:
**Story:**
I changed the wheels and tires in my car. Unfortunately, I didn't read the TPMS ID, which I have to program in the car. I don't want to take the tires off the wheels, so I wanted to use rtl_433 ;)

**Equipment:**
4 TPMS in the wheels and 2 TPMS at home as spares.
RTL.SDR V.3
Laptop with Bunsenlabs Linux (Debian Bookworm)
Compilation from sources:
https://github.com/osmocom/rtl-sdr
https://github.com/merbanan/rtl_433

**Goal:**
Reading the TPMS ID and programming them to the car.

**Problem:**
I don't read any TPMS ID mounted in my wheels. Even rtl_433 doesn't display anything.
I only read the ID of the TPMS that was at home once(!) I can never find it again :/

**What I tried:**
I have to use -F http which works and searches for various "codes", but freezes the browser when set to MODE -> RAW (unknown).
I find other TPMS of other cars.
I drove the car with a laptop and searched for sensors.
I deflated the tires to <1bar and pumped them up again.
I unscrewed the antenna from the dongle and put the TPMS close to it.
I tried the commands:
```
rtl_433 -F http
rtl_433 -f 433.92M -A -F http
rtl_433 -A -F http -l 1000
rtl_433 -vv -Y autolevel -M level -M noise -F http
```
[image]
[image]

zuckschwerdt, 2024-12-01:
It's best to use `-S unknown` then spin the wheels or drive a few minutes. You will have collected some samples in the current directory. Sort through them to find your tpms, upload one as zip here and we can see how to support it.

To sort samples use https://triq.org/pdv/ and confirm that it's short FSK burst, comapre to TPMS samples from https://github.com/merbanan/rtl_433_tests/tree/master/tests

Another way would be to try `-R 0 -X 'n=mytpms,m=FSK_PCM,s=50,l=50,r=150,bits>=32'` or in rare cases instead of the 50 a 100 or 25 -- that's all the used TPMS timings we have seen.

fresh4, 2024-12-06:
I'd suggest trying the same command with a higher sample rate (-s 1000000). I've had similar issues with my BMW 433MHz TPMS sensor where rtl-433 doesn't show me a decoded output unless I do that.

gdt, 2025-07-01:
I'm calling this a mostly-answered inactive question.   Feel free to open a discussion for more help.